MarlinFirmware/MarlinDev #275: Conflicting definition when compiling VIKI display configuration

Wackerbarth, 2015-11-08:
In file included from .../ultralcd.cpp:36:0:
.../ultralcd_implementation_hitachi_HD44780.h:54:0: warning: "LCD_CLICKED" redefined [enabled by default]
       #define LCD_CLICKED (buttons&(B_MI|B_RI))
 ^
In file included from .../ultralcd.cpp:1:0:
.../ultralcd.h:70:0: note: this is the location of the previous definition
     #define LCD_CLICKED (buttons&EN_C)
 ^

AnHardt, 2015-11-13:
Indeed. `LCD_CLICKED` is previously defined from `NEWPANEL`. We really want to redefine this. This is the only difference to `NEWPANEL`.

We have different compiler options in MarlinFirmware/Marlin. There we got no warnings about that.
Fix now included in https://github.com/MarlinFirmware/Marlin/pull/2691

Roxy-3DPrintBoard, 2015-11-13:
And...  Let me just add one more thing for people to think about.    

In the case of where there is no LCD Panel defined, I think we want to include an LCD_CLICKED() function that lets the user touch an end switch (of their choice) to be treated as if they had an LCD Panel.

This let's the user do M600 Filament Change commands even without an LCD Panel!   Also, in my case with the crippled Allen Key Z-Probe on my new Delta I needed a way to pause the firmware to deploy and retact the probe.    In my case I have an LCD Panel.   So I called LCD_CLICKED() to get confirmation from the user that it was OK to continue.    I think they are going to be rare, but there will be Allen Key Delta's that do not have LCD Panels.

I guess what I'm saying is we should have a fall back position to handle getting user confirmation even if the machine does not have an LCD Panel.

Wackerbarth, 2015-11-13:
I don't disagree with the desired functionality.

However, my thinking is that we should be micro-coding these "complex" operations rather than generating additional special purpose sections of code.

Originally, all of the "codes" (G,M, etc.) were primitive operations. It was up to the host to send an appropriate sequence to tell the bot what to do.

That model has since been "corrupted" (with the addition of functions like G29, et. al.)
Perhaps what we should do is return to the primitive concept just as computers went from complex logic to RISC architecture. In that context, we might have a primitive "wait for event" and various ways to generate these events. One of the events might be "USER_SIGNAL" and we could have a primitive to map the "PROXIMITY_SENSOR_#_" to this USER_SIGNAL.
The "LCD_USER_BUTTON_#" could also be mapped in similar fashion.

With such a structure, we can write signal handlers as strings of primitives. We could even store some of them in RAM, ROM, or EPROM since they would be used infrequently and need not execute "quickly".

Were it not for the fact that our memory resources are quite limited, we could have much of the logic reprogrammable. But the limited resources will require that most of the functions at constants at compile time rather than configurable at run time.

Roxy-3DPrintBoard, 2015-11-13:
Well...  Just to be argumentative...  :)    

> Were it not for the fact that our memory resources are quite limited,

Remember...  I'm talking about the case where there is no LCD Panel.   The LCD code is very large.  Without the LCD Code, my PrintRBoard only has 85Kb of its 128KB of flash used up.   And with the 256Kb AVR in my new Delta printer I doubt we are going to run out of room adding these kind of routines.

> However, my thinking is that we should be micro-coding these "complex" operations rather than generating additional special purpose sections of code.

So maybe we need to define an MCode to display a message everywhere possible  and another one to get user confirmation that they have seen the message and are ready for the printer to proceed?   And we can point these at LCD_CLICKED() ?

Wackerbarth, 2015-11-13:
Roxy,
The memory that is most limiting is RAM. If we have a function to take micro-sequences from ROM, EPROM, and from RAM, we can "mix and match". What I was saying is that if I had say 1GB of RAM, I would just load ALL of the potentially variable code into RAM and "execute" it there (much as a typical PC runs any user program.) But I don't think that we have enough RAM to do it that way. But we can still put SOME sequences in RAM if they need to be highly customizable by the end user.

Wackerbarth, 2015-11-13:
As for the creation of an "M-code" for a message, crafting that will require some care since the command parser doesn't expect strings for the message text.

Roxy-3DPrintBoard, 2015-11-13:
> The memory that is most limiting is RAM. If we have a function to take micro-sequences from ROM, EPROM, and from RAM, we can "mix and match". 

Yeah, but if we had some MCode that said "Wait for user confirmation that it is OK to continue" that would take zero RAM.  It needs a little bit of memory on the stack while it is invoked, but it gives that all back when it is done.   And certainly, it won't use much stack space compared to some of the memory pigs we have like G29 doing a 6x6 grid!

> As for the creation of an "M-code" for a message, crafting that will require some care since the command parser doesn't expect strings for the message text.

Agreed.   But even a generic message that was always the same saying "Press the appropriate End Stop to continue." would go a long way.     

I'm not pushing real hard for this.   I've always had a similar thing added to my Prusa i3.  I'm just saying it is something to think about because a lot of users without LCD Panels could benefit from it.

Wackerbarth, 2015-11-14:
I'm not, in any way, opposed to creating the kind of commands that you describe. We certainly should provide the functionality. My discussion is related to to implementation details.

What I am discussing with respect to memory involves the memory which is "programmable" by the end user.  For example, the message to which you refer might be a fixed message. Alternately, it might be a message which the user configures from the host. If it is the latter, then we must provide some memory to store the text of that custom message.

I would prefer to avoid having a code DEDICATED to "Press the end stop to continue". Instead, I would like to see a more generic mechanism which could then be customized to provide that message, and also, at another time, when used in a different context, some other messages.

Roxy-3DPrintBoard, 2015-11-14:
Remember how I started this discussion:

> And... Let me just add one more thing for people to think about.

We are just thinking about things and possible ways to go forward.   Let's explore another path.   Just suppose we had a micro-coding type of instruction set available that we could draw upon.  What type of commands would we want (and need) to accomplish this idea?

Ideally, we would want to have some default messages, but also the ability to customize messages that this micro-coding of instructions can use.   And besides being able to communicate with the outside world, this micro-instruction set needs to be able to check and look at physical events (properties) of the controller board.  In this case, one of the end stop switches.   

And at a slightly higher level, we might want the ability to redirect some of these micro-coded instructions from using the defaults (and in this case point them to a different end stop switch that makes more sense for a particular user).   

What does this look like?

AnHardt, 2015-11-15:
We already have
 * M117 - Display a message on the controller screen
 * M0   - Unconditional stop - Wait for user to press a button on the LCD (Only if ULTRA_LCD is enabled)
 * M42  - Change pin status via gcode Use M42 Px Sy to set pin x to value y, when omitting Px the onboard led will be used.

M117 could be extended by a wait-parameter,
M0 by a text- and or pin-parameter,
M42 by a wait-for-parameter.
In combination with `enqueuecommand()` and companions, i see a lot of possibilitys.

closed 2016-04-26 by thinkyhead