MarlinFirmware/MarlinDev #280: What are the biggest problems to overcome in the RC?

a4jp-com, 2015-11-18:
What are the biggest problems to overcome from this RC?

Roughly how long do you think it will take to get this RC to a full working version that is kind of bug free? 

I used to have to use an older version of Arduino to install my Scoovo 3D printer driver (Marlin), as the LCD needed a plugin to get everything showing up. But will this new version of Marlin easily support RAMBo boards with LCD screens without an extra plugin, or will there still be something needed to get the LCD to work properly this time?

I sent my printer off to the factory to get the z axis ACME rod replaced and after that everything should be perfect. I'm just hoping I can update the firmware soon too.

KiteLab, 2015-11-18:
>What are the biggest problems to overcome from this RC?

That highly depends on whom you are asking. Everyone wants to have killed the bug he met.

>Roughly how long do you think it will take to get this RC to a full working version that is kind of bug free? 

For ever! "Bug free" is a state, only very small programs can reach. If you mean "is printable with", for a lot of configurations the answer is - we are ready.

>the LCD needed a plugin

Do you mean a library? Even in the newest development versions you will have to do some additional clicks to add a library.

> RAMBo

I could not find any RAMBo specific bugs in our issue lists (is:issue rambo ). All issues i could find are specific to some other subsystems, but not only for RAMBo. This can ether mean there are no problems, or the problems are so huge no one tries to use RAMBo and therefore we do not get error reports. I suppose there are no problems. If you already run Marlin on a RAMBo, you know the answer better than me.

>Scoovo 3D

With a Scoovo you seem to be pretty alone. Had a brief look into (http://a4jp.com/3d-files/marlin-2015-01-06th-23-28-00.zip). With a ULTIPANEL type display and no ABL i don't expect major problems.

>What are the biggest problems

The most important errors are that, where safety is concerned. A fire is, what we are afraid for. Luckily the heater protection system currently works as expected. We get some unwanted shutdowns when the protection system isn't tuned well - but that is on the save side. We recently had one case where the heater was turned on directly by a M42 command - that's under investigation.

The second most important errors are that where we can damage the machine. Slamming the nozzle into the bed and driving an axis beyond its limits, are the main cases where we can damage the machine. This is not in a very good shape. 
For almost all setups we have to deactivate the hardware endstops to make it work at all. That means we lose a lot of safety here. I can see a solution here, but the changes may be to much to have them in the current RC - we'll see.
Software endstops seem to work well on Cartesian systems, but not for delta machines. Here some ideas are floating around, waiting for implementation.
All the endstop problems can be avoided by not ordering the nozzle to places it can't go to. So when your marlin-configuration and slicer and host have a proper setup and you are not loosing steps, these problems should never be hit during a normal print.
If G92 is involved, even that is questionable. This potential problem is ignored for now - at least I can't see any activity. For now, just don't use G92 or anything changing the home offsets, or do it at your own risk (like everything doing with Marlin).

All other problems are less important. They ether effect special hardware, special features or combinations of features, convenience ... , but not basic printing. They could be ignored for the next Release, but fixes are welcome. 
If you are using special hardware, effected by a bug, you'll likely have to fix the bug by your selves. It's unlikely you can interest one of the developers to your problem until they have something to test with (Exceptions exist). 

>how long

Hard to say. Depends on how many developers we have. On what they are working on. On how many support questions they answer, instead of working against bugs. How many and how difficult to fix new problems appear during testing. 
Maybe we'll get a Christmas surprise.

How ever. The current `RC` is better and saver then `Release` or `1.0.x` in all aspects. Forget about `Release` and use `RC` or `RCBugFix`. 

Le roi est mort, vive le roi ! - The King is dead, long live The King! - Der König ist tot, lang lebe der König

a4jp-com, 2015-11-18:
What does ABL mean? Is that a probe?

Thank you so much for the amazing reply KiteLab.

I hope the stops get fixed as soon as possible. A broken glass bed would suck big time.

Roxy-3DPrintBoard, 2015-11-18:
> What does ABL mean? Is that a probe? 

Auto Bed Leveling....

Yes, that was a very good status report from KiteLab!

Roxy-3DPrintBoard, 2015-11-18:
This link will get you there:  https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

One thing you can do is change:
~~~~
#define DEFAULT_MAX_FEEDRATE          {1500, 1500, 1500, 45}    // (mm/sec)
~~~~
to something like:
~~~~
#define DEFAULT_MAX_FEEDRATE          {1500, 1500, 50, 45}    // (mm/sec)
~~~~
That will slow down all movement on the Z-Axis and you will have plenty of time to press the reset button if things go crazy.

a4jp-com, 2015-11-19:
I thought the only tricky bit when I loaded the driver before was getting the LCD working with the newest version of Arduino but I was able to load the plugin/URL this time after an internet search. Before I had to load and old version to use the plugin to select my Rambo board.

Just for reference:

http://reprap.org/wiki/Rambo_firmware

Copy this:

https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json

into File > Preferences > Additional Boards Manager URLs:

Click OK

Restart

Tools > Board manager

Search for RAMBo

If there is a small popup link to update files or add-ons click that.

Search again if needed and install the RAMBo board.

It is now selectable from Tools > Board ^^

---------------
Now, all I have to do is wait for the hardware to arrive and hope the hardware endstops can be re-enabled here. Do you have a link to the thread where the endstop problem is? I want to read about the problems. Maybe I can help by testing endstop code of something.

AnHardt, 2015-11-22:
Collected some of the hardware endstop problems in #285

Grogyan, 2015-11-22:
So I'm also keen to know how far the thermal runaway prevention codes being applied to the RC branch. This will assist me in providing more details in tickets for Marlin and OctoPrint, where I have on multiple occasions observed random thermal runaway. Even when the machine was off,  yes off! 
OctoPrint crashed, forcing Marlin into a crashed state,  despite the thermal run away prevention code being in place, thermal run away occurs.

AnHardt, 2015-11-23:
@Grogyan 
Thermal runaway when off? That does not sound very off! For the bed or for the extruders?

Thermal Runaway Protection is implemented for the extrudes and the bed in RC.
Heater Sanity Check is implemented for the extruders only.
Heater Sanity Check for the bed is still not implemented.
Heater Sanity Check PR https://github.com/MarlinFirmware/Marlin/pull/2310 from 18 Jun was rebased and or improved 2 times and closed as https://github.com/MarlinFirmware/Marlin/pull/2352 on 20 Aug from @Wackerbarth without comment. 
No one was willing to test or merge the PR in two month. Maybe someone else will try his or her luck again with this. :-(

Roxy-3DPrintBoard, 2015-11-23:
> So I'm also keen to know how far the thermal runaway prevention codes being applied to the RC branch. 

I'm bringing up RC-2 on my new Delta printer.   I can tell you first hand the Thermal Runaway code is much to aggressive for my taste!  I had to relax it a lot to keep my nozzle's hot.   

> No one was willing to test or merge the PR in two month. Maybe someone else will try his or her luck again with this. :-(

Not quite accurate...   People can only move so fast!    And the next thing on my list is to get Auto Bed Leveling ---CORRECT--- on Delta's !

Grogyan, 2015-11-23:
@AnHardt Yes, the machine was off
Ie, 
the power supply was off 
the extruder was off
the motors was off

About 15minutes after completing a print, with all the above off, Something, either Marlin and/or Octorint crashed hard, which then caused the machine to turn itself back on, by itself, and ramp up the heat uncontrollably.
Keep in mind that the RAMPS board and OctoPrint had been left on in idle mode 24/7 for 2 weeks leading up to that incident. and has occurred again in a similar manner on Saturday.
And has been at least the 5th time it has happened over several months. but the 3rd in about a month as the host interface has been running near flawlessly for about 2 months with OctoPrint on the RPI2, hence why it is being raised now as a ticket.
https://github.com/MarlinFirmware/Marlin/issues/2781

The RC version on the machine was pulled from this repo in September.

a4jp-com, 2015-11-23:
If the power supply is off how does the printer turn back on??? Was it in sleep mode or something?

a4jp-com, 2015-11-23:
Oh I see now M81.

Wackerbarth, 2015-11-23:
@AnHardt -- As a serious question, "Isn't all of the fancy Thermal Runaway code for the BED just the solution to a non-issue?". For the extruders, I can see value in attempting to detect as many failure modes as possible since they that the potential to start a fire.  But I have yet to see a heated bed that can get seriously hot. Perhaps we are wasting effort, and mpu cycles, in the wrong area.

AnHardt, 2015-11-23:
[image]

Roxy-3DPrintBoard, 2015-11-23:
Wow!!!   Those plug directly into wall power???    How hot do they get?

AnHardt, 2015-11-23:
Allowed is up to 260°C. But if not cooled, or limited otherwise, they heat up far beyond self destruction.

Wackerbarth, 2015-11-23:
But I suspect that they don't have issues with the "heating too slowly" problem that is plaguing those little underpowered ones.

AnHardt, 2015-11-23:
Overshoot at low temperatures is the problem with this beasts. Set PID_FUNCTIONAL_RANGE really huge and lower MAX_BED_POWER.

Wackerbarth, 2015-11-23:
Not at all surprising! It's the wrong size tool for the job. As they say, "Don't send a boy to do a man's job". Or, in this case, "a man to do a child's job".

thinkyhead, 2016-04-20:
Closing old issues for housekeeping.

closed 2016-04-20 by thinkyhead