MarlinFirmware/MarlinDev #283: Request: Feedback for baby steps

atunguyd, 2015-11-20:
I recently enabled baby steps in my firmware as I think this will be benificial when that first layer is not looking quite perfect.

Problem is that on the LCD screen there seems to be no feedback as to whether the baby steps is doing anything and in what direction it is doing it.

Can I suggest that this displays the babysteps offset that is being applied, not necessarily in actual measurements but just in steps.

For example I adjust my Z axis down, the baby steps menu should show -3 (or however many steps) when I adjust it, then when I reenter that menu it should start at -3, if I go up it should approach zero going positive if I go above my original Z reference height.

Roxy-3DPrintBoard, 2015-11-20:
I think there would be value to this.   And actually...  I would be interested in writing that code.  But I can't get to it for while.   (I'm in the middle of a bunch of issues for my new Delta printer.)   Maybe somebody else will feel like doing it?

CCS86, 2015-11-22:
Not a bad idea, since the knob clicks aren't always 1:1. 

I always just listen for the servo clicks, and think about directionality as if I was raising/lowering a bolt. CCW brings it up towards you.

atunguyd, 2015-11-23:
Hi, I have implemented this myself. Unfortunatly I am not familiar enough with github to know ho wto submit this via github, so I have reproduced the function and it's changes below. 

This just displays the amount of baby steps applied for each axis, I played with trying to convert the steps to mm's but the sprintf_P function currently being linked in doesnt seem to have support for floating points, so it just shows the raw steps for now.
Also I think resetting the variables that keep track of the steps applied at the end (or beginning) of the print is prudent otherwise consecutive prints will display different values. 

static int BabyStepValue[3] = {0,0,0};
  static void _lcd_babystep(int axis, const char *msg) {
  static char value[8];
    if (encoderPosition != 0) {
      babystepsTodo[axis] += BABYSTEP_MULTIPLICATOR * (int)encoderPosition;
      BabyStepValue[axis] += BABYSTEP_MULTIPLICATOR * (int)encoderPosition;
      sprintf_P(value,PSTR("%d"),BabyStepValue[axis]);
      encoderPosition = 0;
      lcdDrawUpdate = 1;
    }else{
	   sprintf_P(value,PSTR("%d"),BabyStepValue[axis]);
	}
    if (lcdDrawUpdate) lcd_implementation_drawedit(msg,value);
    if (LCD_CLICKED) lcd_goto_menu(lcd_tune_menu);
  }

thinkyhead, 2016-04-20:
The current `RCBugFix` just gained this feature. It displays the number of babysteps for the current babystep command.

closed 2016-05-27 by jbrazio