MarlinFirmware/MarlinDev #284: RAMPS_13_EEF configuration block missing.

gabrielcr78, 2015-11-21:
The new pins_RAMPS_13.h sets servos as default for pins 11,6,5 and 4 as follows:

 #define SERVO0_PIN         11
#define SERVO1_PIN          6
#define SERVO2_PIN          5
#define SERVO3_PIN          4

however many people has the EEB configuration and for fans uses an extender like this one: http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender
that actually need to be configured this way: 
http://www.instructables.com/id/Configuring-and-using-Reprap-Ramps14-RRD-Fan-Exten/step2/Configuring-Marlin-firmware/
the problem is that there is now not specific way to configure the pins only when delected ramps EEB config.
this should be within an IF, allowing those pins to be used for FANS, like before.

thanks!

KiteLab, 2015-11-22:
The servo pins are defined but never used until you define NUM_SERVOS. This makes no sense because all servo pins are blocked by the fan extender. If you define NUM_SERVOS only the defined number of pins are used.
If you don't plug the extender to the RAMPS in the by RRD suggested way, but shifted 2 pins to the right, you are able to use 2 servos and 2 fans, but the fans will have the pins 5 and 4. 
Because you can use the fan outputs for the part-fan, the electronics-fan, the hotend-fan or something completely different, this will always need manual configuration.

KiteLab, 2015-11-22:
> You need Marlin firmware v3 or newer. (cite from RRD)

One day, maybe, we'll get there. But don't expect that for the near future.

Roxy-3DPrintBoard, 2015-11-22:
> http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender

I'm bringing up a G2s Delta printer from Geeetech.   My advice is think twice or three times about ordering ANYTHING from Geeetech.   I can't even get them to reply to my email.

gabrielcr78, 2015-11-22:
thanks for the super quick answer!! :) i sunno the extender has worked ok for me.. didn't buy from geetech tho.. just bought it somewere in ebay.... i know manual config is needed (that0s what i'm doing), but just called my attention that there are no specific configuration anywhere for the EEB configuration. I agree with you that this card takes the 4 ports and only uses 2 for fans, leaving 2 un-usable, but untill i go ahead and design my own fan extender this is what i have :p so i guess i'll just change the original 4 lines (#21 to #24) in the pin_RAMPS_13.h to:
//#define SERVO0_PIN         11  //Gabriel: default was uncommented, commented to use pins with Fan Extender card.
#define FAN_PIN             6  //Gabriel: changed PIN assignation from SERVO01 to FAN_PIN, to use with Fan Extender card.
//#define SERVO2_PIN          5  //Gabriel: default was uncommented, commented to use pins with Fan Extender card.
//#define SERVO3_PIN          4  //Gabriel: default was uncommented, commented to use pins with Fan Extender card.

maybe this is archaic but i guess that's just a temp workaround.. any flaws you see on it?

thanks!

Gabe

KiteLab, 2015-11-22:
You are right. You can't find a literal EEB anywhere in the pins_RAMPS_13.h. The configuration for the FAN_PIN is done in the last else clause here:
```
#if MB(RAMPS_13_EFF) || ENABLED(IS_RAMPS_EFB)
  #define FAN_PIN           8 // (Sprinter config)
  #if MB(RAMPS_13_EFF)
    #define CONTROLLERFAN_PIN  -1 // Pin used for the fan to cool controller
  #endif
#elif MB(RAMPS_13_EEF) || MB(RAMPS_13_SF)
  #define FAN_PIN           8
#else
  #define FAN_PIN           4 // IO pin. Buffer needed
#endif
```

erteclas, 2015-12-21:
could someone have running this adapter can upload files ramps_pins_13.h , configuration_adv if needed another one? so we do not walk sticking groping and avoid burn something

erteclas, 2015-12-21:
i only change fan_pibn 4 // io pin .buffer needed to 6 or 11 and i can control the fan with m106 and m107?

erteclas, 2015-12-23:
no work... fan always on....

thinkyhead, 2016-04-18:
How is `RCBugFix` looking in relation to this issue? That code will be getting migrated over to MarlinDev so if the issue is fixed there, it will soon be fixed here as well.

closed 2016-05-25 by boelle