MarlinFirmware/MarlinDev #285: Hardware Endstops

AnHardt, 2015-11-22:
Hardware Endstops

A small collection of problems with hardware endstops. I'm not sure i'v got all of them.
Sorry. I was to lazy to search for all the original issues. Some of the below questions are constructed.

* 1 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-max endstop. __Before__ homing i can't go in -z direction. What to do?
Workaround: Home the machine.

* 2 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-max endstop. __After__ homing i can't go in -z direction. What to do?
Workaround: Set `#define ENDSTOPS_ONLY_FOR_HOMING`

* 3 Question: I have a probe that is always triggered when not deployed (like a allenkey) and a z-min endstop. I can't go in -z direction and I can't home. What to do?
Workaround: Don't use this configuration.

Reason: The z-min probe acts as a z-min endstop - always when the endstops are checked.
Solution: Add a new variable `bool z_probe_is_ready = false`. Don't test the z-min-probe-switch when false. Set and reset the variable during deploing/undeploing the probe. Make all this configurable with `//#define Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE`. Connect your probe to a probe-pin, not the z-min-endstop-pin.

* 4 Question: I set hardware endstops to on with M120. After G28, G29 or any other g-code using the probe this is forgotten.
Workaround: Use M120 after using the probe.
Reason: The endstop-state is not remembered. After using the probe the state is set depending on `ENDSTOPS_ONLY_FOR_HOMING` only.
Solution: Remember the previous state when switching to "on" for a wanted endstop move, like it is done for the feedrate in `setup_for_endstop_move()`. Restore the previous state in `clean_up_after_endstop_move()`. Disable M120 before homing and give a warning - that's dangerous.

* 5 Question: On my delta the coordinate system is messed up when the probe triggers somewhere between starting G29 and the first probe.
Workaround: Set up your machine so that the probe is not triggered before.
Reason: When on a delta a endstop / probe is triggered unexpected no correction can be made.
Solution: Make software endstops work and narrow down the time the hardware endstops are active to the absolute minimum (`run_z_probe()`, `homeaxis()`). The right reaction on an unexpected hardware endstop hit is to stop the current and all further moves.

* 6 Question: On my delta machine the coordinate system is messed up randomly.
Workaround: Set `ENDSTOPS_ONLY_FOR_HOMING`. 
Reason: Noise on the endstop/probe inputs.
Solution: Fix the endstops, their wiring, ... . Fix software endstops. The right reaction would be to stop the print, instead of making random looking moves.

* 7 Question: When i print the layers shift in some direction. Somewhen later a hardware endstop is hit. Thereafter the layers are shifted a lot in an other direction. The part is a mess but print continues.
Workaround: You are loosing steps in one direction. If you see the layer shifting, stop the print and try to work against loosing steps.
Solution: If a hardware endstop was hit after the machine was homed, something catastrophic has happened before. The software endstops should have prevented that the hardware endstops could have been hit. Even without software endstops the next move will never end where it is expected to. Stop the machine when a hardware endstop hit was detected after homing.

* 8 Question: I homed my machine, send a `G1 X10`, `G1 X0`. I get a message that the endstop was triggered.
Workaround: Set `#define ENDSTOPS_ONLY_FOR_HOMING`
Reason: The x-min-position is set while the switch is triggered. Returning to there will trigger the switch again.
Solution: Back up from the switch before the min-position is set. For example by `X_HOME_BUMP_MM`.

* 9 Question: When i move around i can hit my hardware endstops and get messages. What to do?
Answer: Home your machine before moving around.
Question: If that does not help?
Answer: Activate software endstops.
Question: Even that has not helped. The next move after a endstop hit is strange.
Answer: You have a delta machine. Software endstops are more or less unimplemented there.
Workaround: Don't move to a impossible place.
Solution: Implement proper software endstops for deltas.

* 10 Question: What is the action to do if a hardware-endstop-hit is detected before homing?
Workaround: Homing should always be the first move.
Solution: I don't know. The minimum is to stop the actual move. I guess the strongest is to clean the planner buffer, to prevent further already buffered moves from hitting the other side, where potentially no endstop is.

* 11 Question: I can't move an axis in any direction. What to do?
Workaround: Test your endstops with M119. Maybe both endstops of that axis are triggered at the same time. Configure your endstos correct.
Solution: Send a warning during boot/selftest. Triggered endstops at both ends, at the same time, indicate an error.

* 12 Question: I can't move an axis towards an endstop.
Workaround: Test your endstop and configure it correct.
Solution: During boot/selftest, if an endstop is triggered but not both, back up slowly `?_HOME_BUMP_MM`. If the endstop then is not free, throw an error.

* 13 Question: When a hardware endstop is triggered and the hardware endstops are tested i can move away from them. The move is not stopped. I expected they'd stick on them.
Answer: Thats a feature, not a bug. The endstops are not tested when the move is away from them.
Question: But isn't testing the direction slower than the actual test of the endstop?
Answer: Maybe we should test the endstop first and make the decision what to do with that information later.
Question: But wouldn't sticking be a simple method to stop any pending moves until we actively free the endstop?
Solution: To be discussed.

AnHardt, 2015-11-22:
* 14 Question: When i move down the nozzle stops to go down. I can't go below a that height. 
Answer: Are the hardware endstops on? Test with M119 if your z-min-endstop ore zmin-probe is triggered. If yes, you could have a inductive probe. 
Workaround: In that case you have to switch of the hardware endstops. define `ENDSTOPS_ONLY_FOR_HOMING`
Solution: Connect your probe as a probe. Use the new `#define Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE` feature. This way x and y stops are still checked but not z.

Wackerbarth, 2015-11-22:
"Connect your probe to a probe-pin, not the z-min-endstop-pin." --  Shouldn't this be the proper solution for all cases? If the signal is a probe, then call it that and if you don't actually have an end-stop, don't claim that you do.

Wackerbarth, 2015-11-22:
5 - "On my delta the coordinate system is messed up when the probe triggers somewhere between starting G29 and the first probe." -- How is this an issue different that Q6? An end stop SHOULD never trigger unless we are doing an "align" (homing) operation. If that is true, then it is either "electrical noise" (false triggers) or you are truly trying to exceed the working range of the actuators.
In the former case, you really need to fix the underlying cause. In the latter case, there is some error in the configuration of the mechanical boundaries.

AnHardt, 2015-11-22:
In 5 you have not the choice to set `ENDSTOPS_ONLY_FOR_HOMING` because G29 took control over the endstops. In that special case the the user was deep enough to trigger the z-min-probe, but did not crash and messed up the stepper coordinate system.
And yes, the propper reaction is to stop this and all further moves until rehomed. 

For a while i thought about how to reanimate a sticked to an endstop, or halted for a endstop hit printer. A special g-code, a button, a menu point? A good answer could be G28 - homing - that's needed anyway.

AnHardt, 2015-11-22:
If we'd say, a probe is like a endstop but needs to be activated/deactivated we have a reasonable and logic difference between them. Until now that was a bit blurry in my brain (It was more a second z-min-endstop for me). If all kinds of probes are not tested when not activated we do not need `Z_MIN_PROBE_IS_TRIGGERED_WHEN_INACTIVE`.

Wackerbarth, 2015-11-22:
From my perspective, a probe is a sensor that needs to be tested in only two circumstances:
1) The obvious, probing -- that time during which we are actively moving toward the bed while looking for the proximity detection event.
2) Possibly as a part of the probe deploy/stow procedure -- For example my "Allen Key" probe registers "in proximity" when it is properly stowed.

Both of these circumstances are related to short durations and specific places in to motion code.

Roxy-3DPrintBoard, 2015-11-22:
Just as an FYI:    In the RC-2 I'm bringing up on my Delta, I warmed over Stepper.cpp to do this:

~~~~
           if (z_test && current_block->steps[Z_AXIS] > 0) { // z_test = Z_MIN || Z2_MIN
              endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
              endstop_hit_bits |= BIT(Z_MIN);
              if (!performing_homing || (z_test == 0x3))  //if not performing home or if both endstops were trigged during homing...
                step_events_completed = current_block->step_event_count;
            }
          #else // !Z_DUAL_ENDSTOPS
	    if (z_probe_is_stowed==false)
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
          {
            endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
~~~~

AnHardt, 2015-11-22:
The test, if a Allen Key was deployed and ore stowed  successfully can be done and is done outside the interrupt, asking the pin directly.

Roxy-3DPrintBoard, 2015-11-22:
@AnHardt  I'm not following what you are saying.   update_endstops()  is called at interrupt time.  Can you point me to the code you are talking about?

Wackerbarth, 2015-11-22:
"The test, if an Allen Key was deployed ... done outside the interrupt"
I agree. I've even wondered if we might even take all of the end stop / probing tests out of the stepper interrupt and handle them at a lower priority.

There are only two cases where these sensors are important.
When they are used as a "safety", their sensing of proximity is going to abort the print (hopefully before the mechanism is damaged).  In this case, I suspect that there is some "slop". As long as the actuator doesn't take "too many" additional steps, we should be OK. Once the condition is triggered, we no longer really care about the true position of the mechanism.

The other case is during "alignment" (AKA homing) or "probing".  Under these conditions, accuracy is important. However, done properly, we should be moving very slowly while looking for that accurate edge.  Additionally, we really need to worry about one sensor at a time.

Does anyone have an idea just how quickly we MUST react to an end stop? (The time from when an end stop triggers until the steppers no longer receive pulses)

Roxy-3DPrintBoard, 2015-11-23:
> Does anyone have an idea just how quickly we MUST react to an end stop? (The time from when an end stop triggers until the steppers no longer receive pulses)

This seems like it should be straight forward to calculate.    Maybe in the morning I'll feel like getting out the calculator!   :sunglasses:

AnHardt, 2015-11-23:
@Roxy-3DPrintBoard 
~~~~
           if (z_test && current_block->steps[Z_AXIS] > 0) { // z_test = Z_MIN || Z2_MIN
              endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
              endstop_hit_bits |= BIT(Z_MIN);
              if (!performing_homing || (z_test == 0x3))  //if not performing home or if both endstops were trigged during homing...
                step_events_completed = current_block->step_event_count;
            }
          #else // !Z_DUAL_ENDSTOPS
/*
	    if (z_probe_is_stowed==false)
I'm quite sure the z-min-endstop should not depend on the probe.
If you connect your probe to a z-min-pin because you have no z-min-switch connected, you should #define Z_MIN_PROBE_PIN former_Z_MIN_PIN 
*/
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
//            ^^^^^^^^^^^^^^^^^^^^^^^^^ why testing this twice? how about a '{' behind the if 3 lines above.
          {
/*
          endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
         This is already done in the UPDATE_ENDSTOP(Z, MIN_PROBE) macro.
*/
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
~~~~

AnHardt, 2015-11-23:
@Roxy-3DPrintBoard 
MarlinMain.cpp
```
      #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        bool z_probe_endstop = (READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING);
        if (!z_probe_endstop)
      #else
        bool z_min_endstop = (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING);
        if (!z_min_endstop)
      #endif
        {
          if (IsRunning()) {
            SERIAL_ERROR_START;
            SERIAL_ERRORLNPGM("Z-Probe failed to retract!");
            LCD_ALERTMESSAGEPGM("Err: ZPROBE");
          }
          Stop();
        }
    #endif // Z_PROBE_ALLEN_KEY
```

AnHardt, 2015-11-23:
During probing we need the tests to be in the interrupt, because they are in sync with the steppers there and we recalculate the position. During homing we only want to stop - but as fast as possible. The other time we only need the information if one of the switches was hit.

Wackerbarth, 2015-11-23:
"in sync" only implies that we know the step position to within one step. That is not REQUIRED to be at the interrupt level IF the sampling rate is greater than the step rate.  If, when we are probing, we are moving slowly enough, the sampling can also be relatively slow and can be in a completely independent loop.

Roxy-3DPrintBoard, 2015-11-23:
>  That is not REQUIRED to be at the interrupt level IF the sampling rate is greater than the step rate. If, when we are probing, we are moving slowly enough, the sampling can also be relatively slow 

Cool!   I just realized something.   This is important if I get out my calculator tomorrow.    I certainly could be wrong.   But this sounds a lot like 'Nyquist Theorem' stuff.   If that is true, our sampling rate has to be twice (at a minimum) of what ever frequency we want to measure.   

I'll be reading this tomorrow:  https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem

Wackerbarth, 2015-11-23:
@Roxy-3DPrintBoard  -- No, Nyquist has to do with determining the frequency components of signals.

In our case, we need only have a sampling rate no lower than the stepping rate.  That way, we are ALWAYS assured that for each step taken, we have the ability to sample the sensor at least once (just as would be the case if the sampling is done in the ISR)

a4jp-com, 2015-11-23:
I read in another thread the hardware endstops were disabled but in RC-2 have they been enabled again?

AnHardt, 2015-11-23:
@a4jp-com 
Please reread. The hardware endstops are still deactivated after homing everywhere. 'But it's your choise. You can activate or deactivate `ENDSTOPS_ONLY_FOR_HOMING` by your will.
`ENDSTOPS_ONLY_FOR_HOMING` is activated by default because otherwise it's very likely you'll experience some of the problems described above. Please have an eye on the workarounds.
In this thread we try to find out what is necessary to to change, to get less trouble and more profit from hardware endstops.
You are the one who asked for this thread.

Roxy-3DPrintBoard, 2015-11-23:
@AnHardt 
~~~~
          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
//            ^^^^^^^^^^^^^^^^^^^^^^^^^ why testing this twice? how about a '{' behind the if 3 lines above.
          {
~~~~
I usually catch those kind of easy optimizations!   But I missed that one!

a4jp-com, 2015-11-23:
@AnHardt I reread the top bit but couldn't see the answer to my question but lots of different answers. Your post here clears up my question though ^^ 

Thank you so much for taking the time to write such a detailed answer and thank you also for creating this thread. 

Hopefully I should be getting my printer back in a day or two. Everything is looking good here. I even found a way to get the RAMBo board listed in Arduino so I just need to fill out the details in the Marlin files and do a few test prints after flashing the firmware. 

I physically turn off my printer so I don't think I'll have any fires but it still good to hear about things like that.

@Roxy-3DPrintBoard nice catch.

Wackerbarth, 2015-11-23:
@a4jp-com -- AFAICT, the RAMBo is nothing more than a Mega2560. I suggest that you look at the MarlinDev wiki and obtain the build package for Marlin rather than attempting to use some other version.

Our standard builds make use of modified "cores" and you are likely to have issues unless you carefully mix your components.

AnHardt, 2015-11-23:
Hardware endstop update outside the stepper IRQ?

The IRQ frequency varies withe the speed of the motion.
Outside the IRQ we can not guarantee any timing.
To get the same precision of the timing, we have to check the endstops at least as often as inside the IRQ.
What is the possible win? 

During probing we want that precision, we are measuring a distance.
During homing we want that precision, we want to find a point.
During normal operation we do not need that precision but want to stop the move before the endstop is smashed. When moving fast, because of the mechanical inertia, this means, as fast as possible. 
Especially when moving fast, we are slow outside the IRQs.

Can the work to be done inside the IRQ be reduced, if we only want to stop?

When we take away all the stuff for CORE?? and DUAL_X_CARRIAGE the remaining structure is much more obvious.
```
if (TEST(out_bits, X_AXIS))   // stepping along -X axis if (TEST(out_bits, X_AXIS))   // stepping along -X axis (regular Cartesian bot and DELTAS!)
  {
            #if HAS_X_MIN
              UPDATE_ENDSTOP(X, MIN);
            #endif
  }
  else { // +direction
            #if HAS_X_MAX
              UPDATE_ENDSTOP(X, MAX);
            #endif
  }
for X, Y and Z axis. Only for z- we additionally ask for the z-probe
```
I don't think that can be speed up anyhow. In the best case, moving away from all endstops we check no endstop, in the worst case all. If we'd drop the direction test we'd have to check always all endstops. The direction test at least looks much cheaper than the UPDATE_ENDSTOP()_macro.

Let's have a look into `UPDATE_ENDSTOP()`
```
  #define UPDATE_ENDSTOP(AXIS,MINMAX) \
    SET_ENDSTOP_BIT(AXIS, MINMAX); \
    if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX))  && (current_block->steps[_AXIS(AXIS)] > 0)) { \
      endstops_trigsteps[_AXIS(AXIS)] = count_position[_AXIS(AXIS)]; \
      _ENDSTOP_HIT(AXIS); \
      step_events_completed = current_block->step_event_count; \
    }
```
Hard to say what is going on. Let's expand some macros for AXIS = X and MINMAX = MIN.
Where X_MIN is now a bitmask like 00001000.
```
    if(READ(X_MIN_PIN) ^ X_MIN_ENDSTOP_INVERTING) current_endstop_bits |=  X_MIN:
    if (current_endstop_bits&X_MIN ==  old_endstop_bits&X_MIN)  && (current_block->steps[X] > 0)) { 
      endstops_trigsteps[X] = count_position[X]; 
      endstop_hit_bits |= X_MIN; 
      step_events_completed = current_block->step_event_count; 
    }
...
old_endstop_bits = current_endstop_bits;
```
endstop_hit_bits is the communication to the rest of the world outside the IRQ.
current_endstop_bits and old_endstop_bits build a minimal debouncer.
if(current_block->steps[X] > 0)) = are we still moving?
(step_events_completed = current_block->step_event_count; ) = stop the move!

And hura! (endstops_trigsteps[X] = count_position[X]; ) = the part where we store the position and what could be omitted.

That's disappointing. There is about noting to improve.  :-) 

So why do we get complains about hardware endstops are CPU intensive?
I think this is not a general problem, but it is for CORE?? printers.
There we have additional tests like
```
if ((current_block->steps[A_AXIS] != current_block->steps[B_AXIS]) || (TEST(out_bits, A_AXIS) == TEST(out_bits, B_AXIS))) {
```
before every direction test. That is at least 2-3 times more work than the direction test alone. That means the best case scenario is lasting at least 2-3 times longer.

a4jp-com, 2015-11-23:
RAMBo setup is a bit different to the Mega2560. Even though it is kind of the same, if I set the board to Mega2560 it works, but the LCD doesn't. I have to use the RAMBo setup to get the LCD working.

Wackerbarth, 2015-11-23:
A few comments, “homing” is actually a series of two operations. The first is a rapid movement to (or near) the end stop. The second is to back off and SLOWLY re-approach. This second operation is just like “probing” and, I do agree, does need accuracy.

In reference to  “crash detection”, I agree that you have stated the reason for rapid response. However, I do not see anything more than your assertion that that rapidity MUST be "as fast as possible”.

I’ve worked on massive machinery that would never even consider having its “crash detection” as a part of the controller. When those limit switches are hit, they throw the breaker on the mains.

Every limit switch has SOME overshoot, even the poorly designed ones where the mechanism runs directly into the switch rather than just deflecting a control arm.

The question becomes one of relative scales. How much extra time/steps can you wait and still stop in time. Obviously that depends on mass, speed, and stopping distance.

But, just because you could, doesn’t mean that you need to apply maximum braking.
So, I renew my query concerning just how fast we really NEED to react.

Wackerbarth, 2015-11-23:
@a4jp-com -- Have you tried the MARLIN Mega2560 setup (not the Arduino one)?
I suspect that your issue might be with the libraries rather than the "pins".  In any case, if our setup doesn't work, I need to know EXACTLY where it differs from yours so that I can fix ours to work.

Roxy-3DPrintBoard, 2015-11-23:
> But, just because you could, doesn’t mean that you need to apply maximum braking.
So, I renew my query concerning just how fast we really NEED to react.

This answer doesn't really help, but it is probably accurate:  We need to stop before the switch bottoms out and things crash.   The stepper pulses are relatively slow compared to how quickly we can get into the ISR and start doing things.   

~~~~
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80, 80, 80, 93}
#define DEFAULT_MAX_FEEDRATE          {1500, 1500, 1500, 45}    // (mm/sec)              
~~~~

Does this imply a max step rate of 80*1500 steps / sec ?   Because that is 8 micro-seconds of time before the next step happens.    My micro-switches have at least 1/32" of travel.  So that is going to be 3 or 4 steps.    In my case, if we could stop the motors within 2 or 3 steps I wouldn't have a crash.

AnHardt, 2015-11-23:
@Wackerbarth 
>A few comments, “homing” is actually a series of two operations. The first is a rapid movement to (or near) the end stop. The second is to back off and SLOWLY re-approach. This second operation is just like “probing” and, I do agree, does need accuracy.

You are right. It should be that way, it is this way for Cartesian printer, but not for delta printers - what is not perfect, but works - for now.

>In reference to  “crash detection”, I agree that you have stated the reason for rapid response. However, I do not see anything more than your assertion that that rapidity MUST be "as fast as possible”.

A typical Mini Kossel Kit Endstop Mount:
[image]
Maximum braking distance about a mm from trigger to bang. Typically 80steps/mm, 500mm/s. (Values from our KosselMini/Configuration.h)
s = v^2/(2*a) >= a = v^2/(2*s).
500^2/(2*1) = 250,000/2 = 125,000mm/s^2. Assuming we can break with the same deceleration than during accelerate that is more than factor 100 to much (DEFAULT_MAX_ACCELERATION 9000). As fast as possible is not fast enough. Hopefully we can decelerate much better than accelerate. But our A is adjusted to a value where the steppers just do not skip. Full stop breaking means just skipping. So likely we talk about decelerating as much as possible before the bang.

>I’ve worked on massive machinery that would never even consider having its “crash detection” as a part of the controller. When those limit switches are hit, they throw the breaker on the mains.

Good idea. But here we talk about RepRap (crap).

AnHardt, 2015-11-23:
@Roxy-3DPrintBoard 
You are calculating the distance until the breaking begins, not until the stop.

Roxy-3DPrintBoard, 2015-11-23:
> Assuming we can break with the same deceleration than during accelerate that is more than factor 100 to much (DEFAULT_MAX_ACCELERATION 9000). As fast as possible is not fast enough. Hopefully we can [...]

This good thinking...    It would be nice to not lose position but the most important thing is to slow down the movement as much as possible before it crashes.

Is the force the stepper motors putting out depend much about where the motor is?  For example, my guess is the micro-steps are not as powerful as running the motors in the normal stepping modes.   And what happens if we lose a step?   Does the next step start pulling the motor forward instead of trying to slow down the movement?

AnHardt, 2015-11-23:
@Roxy-3DPrintBoard 
As far as i remember the difference in torque between full and half step is factor sqtr(2) at least less thean factor 2. How ever, we non't know where the next full step is.
If we hit a hardware endstop unexpected, we need to rehome. Remember - we are stopping the move by claiming it has arrived at its target. We did less steps than we claim we did have done. If we do not deal with a planned endstop move, our current and destination variables relate to a completely different move, at the other end of the plannerbuffer. We have no way for a correction - or we'll need something like a complete rewrite of the planner.

AnHardt, 2015-11-24:
Is there any point in the solutions sections above where you agree how to solve the problem?
What for do you want to be the first attempt of a PR?

Roxy-3DPrintBoard, 2015-11-24:
I think everybody agrees that if an end stop gets hit accidentally, the right thing to do is stop the movement before a 'crash' happens.   But what is not clear in my mind is how that is done.  Do we just lock the stepper motors and hope they hold position?   That seems almost as bad as a 'crash'.  It certainly will provide a big jerk to moving axis.   Is it possible to slow the motor down to a stop within the extra travel space the switch has?   That would be the best solution if it was possible but can you imagine the user getting those #define's setup right???

Wackerbarth, 2015-11-24:
Given that many of the inexpensive end stops directly depress the plunger rather than depressing a lever (which would allow the actuator to pass by the actual switch, and travel well beyond the trigger point), and also that most users try to push the limits to get a maximal build volume, in most cases, there isn't really enough space the gracefully decelerate the actuator. I believe that you mentioned something on the order of 1 mm. And I have seen even less.

Therefore, jerk, or not, I think that the right strategy is to lock the stepper rather than trying to achieve a controller braking.

Besides, any part being constructed is already damaged since the triggering indicates that the mechanism is not in its expected position. The efforts to reduce jerk are primarily aimed at creating smooth extrusions, properly placed. The printer parts that we are dealing with should not have enough mass to be damaged by an occasional jerky stop.

Roxy-3DPrintBoard, 2015-11-24:
I think I agree with what you said.   It would be nice to slow down gracefully but that requires a lot of assumptions.

Wackerbarth, 2015-11-24:
"EDITED: Not a good idea. If we have no current, we have no, or low, holding torque with steppers." --
For stepper motors, I perhaps. The motors driving the equipment to which I referred used regular motors and, in addition to disconnecting the mains, there was some mechanism that shunted the motor (turning it into a generator attempting to drive a power dissipating load).

These devices also had quite a bit of available travel beyond the safety stop.

But my whole point is that on industrial equipment, the safety stop mechanism is separate from the controller which is driving the machine during normal operations.

Roxy-3DPrintBoard, 2015-11-24:
I think I agree with what you said.   It would be nice to slow down gracefully but that requires a lot of assumptions.

@AnHardt 
> Is there any point in the solutions sections above where you agree how to solve the problem?
> What for do you want to be the first attempt for a PR?

Can you live with locking the motors?

AnHardt, 2015-11-25:
It's surprisingly hard to find well designed hardware endstop in the RepRap world.
That's how the crap works:
[image]
That's how better designed hardware endstop should work:
[image]
This way you can not crush the endstops and have some more way to deaccelerate. But it needs some more adjustment to get the distances right.
However we have to live with, how the machines do work now. And they do work - mostly. They work because we homing slowly, because the masses are not that high, because the machines are not that stiff, because we use software endstops and usually don't loose steps.
Unfortunately it needs some expirience to set up the machine until that works. So users building their fist machine have the highest risk to break something. That risk can decreased by making the hardware endstops work as early as possible. How ever.

Actually we break by stopping the stepper pulses immediately. That stops the magnetic field from rotating but keeps it on. The permanent magnets and the electric magnets still attract each other, an considerable breaking torque, in about the magnitude of the holding torque is developed. If the inertia of the rotor and the connected masses is to high, the holding torque is exceeded, we will lose steps. The stepper produces AC now, but the stepper driver still tries to keep the current constant  In that case the graph (breaking torque/angle) will have a form between sine and saw-tooth, but still considerable above zero, The exact form depends on the decay settings of the stepper driver, where we normally do not have influence on - at least not on the fly. (still simplified) We could think about turning the field that wa,y the breaking torque is always close to its max. but don't have sensors for that. If we try to calculate the slow down and have only a tiny error, the wave we talked about above will only have a longer period, but not a higher average breaking torque.
I guess just stopping the steps is the best we can do. And best of all is - we already do that.

My plan is to not alter the IRQ at all and wait for an endstop hit in checkHitEndstops() as before. There ill try to empty the planner buffer, set a flag and throw an error. Maybe i have to kill a already running next move. That flag will be checked in the planner to not allow new moves, until the flag is reset at the start of a G28.

a4jp-com, 2015-11-25:
@Wackerbarth I'll check as soon as the machine gets back.

Roxy-3DPrintBoard, 2015-11-25:
> That flag will be checked in the planner to not allow new moves, until the flag is reset at the start of a G28.

Perhaps this is too much complexity.  But shouldn't it be safe to move away from the end stop?   Position will have been lost.   But it is nice to be able to back away from the end stop manually.

Wackerbarth, 2015-11-25:
@Roxy-3DPrintBoard -- What are you suggesting?  We are discussing the case where an actuator has "crashed" because some steps were lost, etc. (Software SHOULD have prevented any proposed move to an unreachable position)  Andreas is proposing to stop movement, issue an error, and ignore commands until we are given a G28. (IMHO, a very reasonable response to the failure)

Personally, I suggest that we return to the "just powered up" state where we assume that we know nothing about the position of any actuator.

Roxy-3DPrintBoard, 2015-11-25:
> What are you suggesting? We are discussing the case where an actuator has "crashed" because some steps were lost, etc.

Well maybe I'm confused.  I thought we were talking about the case where an endstop switch got triggered.  That seems different than the actuator having crashed.   What I was trying to say is it might be too much extra complexity, but it would seem to be safe to back away from the end stop switch and probably even in a crashed state, it should be ok to back away from the end stop switch.  After all, that is what the G28 is going to do.  It is going to back away from the end stop switch.

Wackerbarth, 2015-11-25:
"where an endstop switch got triggered. That seems different than the actuator having crashed." --
Well, ... triggered UNEXPECTEDLY.  But I don't see a distinction between the two.

What I fail to see is why there is any value in attempting to accept ANY motion commands other than some kind of "reset".

Roxy-3DPrintBoard, 2015-11-25:
For me, the value would be when trying to setup the printer.   I can move the nozzle back and forth to find the total allowed travel.  If I bump into an endstop, I can manually back away and know the limits.

Wackerbarth, 2015-11-25:
Sorry, I'm failing to see how you are using this. Are you manually moving the nozzle? If so, how do you know what the step count is which corresponds to your position?  If not, why is starting with the G28 an issue?

Roxy-3DPrintBoard, 2015-11-25:
Don't get me wrong...  I'm not pushing real hard for this.   But when I setup my first printer, I used PronterFace to move the nozzle around.   I needed to do that to find out exactly how big the bed was because both Slic3r and PronterFace wanted to know those numbers.

All's I'm saying is it is nicer (or more user friendly) to let somebody bump up against and end stop switch and then allow them to manually back away and bring the nozzle somewhere else.   With PronterFace's target in the upper left corner you can move 100, 10, 1 and .1 mm.  So it is pretty easy to keep track of where you are even without an LCD Display.

Wackerbarth, 2015-11-25:
I do a similar process using Repetier-Host. The program even tallies the position for me.
But each time I hit the stop, I always re-establish my reference point by doing a G28.

AnHardt, 2015-11-25:
>@Wackerbarth

    A few comments, “homing” is actually a series of two operations. The first is a rapid movement to (or 
    near) the end stop. The second is to back off and SLOWLY re-approach. This second operation is 
    just   like “probing” and, I do agree, does need accuracy.

>You are right. It should be that way, it is this way for Cartesian printer, but not for delta printers - what is not perfect, but works - for now.

Sorry messed that up. It's the __probing__ only done once for deltas.

AnHardt, 2015-11-25:
Did i mention before the the output of `checkHitEndstops()` is crap? 
a. The coordinates are in a format interpreted by RepetierHost to update the display.
b. For Cartesian printers the calculated position where the move was stopped is displayed. Tat's not true. We are at one of the min or max positions. It may have some value for debugging, but RH should not interpret that. We also can not claim to be anywhere, because we do not have the information if we hit a min or max endstop. (ok - we now, if we have only one)
For deltas we get a wired intermix between tower and Cartesian coordinates. For CORE machines the values are completely meaningless (to me).

I'll replace the ':' in the output with a '=' to prevent parsebility by RH.

AnHardt, 2015-11-25:
Did i mention before we don't back up from the endstops when homing? Only for deltas we back up by `endstop_adj[axis]` (wold be nice to have that configurable in the config). In all other cases we set the position for the software endstop with a triggered hardware endstop. That means returning to there (what the software endstop will allow) triggers the hardware endstop again!
I suggest to back up by `home_bump_mm(axis)`. WARNING: That will decrease the max travel distance by `home_bump_mm(axis)`. For deltas we only loose in z, for Cartesian printers we have to decrease the difference between `?_MIN_POS` and `?_MIN_POS`. Old configs have to be changed, else they may crash at the other side. (Not our defaults, but that for real machines)
Actually  `home_bump_mm(axis)` is a bit more than we need. The value of  `home_bump_mm(axis)` can be optimized for the much faster first bump. If then used after the slower bump, we have a bit of safety margin, we could optimize away, with a second set of distances.
But i guess `home_bump_mm(axis)` is just right.

AnHardt, 2015-11-25:
Setting up the min and max positions for a Cartesian not CORE printer:
Set the values in the hosts much to large for this test. They may have their own software endstops.
Set the  `?_MIN_POS`,  `?_MAX_POS` values in the Marlin config to -+ (max travel + a bit) to fool the software endstops.
Shut off the machine, or at least the motors. Move the axis, by hand, to the point most away from the endstop, just touching the other side. Switch on the machine or reboot. All axes coordinates are now 0 for Marlin. Send a "G0 X(maxtravel+a bit) F(HOMING_FEEDRATE_for_that_axis / HOMING_BUMP_DIVISOR)" in direction to the endstop. Take a look into your console window.The abs(value) for the endstop hit position, reduced by  `home_bump_mm(axis)` is your exact maximum travel distance. Calculate your  `?_MIN_POS`,  `?_MAX_POS` entry for the config according to that number. Set the same numbers in the host programs.

That also works for z on deltas, if you already know the mid of your bed. You can also see what tower triggered first. Maybe that's useful to find the mid.

Roxy-3DPrintBoard, 2015-11-25:
> For deltas we get a wired intermix between tower and Cartesian coordinates. For CORE machines the values are completely meaningless (to me).

We might be able to use  void calculate_delta(float cartesian[3])  to map the current coordinates to something meaningful for Delta's.  We would need to give Stepper.cpp access to the global current_position[] array.   But here is the thing...  If a Delta hits its endstops, it is pretty much up at the top.  These are the Max endstops and not Min endstops.  I don't think any position information will get printed.

~~~~
    if (endstop_hit_bits & BIT(X_MIN)) {
      SERIAL_ECHOPAIR(" X:", (float)endstops_trigsteps[X_AXIS] / axis_steps_per_unit[X_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "X");
    }
    if (endstop_hit_bits & BIT(Y_MIN)) {
      SERIAL_ECHOPAIR(" Y:", (float)endstops_trigsteps[Y_AXIS] / axis_steps_per_unit[Y_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "Y");
    }
    if (endstop_hit_bits & BIT(Z_MIN)) {
      SERIAL_ECHOPAIR(" Z:", (float)endstops_trigsteps[Z_AXIS] / axis_steps_per_unit[Z_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "Z");
~~~~
In the case of the Z-Min-Probe the coordinates reported are going to be bogus.  So maybe we want to use void calculate_delta(float cartesian[3])  to get something meaningful.  But I'm not sure it matters.  Maybe it would be better to just not attempt to print the coordinates for Delta's ???
~~~~
   #if ENABLED(Z_MIN_PROBE_ENDSTOP)
      if (endstop_hit_bits & BIT(Z_MIN_PROBE)) {
        SERIAL_ECHOPAIR(" Z_MIN_PROBE:", (float)endstops_trigsteps[Z_AXIS] / axis_steps_per_unit[Z_AXIS]);
        LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "ZP");
      }
    #endif
~~~~

> I'll replace the ':' in the output with a '=' to prevent parsebility by RH.

Will that break Repetier Host?   If it breaks something, we may not want to do that change.  Who is our Repetier Host contact?  Maybe we should flag this post for them to see???

AnHardt, 2015-11-25:
> Will that break Repetier Host? If it breaks something, we may not want to do that change. Who is our Repetier Host contact? Maybe we should flag this post for them to see???

No. That's already tested. It will not break RB. It will prevent it from breaking. 
In the normal case RH tracks the sent g-codes and updates its positions accordingly. So during a program it is ahead of the nozzle anyway. Only when moving manual and not to fast the RH display is the same as the real position. The error distance is the content of the planner buffer. Now sending one corrected coordinate for one axis, to correct a display being several moves ahead is rubbish. It's better than if RH does not understand our output in this case. See my last posting to see what the output is good for.

Please have again a look into https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/stepper.cpp#L302
```
#define _ENDSTOP_HIT(AXIS) endstop_hit_bits |= BIT(_ENDSTOP(AXIS, MIN))
```
MIN is not a parameter of the here defined macro. It just sets a MIN flag - ever - never a MAX.
That means we get the information that a MIN __OR__ MAX endstop was hit, by a MIN flag. For the current calculation in `checkHitEndstops()` the information if it was a MIN or MAX is completely irrelevant.

Right now discovered why the hosts keeps on sending moves. We send:
```
echo:endstops hit:  X:-0.01
```
with `SERIAL_ECHO_START;` instead of `SERIAL_ERROR_START;`
```
Error:endstops hit:  X:-0.01
```
what would stop the host.

>We might be able to use void calculate_delta(float cartesian[3]) to map the current coordinates to something meaningful for Delta's.

I believe that is to much. From all the data we have in Marlin we can only calculate where we assumed to be when the stop was triggered. If we did not fool the hosts and our own software endstops and get a hit on a hardware endstop, i can grantee we are not there where we assumed.
For planned hardware endstop hits we already get all the needed information - when done right. For unplanned hits all we can calculate is rubbish. The only relevant information we get, is that we don't know where we are. Even the information what endstop triggered only indicates that we are close to it. You can't say 'we hit that endstop, so we are homed there' because we don't have the information about the speed and how much overshot we had.

Roxy-3DPrintBoard, 2015-11-25:
Thank You for the detailed explanation!

AnHardt, 2015-11-25:
>In the case of the Z-Min-Probe the coordinates reported are going to be bogus.

For a planned endstop move we do not get this message. We carefully avoid that. More on the side of too careful:
```
    // Move away from the endstop by the axis HOME_BUMP_MM
    enable_endstops(false); // Disable endstops while moving away
    destination[axis] = -home_bump_mm(axis) * axis_home_dir;
    line_to_destination();
    st_synchronize();
```
We now know `enable_endstops(false); ` when moving away is not really needed because when moving away the endstop is not checked in the IRQ

In fact a rigid mounted inductive probe currently gives false positives her and a normal Allen Key probe as well. That's the reason why defining `ENDSTOPS_ONLY_FOR_HOMING` is absolutely mandatory with this probes. That's also the reason why the probes should only be tested when deployed. I'm quite sure we talked about that before.

Roxy-3DPrintBoard, 2015-11-25:
> In fact a rigid mounted inductive probe currently gives false positives her and a normal Allen Key probe as well. That's the reason why defining ENDSTOPS_ONLY_FOR_HOMING is absolutely mandatory with [...]

The Allen Key Probe stuff that was killing me was in void update_endstops() in Stepper.cpp.  That is where I added those lines that had the redundant test. 
~~~~
  if (z_probe_is_stowed==false)
~~~~
But there was nothing to sensitize it to ENDSTOPS_ONLY_FOR_HOMING.  Perhaps that would have kept me from having problems?   Maybe that is a better way to fix my problem than what I did below?

Your thoughts?

~~~~
          #else // !Z_DUAL_ENDSTOPS
	    if (z_probe_is_stowed==false)
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
          {
            endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
~~~~

AnHardt, 2015-11-26:
My mind has not changed.  https://github.com/MarlinFirmware/MarlinDev/issues/285#issuecomment-158824673

~~~~
          #else // !Z_DUAL_ENDSTOPS
             UPDATE_ENDSTOP(Z, MIN); // This is looking for the Z_MIN_PIN only
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
          if (z_probe_is_stowed==false) {
            UPDATE_ENDSTOP(Z, MIN_PROBE); // This is looking for the Z_ MIN_PROBE_PIN only
            if (TEST_ENDSTOP(Z_MIN_PROBE)) {
              endstop_hit_bits |= BIT(Z_MIN_PROBE);
            }
         }
       #endif
~~~~
Don't  make any endstop dependend on (z_probe_is_stowed). Don't make code redundant.

https://github.com/MarlinFirmware/MarlinDev/issues/285#issuecomment-158795313
impies: Don't connect a probe to a Z_MIN_PIN! If you want to use your hardwares Z_MIN_PIN connector to connect your probe,  comment or #undefine your  Z_MIN_PIN and use its number.
```
// #define Z_MIN_PIN 18
...
#undefine Z_MIN_PIN
#define Z_MIN_PROBE_PIN 18
...
#define Z_MIN_PROBE_ENDSTOP
```

To be done: set `z_probe_is_stowed` with all sowing and deploying probes. That is the minimal thing what has to be done with all probes, even with that probes what did not need any handling before.

Likely `#if ENABLED(Z_MIN_PROBE_ENDSTOP)` is wrong then. Have to reeread the description in the config

`ENDSTOPS_ONLY_FOR_HOMING` is a workaround - not the solution. `ENDSTOPS_ONLY_FOR_HOMING` shuts down __all__ endstops and probes after homing. Only for probing moves they are temporarily activated again. When our probe-testing now relies on  `z_probe_is_stowed` we can safely let the endstops on an __win__ a bit of safety for the more often loosing steps __x__ and __y__ axis. (provided some other errors are fixed where ` ENDSTOPS_ONLY_FOR_HOMING` is the workaround for).

AnHardt, 2015-11-26:
>The Allen Key Probe stuff that was killing me

How did the failure look exactly?
No movement in -z before homing?
Problems when deploying the Allen key?
Problems when stowing the Allen Key?
No movement in -z after Probing?
Any console messages? When?

If you told us before, I don't remember. And you are writing to much about your new delta, to search for that info.

AnHardt, 2015-11-26:
If your code worked for you, you have connected the probe to the Z_MIN_PIN. You cold have omitted the changes around the Z_MIN_PROBE_PIN.

Roxy-3DPrintBoard, 2015-11-26:
> My mind has not changed. #285 (comment)

OK.  Good!

> To be done: set z_probe_is_stowed with all sowing and deploying probes. That is the minimal thing what has to be done with all probes, even with that probes what did not need any handling before.

I think we are on the same page.  Right now I have:
~~~~
  static void deploy_z_probe() {
...
  z_probe_is_stowed = false;
  }

  static void deploy_z_probe() {
...
  z_probe_is_stowed = false;
  }
~~~~
>Likely #if ENABLED(Z_MIN_PROBE_ENDSTOP) is wrong then. Have to reeread the description in the config
>ENDSTOPS_ONLY_FOR_HOMING is a workaround - not the solution. >ENDSTOPS_ONLY_FOR_HOMING shuts down all endstops and probes after homing. Only for probing moves they are temporarily activated again.

This might be close to true...  But loop() does a call checkHitEndstops(); and that logic is an exception to what you are saying.

> If your code worked for you, you have connected the probe to the Z_MIN_PIN. You cold have omitted the changes around the Z_MIN_PROBE_PIN.

Yes, after I got it working I got advice telling me that I should not have declared myself as an Allen Key probe and done things differently.   

Mostly the reason for that last post was it seems Delta's pretty much just use Endstops for homing.  Right?   So what I was kind of thinking was if we are going to have an ENDSTOPS_ONLY_FOR_HOMING it might be good if took care of everything.   Once again, I'm not pushing hard for anything.  I'm just saying that is kind of where my mind is wandering to.

[31 more replies]

closed 2016-05-25 by boelle