MarlinFirmware/MarlinDev #290: Using printer like boards wackerbarth's install

iplayfast, 2015-11-24:
Setting up arduino Preferences, additional boards managers urls with 
https://raw.githubusercontent.com/Wackerbarth/MarlinDev/platform/package_marlinfirmware_index.json
will allow boards to be installed.
However it gets a crc error so install fails.

iplayfast, 2015-11-24:
Just found proper json file at 
    https://raw.githubusercontent.com/MarlinFirmware/MarlinDev/platform/package_marlinfirmware_index.json

Same result. CRC doesn't match file is corrupted.

Wackerbarth, 2015-11-24:
I just downloaded it without any problem.

Make sure that your .json file specifies version 1.0.2

I just flushed everything from my Arduino15 support folder (where the IDE stores all of its preferences and downloaded pieces) and reinstalled using the board manager without any problem.

iplayfast, 2015-11-24:
Yup deleting .arduino15 and starting fresh fixed it. Sorry bout that.

closed 2015-11-24 by iplayfast

Wackerbarth, 2015-11-25:
I think think that I see a problem in the IDE. When there is a checksum error, the file needs to be flushed and reloaded.

iplayfast, 2015-11-25:
I think you've got it.