MarlinFirmware/MarlinDev #292: enabled Auto-level makes entire printer move very slowly - before issueing G29 command

dintid, 2015-11-29:
I posted the problem on RepRap Forums in the hope someone could help http://forums.reprap.org/read.php?146,586527
Spent most of the time with people suggesting things to do without them having read my description..or hinted/Said I did something wrong.. Again without having read my description.

Here is the entire issue copied from above place:

I have Googled up and down, and searched on the forums and on Github Issues, and havn't found a similar thread, so I hope I'm not asking something that has allready been solved.

Printer configuration
1. Printer type: UM2 clone type (my source) - I have made lots of changes, but that is not relevant for this issue.
2. Controller: Arduino Mega 2560 R3 + Ramps 1.4 -- Modded to 24v
3.IR-Sensor: I'm using DC42's mini IR Sensor for my Z-height sensor.
4. Firmware: Release Candidate -- Marlin 1.1.0-RC2 - 29 September 2015
5.My firmware: You can see my firmware here on dropbox - I have not setup Z-offset and other Things for printing, as I'm not that far yet.

Problem
1.When Auto-level is disabled everything Works fine.. printer moves fast around and homes as expected -> to the rear left. Can see video here [youtu.be]
2.When Auto-level is enabled everythigng slows Down. I can still move it around and homing Works, but it is extremely slow. Especially when doing small moves (10mm) and especially on the X axis. Y axis is also very slow though. Can see video here: [youtu.be]
3.The issue is very easy to see, but I have no idea how to solve it?

What I've tried:
1.I have tried enabling/disabling my Mini OLED display. It uses I2C and I thought it might cause problems. No change though.
2.I have tried enabling/disabling EEPROM_SETTINGS. No change.
3.I have tried enabling/disabling DISABLE_MAX_ENDSTOPS - this was listed in an old Github issue, but no change.
4.I have tried enabling/disabling min/max-software_endstops - this was listed in an old Github issue, but no change.
5.I have tried enabling/disabling ENDSTOPS_ONLY_FOR_HOMING - no change
6.I have tried enabling/disabling Z_SAFE_HOMING - no change.
7.Tried setting same HOMING_FEEDRATE as XY_TRAVEL_SPEED (speed between probes) - no change
8.Tired upping Z DEFAULT_MAX_FEEDRATE from 45 to 300mm/sec to test for Z axis-movement limiation - no change

Thesis:
 Firmware is corrupt.
 Derived from the Issues on Github it seems a Z-auto-leve issue was introduced with the merging of the Delta printers in march/april 2015
 At best there is some "hidden" speed settings after using G29 which I just can't find in the firmware?

 I really hope someone has some idea, as I have no clue how to proceed!

KiteLab, 2015-11-29:
Workaround: Set your speed manually with a g-code. After sending an 'F' the speed is at your will.

Reason: `setup_for_endstop_move()` is used to save the feed rate, when such a move is planed. `setup_for_endstop_move()` also activates the hardware endstops. `setup_for_endstop_move()` is made for making it's job in one single level, but is used nested under some circumstances

dintid, 2015-11-29:
I didn't undertand any of that really. Aside from you say I have to define movespeed manually always? 
Why is that? It forgets settings when auto-level is enabled, or?

Problem is the entire printer gets SLOW. but at a variable level.. when moving X axis towards maximum it gets progressively slower the farther I go. After 3 times 10mm it hardly moves at all.

My endstops are only enabled for homing.

KiteLab, 2015-12-03:
In several places in Marlin the axis number is limited to 4. You can configure it with less than 4 but not with more axis! Extruders with different steps/mm are not supported.

KiteLab, 2015-12-03:
The workhorse in Marlin, the stepper interrupt (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/stepper.cpp#L601) cant handle more than one extruder (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/stepper.cpp#L669) - for example.

Wackerbarth, 2015-12-03:
I would say "active extruder". We are able to switch between various nozzles to be able to handle situations such as a separate plastic for the support material.

dintid, 2015-12-03:
kiteLab you Can set numaxes to 5 but it really doesnt pertain to my problem. Or at least not to my situation, as I only have it configured with 1 extruder. 
Why mention multiple extruders here at all?

a4jp-com, 2015-12-04:
Got a link to the IR-Sensor?

KiteLab, 2015-12-04:
@dintid 
You can set the number of axis to 20 if you really want, but they are ether not handled, like in the stepper interrupt, or worse, you could miss some array boundary, and write somewhere in the RAM - what could cause any error.
At first you are complaining about, no one is reading your provided materials, than you claim that's not the stuff what causes the error, my config/code is different.
With such a rare error i'd think about a rare (daring) configuration - what num_axis = 5 is.
If your code is changed now - pleas provide it - and don't waste our time.

@a4jp-com 
If you find a link in the top posting - please follow it, read it, and don't ask already answered questions.

@Wackerbarth 
Just to remember. (not related to this issue)
One of the many conceptual problems in Marlin is, that we do use the term 'extruder'  and do not distinguish between 'hotend' (has a thermometer and a heater) and the 'cold end' (has a stepper). In Marlin they are always one unit. That makes the handling of Cyclops, Chimera, Kraken a bit difficult (Workaround: Use the same parameters for the heat at all 'extruders'). Our limit to use only one axis for all extruders makes mixing extruders, where more than one stepper is used at the same time, impossible.

a4jp-com, 2015-12-04:
For some reason the posts weren't loading properly but I reinstalled Chrome and everything is working now. Sorry about that.

dintid, 2015-12-04:
@KiteLab I am not complaining? I said on RepRap forum that I disliked how people jumped to conclusion without having read the case I had spent a lot of time trying to outline.
You are doing the same - jumping to conclusions without asking for more information - you ASSUME I have more than 4 axes.. ENQUIRE about it before you state so.

I am deepbly confused by your entire line of reasoning?

I'm not CLAIMING anything - YOU are claiming something about my setup.

I'm just VERY confused about yoru reasoning. I have only set up 4 axis.. so teliing me the error is caused by me using more than 4 axes is just plain silly.

The wasting time is not me.. seriously. Ask for information before jumping to conclusions. 

I HAD very basic setup.. Only setup temperature-sensors, steps pr mm, endstops and it worked fine.. as soon as I enabled auto-level it defuncted.
In Release Candidate - it Works fine in latest Stable.

KiteLab, 2015-12-04:
You already provided your Configuration.h.
Your firmware in your dropbox https://www.dropbox.com/sh/s4g6sxvk9kmuv2v/AABgcQ7Xw4bC1NiZ0Y1ulnrUa?dl=0
Configuration.h line 608:
```
// #define NUM_AXIS in marlin.h
```
Your Marlin.h line 186:
```
/**
 * The axis order in all axis related arrays is X, Y, Z, E1, E2
 */
#define NUM_AXIS 5 // MDN XYZ+2extruders
```
That's not our Marlin we support here.
Our RC line 186:
```
/**
 * The axis order in all axis related arrays is X, Y, Z, E
 */
#define NUM_AXIS 4
```

If the content of your dropbox is not the same, as what you are using to produce the error - you are wasting my time.

dintid, 2015-12-04:
That is ancient... and yes, I had 5 axes in that one.

However,  as I said, I have tried many since that where only the basics was setup.

Wackerbarth, 2015-12-04:
@dintid -- Please! If you offer some code for us to examine, let that be the code that actually produces the condition to which you refer.

I think that @KiteLab is correct in complaining that the code offered does not reflect the issue that you raise.  You shouldn't leave it to us to guess what you are actually doing. When you offer code, res ipsa loquitur.

dintid, 2015-12-05:
@Wackerbarth and @KiteLab  you are correct that I made ONE mistake... The link to Dropbox was not intented to have been included here....
This post was originally posted on RepRap Forum, and the link was valid at first.... as I posted I tried many different configuration and did not make a seperate upload for each... I could have made a subfolder for each.. maybe I should. Would anyone even go through each?

In any case, I only had 5 Axes defined in the very first example, as the rest were widdled Down versions.

That said, @KiteLab is using a VERY aggressive tone. I appreciate effort to help, but please. Do so in a more friendly manner... maybe write response down somewhere else, and review an hour later before posting it.

This often seen aggressive, very elitist approach is what will kill RepRap as new people are frightened away by the uncompromising tone. 
Try looking through comments made to my total newbie guide and you will notice how normal people feel regarding getting started on this http://www.instructables.com/id/Complete-newbie-step-by-step-3D-printer-with-all-p/

@KiteLab and @KiteLab My post was intentended to DESCRIBE my problem and try to describe all the steps I had taken to try and fix it and narrow it Down.. with ALL the different configurations the conditions (slow axes after auto-level enabled in firmware) was present.

I had hoped to get input on possible causes which I could try and then post the result.

I did not see it as realistic to post 6 different firmwares for review... if I should have, I apologize for that, and I have learned to do that if I encounter any problems again. And do so when I get my stuff unpacked and take up troubleshooting this issue again...

But please @KiteLab try to be more forthcoming and less aggressive. You have a perfect opportunity to educate me and other people in better ways to do things regarding firmware troubleshooting, but instead you thunder threats and are in general being rather derogatory towards me (wasting my time being used more than once).

I see the same trend from established people towards new/er people in RepRap community in general, and it causes a lot of people to stay away, and buy closed firmware-printers instead.

nitewing76, 2015-12-18:
I realize this is 2 weeks old, but.....

@dintid It seems something like this happened to me a long time ago.  It has been so long ago that I can't remember exactly what was wrong, but I have discovered 2 easy fixes for most non-mechanical problems which "unusual and rare"......usually I have accidently placed a stray character somewhere or inverted 2 characters. To locate and fix this, I use the "Compare" plugin for Notepad++...comparing my version with the original version from the ZIP.  Or sometimes it is as simple as uploading the sample "eeprom_clear" sketch, then the "Blink" sketch as a quick way to make sure there are no communication problems, then I recompile+upload the firmware.

@KiteLab As a former U.S. Marine (meaning my definition of "too aggressive" is very broad/liberal), I also agree you comments are very overly aggressive. Not only to @dintid, but also @a4jp-com....who asked a simple question about a link, which was at the top of the page on RepRap forum...not here.  And since it had been over a since the OP posted the dropbox link on another forum, it would have been worth your "valuable time" to ask questions to confirm....versus making assumptions and choosing to waste your own time by treating two forums as one entity. Quick, easy, and efficient would had been to say, " @dintid Please verify and ensure that your number of axis are set to 4 [Marlin.h line 186]. in your code linked on the RepRap forum, it says 5."  Excluding links and copied code....your 380 words vs. 27 words (both separately counted in MS Word). To assume your time is worth more than the time of another human...is arrogance. Is it so hard to communicate efficiently in a professional tone? Ofc not...and it saves time.

a4jp-com, 2015-12-19:
I'm cool. No offence taken at all.

Let's work together and make the software nicer and/or easier to use ^^

dintid, 2015-12-19:
I set it up again in RC3 an everything Works fine.

Regardin the number of axes to I had set to 5 initially: I did it as I read a post where a Company used it when changing tools with different steps. I had no idea this wasn't supported. Not like there is a supported/not-supported configurations guide anywhere :)

I changed it back to default though and it still didn't Work in RC2

closed 2016-04-20 by thinkyhead