MarlinFirmware/MarlinDev #295: Marlin Changelog

thawkins, 2015-12-02:
Would it be possible to maintain a change-log for Marlin, detailing what significant items where added/changed in each release, not every commit. I have a number of friends and colleagues that have complained that they never know what is new from release to release. Note exporting the commit log would achieve what is required. 

In the past i have pointed several people at the commit log, but for most casual users that is a bit beyond them and they struggle to understand what it means. 

all it needs is something like a changelog.md file in the root where large items and significant bug fixes are recorded.

Roxy-3DPrintBoard, 2015-12-02:
I think this would be valuable.   And it really is something we should discuss.

In my case, I would like to have the change-log actually be associated with the file set for the release.  If somebody downloads a .ZIP file of all the files, I would like the change-log to be one of the files in the set.   I would also like there to be a READ_ME.txt file that gives basic instructions for getting the release set up and built.  There may be other places that this same information should be kept.  But the intent is to make sure everything for a given release can pretty much be found within the release.

For sure, not everybody is going to agree with this thinking.   But maybe that is enough of a description that people can start discussing the pro's and con's of it and hopefully offer and even better idea.

Wackerbarth, 2015-12-02:
@Roxy-3DPrintBoard -- Re: "There may be other places that this same information should be kept. But the intent is to make sure everything for a given release can pretty much be found within the release."

I have a major problem with this. It is a direct contradiction of the DRY principal.
So, we really need to decide just how we want to keep the information, do it that way, do it well, and just reference that result in "the other places".

Frankly, from my viewpoint, the source distribution is one of those "other places". For there to be an entry in the README which directs the user to installation documentation is a good thing. But expecting the source code to have all of the instructions is not.

For example, in order to compile the code, you need an appropriate compiler.
I don't think that the details for obtaining and installing that compiler belong in the Marlin source code.
After all, those instructions apply to many other bodies of source code.

It is appropriate for the source to state REQUIREMENTS (for example, "Arduino IDE 1.6.6, or later").
But it is outside the scope of the Marlin SOURCE to discuss the details of actually installing an environment that meets those requirements.

Roxy-3DPrintBoard, 2015-12-02:
Well...   Mostly I want all of the information for a specific release easily available.   If that really was the case, there would not be the fear and concern that it won't be available.  And people like me would not feel the need to put it in a bunch of places.

But right now, we don't have all the information to build and a change long of the revisions available.  So people like me want to put it all in an easy to find place like the source files.

This is an 'easy' problem to fix.   If we really had Revision Specific Build Instructions available and a detailed Change Log for each Release, I bet we can find a single place to store it.

Wackerbarth, 2015-12-02:
I suggest that you look at the information that I started building in the MarlinDev wiki (KiteLab, et.al. have been pushing for having documentation in a wiki rather than a Documentation folder of text files)

It would be nice if others would contribute to the effort.

Roxy-3DPrintBoard, 2015-12-02:
> I suggest that you look at the information that I started building in the MarlinDev wiki (KiteLab, et.al. have been pushing for having documentation in a wiki rather than a Documentation folder of text files)

OK!   Very good start!   Maybe we can address Thawkin's point with a section with a Change-Log section?     

> It would be nice if others would contribute to the effort.

Agreed.   Most developer types do not like doing documentation.   We might have to assign some of the areas to different individuals to get them done.    I think you are going to be on the hook to do the **Compile and Install the Firmware** section.   :relaxed:

Wackerbarth, 2015-12-02:
"the Compile and Install the Firmware section" -- Actually, it might be better if someone less involved in the details start it . Short version -- the Arduino sketch is in the examples as it is for all properly constructed libraries. (You might also look at GRBL, etc. Their code is based on the same organizational design)  Then I can come in and add information about some of the fancier things that are possible (like adding you private custom builds to the IDE menu).

thawkins, 2015-12-02:
While i think the wiki is a good resource, the original intent of this ticket was to discusss a changelog, people want to know what the difference is between release x and release y, because they want to know if it is worth upgrading. traditionaly that has been done by a "changes.txt" file in the root  of the project.

Wackerbarth, 2015-12-03:
@thawkins -
Tim,
I agree that we should have Release Notes / Change Log information in a documentation file that is a part of the source.

Unfortunately, the way changes have been introduced into the code has created a situation where it is difficult for someone other than the original author to determine just what the change was intended to accomplish and, accordingly, whether or not that change warrants specific mention.

Further complicating the situation is the fact that the contributors primarily responsible for the changes have not been "documenting as you go" (a problem common to many projects) and they have since "disappeared".

Thus, we would welcome any assistance you, or others, might provide in helping us to prepare such a document.

Because of the ease of access to make corrections and additions, I think that it might be worth while to utilize the wiki as a place to compose the text. Then we can "copy and paste" the result into a permanent documentation file that is a part of the distribution.

What do you think of that approach?

thawkins, 2015-12-03:
That would work.

KiteLab, 2015-12-03:
Is https://github.com/MarlinFirmware/Marlin/releases what you are looking for?

Roxy-3DPrintBoard, 2015-12-03:
Wow!  I didn't know that page existed!

closed 2016-04-20 by thinkyhead