MarlinFirmware/MarlinDev #296: Help flashing firmware: Moving to Arduino 1.6.6 from 1.0.5 (bootloader error)

a4jp-com, 2015-12-04:
I just updated to Arduino 1.6.6 and get this error:

Bootloader file specified but missing: C:\Users\design\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\bootloaders\stk500boot_v2_mega2560.hex

I don't have that folder (bootloaders) or the file stk500boot_v2_mega2560.hex

What can I do to set up Marlin from 1.6.6? Sorry for asking about this here.

How do I make or replace/delete the bootdriver?

Wackerbarth, 2015-12-04:
... packages\rambo ... is not the build setup that we support.
Load the Marlin package using the board manager. And select the mega2560 processor.

When I compared the variants, I didn't find any difference between "rambo" and a standard mega2560.
But, if I missed something, let me know and I will take care of it in the Marlin package.

a4jp-com, 2015-12-04:
I just tried but but now my printer has a blank screen.

a4jp-com, 2015-12-04:
Pronterface can move the print head though.

Wackerbarth, 2015-12-04:
That sounds like an issue with the LCD configuration.

I suspect that it would also be an issue if you tried to use 1.1.0-RC3 and the IDE 1.0.5 environment.

There is a copy of the bootloader that you sought at ...packages/Marlin/hardware/avr/1.0.3/bootloaders/stk500v2/stk500boot_v2_mega2560.hex
you could copy it into your rambo setup. But I don't expect that you will find any improvement.
(Bypass the Sanity check at the bottom of FirmwareCustomization.h if you don't use the Marlin package)

The question is going to be whether, or not, someone else has managed to bring up 1.1.0-RC on similar hardware.  (I would bet that if they have, there is something different in the configuration)

a4jp-com, 2015-12-04:
I went to:

https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/bootloaders/stk500v2

and copied all the files in the the folder on my computer:

C:\Users\design\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\bootloaders

Then set the board to RAMBo again and now the text is showing but something weird is happening.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

a4jp-com, 2015-12-04:
Where can I set the LCD size?

Wackerbarth, 2015-12-04:
OK, that indicates two separate issues.
1) Why the Rambo build is different from the Marlin build
2) Why the display isn't wrapping properly.

I need to reproduce your results. Where did you get the Rambo package for the IDE?

a4jp-com, 2015-12-04:
I got the board info from:

http://reprap.org/wiki/Rambo_firmware

After that, if you go to the preferences of Arduino and then add the URL below to the Additional Board Manager URLs: https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json
you can get the same board driver.

You will have to open and close the program after adding the URL. Then click update when the program opens.

a4jp-com, 2015-12-04:
1) The RAMBo board can use the Atmega2560 MCU's pins that are not available on Arduino Mega2560. These pins are used for the LCD interface or external stepper drivers.
2) Not sure what is happening here yet. I guess the code doesn't automatically work with 16x4 screens like the one I have.

Wackerbarth, 2015-12-04:
Ok, I added the RAMBo board to my Marlin platform. (Now version 1.0.4)
Please try compiling with that version and see if it makes any difference.

The other difference is that the RAMBo platform is providing the old 1.11 version of U8glib and, at the request of those dealing with displays, I now supply  the latest (version 1.18) of that library.

KiteLab, 2015-12-04:
@a4jp-com 
To set up the LCD_WIDTH/LCD_HEIGHT
Conditionlals.h Line 143+
```
  #if ENABLED(ULTIPANEL)
    #define NEWPANEL  //enable this if you have a click-encoder panel
    #define ULTRA_LCD
    #if ENABLED(DOGLCD) // Change number of lines to match the DOG graphic display
      #define LCD_WIDTH 22
      #define LCD_HEIGHT 5
    #else
      #define LCD_WIDTH 20          // <<<< 16
      #define LCD_HEIGHT 4
    #endif
  #else //no panel but just LCD
    #if ENABLED(ULTRA_LCD)
      #if ENABLED(DOGLCD) // Change number of lines to match the 128x64 graphics display
        #define LCD_WIDTH 22
        #define LCD_HEIGHT 5
      #else
        #define LCD_WIDTH 16
        #define LCD_HEIGHT 2
      #endif
    #endif
  #endif
```

Wackerbarth, 2015-12-04:
Actually, we need to convert this into the default value structure so that  the user can simply add
```
#define LCD_WIDTH  16
```
, or whatever they need to their top level Configuration rather than editing "Conditionals.h".

a4jp-com, 2015-12-04:
Thank you very much @Wackerbarth and @KiteLab. I'm just trying to figure out how to set up or load add the files together in Arduino. The other folder I had was made by the printer company so everything was easy but this is my first time setting up everything from scratch. I understand all the calculations and figures I probably need to put in though. I also figure as long as I can get the info showing up on the LCD everything will work out after that.

Wackerbarth, 2015-12-04:
OK, I just pushed a change for Conditionals.h so that you can perform the override in your top level configuration file.  Try using the "LCD_Display_Dimensions" branch.

a4jp-com, 2015-12-04:
Oh okay. Also can you tell me if this process is correct?

File > New - Any file name ie Marlin-1-0-4

replace 

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

with

#include <LiquidCrystal.h>

#include <LiquidCrystal.h>

/* -*- c++ -*- */

/*
    Reprap firmware based on Sprinter and grbl.
 Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 
 This program is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.
 
 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.
 
 You should have received a copy of the GNU General Public License
 along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

/*
 This firmware is a mashup between Sprinter and grbl.
  (https://github.com/kliment/Sprinter)
  (https://github.com/simen/grbl/tree)
 
 It has preliminary support for Matthew Roberts advance algorithm 
    http://reprap.org/pipermail/reprap-dev/2011-May/003323.html
 */

/* All the implementation is done in *.cpp files to get better compatibility with avr-gcc without the Arduino IDE */
/* Use this file to help the Arduino IDE find which Arduino libraries are needed and to keep documentation on GCode */

#include "Configuration.h"
#include "pins.h"

#ifdef ULTRA_LCD
  #ifdef DOGLCD
    #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://code.google.com/p/u8glib/)
  #else
    #include <LiquidCrystal.h> // library for character LCD
  #endif
#endif

#if DIGIPOTSS_PIN > -1
#include <SPI.h>
#endif

from old file

Sketch > Add file

all new files with the same name as the old set

Wackerbarth, 2015-12-04:
No. Please see https://github.com/MarlinFirmware/MarlinDev/wiki for more details.

But, since you have been dealing with the RAMBo package, you will probably understand these brief hints:

Arduino IDE 1.6.6

Add additional boards manager source
https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json

(Restart IDE)

Load Marlin board package (version 1.0.4)

With Git, clone the marlin source into your sketchbook libraries

git clone https://github.com/MarlinFirmware/MarlinDev.git

git checkout LCD_Display_Dimensions

Select the File > Examples > MarlinDev > MarlinFirmware sketch

Select the FirmwareCustomization.h tab and edit your specific changes there.

ALTERNATELY, create a new file for your configuration items and just place a reference to it in FirmwareCustomization.h.

It is a bit more complicated to set up, but you can also reference your customized configuration by adding a line to the Printers menu so that you don’r need to change any of the files in the Marlin source tree.

PS: I recommend that you name your configuration file something that is unique to yourself (either by using a unique ID or you name, etc.).
That way, without causing a conflict for them, you can share the file with someone such as myself who might be attempting to assist you.

> On Dec 4, 2015, at 2:31 PM, a4jp-com <notifications@github.com> wrote:
> 
> Oh okay. Also can you tell me if this process is correct?
> 
> File > New - Any file name ie Marlin-1-0-4

a4jp-com, 2015-12-04:
Where is Configuration.h? Wow this is so complicated. I just finished adding a heap of files to my new file Marlin-1-0-4-lcd

Wackerbarth, 2015-12-04:
You are doing it backwards.

Assuming that you have installed the Marlin platform AND Marlin source library as I indicated,

look at configurations/custom/e9cbe275/Customization.h

You should be making a similar file (called Customization.h) in an adjacent folder.

That file will have the entries that you need to change.

For example, if you need it,  the 
```
  #define LCD_WIDTH 16
```
entry.

In general, there should not be very many entries because you will be able to get most of what you need from some "stock' setup with the 
```
#include "configurations/transitional_default_configurations/ ..."
```
line.

a4jp-com, 2015-12-04:
(x-x;) I loaded the board from your files.

And tried adding all the files.

then added the configurations/custom/e9cbe275/Customization.h file and tried to get verify to work but there were heaps of errors.

Maybe I need to start again...

If I get this right once everything should be good.

Sorry for any inconvenience.

Wackerbarth, 2015-12-04:
"And tried adding all the files." ----   NO! NO! NO!
Start without ANY of your own files and follow the instructions (above) to compile the "Default Configuration" printer without editing ANYTHING.

Then create your own custom Configuration.h file as I have indicated and put in just the few #defines that you need to do the customization.  To start, don't even consider editing anything other than the files that I directly mentioned.

Now, I do recognize that you may get some errors because something is being redefined.
Those are likely to be the result to the fact that the transitional defaults are still being consolidated.
For those, you will probably want to remove the offending entry until I get a chance to replace the files with newer versions that make the default definitions conditioned on their not being previously defined.

a4jp-com, 2015-12-04:
I started again.

opened arduino

pasted

https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json

restarted

selected RAMBo board

copied downloaded zip folder to 

C:\Users\design\Documents\Arduino\libraries

opened

MarlinFirmware.ino

from Arduino from

C:\Users\design\Documents\Arduino\libraries\MarlinDev-LCD_Display_Dimensions\examples\MarlinFirmware

now two files are showing

marlinfirmware and firmwarecustomization

clicked verify

one error

Multiple libraries were found for "U8glib.h"
 Used: C:\Users\design\Documents\Arduino\libraries\U8glib
 Not used: C:\Users\design\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.4\libraries\U8glib

Does that look better?

a4jp-com, 2015-12-04:
Now do I go to Sketch > Add File... for the configurations/custom/e9cbe275/Customization.h file or do I just edit the configurations/custom/e9cbe275/Customization.h file directly?

a4jp-com, 2015-12-04:
Or add a line like

#include "configurations/custom/a4jp/Customization.h"

to the firmwarecustomization.h file

Wackerbarth, 2015-12-04:
> On Dec 4, 2015, at 3:53 PM, a4jp-com <notifications@github.com> wrote:
> 
> I started again.
> 
> opened arduino
> 
> pasted
> 
> https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json <https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json>
> restarted
> 
> selected RAMBo board
> 
> copied downloaded zip folder to
> 
> C:\Users\design\Documents\Arduino\libraries
> 
> opened
> 
> MarlinFirmware.ino
> 
> from Arduino from
> 
> C:\Users\design\Documents\Arduino\libraries\MarlinDev-LCD_Display_Dimensions\examples\MarlinFirmware
> 
> now two files are showing
> 
> marlinfirmware and firmwarecustomization
> 
> clicked verify
> 

Spot on so far.
> one error
> 
> Multiple libraries were found for "U8glib.h"
> Used: C:\Users\design\Documents\Arduino\libraries\U8glib
> Not used: C:\Users\design\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.4\libraries\U8glib
> 

OK, that says that you have your personal copy of U8glib and that I have also included one in the Marlin package.

What version is the one that you are providing?
If you want to use it, everything is fine.
Otherwise, you should move it out of the way.
> Does that look better?
> 

That should have created a .hex file.  If so, all is well.

CONTINUING …

I went to https://www.uuidgenerator.net/version4 <https://www.uuidgenerator.net/version4>

and obtained this uuid (just for you)

d6a2d588-0a0a-418c-b145-9646e3e37d67

So, let’s use it to set up your personal configuration.

Edit the file in the FirmwareCustomization.h tab.

Make the following line active
```
  #define  SPECIFIC_PRINTER_ID d6a2d588
```

Now select the “Default Configuration” from the Tools > Printers menu

And click verify again.

you should get a fatal error
```
… MarlinFirmware.h:18:65: fatal error: configurations/custom/d6a2d588/Customization.h: No such file or directory
```
Of course that is because we haven’t created your customization file :)

So create the containing folder — C:\Users\design\Documents\Arduino\libraries\MarlinDev-LCD_Display_Dimensions\configurations\custom\d6a2d588\
and use your editor and create the file — Customization.h

```
#define CUSTOMIZATION_H

#define UUID "d6a2d588-0a0a-418c-b145-9646e3e37d67"

#define  STRING_CONFIG_H_AUTHOR “(••••••••••••••)" // Who made the changes.   <== Insert your name here

//  •••••••••  You’ll add more stuff here

#include "configurations/transitional_default_configurations/default/Configuration.h"
```
Instead of “default” in that last line, you may prefer to start from something more like your own printer.

The goal will be to develop the proper entries for your vendor and model and all you will need to do is reference that.

Re: Your next reply:
You can use “a4jp” if you like. Personally, I have gotten in the habit of using UUIDs. But I have quite a few microprocessor boards (belonging to various projects) and need separate names to keep straight what I have flashed into each one.

a4jp-com, 2015-12-12:
I think I'm getting close but I still can't get the files to compile properly.

Wackerbarth, 2015-12-12:
We ran into a problem with the Arduino IDE 1.6.6.

The source would compile. But later in the chain of tools, everything would just stop and you never got to the commands that output the size, etc.

Please try with the Arduino IDE 1.6.7 (hourly build). Things seem to work with that version as well as some of the pre-released 1.6.6 (which you are unlikely to have). But there is a "hole" in the sequence which, unfortunately, includes the 1.6.6 release.

I'm online at the Marlin hangout https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia if you would like to drop in to discuss build issues.

a4jp-com, 2015-12-14:
I still can't figure out why it isn't working, If it okay can I go through this step by step? I'm getting an error with the Marlin board drivers.

I downloaded the nightly build. 1.6.7 2015/12/14 11:21

Opened Arduino and then pasted https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json
into the Additional Boards Manager URLs of the preferences.

Closed it and opened it again.
Clicked on the update link.
Closed it and opened it again.
I went to the board manager and updated the board for Marlin.
Clicked on verify and got this error:

Board rambo (platform avr, package Marlin) is unknown

Error compiling.
Closed and opened again.
Got the same error.

I tried selecting
Arduino Mega or Mega 2560 but got an error:
Board mega (platform avr, package Marlin) is unknown

Error compiling.

If I select the original board Arduino/Genuino Mega 2560 there isn't an error.

Roxy-3DPrintBoard, 2015-12-14:
> *Also which programmer should I choose from the tools menu? AVRISP Mk II?

Wackerbarth will have the full details...   But what I did was just use the exact same settings that I previously used which was AVR-ISP.   I think it depends on who made the board for which programmers work.  With a Mega2560 AVR-ISP does work.

a4jp-com, 2015-12-14:
https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json

Verifies without an error but I haven't tested the newest Marlin files though.

a4jp-com, 2015-12-14:
Thanks for the info @Roxy-3DPrintBoard ^^

Wackerbarth, 2015-12-14:
> 
> I still can't figure out why it isn't working, If it okay can I go through this step by step?

Very good.

PS: Join me in the google hangout if you like.
https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia <https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia>

> I'm getting an error with the Marlin board drivers.
> 
> I downloaded the nightly build. 1.6.7 2015/12/14 11:21
I expect it to be OK. Yesterday was fine. But I haven’t tried today’s build.

> Opened Arduino and then pasted https://raw.githubusercontent.com/marlinfirmware/MarlinDev/platform/package_marlinfirmware_index.json [...]
OK
> Clicked on the update link.
> Closed it and opened it again.
> I went to the board manager and updated the board for Marlin.
Which version of the Marlin platform? I now have 3 different ones posted. The one to use needs to match the version of the IDE
(1.6.7 in your case)

But, if you have the wrong one, I’ve seen it hang in the “make” sequence. I’ve never seen the error that you are reporting.

> Clicked on verify and got this error:
> 
> Board rambo (platform avr, package Marlin) is unknown
That error message would lead me to believe that you are selecting the appropriate board.

Look in your Arduino15  (the preferences) folder and dump the Marlin package from both packages and staging
so that it will have to reload them.

Wackerbarth, 2015-12-14:
The “programmer” is used only if you are flashing a bootloader or if you are not using the bootloader to reflash the Marlin firmware.  The selection should not matter if you use the standard bootloader over the USB cable.

a4jp-com, 2015-12-17:
I got 2 error messages. One said the board wasn't recognized and the other said HEATER_0_PIN was not defined for this board.

This is a bit of the code in the configuration code:

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define BOARD_RAMBO             301  // Rambo
#endif

-------------------------------
Error Message when verifying:

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/a4jp/Customization.h:879,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:132:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

   #error Unknown MOTHERBOARD value set in Configuration.h

    ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:628:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/a4jp/Customization.h:879,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/SanityCheck.h:326:4: error: #error HEATER_0_PIN not defined for this board.

   #error HEATER_0_PIN not defined for this board.

    ^

Multiple libraries were found for "MarlinFirmware.h"
 Used: C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev
 Not used: C:\Users\design\Documents\Arduino\libraries\MarlinDev-LCD_Display_Dimensions
exit status 1
Error compiling.

a4jp-com, 2015-12-17:
I used this code instead of the code that was written in the boards.h file

    #ifndef MOTHERBOARD
    #define MOTHERBOARD 301
    #endif

I think everything might work now ^^

Multiple libraries were found for "MarlinFirmware.h"
 Used: C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev
 Not used: C:\Users\design\Documents\Arduino\libraries\MarlinDev-LCD_Display_Dimensions

Sketch uses 70,496 bytes (27%) of program storage space. Maximum is 258,048 bytes.
Global variables use 4,295 bytes of dynamic memory.

a4jp-com, 2015-12-17:
I uploaded the file but the LCD is blank.

a4jp-com, 2015-12-17:
I was missing some code (>-<) sorry.

    #define ULTRA_LCD  //general LCD support, also 16x2
    //#define DOGLCD  // Support for SPI LCD 128x64 (Controller ST7565R graphic Display Family)
    //#define SDSUPPORT // Enable SD Card Support in Hardware Console
    //#define ULTIMAKERCONTROLLER //as available from the ultimaker online store.
    #define ULTIPANEL  //the ultipanel as on thingiverse

Now it works.

I'll do a few tests prints then onto the newer version of Marlin after that ^^

Thank you so much for your time in Google Chat @Wackerbarth

a4jp-com, 2015-12-17:
Now the card reader doesn't work. There is a double percent mark on the display and the menu that was on the 3D printer before isn't working no matter what button I push. Hmmmmm. New challenges.

[image]

Wackerbarth, 2015-12-17:
FOR
```
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define BOARD_RAMBO             301  // Rambo
#endif

```
USE
```
#define MOTHERBOARD BOARD_RAMBO
```
and you don't even need the "guards" unless you might be using this as a template for something else.

a4jp-com, 2015-12-17:
I don't know what happened but after restarting my computer for an update now I get an error.

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\configurations/custom/a4jp/Customization.h:883,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin.h:26,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin_main.cpp:30:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin_main.cpp:263:54: error: 'DEFAULT_NOMINAL_FILAMENT_DIA' was not declared in this scope

 float filament_size[EXTRUDERS] = ARRAY_BY_EXTRUDERS1(DEFAULT_NOMINAL_FILAMENT_DIA);

                                                      ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:409:50: note: in definition of macro 'ARRAY_BY_EXTRUDERS'

     #define ARRAY_BY_EXTRUDERS(v1, v2, v3, v4) { v1 }

                                                  ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin_main.cpp:263:34: note: in expansion of macro 'ARRAY_BY_EXTRUDERS1'

 float filament_size[EXTRUDERS] = ARRAY_BY_EXTRUDERS1(DEFAULT_NOMINAL_FILAMENT_DIA);

                                  ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin_main.cpp: In function 'void gcode_M200()':

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin_main.cpp:4414:52: error: 'DEFAULT_NOMINAL_FILAMENT_DIA' was not declared in this scope

         if (! filament_size[i]) filament_size[i] = DEFAULT_NOMINAL_FILAMENT_DIA;

                                                    ^

exit status 1
Error compiling.

a4jp-com, 2015-12-17:
How do I get the SD card reader working now and the menu on the 3D printer working next?

KiteLab, 2015-12-17:
Looks like a pinning error.
Can you give us the link to your __original__ firmware again please?

a4jp-com, 2015-12-17:
Sure ^^
http://a4jp.com/3d-files/marlin-2015-01-06th-23-28-00.zip

a4jp-com, 2015-12-17:
I guess I just add the code below to the pins.h file but do I just comment out the #if MOTHERBOARD == 301 line?
---------------------------------------------

#if MOTHERBOARD == 301
#define KNOWN_BOARD
/*****************************************************************
* Rambo Pin Assignments
******************************************************************/

#ifndef __AVR_ATmega2560__
#error Oops!  Make sure you have 'Arduino Mega 2560' selected from the 'Tools -> Boards' menu.
#endif

#define LARGE_FLASH true

#define X_STEP_PIN 37
#define X_DIR_PIN 48
#define X_MIN_PIN 12
#define X_MAX_PIN 24
#define X_ENABLE_PIN 29
#define X_MS1_PIN 40
#define X_MS2_PIN 41

#define Y_STEP_PIN 36
#define Y_DIR_PIN 49
#define Y_MIN_PIN 11
#define Y_MAX_PIN 23
#define Y_ENABLE_PIN 28
#define Y_MS1_PIN 69
#define Y_MS2_PIN 39

#define Z_STEP_PIN 35
#define Z_DIR_PIN 47
#define Z_MIN_PIN 10
#define Z_MAX_PIN 30
#define Z_ENABLE_PIN 27
#define Z_MS1_PIN 68
#define Z_MS2_PIN 67

#define HEATER_BED_PIN 3
#define TEMP_BED_PIN 7 //2014/02/04  0:T0 / 1:T1 / 2:T2 / 7:T3

#define HEATER_0_PIN  9
#define TEMP_0_PIN 0

#define HEATER_1_PIN 7
#define TEMP_1_PIN -1

#define HEATER_2_PIN -1
#define TEMP_2_PIN -1

#define E0_STEP_PIN         34
#define E0_DIR_PIN          43
#define E0_ENABLE_PIN       26
#define E0_MS1_PIN 65
#define E0_MS2_PIN 66

#define E1_STEP_PIN         33
#define E1_DIR_PIN          42
#define E1_ENABLE_PIN       25
#define E1_MS1_PIN 63
#define E1_MS2_PIN 64

#define DIGIPOTSS_PIN 38
#define DIGIPOT_CHANNELS {4,5,3,0,1} // X Y Z E0 E1 digipot channels to stepper driver mapping

#define SDPOWER            -1
#define SDSS               53
#define LED_PIN            13
#define FAN_PIN            8
#define PS_ON_PIN          4
#define KILL_PIN           -1
#define SUICIDE_PIN        -1  //PIN that has to be turned on right after start, to keep power flowing.

//2014/01/20 Add pin definitions
#define FAN_1_PIN   6
#define FAN_2_PIN   2
//

//2014/04/16
#ifdef ULTRA_LCD
  #ifdef NEWPANEL
  //arduino pin which triggers an piezzo beeper
    #define BEEPER 79      // Beeper on AUX-4
    #define LCD_PINS_RS 70 
    #define LCD_PINS_ENABLE 71
    #define LCD_PINS_D4 72
    #define LCD_PINS_D5 73 
    #define LCD_PINS_D6 74
    #define LCD_PINS_D7 75
   
    //buttons are directly attached using AUX-2
    #define BTN_EN1 76
    #define BTN_EN2 77
    #define BTN_ENC 78  //the click
   
    #define BLEN_C 2
    #define BLEN_B 1
    #define BLEN_A 0
   
    #define SDCARDDETECT 46    // Ramps does not use this port
   
    //encoder rotation values
    #define encrot0 0
    #define encrot1 2
    #define encrot2 3
    #define encrot3 1
  #else //old style panel with shift register
    //arduino pin witch triggers an piezzo beeper
    #define BEEPER 33    //No Beeper added
    //buttons are attached to a shift register
    // Not wired this yet
    // #define SHIFT_CLK 38
    // #define SHIFT_LD 42
    // #define SHIFT_OUT 40
    // #define SHIFT_EN 17
   
    #define LCD_PINS_RS 70 
    #define LCD_PINS_ENABLE 17
    #define LCD_PINS_D4 23
    #define LCD_PINS_D5 25 
    #define LCD_PINS_D6 27
    #define LCD_PINS_D7 29
   
    //encoder rotation values
    #define encrot0 0
    #define encrot1 2
    #define encrot2 3
    #define encrot3 1
   
    //bits in the shift register that carry the buttons for:
    // left up center down right red
    #define BL_LE 7
    #define BL_UP 6
    #define BL_MI 5
    #define BL_DW 4
    #define BL_RI 3
    #define BL_ST 2
    #define BLEN_B 1
    #define BLEN_A 0
  #endif 
#endif //ULTRA_LCD

// 2014/04/16
#define LCD_PINS_RS 70      //ext2_5

#define LCD_PINS_ENABLE 71  //ext2_7
#define LCD_PINS_D4 72
#define LCD_PINS_D5 73 
#define LCD_PINS_D6 74     //ext2_13
#define LCD_PINS_D7 75
#define BEEPER -1

//#define SW_DOWN  85 //ext2_6
#define BTN_EN1  85 //ext2_6
//#define SW_UP 84 //ext2_8 
#define BTN_EN2 84 //ext2_8 
//#define SW_ENTER 81 //ext_14
#define BTN_ENC 81 //ext_14
#define BTN_LEFT 83  //ext_10
#define BTN_RIGHT 82  //ext_12
#define BTN_HOME 80  //ext_16
//

#endif

a4jp-com, 2015-12-17:
That is probably the lazy way. I guess I should copy the pins.h file and then include it some how to override the other file.

a4jp-com, 2015-12-17:
I copied the old pins.h file to the directory with my new config file, then replaced BEEPER with BEEPER_PIN in 3 spots. Commented out the top line of code //#if MOTHERBOARD == 301 the also //#endif

Looks like everything might work now. I'll try compiling it now ^^

Wackerbarth, 2015-12-17:
Yes! And it will probably lead to more issues.
Your "old" pins file may need some additions/changes to keep up with the expectations of the newer firmware. (But it is obvious that you have some connections that differ from the "stock" RAMBO setup.

Create a pins_A4JP.h file and make it look as much like the pins_RAMBO.h file as possible.
Then add
```
#define BOARD_A4JP              900  // Rambo -- Derived for a4jp
#define MOTHERBOARD BOARD_A4JP
```
at the top of your Customization.h

Chat?

a4jp-com, 2015-12-17:
Yes, a chat would be great if that is okay.

a4jp-com, 2015-12-17:
I have added as much info to the new pins_A4JP.h file as I can. I also added the code to the top of the Customization.h file in the A4JP folder.

a4jp-com, 2015-12-17:
https://talkgadget.google.com/hangouts/_/uv7ahpejul7tknohrpahppzsgua

a4jp-com, 2015-12-18:
@Wackerbarth I went through the file and removed most of the double entries but pins.h is still referenced somewhere.

This is a bit long but it is the error message now.

Arduino: 1.6.7 Hourly Build 2015/12/16 10:44 (Windows 10), Board: "RAMBo, Printer A4JP"

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:130:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins_A4JP.h:42:0: warning: "E0_MS1_PIN" redefined [enabled by default]

 #define E0_MS1_PIN 65

 ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:15:0: note: this is the location of the previous definition

 #define E0_MS1_PIN -1

 ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:130:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins_A4JP.h:43:0: warning: "E0_MS2_PIN" redefined [enabled by default]

 #define E0_MS2_PIN 66

 ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

[... 4238 lines cut ...]
                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\Marlin.h:26,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\planner.cpp:51:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins_A4JP.h:51:0: warning: "DIGIPOTSS_PIN" redefined [enabled by default]

 #define DIGIPOTSS_PIN 38

 ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 fro

Wackerbarth, 2015-12-18:
Well, there are still a number of duplicate entries in your pins_A4JP.h

But, addressing the first error, that faulty code goes back to 2012.
Unfortunately, Marlin is full of very poor programming practices such as this.

In the past, they thought nothing of redefining symbols on the fly.
And then they would why things don't work as expected :)

a4jp-com, 2015-12-18:
That file looks very nice.

Down to maybe two errors.

Arduino: 1.6.7 Hourly Build 2015/12/16 10:44 (Windows 10), Board: "RAMBo, Printer A4JP"

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Conditionals.h:222:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:627,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:117:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

   #error Unknown MOTHERBOARD value set in Configuration.h

    ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:628:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/SanityCheck.h:326:4: error: #error HEATER_0_PIN not defined for this board.

   #error HEATER_0_PIN not defined for this board.

    ^

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Wackerbarth, 2015-12-18:
Ah! If we are going to define a new MOTHERBOARD, we still need to drag in its pins_XXX.h

so add
```
#elif MB(A4JP)
  #include "pins_A4JP.h"
```
to pins.h

a4jp-com, 2015-12-18:
Thank you but I still get a compiling error.

Part 1 ok
Part 2 #define HEATER_0_PIN  9 in the pins_A4JP.h file
Part 3 activated

These are the errors:

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:628:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/SanityCheck.h:326:4: error: #error HEATER_0_PIN not defined for this board.

   #error HEATER_0_PIN not defined for this board.

    ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Marlin.h:27:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/thermistortables.h:4,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:886,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:119:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

   #error Unknown MOTHERBOARD value set in Configuration.h

    ^

exit status 1
Error compiling.

a4jp-com, 2015-12-18:
I just put #define HEATER_0_PIN  9 into the Customization file and the next file that was missing popped up but the other error went away. ^^

I will go through and make copies of the missing bits in the Configuration.h file and copy them from pins_A4JP.h 

    In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/transitional_default_configurations/default/Configuration_adv.h:628:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:885,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/SanityCheck.h:328:4: error: #error TEMP_0_PIN not defined for this board.

       #error TEMP_0_PIN not defined for this board.

        ^

    In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Marlin.h:27:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/thermistortables.h:4,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:886,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

    C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:119:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

       #error Unknown MOTHERBOARD value set in Configuration.h

        ^

    exit status 1
    Error compiling.

Wackerbarth, 2015-12-18:
I'm sorry, but this isn't working.  It is obvious that the MOTHERBOARD is not being recognized in pins.h

I really need to be able to look at ALL of your EXACT files.

The best way to accomplish that would be for you to fork MarlinFirmware/MarlinDev on Github.
Then set your current files aside, `git checkout dev`,  replace those files with your current work, commit and push the differences back to your Github account.

BTW, copying pieces of a file to work around the issue is not a sustainable approach.
We need to fix the underlying problem. (And, unfortunately, I don't understand why my suggested fix is not seeming to work.

Chat?

a4jp-com, 2015-12-18:
Copying these 

#define HEATER_0_PIN  9
#define TEMP_0_PIN 0
#define E0_STEP_PIN         34
#define E0_DIR_PIN          43
#define E0_ENABLE_PIN       26

into the Customization file gets rid of all but 1 error ^^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/Marlin.h:27:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/thermistortables.h:4,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/configurations/custom/A4JP/Customization.h:890,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/IncludeUserCustomization.h:21,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/MarlinFirmware.h:11,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino:2:

C:\Users\design\Documents\Arduino\libraries\MarlinDev-dev/pins.h:119:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

   #error Unknown MOTHERBOARD value set in Configuration.h

    ^

exit status 1
Error compiling.

a4jp-com, 2015-12-18:
Yes please.

https://talkgadget.google.com/hangouts/_/anjyq2hn6bye3k4qofwcnmtjtya

a4jp-com, 2015-12-18:
I'm finding it difficult to set the pins for the menu controller. Any easy way to find out what pin should be set? I've just been commenting out code and switching numbers from the old file but can't figure out how to get the up, down, left and right buttons to work properly. I can kind of switch the menu button but it is really weird.

a4jp-com, 2015-12-19:
Any pin experts out there? I have a RAMBo board and can't get the click coder pins working properly.

I have this line in my config file:

#define ULTIPANEL  //the UltiPanel as on Thingiverse

And

I can set pin 80 as the kill switch but BTN_HOME wasn't working for some reason when active.

    #define KILL_PIN           80  //Glen//was -1
    //#define BTN_HOME 80  //ext_16//Glen//80 is used above for now as a kill switch

is probably the code for the menu button

    #define BTN_ENC 81 //ext_14 

but this isn't working

    //#define SW_DOWN  85 //ext2_6
    #define BTN_EN1  85 //ext2_6    //Glen//76?
    //#define SW_UP 84 //ext2_8
    #define BTN_EN2 84 //ext2_8      //Glen//77? 
    #define BTN_LEFT 83  //ext_10
    #define BTN_RIGHT 82  //ext_12

Also what excatly does each line of code here do:

    #define LCD_PINS_RS 70      //ext2_5
    #define LCD_PINS_ENABLE 71  //ext2_7
    #define LCD_PINS_D4 72
    #define LCD_PINS_D5 73 
    #define LCD_PINS_D6 74     //ext2_13
    #define LCD_PINS_D7 75

This is the pin file:

/************************************************
 * Rambo pin assignments dev folder
 ************************************************/

// Servo support
#define SERVO0_PIN 22 // Motor header MX1
#define SERVO1_PIN 23 // Motor header MX2
#define SERVO2_PIN 24 // Motor header MX3
#define SERVO3_PIN  5 // PWM header pin 5

#if ENABLED(Z_PROBE_SLED)
  #define SLED_PIN -1
#endif

#undef X_MS1_PIN
#undef X_MS2_PIN
#undef Y_MS1_PIN
#undef Y_MS2_PIN
#undef Z_MS1_PIN
#undef Z_MS2_PIN
#undef E0_MS1_PIN
#undef E0_MS2_PIN
#undef E1_MS1_PIN
#undef E1_MS2_PIN

#undef DIGIPOTSS_PIN
//Fan_2 2

/*****************
#if ENABLED(ULTRA_LCD)

  #define KILL_PIN -1 //was 80 Glen maybe a mistake

[... 159 lines cut ...]
  #endif 
#endif //ULTRA_LCD

// 2014/04/16
#define LCD_PINS_RS 70      //ext2_5 //??CS chip select /??SS chip slave select
#define LCD_PINS_ENABLE 71  //ext2_7
#define LCD_PINS_D4 72               //??SCK (CLK) clock
#define LCD_PINS_D5 73 
#define LCD_PINS_D6 74     //ext2_13
#define LCD_PINS_D7 75
#define BEEPER_PIN -1

//Click encoder pins
//#define SW_DOWN  85 //ext2_6
#define BTN_EN1  85 //ext2_6    //Glen//76?

//#define SW_UP 84 //ext2_8
#define BTN_EN2 84 //ext2_8      //Glen//77? 

//#define SW_ENTER 81 //ext_14
#define BTN_ENC 81 //ext_14  //Glen//center button//78 doesn't work here

#define BTN_LEFT 83  //ext_10
#define BTN_RIGHT 82  //ext_12
//#define BTN_HOME 80  //ext_16//Glen//80 is used above for now as a kill switch

[3 more replies]

closed 2015-12-19 by Wackerbarth