MarlinFirmware/MarlinDev #299: DOGM128 LCD displaying gibberisch now and then

maukcc, 2015-12-07:
We have the DOGM128 LCD hooked up to our electronics and now and then:
- it moves the view up/down
- displays gibberisch 
- does not display anything at all
I have seen this problem being discussed before but never seen a solution.

Normally it is hooked up over HW-SPI and we now tried SW-SPI (sck,mosi,cs and a0 set to other pins then normal, and no pins mixed with SDcard)
This dramatically improved the "now and then" factor, from once a day, to once a month(ish)

What we also found is that the SCK pulses are low in voltage (1.2V) this is borderline for the 4050 (that is in front of the LCD) to pick up and could be the main cause of this problem.
Is there any way to get the pulses higher, or is this a problem of to little time to get the pulses ramped up to 5V or 3V3? In that case, can we use a slower signal?

KiteLab, 2015-12-07:
The problems you describe seem to be, more hardware than software related. Please give us some hints about your hardware.

KiteLab, 2015-12-07:
*  Noise on the encoder lines? Insufficient power supply?
*  Insufficient power supply?
*  We had a problem wit this when killing Marlin before the display was initialized. That's fixed. broken cables? Insufficient power supply?

To much load on that pin?

maukcc, 2015-12-07:
Hardware is Arduino mega with 4 pins(sck,mosi,cs and a0) to a 4050 to make 3V3 pins that go to the LCD. 
The mosi, cs and a0 pin all have 5V output.
For the sck pin I tried a normal pin, analog pin and a pwm pin, all gave the same result, 1V2 high. all with no load connected.
Power supply to the 4050 is a steady 3V3
1meter of flatcable between arduino and 4050.

KiteLab, 2015-12-07:
What is your measuring device for the SCK-pin? Oscilloscope or multimeter?

maukcc, 2015-12-07:
Oscilloscope. top of the pulses are 1.2V

KiteLab, 2015-12-07:
Is SCK pin D13?
Any additional capacitys?
Any other board between Mega and display? (RAMPS?)

If you use the ST7920 device, you can tweak the pulse length in `ultralcd_st7920_u8glib_rrd.h` `ST7920_SWSPI_SND_8BIT()` every other device is directly in u8glib

maukcc, 2015-12-08:
SCK is on pin 49.
The mainboard between the Mega and the LCD is our own board.
The pin goes directly to the flatcable header without any other things in between

We use a DOGM128 with the ST7565 controller

KiteLab, 2015-12-08:
Maybe a defect Mega?
Sorry. What is inside u8glib is outside my scope.

maukcc, 2015-12-22:
Mega is OK.
I found the same for the SCK pin for the SD card(low in voltage)

thinkyhead, 2016-04-20:
Closing old issues for housekeeping.

closed 2016-04-20 by thinkyhead

maukcc, 2016-04-20:
The bug is clearly still present so no need to close this.
What would greatly help is to have the option to reduce the frequency of the signal to the display from 2.5Mhz to 1Mhz or 500Khz. 
Does anybody know how to do this?

thinkyhead, 2016-04-22:
@maukcc Are you still seeing this issue in today's `RCBugFix` code? Please post a bug report over there so we can track it!