MarlinFirmware/MarlinDev #301: MarlinDev not building.  Is it broken or am I broken?

crashmatt, 2015-12-10:
I checked out the Dev branch of MarlinDev. It won't compile with Arduino 1.6.6 or eclipse.  I tried putting the example FirmwareCustomization.h in the root folder.  None of the examples work.  The header includes are mostly broken.

Is there some compiler option I need to fix the broken header includes?
Is there a developers document somewhere?  I checked the wiki and readme and did not find.  I could easily have missed a vital link.

It is not possible to do feature development until MarlinDev is somewhat stable. If MarlinDev/dev is broken, when is it planned to be working/compilable again?

Roxy-3DPrintBoard, 2015-12-10:
The Wiki is getting updated with better instructions.   In particular, it is the Non-GIT users that need clearer direction on how to get setup.   Those better directions might be in place by the end of the day.

Wackerbarth, 2015-12-10:
Matthew,

> I checked the wiki and did not find. I could easily have missed a vital link.

I hope that you are reading the wiki for MarlinDev and not the one for Marlin. There are some CRITICAL differences because the file structures are different.

On the first page of the wiki, there is a paragraph called “Setup”. Similar text will also be found if you follow the “Setup” link in the sidebar.

You cannot expect things to work if you are using the Arduino board definitions, or if you move files around.
Please try to follow the instructions provided and make an initial compile using the Arduino IDE.
After that, you can consider alternate ways to compile.

> If MarlinDev/dev is broken, when is it planned to be working/compilable again?
I compile it all the time, and Travis performs (usually successfully) a set of compilations on every push.

However, I am about to post a major set of changes. So, if you do start developing some code for it, keep it close to the head of “dev” so that you can easily rebase/cherry-pick your changes into the updated version.

Roxy-3DPrintBoard, 2015-12-10:
https://github.com/MarlinFirmware/MarlinDev/wiki/Getting-the-Source-Code

There are directions that work for Windows right now.   But I'm fighting the MarkDown stuff trying to make it into a nice list.   Ignoring that, you should be able to get installed!

crashmatt, 2015-12-10:
I am git compatible so no problems getting the source code.  Marlin/RC compiles ok.

Just found the extra wiki items in the side bar.  I see the description is not complete for the new configuration system.  I will just have to wait until you are done.

Out of interest I am looking at doing auto probe offset calibration.  The auto level probe is one of the all-time best modifications for the printer but it is not perfect. Every time I have a nozzle jam/change the probe offset changes.  Then I have to go figure out all the right gcodes to make the calibration again.
So I will visit an old idea with a slight twist.  First probe a small conductive surface at the side of the build area.  Then lower the nozzle onto the same surface until contact is made.  Record the offset in eeprom and use it when doing bed leveling.  This only needs doing infrequently.  The conductive calibration surface does not need to be a permanent feature. It might simply be placed on the table when it is needed.

/Matt

Wackerbarth, 2015-12-10:
Although the configurations are changing, they are "backward compatable" with RC. (At least until we start getting significant divergence in the code. Today, they are close to identical.)

Read the FirmwareCustomization.h and look in the configurations/ folder. You should be able to figure it out from the examples.  One important thing to remember:
Do not edit any of the entries in configurations/ instead, make your own copy and edit that copy.
(Place your info in configurations/custom/ (preferred) or directly in FirmwareCustomization.h

crashmatt, 2015-12-10:
I have not found a Configuration.h and configuration of it that works.  I tried it in various locations and from various sources.  I tried all the different methods available in FirmwareConfiguration.  There are 96 different Configuration.h files in the dev branch. I tested most in the "my kossel" category guessing that they were kind of live.

I will use Marlin/RC instead.  It should not be too much problem to forward the changes to MarlinDev once it is usable.

closed 2015-12-10 by crashmatt

Wackerbarth, 2015-12-10:
@crashmatt -- What OS are you using? Since you mention eclipse, I would guess Linux.

Are you using the Marlin AVR platform (loaded from the Boards manager)?
Have you installed the root of the MarlinDev repository at ..../Arduino/libraries/MarlinDev? (The place where the IDE expects to find Arduino libraries.

Are you using the samples/MarlinFirmware/MarlinFirmware.ino sketch? (Selected from the File > Examples > Menu) ?

Are you compiling using the IDE?

Since you have git,  git checkout Conforming_2015_12_10 branch.
Our Travis CI run shows that all of those tests run properly on a Linux system.