MarlinFirmware/MarlinDev #303: max Z height reached and printing goes on

maukcc, 2015-12-11:
Say your max Z height in firmware is set to 200mm.
When you print something that is 201mm(or 200000mm) the Z-axis will not go any further.
The printing does not stop at that time but will keep on depositing material on/in the last layer.
Can we stop the printjob when the max Z-height is reached?

KiteLab, 2015-12-11:
The question is, how to handle software endstops. Currently we only clip the move but do not send a message or break the print. (`clamp_to_software_endstops()` in `Marlin_main.cpp`)

Shouldn't the slicers prevent from producing points outside the allowed volume?

maukcc, 2015-12-11:
Both slicer AND machine should do this. 
I am seeing something like the filament runout sensor.
if "z-height is more then Z_MAX_POS" then "stop/pause print".
or even if z-height is Z_MAX_POS - 1 then ....

a4jp-com, 2015-12-12:
+1 for @maukcc 's idea

Wackerbarth, 2015-12-12:
I agree. If we treat the "out of bounds" as an implied "out of filament" type of condition, that should be able to stop the process.

closed 2016-05-25 by boelle