MarlinFirmware/MarlinDev #304: Software Endstops

KiteLab, 2015-12-11:
Some remarks about software endstops.
```
#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.
```
* description of `min_software_endstops` is not correct.
* lower case names for the defined values
```
void clamp_to_software_endstops(float target[3]) {
  if (min_software_endstops) {
    NOLESS(target[X_AXIS], min_pos[X_AXIS]);
    NOLESS(target[Y_AXIS], min_pos[Y_AXIS]);

    float negative_z_offset = 0;
    #if ENABLED(AUTO_BED_LEVELING_FEATURE)
      if (zprobe_zoffset < 0) negative_z_offset += zprobe_zoffset;
      if (home_offset[Z_AXIS] < 0) {
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            SERIAL_ECHOPAIR("> clamp_to_software_endstops > Add home_offset[Z_AXIS]:", home_offset[Z_AXIS]);
            SERIAL_EOL;
          }
        #endif
        negative_z_offset += home_offset[Z_AXIS];
      }
    #endif
    NOLESS(target[Z_AXIS], min_pos[Z_AXIS] + negative_z_offset);
  }

  if (max_software_endstops) {
    NOMORE(target[X_AXIS], max_pos[X_AXIS]);
    NOMORE(target[Y_AXIS], max_pos[Y_AXIS]);
    NOMORE(target[Z_AXIS], max_pos[Z_AXIS]);
  }
}
```
* code could be saved when #if would be used instead of if()
* When the handling for `zprobe_zoffset` in min is correct, why is it different for max.
```
static void _lcd_move(const char* name, AxisEnum axis, int min, int max) {
  if (encoderPosition != 0) {
    refresh_cmd_timeout();
    current_position[axis] += float((int)encoderPosition) * move_menu_scale;
    if (min_software_endstops && current_position[axis] < min) current_position[axis] = min;
    if (max_software_endstops && current_position[axis] > max) current_position[axis] = max;
...
```
* Code could be saved if  #if would be used.

* What do software endstops help us before homing?
** Wouldn't the 'save behaviour' be to allow only moves towards an hardware endstop (if they are working)
* What profit do we get from software endstops?
** Prevention from chrashes on the side where we do not have an hardware endstop.
** Prevention from hitting the hardware endstop (if the border would not be defined while the hardware endstop is triggered)

* What is the best action to take, if a software endstop  is hit?
** Before homing - do nothing, even clipping is unlogic and prevents us from getting some distance from z-max (for example).
** After homing, a pause seems to be a good action. If we sliced a part, not fitting in our build volume, we can't print it. A kill() is to much, because we do not have a safety problem - only the part will be ruined.

* Delta (and SCARA) printers are not handled adequate.

Roxy-3DPrintBoard, 2015-12-11:
> lower case names for the defined values

I have always wondered why these are not uppercase.   It seems very inconsistent. (But the rest of what you said is even more important!)

Wackerbarth, 2015-12-11:
I thin that the intention was to have this be a run-time option that the user could enable or disable at various times.

KiteLab, 2015-12-11:
Yep.
Could be handled with M120, M121. Our description only talks about 'endstops' - not the kind of endstop (same in http://reprap.org/wiki/G-code#M120:_Enable_endstop_detection). Maybe with parameters to distinguish between hard and soft.

KiteLab, 2015-12-11:
On the other side - if we have the endstops working proper (one day), there is no need to switch them off.

Wackerbarth, 2015-12-11:
We could have software end stops active all the time and just set them to trigger at a very large value. (Which would never happen)

AnHardt, 2015-12-11:
Or we could say. Before homing we have valid borders, but we no not know where we are, and therefore do not test. After homing we know where we are, do the test and can react adequate.

Part of the G92 problem is, that we can claim the object coordinate system as homed after a G92, but we do not have a relation to the stepper/tower system. So G92 also needs a previously homed system to work.
A sync. between the coordinate systems is absolutely necessary, at least for deltas, when we want to implement the software max-endstops in the tower coordinate system. (You remember the 3 spheres problem?)

Wackerbarth, 2015-12-11:
@KiteLab -- "Code could be saved if #if would be used."

I'm not sure that this is true. If you 
```
#define min_software_endstops true
``` 
, or declare
```
const bool min_software_endstops = true;
```
I think that you will find that the compiler will remove all of the code just as well as if you had eliminated it with a #if

KiteLab, 2015-12-11:
Maybe - or not. With an #if the compiler has no room for interpretation.

thinkyhead, 2016-03-23:
I'm not sure why these settings were originally created this way, except that perhaps at the time these options were introduced `#define` was not being used for items with `true` or `false` values.

There's some argument for using `const` more often than using `#define`. For one, the code tends to look nicer. For another, you can enforce type safety. But, with `const` you can't check the value at compile-time. (Of course, that is also the case for any `#define` that isn't set to an integer value.)

Anyway, for the record, the compiler is smart enough to recognize that the following source code will never do anything, so it completely leaves it out of the binary:
```cpp
if (const_variable_that_is_false) {
  printf("This code will not be included in the output.");
}
```
But why am I here? Because it seems like this is an important issue to clear up asap! So I'm visiting some of the issues that I've had on the back-burner during this release.

thinkyhead, 2016-03-23:
@KiteLab For reference, what is your `Z_PROBE_OFFSET_FROM_EXTRUDER` setting?

thinkyhead, 2016-03-23:
> When the handling for zprobe_zoffset in min is correct, why is it different for max?
The use of `zprobe_zoffset` in `clamp_to_software_endstops` seems completely incorrect to me. The probe offset should have no bearing on the minimum limit except in one specific case: When the probe is "higher" than the nozzle (because the nozzle acts as a probe and has to go "below" the bed to disengage a contact switch). But even in this case, the movement limits should _only_ be expanded during homing / leveling.

closed 2016-05-27 by jbrazio