MarlinFirmware/MarlinDev #307: createTemperatureLookupMarlin fails: Invalid Syntax

Kriechi, 2015-12-16:
I'm trying to run from [this commit](f6950acb8e93b94f9e32120c1a2468dad23d5020):
```
$ ./createTemperatureLookupMarlin.py --rp=4670 --t1=21.7:110700.0 --t2=190.0:623.0 --t3=259.0:192.0 --num-temps=72
  File "./createTemperatureLookupMarlin.py", line 76
    Tinv = self.c1 + self.c2*l + self.c3* l**3) # inverse temperature
                                              ^
SyntaxError: invalid syntax
```

Did this script change significantly? Because I copied the above command from a 2-ish year old folder I had, and I couldn't get it to work anymore...

KiteLab, 2015-12-16:
A yead ago this line looked like
```
Tinv = self.c1 + (self.c2*lnr) + (self.c3*pow(lnr,3))
```
I guess removing the ')' in the new version will do the job.

Wackerbarth, 2015-12-16:
There is a similar problem a few lines further down.

I'm making a PR to fix them

Wackerbarth, 2015-12-16:
@Kriechi -- Please try the code in PR #308 and let us know if it solves your problem.
If there are still errors, you can make a pull request against the CreateTemperature branch

Kriechi, 2015-12-16:
@Wackerbarth not really...

```
./createTemperatureLookupMarlin.py --rp=4670 --t1=21.7:110700.0 --t2=190.0:623.0 --t3=259.0:192.0 --num-temps=72                                                                                                                                                                                                  [0]
Traceback (most recent call last):
  File "./createTemperatureLookupMarlin.py", line 156, in <module>
    main(sys.argv[1:])
  File "./createTemperatureLookupMarlin.py", line 128, in main
    low_bound = t.temp(ARES-1);
  File "./createTemperatureLookupMarlin.py", line 75, in temp
    l = log(self.resist(adc))
ValueError: math domain error
```

Wackerbarth, 2015-12-16:
Well, it's just a python script. I found another issue that will now let it execute.

However, **** I KNOW NOTHING ABOUT THE CORRECTNESS OF THE CALCULATIONS ****

Please take further discussion over to #308

closed 2015-12-16 by Wackerbarth