MarlinFirmware/MarlinDev #310: Help needed including the correct files from Marlin 1.0.0 and configuring the pins_* file for my board

a4jp-com, 2015-12-20:
I have just moved from Marlin 1.0.0 to MarlinDev and I need help configuring the pins and buttons of my 3D printer. One problem I have is some of the old files may need to be included to make the clicker controller work properly.

I have installed the firmware and I can see the menu on the printer but none of the directional buttons work on the clicker so I can't use the menu yet.

Any help would greatly be appreciated. 

I spent a few days trying to get the pin numbers right before I realizing the line #define BTN_LEFT 83 and  #define BTN_RIGHT 82  etc wasn't connected to anything in the new MarlinDev files. I can see some lines of code that are needed in a file called ultralcd_E_Key_first.cpp but I'm not quite sure what to do with the code or just need help including the files.

Link to original files:

http://a4jp.com/3d-files/marlin-2015-01-06th-23-28-00.zip

Link to the new files:

https://github.com/a4jp-com/MarlinDev/

a4jp-com, 2015-12-20:
I found where the lines of code where missing from. The ultralcd.cpp file. Is it okay just to swap the files?

a4jp-com, 2015-12-20:
I've been going through the errors for hours but still can't get it to compile. Hopefully just removing errors from that file will make everything work again.

KiteLab, 2015-12-20:
The Marlin original menu system was made to use minimal hardware (pins). All what is needed is a method to go UP, DOWN and SELECT. To avoid a left of back button our first menu item points back to the previous menu.
The SCOOVO people made a major rework of that system. Porting back that to a recent Marlin is a major task.
There are extensions for other panels, using 5 buttons, already in Marlin. But non of them uses the LEFT button and most of them use the RIGHT button in the same way the SELECT button works. Sadly all these panels are not connected directly but via I²C or shift registers (to save pins).
As far i can see @japzi in https://github.com/MarlinFirmware/Marlin/pull/2848 shows a way to use directly connected buttons with a few lines of code.
You could try the snippet in https://github.com/MarlinFirmware/Marlin/pull/2848#issuecomment-164297601, Remove the `RIGIDBOT_PANEL` conditions and make the pin names fit. Then up and down should go up and down in the menu or when editing a value ENCODER_STEPS_PER_MENU_ITEM units; the right and left buttons will increase or decrease a value by 1 unit. Your central button should be connected to BTN_ENC.

a4jp-com, 2015-12-20:
Thank you for the wonderful info. Just to make sure I get everything right, which file would this code go in?

   #if BTN_ENC > 0 || Enable Clicker-controller (Panel)
       millis_t now = millis();
    #endif 

      if (now > next_button_update_ms) {
        if (READ(BTN_UP) == 0) {
          encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
          next_button_update_ms = now + 300;
        }
        else if (READ(BTN_DWN) == 0) {
          encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
          next_button_update_ms = now + 300;
        }
        else if (READ(BTN_LFT) == 0) {
          encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
          next_button_update_ms = now + 300;
        }
        else if (READ(BTN_RT) == 0) {
          encoderDiff = ENCODER_PULSES_PER_STEP;
          next_button_update_ms = now + 300;
        }
      }
    

    #if BTN_ENC > 0
      if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
    #endif

KiteLab, 2015-12-20:
ultralcd.cpp in lcd_buttons_update() around line 1846.
Have a look at the complete commit https://github.com/MarlinFirmware/Marlin/pull/2848/files, but reduce it to the buttons relevant part.

a4jp-com, 2015-12-20:
I made something like this but I'm not sure what I'm doing:

    #if ENABLED(ULTIPANEL)

      /**
       * Setup Rotary Encoder Bit Values (for two pin encoders to indicate movement)
       * These values are independent of which pins are used for EN_A and EN_B indications
       * The rotary encoder part is also independent to the chipset used for the LCD
       */
      #if defined(EN_A) && defined(EN_B)
        #define encrot0 0
        #define encrot1 2
        #define encrot2 3
        #define encrot3 1
      #endif

  /**
   * Read encoder buttons from the hardware registers
   * Warning: This function is called from interrupt context!
   */
  void lcd_buttons_update() {
    #if ENABLED(NEWPANEL)
      uint8_t newbutton = 0;
      if (READ(BTN_EN1) == 0) newbutton |= EN_A;
      if (READ(BTN_EN2) == 0) newbutton |= EN_B;
	  
      #if BTN_ENC > 0
        if (millis() > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
      #endif
	  
      buttons = newbutton;
      #if ENABLED(LCD_HAS_SLOW_BUTTONS)
        buttons |= slow_buttons;
      #endif
      #if ENABLED(REPRAPWORLD_KEYPAD)
        // for the reprapworld_keypad
        uint8_t newbutton_reprapworld_keypad = 0;
        WRITE(SHIFT_LD, LOW);
        WRITE(SHIFT_LD, HIGH);
        for (int8_t i = 0; i < 8; i++) {
          newbutton_reprapworld_keypad >>= 1;
          if (READ(SHIFT_OUT)) newbutton_reprapworld_keypad |= BIT(7);
          WRITE(SHIFT_CLK, HIGH);
          WRITE(SHIFT_CLK, LOW);
        }
        buttons_reprapworld_keypad = ~newbutton_reprapworld_keypad; //invert it, because a pressed switch produces a logical 0
        
		
		if (now > next_button_update_ms) {
			if (READ(BTN_UP) == 0) {
				encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
				next_button_update_ms = now + 300;
			}
			else if (READ(BTN_DWN) == 0) {
				encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
				next_button_update_ms = now + 300;
			}
			else if (READ(BTN_LFT) == 0) {
				encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
				next_button_update_ms = now + 300;
			}
			else if (READ(BTN_RT) == 0) {
				encoderDiff = ENCODER_PULSES_PER_STEP;
				next_button_update_ms = now + 300;
			}
		}	  
	  
	  #endif
    #else   //read it from the shift register
      uint8_t newbutton = 0;
      WRITE(SHIFT_LD, LOW);
      WRITE(SHIFT_LD, HIGH);
      unsigned char tmp_buttons = 0;
      for (int8_t i = 0; i < 8; i++) {
        newbutton >>= 1;
        if (READ(SHIFT_OUT)) newbutton |= BIT(7);
        WRITE(SHIFT_CLK, HIGH);
        WRITE(SHIFT_CLK, LOW);
      }
	  
	  
	  
	  
	  #if BTN_ENC > 0
	  if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
	  #endif
	  
	  
	  
	  
      buttons = ~newbutton; //invert it, because a pressed switch produces a logical 0
    #endif //!NEWPANEL

    //manage encoder rotation
    uint8_t enc = 0;
    if (buttons & EN_A) enc |= B01;
    if (buttons & EN_B) enc |= B10;
    if (enc != lastEncoderBits) {
      switch (enc) {
        case encrot0:
          if (lastEncoderBits == encrot3) encoderDiff++;
          else if (lastEncoderBits == encrot1) encoderDiff--;
          break;
        case encrot1:
          if (lastEncoderBits == encrot0) encoderDiff++;
          else if (lastEncoderBits == encrot2) encoderDiff--;
          break;
        case encrot2:
          if (lastEncoderBits == encrot1) encoderDiff++;
          else if (lastEncoderBits == encrot3) encoderDiff--;
          break;
        case encrot3:
          if (lastEncoderBits == encrot2) encoderDiff++;
          else if (lastEncoderBits == encrot0) encoderDiff--;
          break;
      }
    }
    lastEncoderBits = enc;
  }

  bool lcd_detected(void) {
    #if (ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)) && ENABLED(DETECT_DEVICE)
      return lcd.LcdDetected() == 1;
    #else
      return true;
    #endif
  }

  bool lcd_clicked() { return LCD_CLICKED; }

#endif // ULTIPANEL

KiteLab, 2015-12-20:
@a4jp-com 
You could make your code much better readable by using https://guides.github.com/features/mastering-markdown/ Syntax highlighting (first variety).
Do you have a REPRAPWORLD_KEYPAD? if not don't make your new code defendant on that. I really hoped
```
@@ -1843,10 +1854,37 @@ void lcd_reset_alert_level() { lcd_status_message_level = 0; }
    void lcd_buttons_update() {
      #if ENABLED(NEWPANEL)
        uint8_t newbutton = 0;
 -      if (READ(BTN_EN1) == 0) newbutton |= EN_A;
 -      if (READ(BTN_EN2) == 0) newbutton |= EN_B;
 +      #if BTN_EN1 > 0
 +        if (READ(BTN_EN1) == 0) newbutton |= EN_A;
 +      #endif
 +      #if BTN_EN2 > 0
 +        if (READ(BTN_EN2) == 0) newbutton |= EN_B;
 +      #endif
 +      #if ENABLED(RIGIDBOT_PANEL) || BTN_ENC > 0
 +        millis_t now = millis();
 +      #endif
 +      #if ENABLED(RIGIDBOT_PANEL)
 +        if (now > next_button_update_ms) {
 +          if (READ(BTN_UP) == 0) {
 +            encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_DWN) == 0) {
 +            encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_LFT) == 0) {
 +            encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_RT) == 0) {
 +            encoderDiff = ENCODER_PULSES_PER_STEP;
 +            next_button_update_ms = now + 300;
 +          }
 +        }
 +      #endif
        #if BTN_ENC > 0
 -        if (millis() > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
 +        if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
        #endif
        buttons = newbutton;
        #if ENABLED(LCD_HAS_SLOW_BUTTONS)
```
would be clear enough.
```
@@ -1843,10 +1854,37 @@ void lcd_reset_alert_level() { lcd_status_message_level = 0; }
    void lcd_buttons_update() {
      #if ENABLED(NEWPANEL)
        uint8_t newbutton = 0;
 -      if (READ(BTN_EN1) == 0) newbutton |= EN_A;
 -      if (READ(BTN_EN2) == 0) newbutton |= EN_B;
 +      #if BTN_EN1 > 0
 +        if (READ(BTN_EN1) == 0) newbutton |= EN_A;
 +      #endif
 +      #if BTN_EN2 > 0
 +        if (READ(BTN_EN2) == 0) newbutton |= EN_B;
 +      #endif
 +//      #if ENABLED(RIGIDBOT_PANEL) || BTN_ENC > 0
 +      #if BTN_ENC > 0
 +        millis_t now = millis();
 +      #endif
 +//      #if ENABLED(RIGIDBOT_PANEL)
 +        if (now > next_button_update_ms) {
 +          if (READ(BTN_UP) == 0) {
 +            encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_DWN) == 0) {
 +            encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_LFT) == 0) {
 +            encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
 +            next_button_update_ms = now + 300;
 +          }
 +          else if (READ(BTN_RT) == 0) {
 +            encoderDiff = ENCODER_PULSES_PER_STEP;
 +            next_button_update_ms = now + 300;
 +          }
 +        }
 +//      #endif
        #if BTN_ENC > 0
 -        if (millis() > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
 +        if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
        #endif
        buttons = newbutton;
        #if ENABLED(LCD_HAS_SLOW_BUTTONS)
```

a4jp-com, 2015-12-20:
I tried this code below but it compiles with lots of errors:

```
#if ENABLED(ULTIPANEL)
  /**
   * Setup Rotary Encoder Bit Values (for two pin encoders to indicate movement)
   * These values are independent of which pins are used for EN_A and EN_B indications
   * The rotary encoder part is also independent to the chipset used for the LCD
   */
  #if defined(EN_A) && defined(EN_B)
    #define encrot0 0
    #define encrot1 2
    #define encrot2 3
    #define encrot3 1
  #endif

  /**
   * Read encoder buttons from the hardware registers
   * Warning: This function is called from interrupt context!
   */
  void lcd_buttons_update() {
    #if ENABLED(NEWPANEL)
      uint8_t newbutton = 0;
	  #if BTN_EN1 > 0
        if (READ(BTN_EN1) == 0) newbutton |= EN_A;
      #endif
      #if BTN_EN2 > 0
        if (READ(BTN_EN2) == 0) newbutton |= EN_B;
      #endif
 
      #if BTN_ENC > 0
        millis_t now = millis();
      #endif
	  
	  if (now > next_button_update_ms) {
		if (READ(BTN_UP) == 0) {
			encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_DWN) == 0) {
			encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_LFT) == 0) {
			encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_RT) == 0) {
			encoderDiff = ENCODER_PULSES_PER_STEP;
			next_button_update_ms = now + 300;
		}
	  }	  
	  	  
	  #if BTN_ENC > 0
	    if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
	  #endif
	  	  
	  
      buttons = newbutton;
      #if ENABLED(LCD_HAS_SLOW_BUTTONS) //!NEWPANEL

    //manage encoder rotation
    uint8_t enc = 0;
    if (buttons & EN_A) enc |= B01;
    if (buttons & EN_B) enc |= B10;
    if (enc != lastEncoderBits) {
      switch (enc) {
        case encrot0:
          if (lastEncoderBits == encrot3) encoderDiff++;
          else if (lastEncoderBits == encrot1) encoderDiff--;
          break;
        case encrot1:
          if (lastEncoderBits == encrot0) encoderDiff++;
          else if (lastEncoderBits == encrot2) encoderDiff--;
          break;
        case encrot2:
          if (lastEncoderBits == encrot1) encoderDiff++;
          else if (lastEncoderBits == encrot3) encoderDiff--;
          break;
        case encrot3:
          if (lastEncoderBits == encrot2) encoderDiff++;
          else if (lastEncoderBits == encrot0) encoderDiff--;
          break;
      }
    }
    lastEncoderBits = enc;
  }
  }
  }

  bool lcd_detected(void) {
    #if (ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)) && ENABLED(DETECT_DEVICE)
      return lcd.LcdDetected() == 1;
    #else
      return true;
    #endif
	}

  bool lcd_clicked() { return LCD_CLICKED; }

#endif // ULTIPANEL
```

a4jp-com, 2015-12-20:
Here are the errors:

```
C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1813:0: error: unterminated #if

 #if ENABLED(ULTIPANEL)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:2:0: error: unterminated #if

 #if ENABLED(ULTRA_LCD)

 ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:36:0:

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd_implementation_hitachi_HD44780.h:799:15: warning: unused parameter 'pstr' [-Wunused-parameter]

   static void lcd_implementation_drawmenu_sd(bool sel, uint8_t row, const char* pstr, const char* filename, char* longFilename, uint8_t concat, char post_char) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1003:8: warning: unused parameter 'e' [-Wunused-parameter]

   void copy_and_scalePID_i(int e) {

        ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1007:8: warning: unused parameter 'e' [-Wunused-parameter]

   void copy_and_scalePID_d(int e) {

        ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1475:15: warning: unused parameter 'longFilename' [-Wunused-parameter]

   static void menu_action_sdfile(const char* filename, char* longFilename) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1485:15: warning: unused parameter 'longFilename' [-Wunused-parameter]

   static void menu_action_sddirectory(const char* filename, char* longFilename) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1492:13: warning: unused parameter 'pstr' [-Wunused-parameter]

 static void menu_action_setting_edit_bool(const char* pstr, bool* ptr) { *ptr = !(*ptr); }

             ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1742:6: warning: unused parameter 'persist' [-Wunused-parameter]

 void lcd_finishstatus(bool persist=false) {

      ^

In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev\Marlin.h:25:0,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.h:4,

                 from C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1:

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp: In function 'void lcd_buttons_update()':

C:\Users\design\Documents\Arduino\libraries\MarlinDev\fastio.h:25:27: error: 'DIOBTN_UP_RPORT' was not declared in this scope

 #define _READ(IO) ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))
[... 283 lines cut ...]
   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1376:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float5, ftostr5, 0.01)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1367:15: warning: 'void menu_action_setting_edit_callback_float51(const char*, float*, float, float, menuFunc_t)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1377:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float51, ftostr51, 10)

 ^

exit status 1
Error compiling.
```

KiteLab, 2015-12-20:
I really don't know why you did this
```
@@ -1808,65 +1808,99 @@ void lcd_reset_alert_level() { lcd_status_message_level = 0; }
   void lcd_buttons_update() {
     #if ENABLED(NEWPANEL)
       uint8_t newbutton = 0;
       if (READ(BTN_EN1) == 0) newbutton |= EN_A;
       if (READ(BTN_EN2) == 0) newbutton |= EN_B;
+
       #if BTN_ENC > 0
         if (millis() > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
       #endif
+
       buttons = newbutton;
       #if ENABLED(LCD_HAS_SLOW_BUTTONS)
         buttons |= slow_buttons;
       #endif
       #if ENABLED(REPRAPWORLD_KEYPAD)
-        // for the reprapworld_keypad
-        uint8_t newbutton_reprapworld_keypad = 0;
-        WRITE(SHIFT_LD, LOW);
-        WRITE(SHIFT_LD, HIGH);
-        for (int8_t i = 0; i < 8; i++) {
-          newbutton_reprapworld_keypad >>= 1;
-          if (READ(SHIFT_OUT)) newbutton_reprapworld_keypad |= BIT(7);
-          WRITE(SHIFT_CLK, HIGH);
-          WRITE(SHIFT_CLK, LOW);
+    // for the reprapworld_keypad
+    uint8_t newbutton_reprapworld_keypad = 0;
+    WRITE(SHIFT_LD, LOW);
+    WRITE(SHIFT_LD, HIGH);
+    for (int8_t i = 0; i < 8; i++) {
+      newbutton_reprapworld_keypad >>= 1;
+      if (READ(SHIFT_OUT)) newbutton_reprapworld_keypad |= BIT(7);
+      WRITE(SHIFT_CLK, HIGH);
+      WRITE(SHIFT_CLK, LOW);
+    }
+    buttons_reprapworld_keypad = ~newbutton_reprapworld_keypad; //invert it, because a pressed switch produces a logical 0
+
+
+    if (now > next_button_update_ms) {
+        if (READ(BTN_UP) == 0) {
+            encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
+            next_button_update_ms = now + 300;
         }
-        buttons_reprapworld_keypad = ~newbutton_reprapworld_keypad; //invert it, because a pressed switch produces a logical 0
-      #endif
-    #else   //read it from the shift register
-      uint8_t newbutton = 0;
-      WRITE(SHIFT_LD, LOW);
-      WRITE(SHIFT_LD, HIGH);
-      unsigned char tmp_buttons = 0;
-      for (int8_t i = 0; i < 8; i++) {
-        newbutton >>= 1;
-        if (READ(SHIFT_OUT)) newbutton |= BIT(7);
-        WRITE(SHIFT_CLK, HIGH);
-        WRITE(SHIFT_CLK, LOW);
-      }
-      buttons = ~newbutton; //invert it, because a pressed switch produces a logical 0
-    #endif //!NEWPANEL
-
-    //manage encoder rotation
-    uint8_t enc = 0;
-    if (buttons & EN_A) enc |= B01;
-    if (buttons & EN_B) enc |= B10;
-    if (enc != lastEncoderBits) {
-      switch (enc) {
-        case encrot0:
-          if (lastEncoderBits == encrot3) encoderDiff++;
-          else if (lastEncoderBits == encrot1) encoderDiff--;
-          break;
-        case encrot1:
-          if (lastEncoderBits == encrot0) encoderDiff++;
-          else if (lastEncoderBits == encrot2) encoderDiff--;
-          break;
-        case encrot2:
-          if (lastEncoderBits == encrot1) encoderDiff++;
-          else if (lastEncoderBits == encrot3) encoderDiff--;
-          break;
-        case encrot3:
-          if (lastEncoderBits == encrot2) encoderDiff++;
-          else if (lastEncoderBits == encrot0) encoderDiff--;
+        else if (READ(BTN_DWN) == 0) {
+            encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
+            next_button_update_ms = now + 300;
+        }
+        else if (READ(BTN_LFT) == 0) {
+            encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
+            next_button_update_ms = now + 300;
+        }
+        else if (READ(BTN_RT) == 0) {
+            encoderDiff = ENCODER_PULSES_PER_STEP;
+            next_button_update_ms = now + 300;
+        }
+    }     
+
+  #endif
+#else   //read it from the shift register
+  uint8_t newbutton = 0;
+  WRITE(SHIFT_LD, LOW);
+  WRITE(SHIFT_LD, HIGH);
+  unsigned char tmp_buttons = 0;
+  for (int8_t i = 0; i < 8; i++) {
+    newbutton >>= 1;
+    if (READ(SHIFT_OUT)) newbutton |= BIT(7);
+    WRITE(SHIFT_CLK, HIGH);
+    WRITE(SHIFT_CLK, LOW);
+  }
+
+
+
+
+  #if BTN_ENC > 0
+  if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
+  #endif
+
+
+
+
+  buttons = ~newbutton; //invert it, because a pressed switch produces a logical 0
+#endif //!NEWPANEL
+
+//manage encoder rotation
+uint8_t enc = 0;
+if (buttons & EN_A) enc |= B01;
+if (buttons & EN_B) enc |= B10;
+if (enc != lastEncoderBits) {
+  switch (enc) {
+    case encrot0:
+      if (lastEncoderBits == encrot3) encoderDiff++;
+      else if (lastEncoderBits == encrot1) encoderDiff--;
+      break;
+    case encrot1:
+      if (lastEncoderBits == encrot0) encoderDiff++;
+      else if (lastEncoderBits == encrot2) encoderDiff--;
+      break;
+    case encrot2:
+      if (lastEncoderBits == encrot1) encoderDiff++;
+      else if (lastEncoderBits == encrot3) encoderDiff--;
+      break;
+    case encrot3:
+      if (lastEncoderBits == encrot2) encoderDiff++;
+      else if (lastEncoderBits == encrot0) encoderDiff--;
           break;
       }
     }
     lastEncoderBits = enc;
   }
```
but it's completely different from my suggestion.
Maybe one can go your (maybe better) way, but me and the compiler don't understand what your way is.
I suggest restarting with a fresh copy of ultralcd.cpp

a4jp-com, 2015-12-20:
I'm very sorry. This is really confusing for me. I've tried copying what you wrote with the bits taken away (minus marks) but then I couldn't figure out what went wrong. The only thing I can do is try adding #endif or the closing curly brackets if lines look like they are missing the ends. I've never studied programming like this before so I'm just doing the best I can at the moment. Sorry about that. I'm just used to lines that look like this: if (x) {do something}; not #if x ...

a4jp-com, 2015-12-20:
I loaded the older code before and the compiler is working now I just have to go through the bits to edit one by one and maybe it will work. Thank you so much for your help.

```
#if ENABLED(ULTIPANEL)
  /**
   * Setup Rotary Encoder Bit Values (for two pin encoders to indicate movement)
   * These values are independent of which pins are used for EN_A and EN_B indications
   * The rotary encoder part is also independent to the chipset used for the LCD
   */
  #if defined(EN_A) && defined(EN_B)
    #define encrot0 0
    #define encrot1 2
    #define encrot2 3
    #define encrot3 1
  #endif

  /**
   * Read encoder buttons from the hardware registers
   * Warning: This function is called from interrupt context!
   */
  void lcd_buttons_update() {
    #if ENABLED(NEWPANEL)
      uint8_t newbutton = 0;
	  #if BTN_EN1 > 0
        if (READ(BTN_EN1) == 0) newbutton |= EN_A;
      #endif
      #if BTN_EN2 > 0
        if (READ(BTN_EN2) == 0) newbutton |= EN_B;
      #endif
 
      #if BTN_ENC > 0
        millis_t now = millis();
      #endif
	  
	  #if (now > next_button_update_ms) 
		if (READ(BTN_UP) == 0) {
			encoderDiff = -1 * ENCODER_STEPS_PER_MENU_ITEM;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_DWN) == 0) {
			encoderDiff = ENCODER_STEPS_PER_MENU_ITEM;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_LFT) == 0) {
			encoderDiff = -1 * ENCODER_PULSES_PER_STEP;
			next_button_update_ms = now + 300;
		}
		else if (READ(BTN_RT) == 0) {
			encoderDiff = ENCODER_PULSES_PER_STEP;
			next_button_update_ms = now + 300;
		}
	  #endif  
	  	  
	  #if BTN_ENC > 0
	    if (now > next_button_update_ms && READ(BTN_ENC) == 0) newbutton |= EN_C;
	  #endif
	  	  
	  
      buttons = newbutton;
      #if ENABLED(LCD_HAS_SLOW_BUTTONS) //!NEWPANEL

    //manage encoder rotation
    uint8_t enc = 0;
    if (buttons & EN_A) enc |= B01;
    if (buttons & EN_B) enc |= B10;
    if (enc != lastEncoderBits) {
      switch (enc) {
        case encrot0:
          if (lastEncoderBits == encrot3) encoderDiff++;
          else if (lastEncoderBits == encrot1) encoderDiff--;
          break;
        case encrot1:
          if (lastEncoderBits == encrot0) encoderDiff++;
          else if (lastEncoderBits == encrot2) encoderDiff--;
          break;
        case encrot2:
          if (lastEncoderBits == encrot1) encoderDiff++;
          else if (lastEncoderBits == encrot3) encoderDiff--;
          break;
        case encrot3:
          if (lastEncoderBits == encrot2) encoderDiff++;
          else if (lastEncoderBits == encrot0) encoderDiff--;
          break;
      }
    }
    lastEncoderBits = enc;
  }
  }
  }

  bool lcd_detected(void) {
    #if (ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)) && ENABLED(DETECT_DEVICE)
      return lcd.LcdDetected() == 1;
    #else
      return true;
    #endif
	}

  bool lcd_clicked() { return LCD_CLICKED; }
  #endif

#endif // ULTIPANEL
```

a4jp-com, 2015-12-20:
I understand what you mean now. I have just reloaded the file and added the code you suggested. Also removed the lines that weren't needed. Thank you for you clear instructions. I just needed a bit of thinking time to understand everything.

a4jp-com, 2015-12-20:
The directional buttons aren't doing anything yet but the code compiled. Here are the latest warnings:

```
In file included from C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:36:0:

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd_implementation_hitachi_HD44780.h:799:15: warning: unused parameter 'pstr' [-Wunused-parameter]

   static void lcd_implementation_drawmenu_sd(bool sel, uint8_t row, const char* pstr, const char* filename, char* longFilename, uint8_t concat, char post_char) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1003:8: warning: unused parameter 'e' [-Wunused-parameter]

   void copy_and_scalePID_i(int e) {

        ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1007:8: warning: unused parameter 'e' [-Wunused-parameter]

   void copy_and_scalePID_d(int e) {

        ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1475:15: warning: unused parameter 'longFilename' [-Wunused-parameter]

   static void menu_action_sdfile(const char* filename, char* longFilename) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1485:15: warning: unused parameter 'longFilename' [-Wunused-parameter]

   static void menu_action_sddirectory(const char* filename, char* longFilename) {

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1492:13: warning: unused parameter 'pstr' [-Wunused-parameter]

 static void menu_action_setting_edit_bool(const char* pstr, bool* ptr) { *ptr = !(*ptr); }

             ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1742:6: warning: unused parameter 'persist' [-Wunused-parameter]

 void lcd_finishstatus(bool persist=false) {

      ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1363:15: warning: 'void menu_action_setting_edit_float43(const char*, float*, float, float)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1375:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float43, ftostr43, 1000)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1363:15: warning: 'void menu_action_setting_edit_long5(const char*, long unsigned int*, long unsigned int, long unsigned int)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1379:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(unsigned long, long5, ftostr5, 0.01)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1367:15: warning: 'void menu_action_setting_edit_callback_float3(const char*, float*, float, float, menuFunc_t)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1373:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float3, ftostr3, 1)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1367:15: warning: 'void menu_action_setting_edit_callback_float32(const char*, float*, float, float, menuFunc_t)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1374:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float32, ftostr32, 100)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1367:15: warning: 'void menu_action_setting_edit_callback_float5(const char*, float*, float, float, menuFunc_t)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1376:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float5, ftostr5, 0.01)

 ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1367:15: warning: 'void menu_action_setting_edit_callback_float51(const char*, float*, float, float, menuFunc_t)' defined but not used [-Wunused-function]

   static void menu_action_setting_edit_callback_ ## _name (const char* pstr, _type* ptr, _type minValue, _type maxValue, menuFunc_t callback) { \

               ^

C:\Users\design\Documents\Arduino\libraries\MarlinDev\ultralcd.cpp:1377:1: note: in expansion of macro 'menu_edit_type'

 menu_edit_type(float, float51, ftostr51, 10)

 ^

Sketch uses 98,186 bytes (38%) of program storage space. Maximum is 258,048 bytes.
Global variables use 4,555 bytes of dynamic memory.
```

a4jp-com, 2015-12-20:
I tried commenting out the RIGIDBOT_PANEL lines and that is where the error occurs. Besides that everything looks okay.

a4jp-com, 2015-12-20:
Is there standard code for these buttons?

Is BTN_EN1 is a term the company just made up or the term that should be used.

Also is there a physical way to check the correct pin numbers? At least if I can check the pin number on the actual board it would be better than just guessing. I do have the original files so I think those numbers are correct though.

Roxy-3DPrintBoard, 2015-12-20:
I would probably put a helper function in place.   Something like:

~~~~
    case 45: //M45 - Scan pin status via gcode
      {
        int i, j, k;
	while (1)
	for(j=0; j<100; j++) {
		for(int8_t i = 0; i < (int8_t)(sizeof(sensitive_pins)/sizeof(int)); i++) {
		  if (sensitive_pins[i] == j )
		    continue;
		}
		pinMode(j, INPUT);
		k = digitalRead(pin_number);
        	SERIAL_PROTOCOL( k );
		if ( j%10 == 0 )
        		SERIAL_PROTOCOL( ' ' );
		if ( j%30 == 0 )
        		SERIAL_PROTOCOL( '\n' );
	}
        SERIAL_PROTOCOL( '\n' );
      }
     break;
~~~~

With a debugged version of that, you would be able to see what pin toggled when you pressed a button or end stop.

a4jp-com, 2015-12-20:
Very nice. Which file do I put that in? I tried the pins_A4JP.h file and ultralcd.cpp file but maybe I had it in the wrong spot.

Roxy-3DPrintBoard, 2015-12-20:
That would go in Marlin_main.cpp.    You would put it right in front (or after) another Mxx command.   That code is close, but almost for sure something will have to be tweaked.

a4jp-com, 2015-12-21:
Thank you for the hint but I still need help.

If I have a fresh copy of the file could you tell me which line I should put the code into? Or any other ideas?

```
/** Button and pin checker **/
#define one 0
#ifdef one
    printf("one is defined ");
	case 45: //M45 - Scan pin status via gcode
      {
        int i, j, k;
    while (1)
    for(j=0; j<100; j++) {
        for(int8_t i = 0; i < (int8_t)(sizeof(sensitive_pins)/sizeof(int)); i++) {
          if (sensitive_pins[i] == j )
            continue;
        }
        pinMode(j, INPUT);
        k = digitalRead(pin_number);
            SERIAL_PROTOCOL( k );
        if ( j%10 == 0 )
                SERIAL_PROTOCOL( ' ' );
        if ( j%30 == 0 )
                SERIAL_PROTOCOL( '\n' );
    }
        SERIAL_PROTOCOL( '\n' );
      }
     break;
#endif
#ifndef one
    printf("one is not defined ");
	case 45: //M45 - Scan pin status via gcode
      {
        int i, j, k;
    while (1)
    for(j=0; j<100; j++) {
        for(int8_t i = 0; i < (int8_t)(sizeof(sensitive_pins)/sizeof(int)); i++) {
          if (sensitive_pins[i] == j )
            continue;
        }
        pinMode(j, INPUT);
        k = digitalRead(pin_number);
            SERIAL_PROTOCOL( k );
        if ( j%10 == 0 )
                SERIAL_PROTOCOL( ' ' );
        if ( j%30 == 0 )
                SERIAL_PROTOCOL( '\n' );
    }
        SERIAL_PROTOCOL( '\n' );
      }
     break;
#endif
```

a4jp-com, 2015-12-21:
@Wackerbarth the c49dc7e file compiled so it's looking good. Can't wait to get the buttons working again. I made a commit of the files and synced them in GitHub. 

Just one quick question, will I be able to print with this version of the software in the meantime? 

Time to get a bit of sleep, it's 5:49 am here now.

Wackerbarth, 2015-12-21:
I can't say for sure. But I think that it should work.

a4jp-com, 2015-12-22:
I was able to print through the USB cable. 

There was a bit of corner ringing but this could be from either the updated version of Slic3r I'm using or the Marlin software. I'll have to do some more testing to figure out what is happening there.

a4jp-com, 2015-12-23:
@Wackerbarth I've just finished work and can continue testing pins/buttons from now. 

https://hangouts.google.com/call/jgx5ywnjhtjqdl4tf76f6o3jlqa

a4jp-com, 2015-12-23:
I just added one extra line to the end of the code in M45_Check_Buttons to make checking the code a little easier.

```
// M115 Capabilities String

#include "MarlinFirmware.h"
#include "host_interface/host_io.h"
#include "fastio.h"

const int sensitive_pins[] = SENSITIVE_PINS; ///< Sensitive pin list for M45

void gcode_M45() {
  millis_t stop_ms = millis() + (5 * 1000UL);
  while ((unsigned int)(millis()) < stop_ms) {
	millis_t last_ms = millis();
	void idle();
    while (((unsigned int)(millis() - last_ms)) < (500UL)) idle();
    for (uint8_t pin_number = 50; pin_number < NUM_DIGITAL_PINS; pin_number++) {
      uint8_t v;
      // Arduino pin numbering
      pinMode(pin_number, INPUT);
      v = digitalRead(pin_number);
      if (v)  SERIAL_PROTOCOL('1');
      else  SERIAL_PROTOCOL('0');
      if (pin_number % 10 == 9)
        SERIAL_PROTOCOL(' ');
      //if ((pin_number-50) % 30 == 0)
       // SERIAL_PROTOCOL('\n');
    }
    SERIAL_PROTOCOL('\n');
  }
  SERIAL_PROTOCOL('ended');
}```

Default loop:
1101011000 0000000000 1001110000 11

Home Button: 
1101011000 0000000000 1101110000 01

Controller's Center Button:
1101011000 0000000000 1101110000 10

Up:
1101011000 0000000000 1101110000 11

Down:
1101011000 0000000000 1101110000 11

Left:
1101011000 0000000000 1101110000 11

Right:
1101011000 0000000000 1101110000 11

KiteLab, 2015-12-23:
Maybe the missed pins need pullups. Add a `digitalWrite(pin_number);` before the `digitalRead(pin_number);` . Be careful! This may have unwanted effects like running fans, or heating heaters!

KiteLab, 2015-12-23:
At the moment you are testing pins up to 81. In your original pin file you have higher numbers.
```
//#define SW_DOWN  85 //ext2_6
#define BTN_EN1  85 //ext2_6
//#define SW_UP 84 //ext2_8 
#define BTN_EN2 84 //ext2_8 
//#define SW_ENTER 81 //ext_14
#define BTN_ENC 81 //ext_14
#define BTN_LEFT 83  //ext_10
#define BTN_RIGHT 82  //ext_12
#define BTN_HOME 80  //ext_16
//
```

KiteLab, 2015-12-23:
ArduinoIO != FastIO.
digitalRead(pin_number);  != READ(pin_number);
Neither the pin_number_range nor the sequence is necessary the same.

Here likely 
https://github.com/MarlinFirmware/MarlinDev/blob/MarlinPlatform-1.6.7/variants/rambo/pins_arduino.h#L30
should be changed to 85.

Wackerbarth, 2015-12-23:
@KiteLab -- I agree with your last. Knowing nothing about a real RAMBo card, I took the variants file that was used for this card in early versions of RAMBo.

That file has
```
#define NUM_DIGITAL_PINS            82
#define NUM_ANALOG_INPUTS           16
```
But, interestingly, the file also has
```
	PD 	, // PD 4 ** 81 ** D81
	PD 	, // PD 5 ** 82 ** D82
	PD 	, // PD 6 ** 83 ** D83
	PH 	, // PH 2 ** 84 ** D84
	PH 	, // PH 7 ** 85 ** D85
```
So, it would seem that we should be using 85 for NUM_DIGITAL_PINS.

@a4jp-com -- try making that substitution and see if you can find the additional buttons at the end of the line

If anyone has an actual schematic, perhaps we can reconsider the mappings.

KiteLab, 2015-12-23:
86 we also have pin0

Wackerbarth, 2015-12-23:
@KiteLab -  Correct. I'll set up a PR to make that change in the variant file.

Roxy-3DPrintBoard, 2015-12-23:
> I just added one extra line to the end of the code in M45_Check_Buttons to make checking the code a little easier.
~~~~
Default loop:
1101011000 0000000000 1001110000 11

Home Button: 
1101011000 0000000000 1101110000 01

Controller's Center Button:
1101011000 0000000000 1101110000 10

Up:
1101011000 0000000000 1101110000 11

Down:
1101011000 0000000000 1101110000 11

Left:
1101011000 0000000000 1101110000 11

Right:
1101011000 0000000000 1101110000 11
~~~~
Do we get to see the output with NUM_DIGITAL_PINS set at 86 ?   Do all of the different buttons show up?

a4jp-com, 2015-12-23:
I found out my controller can sometimes have a diagonal value when 2 of the contacts get hit.

a4jp-com, 2015-12-23:
After #define NUM_DIGITAL_PINS was set to 86 in the C:\Users\design\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.6.7\variants\rambo\pins_arduino.h file all the values could be seen.

```
// M115 Capabilities String

#include "MarlinFirmware.h"
#include "host_interface/host_io.h"
#include "fastio.h"

const int sensitive_pins[] = SENSITIVE_PINS; ///< Sensitive pin list for M45

void gcode_M45() {
  millis_t stop_ms = millis() + (8 * 1000UL);
  while ((unsigned int)(millis()) < stop_ms) {
	millis_t last_ms = millis();
	void idle();
    while (((unsigned int)(millis() - last_ms)) < (500UL)) idle();
    for (uint8_t pin_number = 80; pin_number < NUM_DIGITAL_PINS; pin_number++) {
      uint8_t v;
      // Arduino pin numbering
      pinMode(pin_number, INPUT);
      digitalRead(pin_number);
	  v = digitalRead(pin_number);
      if (v)  SERIAL_PROTOCOL('1');
      else  SERIAL_PROTOCOL('0');
      if (pin_number % 10 == 9)
        SERIAL_PROTOCOL(' '); //Adds space after 10 characters
    }
    SERIAL_PROTOCOL('\n');
  }
  SERIAL_PROTOCOLPGM("Button Test Ended"); //Double quotes for a text string
  SERIAL_EOL;
}
```

I tried adding digitalWrite(pin_number); but it caused an error.

This code works but you will need to disconnect and reconnect each time you run M45. If you run it 2 times in a row there will be an endless loop for the second M45 for some reason.

Roxy-3DPrintBoard, 2015-12-23:
Well...   That isn't really code to be added to Marlin.  It was just to see what pin numbers gave you values when you pressed different buttons on the LCD Panel.   The problem is the code has no way of knowing what pins are input and which are output.   And you are messing with pins that are defined as 'Sensitive' now.   It is very likely one of those Sensitive pins doesn't like being switched to an input and after you do that Marlin's behavior is going to be unpredictable.

Wackerbarth, 2015-12-23:
@Roxy-3DPrintBoard  -- The issue of "sensitive pins" is MUCH worse than you are imagining.
But they are not an issue here because there are no MARLIN pins numbers which are that high.

First, there are at least three different interpretations of pin = 5.  It could be the fast_io numbering system that Marlin uses, the Arduino digital pin number or the Arduino analog pin number.

From context, we can probably keep the latter two straight. However, in the case of digital pins, we (supposedly) have a list of the MARLIN pin numbers, but no way to relate them to the ARDUINO digital pin numbers.  To further complicate matters, I have seen pins_* files which are attempting to set sensitive pins (like additional stepper drivers) using the ARDUINO numbers for which there is no MARLIN fast_io mapping available.

As to the applicability of code of this nature as a part of "MARLIN", I will agree that it is not code which would be a part of "normal" firmware.  However, I do see a place for it (just like the M100 memory checker) as an optional "debug" feature that can readily be chosen at configuration.

Roxy-3DPrintBoard, 2015-12-23:
> As to the applicability of code of this nature as a part of "MARLIN", I will agree that it is not code which would be a part of "normal" firmware. However, I do see a place for it (just like the M100 memory checker) as an optional "debug" feature that can readily be chosen at configuration.

Yes...   That might make sense because people keep running into trouble trying to find a GPIO pin on their controller board.

> From context, we can probably keep the latter two straight. However, in the case of digital pins, we (supposedly) have a list of the MARLIN pin numbers, but no way to relate them to the ARDUINO [...]

I had some back and forth emails with Lincomatic a long time ago.  He is the one that came up with the board definitions for the PrintRBoard that I have on my first 3D-Printer.   And he got the PrintRBoard to work with Arduino v.22.    He submitted some changes to get the Fast-IO and the Arduino Digital Pin numbers to be consistent.   But his Pull Request was never merged (and he is a bit miffed about that).   I think this was at the time that Marlin development had stopped and nobody was adding to the Marlin code base.

I'm wondering how hard it would be to get the differences cleaned up so every thing is consistent???

Here are some of the things that he published:

https://github.com/MarlinFirmware/Marlin/issues/112

https://github.com/kliment/Sprinter/commit/8b8f5bf47420238c525b2990747f47c5e3248369

http://blog.lincomatic.com/?p=1508

http://blog.lincomatic.com/?p=537

AnHardt, 2015-12-24:
Compared pin mappings as a .csv https://gist.github.com/AnHardt/55198c7ff75f6b3d0229

Roxy-3DPrintBoard, 2015-12-24:
@AnHardt    Can we consolidate the two different sets of pin numbers?

AnHardt, 2015-12-24:
Simple conversion arrays will take up to 88 bytes per direction.
But at least for the analogWrite()_cals __I__ have no better idea.
Maybe someone else has an idea for a FastIO analogWrite replacement macro.

More difficult is to find out what pin_file used what system - and if there are inconsistencys in.

Roxy-3DPrintBoard, 2015-12-24:
Can somebody explain to me why it makes sense to have two different sets of pin #'s to describe the GPIO pins?   That just seems like it is asking for trouble.  Why did they do that?

a4jp-com, 2015-12-24:
This code was very helpful trying to figure out what pins were being used but I also agree it should be in a debug area or folder.

AnHardt, 2015-12-24:
@Roxy-3DPrintBoard 
As the name suggests fastio produces very fast and short code when the pin number is known at compile time.
While the Arduino standard functions look_up/calculate the right registers and positions at runtime, that work is done at compile time with fastio.
Why two systems?
To be honest, this conversion from pin number to register/position is also done at a third place - in `Sd2PinMap.h`.
Because most of our programmers are unaware of the problem. They follow the tracks they are used to.
Because to few (no one) of the regulars here have a wide enough view of Marlin.

a4jp-com, 2015-12-24:
I know the down button is only on pin 85. Is there some code that could easily make that button work?

Roxy-3DPrintBoard, 2015-12-24:
> Also is there a way to find out what type of controller I have through the Marlin Firmware?

I am not aware of any way to do that.  I do believe you can execute some instruction sequences to find out what type of processor is there.  But I don't think you can easily map that to a particular board type.

> This code was very helpful trying to figure out what pins were being used but I also agree it should be in a debug area or folder.

I like the changes you did to get it working.  Everybody agrees it is not in the main line code!  I'm thinking I'm going to expand and polish it.  The features I'm thinking about adding are:

1. The ability to set the start and end pin numbers for a scan.  But have them default to what it is doing now.
2. The ability to set the pause or delay between scans.
3. The ability to set the number of scans across the range of pin numbers.
4. The ability to pulse (non-sensitive) pins so they can be identified with a logic probe or volt meter (instead of reading values, it would pulse the pin as an output)

a4jp-com, 2015-12-24:
Very nice. 

@Wackerbarth is the super guy behind most of the the code here though. Not me. Also @KiteLab. I just did some testing and asked for help moving from Marlin 1.0.0. Thank you everyone for all your help.

Roxy-3DPrintBoard, 2015-12-24:
> As the name suggests fastio produces very fast and short code when the pin number is known at compile time. While the Arduino standard functions look_up/calculate the right registers and positions at runtime, that work is done at compile time with fastio.  Why two systems?

@AnHardt  THANK YOU! That explanation really helps.   I guess I'm going to be crawling through the code looking for places where we don't know the pin numbers at compile time.    I'm curious if we can switch over to a single method.    Does anybody know off of the top of their head places where we don't know the pin numbers at compile time?

KiteLab, 2015-12-24:
M42, the new test loop

a4jp-com, 2015-12-24:
I just tried uploading some of the newer code from https://github.com/Wackerbarth/MarlinDev/tree/Dec24 and now my screen is blank.

Roxy-3DPrintBoard, 2015-12-24:
> I just tried uploading some of the newer code from https://github.com/Wackerbarth/MarlinDev/tree/Dec24 and now my screen is blank.

Wow!   I just looked through all the commits on that branch.   Wackerbarth has been busy with the chainsaw!

Wackerbarth, 2015-12-24:
@KiteLab  -- But M42 is wrong code.
It tests for "SENSITIVE PINS" in the fast_io domain and then does "digital_write" in the Arduino domain.

We need to replace "digital_write" with code whose input is in the fast_io domain.
We cannot use the fast_io macros directly because they require that the pin number be known at compile time.

a4jp-com, 2015-12-24:
@Wackerbarth saved the day. The code is compiling now and my screen going blank had something to do with Windows 10. After I though it might be a good idea to uninstall the board driver when the problem started happening it just kept bring up an error and wouldn't load the driver. An upgrade of Windows 10 over the broken copy using MediaCreationTool.exe (upgrade) fixed everything though. If you have a RAMBo board and come across this problem hopefully that is how you can fix it. It's 6:44 am now. Maybe I should get a bit of sleep soon lol.

KiteLab, 2015-12-25:
@Wackerbarth 
As far as i know FastIO works well with variables, but is not that fast any more.

Wackerbarth, 2015-12-25:
I don't think so.  Changing digitalRead() to READ()

n file included from /Users/wacky/Documents/Arduino/libraries/MarlinDev/MarlinFirmware.h:29:0,
                 from /Users/wacky/Documents/Arduino/libraries/MarlinDev/gCodes/M45_Check_Buttons.cpp:5,
                 from /Users/wacky/Documents/Arduino/libraries/MarlinDev/M45_link.cpp:3:
/Users/wacky/Documents/Arduino/libraries/MarlinDev/gCodes/M45_Check_Buttons.cpp: In function 'void gcode_M45()':
/Users/wacky/Documents/Arduino/libraries/MarlinDev/fastio.h:55:27: error: 'DIOpin_number_RPORT' was not declared in this scope
 #define _READ(IO) ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))
                           ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/fastio.h:95:19: note: in expansion of macro '_READ'
 #define READ(IO)  _READ(IO)
                   ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/gCodes/M45_Check_Buttons.cpp:20:11: note: in expansion of macro 'READ'
       v = READ(pin_number);
           ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/fastio.h:55:54: error: 'DIOpin_number_PIN' was not declared in this scope
 #define _READ(IO) ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))
                                                      ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/fastio.h:46:28: note: in definition of macro 'MASK'
   #define MASK(PIN)  (1 << PIN)
                            ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/fastio.h:95:19: note: in expansion of macro '_READ'
 #define READ(IO)  _READ(IO)
                   ^
/Users/wacky/Documents/Arduino/libraries/MarlinDev/gCodes/M45_Check_Buttons.cpp:20:11: note: in expansion of macro 'READ'
       v = READ(pin_number);
           ^

a4jp-com, 2015-12-27:
I've updated my printer to this version of Marlin (https://github.com/Wackerbarth/MarlinDev/tree/Glen) and it compiles. I just can't access the menu as the buttons aren't set up yet. Is there anyone here that can help set up the buttons or code needed? I would love to be able to use the menu on the printer instead of just being able to use the usb cable only.

Roxy-3DPrintBoard, 2015-12-27:
I just looked and the UltraLCD.cpp file has not changed from what I'm using and the menu scrolls.  Do you have a rotary encoder on your LCD Panel?   Is the problem that you have to use direction buttons to move around?

a4jp-com, 2015-12-27:
I don't have a rotary controller just directional buttons.

a4jp-com, 2015-12-27:
[image]

Wackerbarth, 2015-12-27:
On Dec 27, 2015, at 3:11 PM, Roxy-3DPrintBoard <notifications@github.com> wrote:
> I just looked and the UltraLCD.cpp file has not changed from what I'm using and the menu scrolls. Do you have a rotary encoder on your LCD Panel? Is the problem that you have to use direction buttons to move around?
> 
This version includes the Rigidbot code that Ruwan J Egoda Gamage contributed.

Have you set all of the #define’s to enable it?

Richard

Roxy-3DPrintBoard, 2015-12-27:
Looking through the stander UltraLCD.cpp code...  Besides Rigidbot...  There is this REPRAPWORLD_KEYPAD stuff that lets up/down/left/right work:

~~~~
/**
 *
 * Handlers for RepRap World Keypad input
 *
 */
#if ENABLED(REPRAPWORLD_KEYPAD)
  static void reprapworld_keypad_move_z_up() {
    encoderPosition = 1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_z();
  }
  static void reprapworld_keypad_move_z_down() {
    encoderPosition = -1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_z();
  }
  static void reprapworld_keypad_move_x_left() {
    encoderPosition = -1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_x();
  }
  static void reprapworld_keypad_move_x_right() {
    encoderPosition = 1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_x();
  }
  static void reprapworld_keypad_move_y_down() {
    encoderPosition = 1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_y();
  }
  static void reprapworld_keypad_move_y_up() {
    encoderPosition = -1;
    move_menu_scale = REPRAPWORLD_KEYPAD_MOVE_STEP;
    lcd_move_y();
  }
  static void reprapworld_keypad_move_home() {
    enqueuecommands_P((PSTR("G28"))); // move all axis home
  }
#endif // REPRAPWORLD_KEYPAD
~~~~

[11 more replies]

closed 2016-04-18 by thinkyhead