MarlinFirmware/MarlinDev #311: There is a double percent make on my screen.

a4jp-com, 2015-12-20:
[image]

This is not a big thing but how do I remove the extra percentage mark?

Roxy-3DPrintBoard, 2015-12-20:
Quick question:   What happens if you adjust the feed rate to 100.1% and 99.9%   ?   I'm wondering if that extra % is left over from a previous refresh cycle of the display?

a4jp-com, 2015-12-20:
I'm trying to fix the button code here (https://github.com/MarlinFirmware/MarlinDev/issues/310#issuecomment-166125362) and can't compile because of an error, but as soon as I get that figured out I can try to set the feed rate. Sorry about that.

KiteLab, 2015-12-20:
```
  #if LCD_HEIGHT > 2

    #if LCD_WIDTH < 20

      #if ENABLED(SDSUPPORT)
-        lcd.setCursor(0, 2);
+        lcd.setCursor(0, 1);
        lcd_printPGM(PSTR("SD"));
        if (IS_SD_PRINTING)
          lcd.print(itostr3(card.percentDone()));
        else
          lcd_printPGM(PSTR("---"));
          lcd.print('%');
      #endif // SDSUPPORT

    #else // LCD_WIDTH >= 20

      lcd.setCursor(0, 1);
```

a4jp-com, 2015-12-20:
@KiteLab that tiny change of the number 2 to 1 fixed 2 bugs ^^ Thank you so much. I can now see the SD card line.

[image]

@Roxy-3DPrintBoard would you like me to test the feed rate or is this looking okay now?

Roxy-3DPrintBoard, 2015-12-20:
No!   I was just guessing at possible causes.  KiteLab nailed this one!

a4jp-com, 2015-12-20:
Okay. Nice. 

If you could add this to the updated dev files it would be great.

Thank you so much @KiteLab .

closed 2015-12-20 by a4jp-com