MarlinFirmware/MarlinDev #313: Analog temp input pins with pull up/down resistor ?

maukcc, 2015-12-22:
Is the input pin for temp0 without internal pull up or down resistor, and the others with?
As I get different readings for the inputs when nothing is connected.
I use a thermocouple with a user defined thermistor table.
If I put a 1Mohm resistor on the input pins and connect it to GND the temp0 reads zero degrees(correct), the other temps read 300(ish)

maukcc, 2015-12-22:
It is only for TEMP_1 that is gives strange readings
And it is not the physical pin, because when I switch the TEMP_0 and TEMP_1 pin, the fault switches as well.
so,
normal:
 #define TEMP_0_PIN             0 
 #define TEMP_1_PIN             9
switched:
 #define TEMP_0_PIN             9 
 #define TEMP_1_PIN             0

So it does not matter which physical pin I assign to TEMP_1, it always gives strange values.

Wackerbarth, 2015-12-22:
What version of the source are you using? And how do you have things configured?
(The preferred method for presenting the requested information is to fork MarlinDev on Github and push your changes to a personal branch. Then you can just provide the reference to that repository and we can then duplicate your effort to look for the cause.)

maukcc, 2015-12-22:
Also, if I have the temps connected with 2 extruders:
T0=18
T1=33
with 3 extruders
T0=18
T1=23
T2=18
Voltage on the input pins is identical all over, but T1 drops 10 degrees when T2 is added.

maukcc, 2015-12-22:
@ Wackerbarth : that is a good idea, I will do that, as I made a lot of changes.
none of them should interfere with this I think.

thinkyhead, 2016-04-22:
If this is resolved, it can be closed.

closed 2016-05-25 by boelle