MarlinFirmware/MarlinDev #314: G28 homing improvements

in2-arduino, 2015-12-23:
I'm looking at the way homing currently works for non-delta configurations and would like to propose couple of improvements. Main issue is Z_RAISE_BEFORE_HOMING which suggests the head will always raise before homing which is not always true. 

1 - Current implementation sets Z destination like this:

    // Raise Z before homing any other axes
    // (Does this need to be "negative home direction?" Why not just use Z_RAISE_BEFORE_HOMING?)
    destination[Z_AXIS] = -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS);

which means it depends on the direction the machine homes to. Nearby comment suggests somebody else thought this was questionable. I can’t think of any situation where it would be safe to bring head CLOSER to the bed during homing. However, if the machine has a Z endstop, then raising the head should always be a safe move. **I’m proposing to use just:**

    destination[Z_AXIS] = Z_RAISE_BEFORE_HOMING;

2 - Next issue is that head is not raised if only X or Y is homed. Also, if Z_SAFE_HOMING is enabled, head will be raised only after X are Y are homed. Both behaviors are IHMO wrong, and will crash the head into clamps and other protrusions while homing X and Y axis. **I’m proposing to implement head raise as first step in any homing procedure regardless of axis being homed and Z_SAFE_HOMING parameter.** This by itself has a side effect of multiplying head rises if several homing commands are given, but it’s non issue – please read on.

3 - Then we have the problem that current implementation doesn’t raise the head relatively to current position but sends it to absolute height specified by Z_RAISE_BEFORE_HOMING. Since machine is usually at Z=0 at the time of homing, the move seems to be relative but in reality it is not. If the user jogs the head up to a Z > Z_RAISE_BEFORE_HOMING prior to  homing, the “raising” move will actually bring the head back down to Z = Z_RAISE_BEFORE_HOMING. I didn’t try what happens if Z endstop is hit during jog, but depending on the implementation, it’s possible the raise might actually bring the head closer to table than Z_RAISE_BEFORE_HOMING and crash it in process. **I propose to change the homing procedure to only perform raising movement if current Z position is less then Z_RAISE_BEFORE_HOMING.** This is safe since current position is initialized to 0 during startup. Any jogging user performs before homing can only move the head away from bed and any change of Z position will be reflected in current_position array. If user already raised the head at least to Z_RAISE_BEFORE_HOMING no further raising before homing is necessary. This also solves the problem of multiplying head rises mentioned before since only first homing command will need to perform raise.

4 - Current homing implementation can throw error (MSG_ZPROBE_OUT, MSG_POSITION_UNKNOWN) and abort homing without aborting the print (didn’t try this, so I might be wrong). That means that print could continue with un-homed Z axis. Homing also currently ignores IsRunning() check that other move commands seem to respect. **I’m proposing to abort print completely in case a homing command fails.**

5 - Homing toward the Z minstop, with Z_SAFE_HOMING flag enabled currently only  checks that probe is positioned inside X and Y endstop bounds which is OK for most mechanical probes and for machines where bed is equal or bigger than endstop bounds, but inductive and some other non-contact probes must be well inside the bed to properly function and also there are machines which can be positioned outside bed. **I propose to change the check to include a clearance from bed edge or only allow Z probing at the exact Z_SAFE_HOMING_X/Y_POINT coordinates. Alternatively, head could be moved to safe X and Y position before probing, possibly by enabling additional flag in configuration file.**

I implemented most of this already, but I'm not familiar with the codebase so I'd like more input on these ideas before actually devoting time to do them properly and making a pull request.

Also, I’d be grateful if someone could explain (or point me towards documentation) exactly what the following variables and functions in marlin_main.cpp should contain and do. I can guess most of it by looking at the code but it’s time consuming and error prone. Finer details and gotchas would be especially useful, for instance what happens with current position if endstop is hit during a move, etc… 

    bool Running
    float current_position[]
    static float destination[]
    float home_offset[]

    inline void line_to_current_position()  
    inline void line_to_destination()
    inline void sync_plan_position()

Roxy-3DPrintBoard, 2015-12-27:
There is a lot of history you don't seem to be aware of.   For example, in #3, the Z_RAISE_xxxx values happened incrementally.  In hindsight, we would have done that differently.  It started with a Z_RAISE_BEFORE_HOMING.   Soon, a Z_RAISE_BETWEEN_PROBES was added.  Later, a Z_RAISE_AFTER_PROBING happened.   Incrementally, they all made sense.   

Right now we are in a transition.   Pretty much, we don't want to 'Raise'.  We want to go to a specified height.   But we don't want to change the names just yet because of all the confusion it will cause.   If it helps you, translate in your mind every Z_RAISE_xxxx to Z_HEIGHT_xxxx until we feel comfortable changing the names.

in2-arduino, 2015-12-28:
Well, I expected as much in regards to incremental development. The general gist I get is that everyone developed stuff they needed to happen and the repository did it's best to assimilate it.

I quite need this change to happen and it will definitely be done for my machine. The big question is: can it make way back into this repository? Because if it can't I wont bother making it pretty and working for anything other then my machine. In other words, do you want me to contribute it?
Please, just yes or no will suffice for start, we can go on from there:)

In regards to raise, going to specified height is not very safe IHMO.

KiteLab, 2015-12-28:
There is absolutely __no__ save way to move away from an endstop, except you have endstops on both ends - before homing. That includes the impossibility to deploy a servo probe in a save manner.
All we know at boot time is, either we are touching an endstop, then we know we are __about__ at the endstop position, or we are __somewhere__ in the legal range. That means the position of the z-axis after Z_RAISE_BEFORE_HOMING is a bit above the position we booted - potentially Z_RAISE_BEFORE_HOMING mm above zmax. Because the likelihood of ending the last print to close to the top is small, this works in most cases - but it's not save. Making a move always upward (relative to the start position of G28 instead of relative to our boot position) increases the risk.
If we really have to raise, that indeed makes most sense at the beginning of the G28 sequence.
Go for it.

Roxy-3DPrintBoard, 2015-12-28:
Z_RAISE_BEFORE_HOMING can be dangerous.   But some people do need it.  If they need to use it, they are required to have extra clearance in their Z-Axis to accommodate it.  In the case of a machine that homes towards the Max Z-Axis, this can be problematic.   Delta's do that.  But the good news is Delta's do not need to raise the nozzle before homing.

Anyway... Back to your points...  

I think I agree with the first point.  I don't think the Z-Raise direction should depend upon which direction the axis homes.   But some discussion would probably be wise.

I understand why you want to raise the nozzle prior to any home operation.  But I think a lot of people are going to object to that.   It is going to cause a lot of unwanted movement for those people.   And one more concern is this:   Part of the problem with Z_RAISE_xxxx is some code paths would do multiple raises, and the nozzle was ending up at the wrong place.  That is part of why a switch to Z_HEIGHT_xxxx  is considered by some to be a better solution.   

 > The big question is: can it make way back into this repository? Because if it can't I wont bother making it pretty and working for anything other then my machine. In other words, do you want me to contribute it?  Please, just yes or no will suffice for start, we can go on from there:)

Yes.   It is very easy for these 'corrections' or 'improvements' to make it back into the main line code.  There are a couple of ways to proceed.   One way would be to take the Marlin/MarlinDev branch and make all of your changes.   If they look good and don't obviously break any of the machine types, we will get more and more test time as people pull in that commit.   Once there is enough test time on the MarlinDev branch, it will get promoted to a Release Candidate status and eventually end up on the Stable side of the fence.

The problem is this section of the code has gotten too complex.   And we need to move carefully as we clean it up.   And we really do need a 'Subject Matter Expert' in this area to watch over things.    It would be great if you could clean this (important!!!) area up.

And if you are willing to do that, there are a couple of other issues with G28 that you didn't mention in your list.   One of the most annoying issues is the G28 Z-Height numbers don't always agree with and match the G29 Z-Height numbers.   It would be really nice to clean that up at the same time.

Roxy-3DPrintBoard, 2015-12-28:
Z_RAISE_BEFORE_HOMING can be dangerous. But some people do need it. If they need to use it, they are required to have extra clearance in their Z-Axis to accommodate it. In the case of a machine that homes towards the Max Z-Axis, this can be problematic. Delta's do that. But the good news is Delta's do not need to raise the nozzle before homing.

Anyway... Back to your points...

I think I agree with the first point. I don't think the Z-Raise direction should depend upon which direction the axis homes. But some discussion would probably be wise.

I understand why you want to raise the nozzle prior to any home operation. But I think a lot of people are going to object to that. It is going to cause a lot of unwanted movement for those people. And one more concern is this: Part of the problem with Z_RAISE_xxxx is some code paths would do multiple raises, and the nozzle was ending up at the wrong place. That is part of why a switch to Z_HEIGHT_xxxx is considered by some to be a better solution.

The big question is: can it make way back into this repository? Because if it can't I wont bother making it pretty and working for anything other then my machine. In other words, do you want me to contribute it? Please, just yes or no will suffice for start, we can go on from there:)
Yes. It is very easy for these 'corrections' or 'improvements' to make it back into the main line code. There are a couple of ways to proceed. One way would be to take the Marlin/MarlinDev branch and make all of your changes. If they look good and don't obviously break any of the machine types, we will get more and more test time as people pull in that commit. Once there is enough test time on the MarlinDev branch, it will get promoted to a Release Candidate status and eventually end up on the Stable side of the fence.

The problem is this section of the code has gotten too complex. And we need to move carefully as we clean it up. And we really do need a 'Subject Matter Expert' in this area to watch over things. It would be great if you could clean this (important!!!) area up.

And if you are willing to do that, there are a couple of other issues with G28 that you didn't mention in your list. One of the most annoying issues is the G28 Z-Height numbers don't always agree with and match the G29 Z-Height numbers. It would be really nice to clean that up at the same time.

in2-arduino, 2015-12-29:
@KiteLab 
Of course, if you have no max Z endstop, any raise operation is unsafe - I mentioned that in point 1, but that's nor here nor there because it's how current code actually works. Bigger issue IHMO is that it'll sometimes raise towards minstop and sometimes towards endstop depending on any jog movements user might have done before homing. Users who don't have max endstop on Z axis can either disable raise option, or pay attention on how they power-up and home the machine. I also think that decoupling min Z stop from bed probe is excellent idea tough I imagine it won't be used much on account of additional sensor and wiring.

@Roxy-3DPrintBoard 
My solution is invariant to homing order or internal code path. It simply rises the head if current height is less then the one defined in configuration. It would also be simple to add couple of configuration flags that control which homing operations should perform raise, giving the user the option to configure the firmware to his/her needs.
I can take a look at G28/29 height mismatch you mentioned, but please point me toward the issue that documents the problem and how to reproduce it.

OK, I'll start a fork from MarlinDev and proceed with my G28 implementation. From there we can do some beta testing and see if it works for other people.

Wackerbarth, 2015-12-29:
@in2-arduino -- when you do the fork, carve out the G28 code and place it in a separate file as I have already done for G29.

I agree that we need to change the way that G28 behaves. However, rather than proliferating more and more configuration flags, first try to see if you can define things in a unified manner.

One of the things that we nee to get rid of is trying to trace the nozzle height in the Object coordinates.
Instead, we should track it in the FRAME coordinates

When we power up, we know (only) that the nozzle is above min(probe trigger height, 0)
Note that, some probes actually trigger when the bed is below the bed height. For these, in particular, we need to raise the probe enough to assure that we are at least some travel height above the bed.

And I agree that you should take into account any manual jogging that has occurred.

Rather than just writing code, first try to develop a flow diagram that describes the motion behavior.
Once we have agreement that the proposed process is appropriate, actually writing the code should be fairly easy.

Roxy-3DPrintBoard, 2015-12-29:
@in2-arduino

> I can take a look at G28/29 height mismatch you mentioned, but please point me toward the issue that documents the problem and how to reproduce it.

I can dig up lots of threads with this complaint.  But it won't help much.   The problem stems from G28 trying to home and G29 trying to correct for bed tilt.   And one thing that will eliminate this problem is to use the same probing methodology that G29 uses.   G28 uses a HOMEAXIS() macro that expands to static void homeaxis(AxisEnum axis).   But G29 uses probe_pt() which calls run_z_probe().    

Please consider using probe_pt() or run_z_probe() for the Z-Axis of G28 if Auto Bed Leveling is enabled.   That will insure the two routines come up with a consistent vision of where the Z-Axis is located.

Wackerbarth, 2015-12-29:
On Dec 29, 2015, at 10:06 AM, Roxy-3DPrintBoard <notifications@github.com> wrote:

> @in2-arduino <https://github.com/in2-arduino>
> I can take a look at G28/29 height mismatch you mentioned, but please point me toward the issue that documents the problem and how to reproduce it.
>
> I [...]
I would suggest that we have MANUAL_PROBE and AUTOMATED_PROBE

In the automated case, the code moves the probe around until it triggers.
In the manual case, the user uses panel control (or the equivalent gCode) to move the nozzle to the “trigger height” and then signals (the “enter key”, or equivalent) that the probe is in its trigger position.

In both cases, we track the location of the head in FRAME coordinates and use that reading to adjust the transform which will be applied when converting from object coordinates to frame coordinates.

They really both need to be calling exactly the same routines to perform the adjustments.

in2-arduino, 2015-12-30:
@Roxy-3DPrintBoard 
What do you mean by object and frame coordinates? Obvious searches didn't produce definitive answers, but I'm guessing frame would refer to actual position of each axis, while object would be ideal untransformed position that is set by G code?

Roxy-3DPrintBoard, 2015-12-30:
This is probably not perfectly accurate, but it will get you close:   Think 'Corrected vs Uncorrected' coordinate systems.   Pretty much, everything you would be doing to replace the G28 command would be in the Uncorrected coordinate system.   

If there is any correction in place, you would set the correction matrix to unity.  G28 already does that.  And then the machine should be left in the defined Home Position at the end of the command.   

I could have this a little bit wrong.   But the Frame vs. Object thinking probably doesn't really affect G28 because that is at the front end of getting the printer setup and all of the coordinate transformation in place and going.   It would seem logical for G28 to do everything based on uncorrected coordinates.

Wackerbarth, 2015-12-30:
@in2-arduino -- I'm probably the one using those terms.

I identify 4 different coordinate systems that can be used to specify a location. The mathematical transformations which convert from one system to another are the basis of the calculations that are performed in the printer.  In many cases, these transformations are "trivial".

To begin, let's refer to the ACTUATOR coordinates. These are some linear scale and zero point that specify where, within the limits of motion, the physical actuators are, or are requested to be. As an example, we might refer to the position of a stepper motor by its distance, in steps, from its "homing" switch. These positions, through the kinematics of the printer, translate into a FRAME coordinate system.
This system is the global position in space. I choose to describe it in such a manner that the Z-axis is parallel to the vertical movement of a nozzle. This is actually the true reference to which all of the other systems are related. The XY plane is perpendicular to that Z-axis and the orientation and zero position are fixed, but chosen in a sensible manner related to the printer.  For example, we might choose Z=0 such that the nozzle is just touching the bed at the origin. And, for a delta printer, we might choose to have the rear tower somewhere on the Y-axis.  Note that exactly how we choose to represent this system, it is, mathematically equivalent to any other scheme. And you can transform coordinates in one frame representation to another by the application of a matrix multiplication.
The next coordinated system would be that of the BED. Here, we want to fix the origin on the surface of the bed. But the Z-axis might be slightly tilted.
And the final coordinate system is the OBJECT. These are the relative positions on, and within, the actual object that we wish to construct. This is the system within which the G-Code commands direct us to move.

Wackerbarth, 2015-12-30:
Notice that G28 (when used to locate the printhead rather than just "parking" it in a particular position) is really just locating the print head in relation to it's alignment sensors (either automatically or manually).
(Which we might view as "0-point adjustment"). When the endstop is triggered, we know where the actuator is located. And we already have the parameters to translate that position into the FRAME coordinates.

If it involves any "probing", it MAY also be determining the position of the bed relative to those sensor positions. (What I might call "1-point adjustment") There, it would be setting the height of the bed origin relative to the FRAME.

in2-arduino, 2015-12-30:
OK, that's quite detailed and makes perfect sense. Thanks.

I think such explanations are sorely missing from the code, so I'm probably gonna work it into a comment somewhere.

There is however one more CS that is neglected. In multi extruder systems, each nozzle has offset from the primary, and as far as I can tell, T0-T3 tool change commands simply add this offset to current (frame) position, which in turn invalidates homing and soft endstops until primary tool is reselected. In the long run it should probably be added as one more transformation if MCU can handle it. Transformation pipeline would then look like this:

ObjectCS x ToolOffsetMatrix =
ToolCS x BedLevelingMatrix =
BedCS x FrameMatrix =
FrameCS x ActuatorMatrix =
ActuatorCS

Roxy-3DPrintBoard, 2015-12-30:
> In multi extruder systems, each nozzle has offset from the primary, and as far as I can tell, T0-T3 tool change commands simply add this offset to current (frame) position, which in turn invalidates [...]

My guess (without going back and looking) is this is another example of incremental growth of the code base.   In a single extruder system, it makes sense to have that single nozzle track the location of the carriage.  When a 2nd extruder was added, it made sense to have an offset from the 1st extruder.

My point is, we should move away from the assumption that the 1st nozzle is what everything is based off of.   Probably, the 1st extruder should be allowed have offsets specified for it and still have everything work.    I suspect it would not take that much effort to change that.  But I haven't looked at that carefully to know for sure.

KiteLab, 2015-12-30:
```
ObjectCS x ToolOffsetMatrix =
ToolCS x BedLevelingMatrix =
BedCS x FrameMatrix =
FrameCS x ActuatorMatrix =
ActuatorCS
```
is a bit slow.

When some of the envirenment data changes.
```
UnifiedMatrix = ActuatorMatrix x FrameMatrix x BedLevelingMatrix x ToolOffsetMatrix x ... 
```
When you actually want to calculate a coordinate and speed matters
```
ActuatorCS = ObjectCS x UnifiedMatrix
```

Roxy-3DPrintBoard, 2015-12-30:
@KiteLab   @Wackerbarth   Quick question:  Can that simplified UnifiedMatrix be made by combining the other matix's if the transformations are non-linear.   For example, the bowl shaped bed of a Delta printer.   Can that be corrected with a matrix multiplication?   My guess is, if it can be corrected by a matrix multiplication, the elements of the matrix are non-linear.  They probably have trig functions in them.  Right?

in2-arduino, 2016-01-01:
@Wackerbarth 
You said to carve G28 into new file like you did with G27, but G27 doesn't exist. What code did you actually mean?

Roxy-3DPrintBoard, 2016-01-01:
@in2-arduino    You don't want to use this branch yet.  @Wackerbarth is in the process of cleaning up the organization of the source files.   Some of the files have grown much too big.  In particular, Marlin_main.cpp.   And that is where the G28 code lives.

Let's wait until Wackerbarth can give clearer directions.  But what he is talking about is here:  https://github.com/Wackerbarth/Marlin/tree/Dec31

Some of the GCodes have been broken out of the Marlin_main.cpp file.   I think what Wackerbarth is suggesting is we don't want to lose the original (and convoluted) G28 just yet.  My guess is he will suggest something like separate the original G28 and make sure everything still builds in different configurations.   Then start modifying it.   And very possibly, it would be good to keep the original separated G28 too just because it will provide us a way to do regression testing if something is acting suspicious.

Wackerbarth, 2016-01-01:
@Roxy-3DPrintBoard -- Spot on!  I have my carving knives out at the moment and am doing a real number on G27, G28, G29.  I hope to get everything back to compiling within a few hours.

So, I will get the initial carving done. Then, hopefully, @in2-arduino can go to work on building the functionality into "homing.cpp" and have the functionality in another G# which becomes nothing more that "home" (which works in FRAME coordinates) followed by G0 [X_HOME] [Y_HOME] [Z_HOME];

From there, it will be trivial to swap the codes and then, when we are satisfied that the new code really works, drop the current implementation of G28.

Wackerbarth, 2016-01-01:
@in2-arduino @Roxy-3DPrintBoard -- Well, I've done a bit of chopping.  See https://github.com/Wackerbarth/Marlin/tree/Jan1
It now compiles with all of the pieces. But I fear that I have broken some global linkage between the parts because Glen (@a4jp) tried an earlier version a few hours ago and the G28 command went off "drilling to China".

If I had to guess, I now have, somewhere, two "local" variables which need to be one "global" one.
But that's what happens when you have code which has no well defined interfaces. If someone spots a problem, we can fix it and be in a position to start cleaning up the pieces individually.

closed 2016-05-25 by boelle