MarlinFirmware/MarlinDev #315: Configuration Instructions

ruggb, 2015-12-26:
So far, if I ran a virgin build it compiled. 
dev.d8d0859 does not do that and I am at a loss as to how to set it up so that it will.
There isn't even a Marlin.ino file to run so I tried this one
.....\MarlinDev-dev\examples\MarlinFirmware\MarlinFirmware.ino

Obviously, I need to do my configs for my machine, but after searching for hours I am lost there also.
Where can I find the instructions for the new process?

My machine is a homebrew coreXY so I can't use any included configs as they r.

Wackerbarth, 2015-12-26:
Have you tried the MarlinDev wiki? https://github.com/MarlinFirmware/MarlinDev/wiki

I will also be online for a while and will try to provide additional assistance, if needed.
https://hangouts.google.com/hangouts/_/jgx5ywnjhtjqdl4tf76f6o3jlqa

 
From the README --

This is the source code which produces the Marlin Firmware

Complete instructions for setting up the build environment and configuring the firmware can be found at the wiki. In order to build the firmware, you will also need to setup the build environment. The official build environment is based on the Arduino IDE 1.6.6.

ruggb, 2015-12-26:
richard
I have been using Marlin for a while. Y'all changed things since the last dev I loaded and I have no clue how to run this commit. There is no MARLIN.INO file. The config files do not reside in the marlin folder. It appears that I can create a config file (Customization) and somehow use that across future commits but it is a mystery to me how to do that. Some of the items I need to change have an IF DEFINED stmt around them some of them do not. If I define the ones that do in a custom file that should work if I can understand how to link all the files up and where to put them. But won't the straight defines just overwrite my changes?

Wackerbarth, 2015-12-26:
There is a Marlin.ino, and you found it.
Did you read the wiki?

Marlin has been changed into  library.

You need to place your configuration in either the same folder as the .ino or place a reference to it in either the FirmwareCustomization.h or in the Boards.local.txt.

If you choose to use the “include the whole configuration file” scheme, then, it works much like the previous scheme.
In that case, except for a few lines, your configuration file should look just like one of the deprecated configurations.

The intention is to move to a situation where all you need are the lines that you want to change, and a reference to a transitional configuration.  You are correct that, currently, there are conflicts because I haven’t been able to complete the horrendous task of figuring out just what part of the configuration file needs to be preserved and which part needs to be factored into the more general level.  The configuration files that we have “inherited” for the earlier work are virtually impossible to decypher because they have “hidden” effects in Conditionals, etc. and #define something, only later to have it #undef and redefined. Ad that is just one of the issues that makes to old scheme unmaintainable.

But in short term, you can either avoid the transitional base files or help “fix” them by adding “guards” around their definitions.

#ifndef SOMETHING
  #define SOMETHING value
#endif

Eventually, all of the entries that can be overridden by the user will be in that format.

> On Dec 26, 2015, at 1:29 PM, ruggb <notifications@github.com> wrote:
> 
> richard
> I have been using Marlin for a while. Y'all changed things since the last dev I loaded and I have no clue how to run this commit. There is no MARLIN.INO file. The config files do not reside in the marlin folder. It appears that I can create a config file (Customization) and somehow use that across future commits but it is a mystery to me how to do that. Some of the items I need to change have an IF DEFINED stmt around them some of them do not. If I define the ones that do in a custom file that should work if I can understand how to link all the files up and where to put them. But won't the straight defines just overwrite my changes?

ruggb, 2015-12-26:
Well, it is not Marlin.ino but MarlinFirmware.ino, right?
I did read the wiki, but it has no specific info re my issue.
what I did was to unzip the dnld and run the ino but it would not compile.

D:\Users\Bill\Documents\##3D 
PRINTING\#Firmware\##Marlin\Marlin-Development\12-04-15\MarlinDev-dev.d8d0859\examples\MarlinFirmware\MarlinFirmware.ino:2:28: 
fatal error: MarlinFirmware.h: No such file or directory

  #include <MarlinFirmware.h>

I moved the ino and custom file to the main folder (and chg the folder 
name) I get this error

Unable to find D:\Users\Bill\Documents\##3D 
PRINTING\#Firmware\##Marlin\Marlin-Development\12-04-15\Marlinfirmware\Marlinfirmware.ino 
in D:\Users\Bill\Documents\##3D 
PRINTING\#Firmware\##Marlin\Marlin-Development\12-04-15\Marlinfirmware

but the file is there, I just ran it from there.

Wackerbarth, 2015-12-26:
From this, I am guessing that you did not use the board manager to install the Marlin board definitions, but are trying to use the Arduino board definitions instead.

> On Dec 26, 2015, at 2:47 PM, ruggb <notifications@github.com> wrote:
> 
> Well, it is not Marlin.ino but MarlinFirmware.ino, right?
Yes, my mistake.

> I did read the wiki, but it has no specific info re my issue.
> what I did was to unzip the dnld and run the ino but it would not compile.
>
> D:\Users\Bill\Documents\##3D [...]

ruggb, 2015-12-26:
I never have used the "board manager" in the past.
why does it not find a file that is there?

ruggb, 2015-12-26:
i looked into MarlinFirmware.h and it has
#include "IncludeUserCustomization.h"
which does not exist anyplace.
should that be the file that is with MarlinFirmware.ino??
FirmwareCustomization.h

Wackerbarth, 2015-12-26:
> On Dec 26, 2015, at 3:17 PM, ruggb <notifications@github.com> wrote:
> 
> i looked into MarlinFirmware.h and it has
> #include "IncludeUserCustomization.h"
> which does not exist anyplace.

Actually, that file is there!

I just added some additional tests to provide better diagnostic error messages if you fail to
select the correct options when doing the build in the Arduino IDE.

Try downloading the Dec26 branch and see if that helps you determine just what selection you have missed.

ruggb, 2015-12-26:
I searched for "usercustom" and found nothing.
where did u find it?

ruggb, 2015-12-26:
never mind. a windows issue entirely

Wackerbarth, 2015-12-26:
“IncludeUserCustomization.h"

It’s right in there with everything else, in the same folder as “Marlin_main.cpp.

It appears right before the language files in my alphabetical listing.

> On Dec 26, 2015, at 5:06 PM, ruggb <notifications@github.com> wrote:
> 
> I searched for "usercustom" and found nothing.
> where did u find it?

ruggb, 2015-12-26:
OK - I dnld dec26, unzipped it, changed the folder name to 
*MarlinFirmware*, copied contents of *examples\MarlinFirmware* to 
*MarlinFirmware *folder, and ran the *ino *file. It did not compile. *Is 
it suppose to?* ALL the other versions have compiled without moving any 
files or making any changes. I am using that to test if things r in the 
right place before I try to change things.

I have used nothing from any other distribution. There r no config files 
in the folder, so what config file is it using and not finding a 
motherboard define? It is also saying the config files r outdated. What 
files? If it is using files in the dist how can they be outdated?

Sorry Richard, I am not a programmer. I know enough to be dangerous, but 
I am an engineer. I figured this out in the past, but this is not 
responding to my beatings.

error message.

Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, 
ATmega2560 (Mega 2560)"

In file included from sketch\IncludeUserCustomization.h:9:0,

                  from sketch\MarlinFirmware.h:12,

                  from sketch\Marlin.h:26,

                  from sketch\M100_Free_Mem_Chk.cpp:24:

FirmwareCustomization.h:82: error: #error Compilation using cores other 
than those provided in this distribution are unsupported

    #error Compilation using cores other than those provided in this 
distribution are unsupported

     ^

In file included from sketch\Marlin.h:26:0,

                  from sketch\M100_Free_Mem_Chk.cpp:24:

MarlinFirmware.h:16: error: #error Failed to properly include 
configuration parameters

    #error Failed to properly include configuration parameters

     ^

In file included from sketch\Marlin.h:27:0,

                  from sketch\M100_Free_Mem_Chk.cpp:24:

pins.h:132: error: #error Unknown MOTHERBOARD value set in Configuration.h

    #error Unknown MOTHERBOARD value set in Configuration.h

     ^

In file included from sketch\M100_Free_Mem_Chk.cpp:24:0:

Marlin.h:30: error: #error Your Configuration.h and Configuration_adv.h 
files are outdated!

    #error Your Configuration.h and Configuration_adv.h files are outdated!

     ^

exit status 1
#error Compilation using cores other than those provided in this 
distribution are unsupported

   This report would have more information with
   "Show verbose output during compilation"
   enabled in File > Preferences.

-- 
"Too often we hold fast to the cliches of our forebears.....
  We enjoy the comfort of opinion without the discomfort of thought."
  —John F. Kennedy

Wackerbarth, 2015-12-26:
> On Dec 26, 2015, at 5:38 PM, ruggb <notifications@github.com> wrote:
> 
> OK - I dnld dec26, unzipped it, changed the folder name to 
> *MarlinFirmware*,

Hopefully you put that folder in your <sketchbook>/libraries/ folder.

> copied contents of *examples\MarlinFirmware* to 
> *MarlinFirmware *folder, and ran the *ino *file. It did not compile. *Is 
> it suppose to?*

NO! DON’T MOVE ANYTHING. Please follow the instructions to open the example file where it sits. (using the File -> Examples -> MarlinDev -> MarlinFirmware menu entry)

> Sorry Richard, I am not a programmer. I know enough to be dangerous, but 
> I am an engineer. I figured this out in the past, but this is not 
> responding to my beatings.
> 
> error message.
> 
> Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, 
> ATmega2560 (Mega 2560)”

Also, the instruction specifically say the select the Board from the Marlin AVR section of the boards menu.

You did load that, AS INSTRUCTED, using the Boards Manager. Didn’t you?

> In file included from sketch\IncludeUserCustomization.h:9:0,
>
> from sketch\MarlinFirmware.h:12,
>
> from sketch\Marlin.h:26,
>
> from sketch\M100_Free_Mem_Chk.cpp:24:
>
> FirmwareCustomization.h:82: error: [...]

Roxy-3DPrintBoard, 2015-12-27:
More evidence we need a 'Cheat Sheet' !     :)

ruggb, 2015-12-27:
Richard
I started this all over again. It has been over 30 sec so I forgot a lot 
already.
I am at the *manual installation* and I am unsure of which folder to 
install as a library.*
*Is it the *MarlinFirmware *folder or the *MarlinDev* folder that I cloned?

I have installed the u8glib - ver 1.18.1. It shows as installed but as 
an *unknown *version - the previous install showed as ver 1.18.0.

thx

ruggb, 2015-12-27:
OK, I got it.
The *Where to install the source *section is unclear.
As it stands it is telling me to manually install something I don't have 
yet.

It is followed by the Preferred method to get the folder I was already 
suppose to have installed. I would suggest replacing it with this.

/The source for the Marlin Firmware is contained in a single folder 
(with sub-folders) which needs to be installed as an Arduino library. 
Refer to the Preferred/Alternate methods below to obtain the firmware 
folders. See the //Arduino Library guide 
<https://www.arduino.cc/en/Guide/Libraries>//"Manual Installation" for 
instructions on installing the expanded source files in the appropriate 
location.//
/
thanks

Wackerbarth, 2015-12-27:
Step by step —
1) I assume that you have already installed Arduino IDE 1.6.6 (or a more recent “nightly" 1.6.7)
2) If you have the .zip file from the download of MarlinDev — branch “dev”, unpacked it and renamed it to MarlinDev,
you should place that in your <sketchbook>/libraries/ folder.
3) Open the IDE. In the preferences, add 

https://raw.githubusercontent.com/MarlinDev/MarlinDev/platform/package_marlinfirmware_index.json

to the Additional Boards Manager URLs.  Save that and Quit the IDE.

 4) Re-open the IDE — Go to the Tools menu and select Board Manager from the Boards submenu.
Down at, or near the bottom, you should find an entry “Marlin by Richard Wackerbarth”. Select that.
Select the version that corresponds to the IDE (1.6.6 or 1.6.7) Install it.

5) (Unknown, but it won’t hurt) Restart the IDE again.

6) Go to the File -> Examples -> MarlinDev menu and open the MarlinFirmware.
You should have two tabs: MarlinFirmware and FirmwareCustomization.h

7) Select the “Arduino Mega or Mega 2560” FROM the Marlin AVR section of boards.

8) You should be able to compile the “Default Configuration”

9) Now, you can edit your own custom configuration and compile and install it.

Wackerbarth, 2015-12-27:
> On Dec 26, 2015, at 8:45 PM, ruggb <notifications@github.com> wrote:
> 
> OK, I got it.
> The *Where to install the source *section is unclear.

It’s a wiki. Feel free to improve the description yourself.

> As it stands it is telling me to manually install something I don't have
> yet.
>
> It is followed by the Preferred method to get the folder I was already
> suppose to have installed. I would suggest [...]

Roxy-3DPrintBoard, 2015-12-27:
> It’s a wiki. Feel free to improve the description yourself.

The Wiki now has a 'Quick Start' section...

ruggb, 2015-12-28:
I have the firmware compiling the default, but it isn't perfectly happy.
===================
Multiple libraries were found for "U8glib.h"
 Used: d:\Users\Bill\Documents\Arduino\libraries\U8glib
 Not used: C:\Users\Bill\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.6.7\libraries\U8glib
Multiple libraries were found for "LiquidCrystal.h"
 Used: d:\Users\Bill\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Users\Bill\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.6.7\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\3D Printing\Arduino\libraries\LiquidCrystal
===================
The question is, r all these the same, and/or is it using the correct one?
They were all installed by the process.

NOW I have to figure out how to get my configs into this.
Only one config is mentioned in the docs. What do I do with the configuration_adv.h file?

This, from "Installing Customized Configurations", raises a couple ???
===========
Whether the individual configuration is simple or complex, the user is encouraged to create that configuration in a file which is added to the source tree. It is then accessed by placing a reference to the file in the appropriate location.
=================
What do I call this file? What is the SOURCE TREE? Where is the APPROPRIATE LOCATION to place the reference. AND What is the reference I need to add?

And finally, should I just copy/paste both my current config files into the above file?

ruggb, 2015-12-28:
and I have no idea why he above text is so big and bold. It wasn't intentional

Wackerbarth, 2015-12-28:
> I have the firmware compiling the default, but it isn't perfectly happy.
>
> Multiple libraries were found for "U8glib.h"
> Used: d:\Users\Bill\Documents\Arduino\libraries\U8glib
> Not used: [...]
The IDE has found two copies of the U8glib library.  One, on the C:\ drive, is in our distribution. The other, on your D:\ drive, was installer by the user (you).  You have 3 choices. 1) This is just a warning, you can just ignore it each time you compile.
2) You can delete the version on the D:\ drive that you added. That might not be terribly desirable if you are also dealing with other AVR projects. 3) You can delete the version from our distribution. The path is given in the above message.

> Not used: C:\Program Files (x86)\3D Printing\Arduino\libraries\LiquidCrystal
This appears to be yet another copy. (Perhaps you downloaded it from the IDE library manager.)
> The question is, r all these the same, and/or is it using the correct one?
I suspect that any of them will be OK. However, there should be a version number on at least some of them.
> They were all installed by the process.
??? I doubt that they ALL got there automatically.

> NOW I have to figure out how to get my configs into this.
> Only one config is mentioned in the docs. What do I do with the configuration_adv.h file?
That all depends on how YOU wish to handle it.  You can install both files. Or you can concatenate them into one file. Or you can strip out just the entries that you changed and use those with a “transitional” configuration.

All of the configuration files that we supply are in sub-folders of the “configurations” folder.
You can create your own personal folder in the configurations/custom/ folder and put things there (That’s what is referred to in your next comment). You can also place them in, or adjacent to the FirmwareCustomization.h file in the examples/MarlinFirmware/ folder.

> This, from "Installing Customized Configurations", raises a couple ???
>
> Whether the individual configuration is simple or complex, the user is encouraged to create that configuration in a file which [...]
It's just a standard #include. You can place them ANYWHERE that the compiler can reach them.  However, I suggest that you use one of the suggested locations.  Read the remarks in FirmwareCustomization.h and do something similar to one of the existing configurations.
> And finally, should I just copy/paste both my current config files into the above file?
That is also an option. However, to avoid having a large number of entries overwritten when you next update the source tree, I recommend that you place your values in your own file (or files).  And I also suggest that you stash a copy of them somewhere “safe” so that you do not accidentally clobber them by mistake.

ruggb, 2015-12-28:
d:\Users\Bill\Documents\Arduino\libraries\U8glib
and
C:\Program Files (x86)\3D Printing\Arduino\libraries\LiquidCrystal
were part of the Arduino installation. The latter is the program folder, the former is the Document folder created by the install. Neither were installed separately. 

I guess I will just ignore them. I might break something removing one or both.

Or you can strip out just the entries that you changed and use those with a “transitional” configuration.
I am not sure what u mean by this?

Wackerbarth, 2015-12-28:
> On Dec 27, 2015, at 10:05 PM, ruggb <notifications@github.com> wrote:
> 
> d:\Users\Bill\Documents\Arduino\libraries\U8glib
> and
> C:\Program Files (x86)\3D Printing\Arduino\libraries\LiquidCrystal
> were part of the Arduino installation.
> 
> The latter is the program folder, the former is the Document folder created by the install.
> 
Yes, the former is the “Document folder” (AKA <Sketchbook>) which was formed as a part of the running of the IDE.

But the issue is “who, when, and why” was a copy of the LiquidCrystal library installed there?
I suspect that YOU installed it some time in the past for use with another version of the IDE, etc.

Since there is now a copy of this library in the IDE itself, I will be dropping it from our distribution.
Please delete the copy that we provided.
> Neither were installed separately.
> 
> I guess I will just ignore them. I might break something removing one or both.
> 
Don’t mess with the one in the IDE (Program Files).  But you will not hurt anything by deleting the one in the sketchbook (D:| drive)

> Or you can strip out just the entries that you changed and use those with a “transitional” configuration.
> I am not sure what u mean by this?
> 
The entire direction that the configuration files is going is to avoid having many copies of ALL of the possible options floating around.  Instead, you will just have a short file with the few entries that make your installation different from the next.
Then you will use the C++ preprocessor #include to draw in a more generic description of a similar printer and use its values as the default values for your build.

In order to figure out just what values are peculiar to your printer, perform a “diff” against the “deprecated” configuration file that you used as a base in the Marlin 1.1RC builds.  Anything that is the same can be picked up from the “default” mechanism.

Although they still need quite a bit of work, the “transitional” configuration files form the defaults corresponding to the corresponding “deprecated” configuration version.

Eventually, all of these files will be phased out in favor of a newer organization of the information.  But, one of the good things about Marlin 1.2 is that we get to use the #include mechanism and that gives us many choices from which we can build the set of #defines that describe a particular printer.

PSimonovich, 2015-12-29:
The Board Manager JSON file specified in the wiki documentation (i.e., https://raw.githubusercontent.com/MarlinDev/MarlinDev/platform/package_marlinfirmware_index.json) cannot be found. What is the correct path?

Wackerbarth, 2015-12-29:
Thanks for catching the error.
It's 
https://raw.githubusercontent.com/MarlinFirmware/MarlinDev/platform/package_marlinfirmware_index.json
I'll switch to the wiki and fix that.

ruggb, 2015-12-30:
I have arduino 1.6.7 installed.
When I go to boards manager there r 2 entries for marlin for 1.6.6 and 1.6.7
If i try to install 1.6.7 / 0.0.2, I get this
--------
Tool versioning is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool versioning is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:167)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Tool versioning is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:88)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:164)
	... 1 more
--------
I can install 1.6.6

I thought IDE 1.6.7 referred to my version - is there a problem here?

Wackerbarth, 2015-12-30:
The problem is in the package that I set up.  I'm guessing that you have Windows.

I don't know if it will fix the issue. But I just pushed a new json file describing the platform.
Please quit the IDE and restart IDE1.6.7 and see if you can upload  the package based on this new description.

Unless I am badly mistaken, the error is just in my packaging and not in any of the components (like java or the IDE) on your system.

Please let me know if it works or the error if you get one.

Thanks,
Richard

ruggb, 2015-12-30:
Got it - I removed 1.6.6 and installed 1.6.7 = works now, thx
I guess u didn't remove the u8glib copy yet from Marlin. It defaults to using the Arduino copy, so I guess that is good.

comment. I have reinstalled this from scratch again (about 4 times now) to make sure I have it right.
All the info is there, someplace, but it is a real job finding it in the right sequence even after the 4th time. It is definitely a developer thing I quess.
I would write a cheat sheet if I knew how to publish it and no one else was up to it now.
I would be surprised if u find any time to develop between questions.

Wackerbarth, 2015-12-30:
Thanks for your understanding -- and help. As you note, all the instructions are probably "around", but not well organized. Roxy tossed in a Quick Start up front on the wiki and there is more detail if you dig into the pages.

But remember that the documentation is in a wiki so that folks like yourself can go in and edit it.
If you think that it can be better organized, give it a shot. As long as everyone is careful to avoid adding incorrect information, the experience should improve over time.

Wackerbarth, 2015-12-30:
@ruggb -- I don't have multiple copies of U8glib on my system.
Where are yours located? If it is in the <sketchbook> then that is because you did something to add it.

ruggb, 2015-12-30:
This is the problem and I don't know how to fix it.

The Wiki start page has SETUP on it, but has SETUP listed in the right sidebar. The content is the same so that should be removed.

and what was confusing me..........
Under setup there r 2 links. The first goes to The Development Environment. This has a continue link on the bottom to Obtaining The Arduino IDE. And then a link to Install The Marlin Library Source, then to Customizing the Firmware. 

The 2nd link goes to Getting the Source Code, which is an expanded, clearer (IMHO) description of the last 2 paragraphs of Install The Marlin Library Source. This is where the flow breaks down.

I inserted Getting the Source Code after The Marlin Library Source and Customizing the Firmware after that.

Technically, the Getting the Source Code files link under Setup is redundant because the follow on links now take one through all that. I left it b/c I don't know how to put links in the sidebar (see below).
I changed the name of the link to make it clear where it was going.

My preference would be to remove Setup completely and put that first link at the bottom as a continue page. I would put links to all the pages in the sidebar and include a resources page.
But as above, I don't know how to put those links on the sidebar yet.

Roxy-3DPrintBoard, 2015-12-30:
> The Wiki was jumbled up - all the info was there, it just didn't flow.
> Please check out the Wiki - especially the sidebar and the flow of the pages.

I had the same problem.  I didn't realize those little links had important information.

ruggb, 2015-12-30:
If by "little links" u mean the sidebar links - I just created them as shortcuts to specific pages in the wiki.
Those were also incomplete and jumbled up.
If u start at HOME, there r links at the bottom of the pages to get to the next page or u can use the sidebar link. I also revised a couple pages to keep it flowing.

I hope that is working for you.

Roxy-3DPrintBoard, 2015-12-30:
Good job!  Thank You!

thawkins, 2016-01-13:
I have to say this guys, the whole deal with the new layout and build method has created some horrendous complexity in what usex to be a very simple process. 

The process used to be simple, but now it requires a load of skills that are i suspect beyond many of the less software engineering focused folks,  im a software developer with 40 years behind me, still working at the core of building large complex projects, and even i find this new method confusing and restrictive. 

And before people start to claim it is not, you know that you will loose 50% of your user base the moment you mention "boards.txt", "library" or "sketchbook". Most peoples use of the arduino ide is just to compile marlin, they have little or no understanding of the ins and outs of the package beyond that task, and have virtualy no familiaritly with it. gawd, most wont even know where it has been installed, and certainly wont know where to put these components. To them its an icon in a menu.

ruggb, 2016-01-13:
@thawkins 
Thx - I am in total agreement. However, the current process is a PITA if u want to keep up to date with the builds and features. I keep missing things every time I chg. I can also understand the developer issues. The question is how can it be improved for both sides. I have attempted to update the wiki so less technical ppl can follow it. It certainly isn't finished - for one I haven't figured out how I setup my config file so that I don't need to redo the config every time I update. 

As @Wackerbarth admits, the developers r too close to it to be able to translate it to users. I am an engineer and have written lots of technical manuals and app notes so I feel somewhat qualified, IF I could only understand it myself. But I have to say it is a struggle for me to figure out what some of the developers r saying and, probably, most users r not engineers.

It seems to me that it is not possible to keep the current config files from changing such that my config won't need to chg. There r also other files that have variables that one may want to change, so somehow ALL these items need to be in the users config file. BUT, those files seem to me to be complicated with IF statements and other code. IMHO, it should only have defines and parameters. The code files should determine if it uses a default parameter or one in the config file.

For example:
//===========================================================================
//========================= Filament Runout Sensor ==========================
//===========================================================================
//#define FILAMENT_RUNOUT_SENSOR // Uncomment for defining a filament runout sensor such as a mechanical or opto endstop to check the existence of filament
                                 // In RAMPS uses servo pin 2. Can be changed in pins file. For other boards pin definition should be made.
                                 // It is assumed that when logic high = filament available
                                 //                    when logic  low = filament ran out
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  const bool FIL_RUNOUT_INVERTING = true;  // Should be uncommented and true or false should assigned
  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif

Is too much to be in a config file.
even this
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif
is too much
How is MOTHERBOARD going to be defined unless I define it.
The test should be someplace, and trigger an error, but not in the config file.

I for sure do not minimize the effort required to get it right, but I think that should be the goal.
simple for the user - leave the complexity to the developer.
thx

Sniffle, 2016-01-13:
I stepped away for a while but i would think that we could have a simple
installer.exe that checks to see if the ide is installed and updates if
necessary and sets the package up in the ide so that when you open the ide
make a couple clicks put in your settings and it just works... If it cant
be setup to this degree youll lose users because thats all they have to do
now and ive been attempting to test repetier today but in comparison to
marlin even with its configurator its convoluted and harder to setup mainly
because where things are different there isnt a good enough description to
figure out its easier to read counterpart in marlin

thawkins, 2016-01-13:
Great idea, so long as it also works on linux and macosx.

Sniffle, 2016-01-13:
For mac and linux it would be even easier because it would just be scrpt
that would need to be ran.  Unless you wanted fancy graphics

thawkins, 2016-01-13:
Nope, script is fine for linux

Roxy-3DPrintBoard, 2016-01-13:
Your comments about complexity are noted.   I'm not sure what the answer is going to be.   But remember a couple of things.   First, this is all happening in the MarlinDev branch.  It is assumed that the Developers can handle a little bit of extra complexity.  

When we finally have the code broken apart and modulized (compared to what it started as),  we can make it easier to install.   It may be possible to just unzip a .ZIP file and everything lands in the right place.   The user customization instructions are going to have to be dealt with also.   It may be that we need a configuration utility to help.  I don't know.   

But the honest truth is we have to do something to break the code apart and get it into smaller pieces.   We can't have #ifdef's nested 37 levels deep.   Any suggestions are welcome.   And much of this complexity is because we are trying to keep using Arduino.  It is possible we would be better off telling the Window's users that they have install the Cygwin system.  But that introduces a lot of complexity too.

Well thought out criticisms and suggestions are much appreciated.

Wackerbarth, 2016-01-13:
Gentlemen:

There are two, rather separate, issues here.  One is obtaining and setting up a current copy of the build environment and source for the firmware. The second is how you generate, understand, and maintain your individual custom configuration.

At present, neither of these works very well in either the "Marlin" or the "MarlinDev" formulation of the source.

However, keep in mind that the structural changes being introduced are being done incrementally in order to provide a way for KNOWLEDGABLE DEVELOPERS to continue to fix functional issues in the firmware while the "remodeling" is going on.

So, first, let's look at the intended result.

Starting from "scratch", naive user should be able to download and install the Arduino IDE (to get the toolset needed to build and install new firmware). Then, the user will use the Board Manager menu to install the Marlin specific board specifications. Then, using the Library Manager menu, they will install the latest Marlin source code. Finally, they will select the Example sketch (again from an IDE menu) and be ready to "customize" things.

Rather than the huge, confusing, and inconsistent files that we currently have, I would hope that we can reduce the entries to just a few lines. In particular, for any manufactured printer (or kit) it should be sufficient to simply specify a make and model.
For those assembling components to form a system, each subsystem should be reduced to a single #include (or equivalent).

There are two reasons why the user's portion of the configuration needs to be "sparse". First, it is much easier to understand when you can see the "concepts" rather than being mired down in the "details".
Second, as the underlying firmware changes over time, the user's portion of the configuration should not change unless he wishes to add some new optional feature.

We cannot "deploy" the above described system until we get at least most of it implemented and have good, stable, firmware generated.

For the "power user", and that should include any "developer", there are a number of features designed into the structure. Most of those are available now. However, they do require some understanding of the IDE, file editing, etc. For those individuals, "sketchbook" and "boards.txt" better not be an insurmountable issue. Particularly with help from people like @ruggb, I would hope that we can document the process sufficiently for these users to be using, and improving, the code so that we can get to a point that we have something to deploy.

I'm sure that it would be possible to write a script which edits the Arduino IDE preferences and, perhaps, executes the steps necessary to do the installation of the components. However, I am not convinced that it is worth the effort to generate something with such a limited life. I feel that the effort would be better spent in generating a clear step-by-step document which would have residual value to future power users.

Sniffle, 2016-01-13:
Just to clarify my own position, i was referencing when marlindev is
released and merged into the main system, not that it needs to happen in
the next 24 hours or anytime soon.  I understand what is occuring is a
massive undertaking by only a few people, i was just making a suggestion
and trying to be an active participant in the project because its the
simple truth of the end user, any change made should be the same complexity
or less because people in general dont like change.

I applaud your work and effort im just popping in after several months of
inactivity due to burn out, i expected a little flack for being so bold as
to come back making suvgestions out of nowhere but i know you understand
what i mean as well because several of you write code cor a living and
simplicity for the end user is very important when the end application is
finished

Roxy-3DPrintBoard, 2016-01-13:
@Sniffle :  You always have been one of the level headed people in the RepRap world.  If you say something, I listen.

ruggb, 2016-01-13:
The current release version user process, IMHO, is only difficult in re-editing the config files for one's printer - and mine is home built, so I can't select a preconfigure. Since there is code here and things change with these files, I MUST go thru and edit each new one rather than copying my old one to the new load.

I agree with the Repetier analysis. I tried it a couple times and gave up. Let's not think that this process is simple for non-std printers or even std ones that the user wishes to tweak.

As far as making chgs, additions, & improvements to the code I understand ur pain and the need to make it easier. Separating ur code from the config file is ideal. Making each function a module is ideal. Interactions r inevitable, but I think the separation makes it easier to deal with. Don't flame me if this is not true, I'm just a hardware engineer with minimal s/w experience - though I did redo the assy code on CPM for my first computer. Anyone remember that?

I think I would start with a custom folder and have separate files for MB, graphics, motion, bed leveling, some pin tweaking AND one for new/added features - that way the old ones won't change unless a new MB or graphic support is added - which is trivial. Searching thru both config files for these items is a little daunting for me, let alone a less technical person. There is a need to have default selections, but commenting those out and uncommenting what one needs is easy if there isn't a bunch of code wrapped around it and u r only looking at 1 file for each category and only doing it once. Repetier tries to automate this, which is great, IF they can get it to flow smoothly - which I am sure they will if they keep working on it. - This structure might then support a similar configurator for Marlin.

What would be cool, is to run a program that inspected those files and printed out what was configured and what was not. Things that were not configured but required would be flagged.
It might just be in the first part of the ino so that it does not even try to compile if something is wrong.. Sometimes there r misconfigs that generate unintelligible errors only a developer would understand.

@Wackerbarth the main reason I am trying to get this dev working is that it appears that BABYSTEPPING for CoreXY may now work in the Z axis. Is that true??

Sniffle, 2016-01-13:
Thank you Roxy, I've had enough CS education to know the massive
undertaking that is underway, but not enough to really help.  So I do what
I can by trying to remind you coders that are neck deep in 0's and 1's what
the rest of the world needs to see to see it as a truly advantageous
change(in the end result) instead of immediately coming after you all with
torches and pitchforks :-)

Because, in reality, the end user doesnt give two shits about what goes on
under the hood so long as you fix problems and make it easier to implement
settings for the firmware to be uploaded, and in all honesty the way I see
things right now a web interface like repetier will make the end user very
happy, because they wont like having to copy/paste from a wiki for things
they want to change.  Whereas a web interface that gives you a
firmwarecustomization.h that has all of the changes you need and none of
the stuff you don't that you have picked from a list will be a boon to the
community(once you are done)

thinkyhead, 2016-04-20:
Closing old issues for housekeeping. Support for CoreXY is actually complete now in `RCBugFix`.

closed 2016-04-20 by thinkyhead