MarlinFirmware/MarlinDev #316: Printrbot planning new fork

thawkins, 2015-12-27:
The printrbot team is planning a fork of Marlin,  Brooke Drumm has announced this on thier google++ feed. 

The Marlindev community should probaly talk to them to determine what thier beef is. And why they feel that they need to fork the firmware. 

https://plus.google.com/u/0/+BrookDrummpb/posts/gWupi2KvLiC?cfem=1

Roxy-3DPrintBoard, 2015-12-27:
I read the whole blog.   It is unfortunate that they are doing this.   But in the bigger picture, that is what Open Source is all about.   They feel a need to go their own way.   So be it.

The shame of the matter is we are well along to getting the code base broken up so it can be cleaned up.   Instead, they are just throwing anything they don't need over board to reduce the complexity.  It is unfortunate, but my guess is they will have a hard time keeping up once we are 'Clean' again.

iplayfast, 2015-12-29:
I'm once again playing with the code, and am finding it very difficult to debug when in the library format. Maybe it's just me, but I'm thinking of forking as well. For example, I can use the LiquidCrystal library to access the lcd display on my mini-kossel, but I can't figure out how to get the marlindev code to. I've defined 
```
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define ULTIPANEL
#define ULTRA_LCD 
#define NEWPANEL
#define SDSUPPORT
```
which from reading the code should get the ULTRA_LCD stuff going. It doesn't. So I've been reduced to putting #error here statements in the included code to try to figure out where it is getting lost. 

I feel that putting the kitchen sink in the library to be recompiled when the configuration is changed is problematic, as it doesn't always get recompiled. 

The library should consist of low level code like
gcode interperter
A class that can be setup to handle values of printer sizes and translate to stepper values.
A class that can handle the menu systems initialized by the size of the screens

The high level stuff should have the setup and loop routines common to arduino code. 
Examples of that would have uses of the above classes, and they would all look very similar.

I've obviously not thought about all this as much as you folks, so I don't know the thought process that led you to do things the way you've done them, but it feels like it's going against the grain.

Roxy-3DPrintBoard, 2015-12-30:
> I feel that putting the kitchen sink in the library to be recompiled when the configuration is changed is problematic, as it doesn't always get recompiled.
...
> I've obviously not thought about all this as much as you folks, so I don't know the thought process that led you to do things the way you've done them, but it feels like it's going against the grain.

The problem is that we are trying to use Arduino to make things easy for the user.  If that was not the case, we would have the full power and scope of the GCC compiler (and makefiles) available to us.   And much of what you say would be true.    But Arduino imposes a lot of restrictions and we are trying to keep things easy for the user.

I understand that right now, the MarlinDev branch is difficult.  But we are working on it and soon it will be as easy as it always was.  In the mean time, you have several choice.  You can use the Stable Branch on the MarlinFirmware/Marlin side which is unaltered.  Or, there is a provision to just drop your existing Configuration.h file in the proper place on the MarlinFirmware/MarlinDev side of the fence.

thinkyhead, 2016-04-18:
The "new direction" for MarlinDev is to adopt the file layout of MarlinKimbra, bringing over the latest code from `RCBugFix`. The library format for Marlin is not going forward until it becomes obvious to us all that it's a really swell idea. For the moment, now that Arduino 1.6.x allows subfolders, we're going to do a nice folder-based reorganization and split up some of the longer code files into more manageable chunks.

As for Printrbot making their own fork, that is fine. The main Marlin branch can be installed on a Printrbot if users want features that we've done, but which their fork currently lacks. And if we find something in their fork that's clearly better than our work, we can adopt it. Maybe they'll eventually prefer our newer prettier main branch over their stripped-down fork.

thawkins, 2016-04-18:
I suspect the main reason why Printrbot wants a stripped down version is
they ship a at90usb1280 based controller that only has half the flash that
most Marlin controllers have. A full featured Marlin build, with the teensy
2.0 libs added just won't fit.

thinkyhead, 2016-04-22:
> …only has half the flash…

@thawkins Makes sense. Some of our code is probably a little larger for being more generalized too.

closed 2016-05-25 by boelle