MarlinFirmware/MarlinDev #317: Marlin support for Due

Roxy-3DPrintBoard, 2015-12-27:
There seem to be a few places that have gotten (older versions of) Marlin up and running on Due microprocessors.   Once we get the current code base cleaned up, we may want to start looking and putting support in place for a higher performance processor.    Repetier has put a HAL (Hardware Abstraction Layer) in place so their firmware can run on the Due microprocessor.  

When the time is right, we will want to be knowledgeable about these efforts.

For example:   https://github.com/bobc/Marlin/tree/Marlin_v1/Due/Marlin
and
https://github.com/repetier/Repetier-Firmware/tree/master/src/ArduinoDUE/Repetier

Wurstnase, 2016-01-13:
@ me if you need some help. I've done a lot in my fork, but this has actually no good HAL an I mess a lot in the code to bring it to work. Would I start again, I would do the stuff better.

Roxy-3DPrintBoard, 2016-01-13:
@Wurstnase  Are you saying you have a version of the Marlin firmware running on Due ?

I have a couple of questions:

1) Would a functioning HAL have made the port cleaner and easier?

2) Are you seeing a performance increase?

3) How bad where the compiler problems?  Did you have to fight to get a usable .HEX file that would run on the board?

4) How bad is it to get your Pin definitions correct so the firmware can talk to your hardware?

Wurstnase, 2016-01-13:
1) We need something like a HAL. I've done a lot of word to get this up to date. But I'm stopping with that some month ago. 
2) Sure, there is a big performance increase. My printer run without double or quadstepping with 160 steps/mm on a coreXY with 250+ mm/s. The limit on the Due with quadstepping is approx. 300kHz +/-.
3) I use most time 1.5.8 with Eclipse.
4) Pin definitions is the same. Nothing special. I made a fastio.h like the AVR-Version.

Roxy-3DPrintBoard, 2016-01-13:
@Wurstnase    Repetier ( https://github.com/repetier/Repetier-Firmware/tree/master/src ) used 
"adjusted HAL files from John Silvia". 

The HAL files are here : https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoDUE/Repetier/HAL.cpp
and
here: https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoDUE/Repetier/HAL.h

There really isn't that much to them.  

One more quick question:   Knowing what you know now, if you were to leverage the above files, how long would it take to get a 'fresh' version of Marlin up and running on Due ?

Wurstnase, 2016-01-13:
In that HAL-files are temperature-stuff and the ISR. Don't forget the fastio-file. Repetier has also two different ;)

The HAL-file from Repetier are not compatible for Marlin. We need something different. Not much but a bit. The main-task is, to make a HAL-file for AVR. If we don't have any MCU-related thing not in a HAL-file we don't need it.

Roxy-3DPrintBoard, 2016-01-13:
Isn't it the opposite?  Marlin already runs on AVR's.  Don't we need a HAL file that maps those hardware functions to the Due ?        

Anyway...  How close is you fork to working?  Does it actually move the nozzle around and process some GCodes?   If so, it would seem it has most of the magic figured out.

CONSULitAS, 2016-01-14:
@Roxy-3DPrintBoard 
perhaps read https://github.com/MarlinFirmware/Marlin/issues/626

you will perhaps find
* RAMPS FD (e.g. http://www.ebay.de/itm/Pololu-Shield-Ramps-fd-Controller-Control-Board-for-Reprap-Arduino-3D-Printer-/221818365964?hash=item33a567a80c:g:lOsAAOSw8d9UxY4e)
* RADDS (e.g. http://www.ebay.de/itm/RADDS-v1-5-3D-Printer-Controller-for-REPRAP-MENDEL-PRUSA-/281790122423?hash=item419c0009b7:g:lLwAAOSw3ydV5~oF)

Wurstnase, 2016-01-14:
My fork is the closest to the actual RC. Just 2 month or so behind. It runs on my Printer for some month now. Also with ABL.

Roxy-3DPrintBoard, 2016-01-14:
Thank You to both of you!   I'll be reading for a while!

Roxy-3DPrintBoard, 2016-03-30:
@Wurstnase @CONSULitAS 

The Due's don't provide any EEPROM ?    That seems like a big limitation.   I don't really care about the EEPROM because I just keep my settings in the firmware and re-flash if I need to.  But most users do tweak their settings and keep them stored in the EEPROM.

Do you think that is a problem that would limit the acceptance of the Due in a Marlin environment?

Sniffle, 2016-03-30:
There is a fairly popular board called "radds" i think basically a duet
tamps board... I know for some on a delta its a fairly popular option.

Roxy-3DPrintBoard, 2016-03-30:
@Sniffle
I found two RADDS boards on eBay.  But they are add on boards for RADDS.   One is an LCD Panel.  The other board is 2x Stepper controller board.   

Can you check out this link:   http://reprap.me/catalogsearch/result/?q=RADDS

That RADDS board still needs a Due board plugged into it, right?   And I presume a bunch of Step Stick boards, right?

Sniffle, 2016-03-30:
http://reprap.org/wiki/RADDS

Yes ma'am thats the board.

Roxy-3DPrintBoard, 2016-03-30:
OK...  I'm confused.  If the RADDS board still needs a Due board plugged into it, why not go with a RAMPS board and just plug a Due into that?

For example, I think this is everything I would need to convert a printer to a Due, right?

http://www.ebay.com/itm/3D-printer-kit-Ramps-fd-Dashboard-MPU-12864LCD-Kits-can-use-for-Arduino-Due-/121933350465?hash=item1c63cb4641:g:czwAAOSwBnVW8hNA

(It says it ships with a Due, but check out the processor board.  It says:  Mega 2560 which is an 8-Bit AVR)

Sniffle, 2016-03-30:
I honestly dont know... I dont have a due to test...  But one major
difference is 3 extruders vs 2.

Roxy-3DPrintBoard, 2016-03-30:
I'll keep studying things.  But I'm wondering if one of the reasons a RADDS board is better is because the Due needs 3.3 volts and the Mega 2560 uses 5 volts.  I wonder if there is a problem interfacing with stuff like step sticks if you try to use the Due on a RAMPS board?

Sniffle, 2016-03-30:
That is a very real possibiility, specific power distribution for due vs
megs makes logical sense... But we all know how logical the world is...

Endlesshunt, 2016-03-30:
RADDS board provides for the 3.3V/5V supply issue with DUE. It also carries an add-on EEPROM for the DUE. The one problem that I have with the RADSS is that it is not open source.

Endlesshunt, 2016-03-30:
PS. If you plug a RAMPS onto a DUE you will immediately pop the DUE. The RADDS was specifically developed to be a RAMPS for the DUE

Roxy-3DPrintBoard, 2016-03-30:
@Endlesshunt  Thank You for the help!     How do they solve the problem where the output signals on the Due are 3.3 volts but the Step Sticks need 5 volt TTL signals to function correctly?   The Step Sticks are interchangeable between RADDS and RAMPS boards, right?  

And I guess I have the same question about the LCD Panels.  You can use the same LCD Panel on both a RAMPS and RADDS board, right?

Roxy-3DPrintBoard, 2016-03-30:
> The one problem that I have with the RADSS is that it is not open source.

Yeah, It looks like they goofed up on the compliance when they declared they were Open Source.  But they did try and they did publish enough stuff that anybody can follow easily in their foot steps:

http://reprap.org/mediawiki/images/8/81/RADDS_26_sch.pdf
http://reprap.org/mediawiki/images/6/6b/Anschlussbelegung.pdf

CONSULitAS, 2016-03-30:
@Roxy-3DPrintBoard 
Yes, the Arduino Due does not have an EEPROM. But that is only a limitation based on our implementation.

An extension i think about is to make an more abstract implementation of the save/read parameters commands (M5xx) so that they provide multiple paths (EEPROM, SD-Card etc.).
So you could save to SD (e.g. EEPROM.g) which would be read  while booting before auto0.g and be an replacement for the EEPROM. If we do it human readable, it would be great for debugging, too.

BTW: RADDS is only one Board. There is RAMPS FD (**f**or **d**ue), too. E.g. http://www.ebay.de/itm/Pololu-Shield-Ramps-fd-Controller-Control-Board-for-Reprap-Arduino-3D-Printer-/221818365964

Together with an Due clone (e.g. http://www.ebay.de/itm/New-Due-R3-Board-SAM3X8E-32-bit-ARM-Controller-Cortex-M-Compatible-For-Arduino-/161271045825) you can start for aprox. 40$/€.

Endlesshunt, 2016-03-30:
@Roxy-3DPrintBoard They drive everything with 3.3V and limit any inputs to 3.3V. Things like the servo PWM output they supply a 5V power line but drive the PWM signal with 3.3V. They are very careful to keep the 3.3V and 5V rails separate. TTL can work on 3.3V input as long as you keep the TTL output away from the DUE. If you do want to use a widget with TTL output you have to put a level converter between it and the DUE circuitry  

TTL Voltages:
A TTL input signal is defined as "low" when between 0 V and 0.8 V with respect to the ground terminal, and "high" when between 2.2 V and VCC (5 V), and if a voltage signal ranging between 0.8 V and 2.0 V is sent into the input of a TTL gate, there is no certain response from the gate and therefore it is considered "uncertain"

Roxy-3DPrintBoard, 2016-03-30:
> BTW: RADDS is only one Board. There is RAMPS FD (for due), too.

Ah!!!!   OK, that helps a lot!   Now I know what to look for!    And Thanks for the explanation about how they make it work with 3.3 volts (@Endlesshunt).

Endlesshunt, 2016-03-30:
Here is some more data on the RAMPS-FD
http://www.geeetech.com/wiki/index.php/Ramps-FD
Make sure you read the warning at the top of the page

Blue-Marlin, 2016-03-30:
If you really want to warn link to http://reprap.org/wiki/RAMPS-FD.
After reading that, RAMPS-FD is not under consideration any more.

Endlesshunt, 2016-03-30:
@Blue-Marlin Sorry, pasted the wrong link, wanted to paste the one to reprap

CONSULitAS, 2016-03-30:
@Roxy-3DPrintBoard @Endlesshunt 
and another warning: The normal LCD adapter would fry the mcu, too. 
So you have to use e.g this one: http://www.ebay.de/itm/221680360756
Than you can connect this http://www.ebay.de/itm/2004-LCD-Display-3D-Printer-Controller-With-Adapter-For-RAMPS-1-4-Reprap-Mendel-/331799311155 or that http://www.ebay.de/itm/12864-Display-LCD-3D-Printer-Controller-Adapter-For-RAMPS-1-4-Reprap-Mendel-/400994571418 . 

for own hardware you have to use a logic level converter like http://www.ebay.de/itm/IIC-I2C-Logic-Level-Converter-Bi-Directional-Module-5V-to-3-3V-/171890679333

@Blue-Marlin
Thanks for the warning!

Most serious for me is the inverted MOSFET logic which is unsafe and could cause uncontrolled heating.

Roxy-3DPrintBoard, 2016-03-30:
Damn!  Nothing is ever easy, is it?

CONSULitAS, 2016-03-30:
fun: mod a 7$ RAMPS for Due in less than 1 hour
you find this http://forums.reprap.org/read.php?219,479626,486426#msg-486426
reading here http://forums.reprap.org/read.php?219,449522,494255#msg-494255

CONSULitAS, 2016-03-30:
AHHH!
http://forums.reprap.org/read.php?219,479626,505774#msg-505774 tells the reason for the issues with inverted logic:

Arduino Due IO ports are HIGH by default, because of internal pull-ups enabled by default. So the port has to be inverted to be safe.

WARNING: The modified RAMPS 1.4 has the same safety issue like RAMPS FD 1.4!

It is because of different port logic on the MCU.

jbrazio, 2016-03-31:
> Yes, the Arduino Due does not have an EEPROM. But that is only a limitation based on our implementation.

I do agree with you @CONSULitAS if the settings are abstracted, the DUE could use SDcard as the storage media for settings, and MEGA could continue to use EEPROM. The only drawback I see is we would make mandatory for any Marlin32 user to have an SDcard or re-flash every time we wants to change a setting.

jbrazio, 2016-03-31:
Did you guys read the links @Blue-Marlin posted about RAMPS-FD ?

> The only safe version of Ramps-FD is the V2.A, issued in February 2014. Former versions present high safety risks which were solved by the V2 issue. Unfortunately, The only Ramps-FD available on the market is yet (february 2016) on revision 1.2, while its manufacturer was repeatedly advised that the only safe version is the 2.A.

> - Unprecise Thermistors
- Inverted Heater/Mosfet-Logic, which causes Heater turn-on during Reset
- get rid of cheap Mosfets, which get too hot
- weak 5V-Supply from Arduino Due

Sniffle, 2016-03-31:
Roxy i think your building an army to move marlin forward :-D

Has anyone looked into the cost to manufacture ramps fd V2.xxx boards?  I
realize it wouldnt be cheap... Just a thought... Id dedicate a printer to
it... I just dont have what is needed to make one

Roxy-3DPrintBoard, 2016-03-31:
Wow!  I didn't realize the 32-Bit side of the world for 3D-Printers was so chaotic.  Reading all of those threads I have some more questions:

Q)  What is a Duet package?  It is some enhancement to the Due but I can't find more information.

Q)  What is and where do you get a Smoothie Board?    Is this a viable 32-Bit platform for Marlin?

Q)  What are the other viable 32-Bit processors that have boards with lots of GPIO ports to plug in different things?

jbrazio, 2016-03-31:
[Smoothie Project](http://smoothieware.org/) -- they have firmware and hardware, don't seem a good bet for us in terms of first choice. In fact one of the great things about Marlin is the multiple boards we support, for Marlin32 we should select (at least for the beginning) the uC present on the most possible number of boards ?

Roxy-3DPrintBoard, 2016-03-31:
>  for Marlin32 we should select (at least for the beginning) the uC present on the most possible number of boards ?

Yes...  But so far, we haven't even found one board that looks fully baked and ready to use.   The Due is OK, but there is nothing trust worthy to plug it into.

closed 2016-05-25 by boelle

Drakelive, 2016-05-25:
Hi Boelle

Why did you close this request? This is a hot topic, to start on Branch Development. This is desired by many people, including me.

Drk

boelle, 2016-05-25:
it has been moved not closed

boelle, 2016-05-25:
https://github.com/MarlinFirmware/Marlin/issues