MarlinFirmware/MarlinDev #319: suggestion

iplayfast, 2015-12-29:
Put the install instructions url in the readme so it is more easily locateable.

https://github.com/MarlinFirmware/MarlinDev/wiki

Roxy-3DPrintBoard, 2015-12-30:
I think this is a reasonable request.

Roxy-3DPrintBoard, 2015-12-30:
Please help me out.  I don't understand.  I went to the Readme file to make the change and it appears it has always been there:

####This is the source code which produces the Marlin Firmware

Complete instructions for setting up the build environment and configuring the firmware can be found at the wiki. In order to build the firmware, you will also need to setup the build environment. The official build environment is based on the Arduino IDE 1.6.6.

thinkyhead, 2016-04-18:
We have a number of helpful links in the `Configuration.h` file now. The "final" location of Marlin documentation is in flux, but going forward we should be able to use http://www.marlinfw.org/ and link to the main topic for each section of the config.

closed 2016-05-25 by boelle