MarlinFirmware/MarlinDev #322: FirmwareCustomization.h interferring with user customization

iplayfast, 2016-01-02:
Finally figured out what was happening. None of my customizations were being done. 
Turns out that when you create the example and get the Marlin.ino and the FirmwareCustomization.h the modifications you do to FirmwareCustomization.h are overwritten by the FirmwareCustomization.h that is in the libraries/MarlenDev/ directory. 

Deleting FirmwareCustomization.h in the MarlineDev Directory removes the problem.

Wackerbarth, 2016-01-02:
I must question if you are following the instructions?  There is no Marlin.ino file. It is called MarlinFirmware.ino  AND it resides AND STAYS in the examples folder. You should either edit the FirmwareCustomization.h next to it or use an entry in the Marlin Boards menu to select the particular build that you wish to do.

Wackerbarth, 2016-01-02:
My apologies. I was looking at a different branch than the one which you seem to be referencing.
The compiler is not handling #includes as it should. I have a new version proposed.

(And my "very latest" is now "Jan2" in my copy of the repository.

thinkyhead, 2016-04-22:
Closing this, as we will not be mandating the use of the new configuration concept. At least not yet. Still much to do in bringing the code into arrangement.

closed 2016-04-22 by thinkyhead