MarlinFirmware/MarlinDev #324: I can't see an option from the menu to stop printing while printing.

a4jp-com, 2016-01-02:
I can't see an option from the menu to stop printing while printing.

I have no kill button but it would be nice to have to be able to stop the printing from the menu. 

At the moment I just turn off the power but the print head is touching the model and it leaves a small bump.

Wackerbarth, 2016-01-02:
Glen,

Perhaps you should reconsider your "home" button and make it a "kill" instead.

a4jp-com, 2016-01-02:
Okay let's do that for now.

The only problem for other people with this printer is the word home is printed onto the metal plate and they probably don't want to have to make the home button a kill button if possible. I would also prefer the home option over a kill option if the code connecting to that pin works. The code I'm using now seems really good and I spent the day tuning up all the calculations and I also made two other changes to the en_16 file.

```

#define MSG_PREHEAT_ABS                     "Heat ABS" //changed
#define MSG_PREHEAT_PLA                     "Heat PLA" //changed

```  

I had to do that because the rest of the message got cut off with those two.

If the home button ends up being a kill button in the meantime I'm okay with that but can "Stop printing" still be added to the menu as a feature later on? I think it would be a nice addition. Having it as an option that can be turned on in the setup file would be great too. It would probly need one other setting though. An option to turn off a default "Are you sure?" message, for people that just want to quickly quit when printing.

This stop printing setting would just have to lift the head up a little bit with a G-code movement then activate the kill setting.

a4jp-com, 2016-01-02:
I just pushed the updates I made to a4jp-com/MarlinDev-a4jp (https://github.com/a4jp-com/MarlinDev-a4jp/commit/1eec58964c5c798a0d860d2757952f5221a72a39)

Wackerbarth, 2016-01-02:
"If the home button ends up being a kill button in the meantime" -- Yes, I was just looking for a stop-gap measure to give you the function (which is probably more critical at this time).

It shouldn't be too difficult to make some changes to the menu. As you note, the operational functionality is already there. All we need to do is link it up in the menu.

a4jp-com, 2016-01-02:
Very nice ^^

KiteLab, 2016-01-02:
If printing from SD-card use the already existent menu-item. (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.cpp#L422)
If printing from a host use that to stop.

a4jp-com, 2016-01-04:
Nice. I will try some SD cards prints to check if that code is good tomorrow.

So far everything is going really well. I just need a pause/stop option that lifts the print head a bit. Maybe 10mm would be good. 

Does the current kill or suicide option move the head up, or just stop the print and leave the head touching the model?

closed 2016-04-19 by thinkyhead