MarlinFirmware/MarlinDev #325: How to I keep the menu open for move axis?

a4jp-com, 2016-01-02:
How to I keep the menu open for move axis? The menu always times out and goes to the top default screen but I would like to disable the time out just on this menu item.

a4jp-com, 2016-01-02:
Even just an increase of the timeout would be nice if this is difficult. Where is the value that sets the length of the timeout?

KiteLab, 2016-01-02:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.h#L36

a4jp-com, 2016-01-02:
Thank you @KiteLab, 15000 milliseconds seems like a lot of time but I'll double it for now. I always seem to run out of time when trying to calibrate.

If I want to set the LCD_TIMEOUT_TO_STATUS for one menu item. Is that tricky or would I just need to add an if statement to one of the menu items in ultralcd.cpp?

KiteLab, 2016-01-02:
You could make an other exception about here:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.cpp#L1710

thinkyhead, 2016-04-18:
The latest code now has a flag `defer_return_to_status` that can be set by any menu or menu item that wants to prevent the timeout. It should be cleared when the menu is done. Currently it's only used for the Manual Bed Leveling items, because they are multi-step. But they could be applied to any edit item, if that makes it work better.

closed 2016-05-25 by boelle