MarlinFirmware/MarlinDev #327: The feed rate or flow rate percentage mark isn't changing on the LCD panel.

a4jp-com, 2016-01-04:
The feed rate or flow rate percentage mark on the status screen isn't changing when I go into tune during a print and adjust the flow rate / feed fade. Does everyone else have this problem or is it just my machine?

a4jp-com, 2016-01-07:
How do we change the value on the LCD screen?

a4jp-com, 2016-01-10:
Any ideas on this?

KiteLab, 2016-01-10:
How about changing Feedrate and or Flow in the Tune-Menu?
Changing Feedrate on the Status screen may be broken, for you, because of your missing encoder-weel.

https://github.com/MarlinFirmware/MarlinDev/blob/dev/ultralcd.cpp#L346

```    #if ENABLED(ULTIPANEL_FEEDMULTIPLY)
      // Dead zone at 100% feedrate
      if ((feedrate_multiplier < 100 && (feedrate_multiplier + int(encoderPosition)) > 100) ||
          (feedrate_multiplier > 100 && (feedrate_multiplier + int(encoderPosition)) < 100)) {
        encoderPosition = 0;
        feedrate_multiplier = 100;
      }
      if (feedrate_multiplier == 100) {
        if (int(encoderPosition) > ENCODER_FEEDRATE_DEADZONE) {
          feedrate_multiplier += int(encoderPosition) - ENCODER_FEEDRATE_DEADZONE;
          encoderPosition = 0;
        }
        else if (int(encoderPosition) < -ENCODER_FEEDRATE_DEADZONE) {
          feedrate_multiplier += int(encoderPosition) + ENCODER_FEEDRATE_DEADZONE;
          encoderPosition = 0;
        }
      }
      else {
        feedrate_multiplier += int(encoderPosition);
        encoderPosition = 0;
      }
    #endif // ULTIPANEL_FEEDMULTIPLY

    feedrate_multiplier = constrain(feedrate_multiplier, 10, 999);
```
Test if ULTIPANEL_FEEDMULTIPLY is defined.
Test if a smaller ENCODER_FEEDRATE_DEADZONE can help you.

a4jp-com, 2016-01-10:
I can change the feed rate number while printing with the up, down, left and right buttons but the status screen values don't change.

How can I get this code working with my buttons? I tried #define ULTIPANEL_FEEDMULTIPLY in my Customization.h file but nothing changed.

a4jp-com, 2016-01-11:
@KiteLab when you say "How about changing Feedrate and or Flow in the Tune-Menu?" do you mean in the code?

KiteLab, 2016-01-12:
I mend, during a print you should have a 'Tune' menu in the main menu, where you can tune Feeredrate and Flow. The code i mentioned above is the place where the Feedrate is tuned on the Status-screen.
But all that seems to be irrelevant since you clarified that you can change the values, only the __display__ for the Feedrate on the Status-screen is messy.
This evening i investigated if FR could be written over from an overflow of an previous line. But since everything is written top down on the display that is unlikely. Sorry i have no further ideas. Even if you messed up #311 it should work. But without having a 16x4 display i'm not absolutely sure..

thinkyhead, 2016-04-22:
@a4jp-com Any luck? Still need this issue open?

a4jp-com, 2016-04-22:
It still doesn't change.

Blue-Marlin, 2016-04-22:
Who wonders. It would need about the same changes for your up/down buttons as in the menu-code, but at a different place (checked in the status screen).

thinkyhead, 2016-05-03:
> It still doesn't change.

Using `RC6` or `RCBugFix`? I will take a look at the code and try it on my machine. I haven't used this option lately.

a4jp-com, 2016-05-03:
Thank you for checking Thinkyhead. I'm not sure what version of the software I'm on but @Wackerbarth helped me set up everything and the button code for the printer was working after some super programming on his part. I'll see if I can find out from him. Hopefully all the updates from the version I'm on now got moved into the latest version here.

thinkyhead, 2016-05-03:
> got moved into the latest version here

The latest version is actually here: https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

Blue-Marlin, 2016-05-03:
@thinkyhead 
> and try it on my machine

@a4jp-com has no encoder-wheel but a bunch of buttons. Wackerbarth made a kind off custom version for him. It's very unlikely that RCBugFix will work for him. Search for "Scoovo" in Dev.
[image]

thinkyhead, 2016-05-03:
@a4jp-com Do you have that code posted? If Marlin supports the Scoovo, and the Scoovo has this controller standard, then it makes sense to add that controller as an option.

thinkyhead, 2016-05-03:
So… Looks like some fairly simple changes are needed. Currently Marlin only acknowledges the `BTN_UP` (etc.) pins for the `RIGIDBOARD` controller. If we relax that condition and simply initialize these pins in `lcd_init` when they exist, and then check them in `lcd_buttons_update` (when `MB(AJ4P)`?) then it should just work.

a4jp-com, 2016-05-04:
Hi. Sorry for the late reply. The version that works so far on my machine is here: [https://github.com/a4jp-com/MarlinDev-a4jp/](https://github.com/a4jp-com/MarlinDev-a4jp/)

thinkyhead, 2016-05-05:
@a4jp-com Yep. I looked at your branch first, then I created this PR which makes the necessary changes to add support to the current Marlin codebase: MarlinFirmware/Marlin#3670 Please test at your earliest convenience.

closed 2016-05-25 by boelle