MarlinFirmware/MarlinDev #328: Homing before and after G29

DeathmannX, 2016-01-05:
//English

Hi Guys,

first things first.
A big THX for devolping this awsome Firmware!

I've got a Problem with the G29 Command.
If i run it the Heatbed will be homed in 3 Points (grid) ok.
But after this homing i cann't move back to the Endstops.
I can only move back to them with homing the X and Y Axis.

But if i do this i think the G29 points will be discarded.

//German

Hallo Leute,

zuerst einmal DANKE für die unglaubliche Firmware die Ihr geschaffen habt.

Leider jedoch habe ich genau mit dieser ein Problem.
Ich kann nach dem ich G29 ausgeführt habe nicht mehr zurück zum Endstop fahren
und muss erst die X und Y Achse wieder homen bevor das geht.

Ich befürchte jedoch das dann auch die G29 Punkte verworfen werden.

Roxy-3DPrintBoard, 2016-01-06:
Please post a copy of your Configuration.h file.    We can see how you have things setup with that.

DeathmannX, 2016-01-06:
And a other Mistery for me is if i set the Z Probe offset to -0.12 and home the Z Axis and type in M114 it shows me Z:0.12 if i go manual to Z0 it is ok.

configuration.h http://pastebin.com/JLCu7zgg

closed 2016-05-25 by boelle