MarlinFirmware/MarlinDev #330: G-code 1 might be broken

a4jp-com, 2016-01-11:
When I send the command G1 Z14, this should keep moving the bed down 14mm, shouldn't it? It goes down and just stays at 14mm. Is this normal? How do I get the bed to keep going down 14mm when I send G-code? G0 seems the same.

Nandox7, 2016-01-11:
You need to change from absolute to relative move.
Check GCodes G91 and G92.

a4jp-com, 2016-01-11:
I typed in G91 Z5 but nothing happened. It's also the same for G92.

a4jp-com, 2016-01-11:
Do I have to use a combination of the G-codes?

Nandox7, 2016-01-11:
Nop. You use G91/92 independently.
Example:
---
G91
G1 Z10
---
or
---
G92
G1 Z10
---

a4jp-com, 2016-01-11:
I see. Thank you so much for that @Nandox7 . 

I had a bad sample of code and now I know how to get everything working ^^

G91 ; Change movement system to relative position system - calculates from current position
G1 Z25 ; Z25 Moves platform down 25mm

closed 2016-01-11 by a4jp-com