MarlinFirmware/MarlinDev #331: endstop inverting customization causing compiler crash

Sniffle, 2016-01-13:
endstop inverting using FirmwareCustomization.h causes compiler crashes, 

setting const bool, or #define and editing the rest of the X(Y)_MIN_ENDSTOP_INVERTING string properly per syntax causes the compiler to crash no matter what.

My theory is that it is set as a constant, but after removing the const, it still fails, so it may need to be changed to a #define in the large config file until it can be sorted properly.

I'm out of steam though, going to bed...

Wackerbarth, 2016-01-13:
That has already been addressed in the ./transitional_default_configurations/default/Configuration.h file (and by extension ALL configurations that derive from that series.

#ifdef X_MIN_ENDSTOP_INVERTED
  const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
#else
  const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
#endif

… and similarly for all of the others.

ALL entries in the FirmwareCustomization.h (or any other customization) will be in the form of #define’s.

I just pushed my development branch (Jan10) over to MarlinDev. It still has some issues to be straightened out before I propose it as a permanent part of the dev branch. But it does contain, as have all of my branches for some time, the first cut toward factoring the configuration file.

The “transitional_default_configurations“ are only a first step in that factoring.
*** WARNING*** — If you get any compiler errors about things being redefined, you MUST resolve them by eliminating the “default” version. As a “work around”, I recommend that you simply make a copy of the default file and place that copy in a custom configuration folder and reference your version instead of the transitional ones. Then where there is a conflict, just eliminate the offending entries. I do have a migration process that will allow you to phase out that work-around file once the new base files stabilize. Eventually, all of that will occur automatically by having “guards” around each entry. However, I haven’t done that on a wholesale effort because it gets in the way of my effort to find the commonality and factor it out. So, the condition is temporary.

> On Jan 13, 2016, at 2:34 AM, Todd Swindoll <notifications@github.com> wrote:
> 
> end-stop inverting using FirmwareCustomization.h causes compiler crashes,
> 
> setting const bool, or #define and editing the rest of the X(Y)_MIN_ENDSTOP_INVERTING string properly per syntax causes the compiler to crash no matter what.
> 
> My theory is that it is set as a constant, but after removing the const, it still fails, so it may need to be changed to a #define in the large config file until it can be sorted properly.
> 
> I'm out of steam though, going to bed...
>

Sniffle, 2016-01-13:
awesome, thank you.  It's been a while since I've touched anything just getting back up to speed.

Sniffle, 2016-01-13:
looks like the docs are already out of date. I'll just wait to figure this out until the config is in a more permanent state.  then I'll figure it out and help with documentation.

closed 2016-01-13 by Sniffle