MarlinFirmware/MarlinDev #332: Change Temperature noozle while is printing

fbarcenas, 2016-01-14:
Hi,
i use display VIKI2.
is easy, while is printing if i go to
       Menu-> Tune-> noozle.
Show me:
      FLOW:   100
In this moment i can move this value but never i can press OK. I must wait seconds to out automatically.

Also while i printing if i go to:
   Menu-> Control->Temperature-> noozle
Show me
    PID-P: 12.5
In this moment i can move this value but never i can press OK. I must wait seconds to out automatically.

Some idea?

Thanks!

zvevqx, 2016-01-22:
Hi same for me 
( marlin RC3 / ramps 1.4 / reprapDiscountDisplay / sd Enable ) 

menu goes like TUNE > NOZZLE > FLOW ....  with value of 10 and then i can change to whatever ans clik nothing change ... 
no idea already try to Re-upload the firmware without change

fbarcenas, 2016-01-22:
This problem is supposed fixed her:
https://github.com/MarlinFirmware/Marlin/pull/2823#issuecomment-173186191

but not....

thinkyhead, 2016-04-20:
If we're not seeing this with `RCBugFix` then we can close this. That code will be migrating to MarlinDev before too long.

fbarcenas, 2016-04-20:
Is fixed, please close!

closed 2016-04-20 by fbarcenas