MarlinFirmware/MarlinDev #333: Marlin_main.cpp down to 5000 lines in Wackerbarth/Jan10 fork...

Roxy-3DPrintBoard, 2016-01-18:
I noticed a bunch more GCodes have been extracted.   I was going to start messing with my ideas for Delta Mesh leveling.  Is it safe to use this branch?

Wackerbarth, 2016-01-18:
Yes, and no.  Please review the individual commits individually. In particular, there are some which are titled to indicate that they need review. (And some which are there to help conform to your earlier code but which probably do not really belong in the permanent firmware.)

Please provide feedback so that we can removed the uncertainties and also make those changes which belong to your particular situation into a patch to be applied to the end of the working branch.

Roxy-3DPrintBoard, 2016-01-18:
A couple of problems and a question:

First, can I only have one branch in my C:\Users\Roxy\Decuments\Arduino\libraries\ folder?   It seems I get errors if I two different branches there.   It doesn't seem like I should get errors just because I have two different branch of the tree there???

~~~~
Multiple libraries were found for "MarlinFirmware.h"
 Used: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10
 Not used: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev
Using library Marlin-RoxyDev-Jan10 at version 1.2.0-dev in folder: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10
~~~~

Also, it seems we have a scope problem in Marlin_Main.cpp right now:

~~~~
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DSPECIFIC_PRINTER_ID=RoxyDeltaDevelopment -DMARLIN  -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR  -IC:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\sketch "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\cores\arduino" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\variants\mega" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2" "-IC:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10" "C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\MarlinSerial.cpp" -o "C:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\libraries\Marlin-RoxyDev-Jan10\MarlinSerial.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DSPECIFIC_PRINTER_ID=RoxyDeltaDevelopment -DMARLIN  -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR  -IC:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\sketch "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\cores\arduino" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\variants\mega" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2" "-IC:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10" "C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\Marlin_main.cpp" -o "C:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\libraries\Marlin-RoxyDev-Jan10\Marlin_main.cpp.o"
C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\Marlin_main.cpp: In function 'void gcode_T(uint8_t)':

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\Marlin_main.cpp:3436:36: error: 'sync_plan_position_delta' was not declared in this scope

           sync_plan_position_delta();

                                    ^

Multiple libraries were found for "MarlinFirmware.h"
 Used: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10
 Not used: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev
Using library Marlin-RoxyDev-Jan10 at version 1.2.0-dev in folder: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10 
Using library LiquidCrystal at version 1.0.1 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal 
Using library U8glib at version 1.18 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib 
Using library LiquidCrystal_I2C at version 1.1 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C 
Using library Wire at version 1.0 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2 (legacy)
exit status 1
Error compiling.
~~~~

Wackerbarth, 2016-01-18:
> On Jan 18, 2016, at 10:19 AM, Roxy-3DPrintBoard <notifications@github.com> wrote:
> 
> A couple of problems and a question:
> 
> First, can I only have one branch in my C:\Users\Roxy\Decuments\Arduino\libraries\ folder?
> 
Yes. Only one at a time.
> It seems I get errors if I two different branches there. It doesn't seem like I should get errors just because I have two different branch of the tree there???
> 
> Multiple libraries were found for “MarlinFirmware.h"
So, how would the IDE know which one you want? They have the same name. You can either use a different, more powerful, IDE or, as I do, swap versions of the source in and out using `git` to swap branches.
> C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\Marlin_main.cpp: In function 'void gcode_T(uint8_t)': [...]

That code is activated by multiple EXTRUDERS (which I haven’t tested recently).  

For now, add
“#include “movement.h”
to Marlin_main.cpp

But, considering the kind of code that is involved, it, too, is a good candidate to be moved out of that file.

Roxy-3DPrintBoard, 2016-01-18:
> So, how would the IDE know which one you want? They have the same name. 

If the .INO file generated a different library name, would there be a conflict?   I'm wondering if there is an easy way to work around the limitation.   It would be kind of nice to be able to have everything in place and just load the appropriate .INO file to work on the version associated with it.

I'll take a look at the scope issue and see if I can clean that up nicely.

Roxy-3DPrintBoard, 2016-01-18:
Problems...  Problems....   Arduino is complaining about an ambiguous reference to set_plan_position();   But it sure looks to me like there is only one version defined.

~~~~
// planner motion
#if ENABLED(HAS_LEVELING)
  void plan_set_position(float x, float y, float z, const float& e);
#else
  void plan_set_position(const float& x, const float& y, const float& z, const float& e);
#endif // HAS_LEVELING
~~~~

If I comment out the second declaration, the error goes away.  The problem is in G28.cpp.   Some how it isn't getting HAS_LEVELING as #define'd.   So it generates the 2nd reference and that causes problems later during the link.

Error messages follow:

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DSPECIFIC_PRINTER_ID=RoxyDeltaDevelopment -DMARLIN  -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR  -IC:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\sketch "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\cores\arduino" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\variants\mega" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2" "-IC:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib\src" "C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G27_link.cpp" -o "C:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\libraries\Marlin-RoxyDev-Jan10\G27_link.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DSPECIFIC_PRINTER_ID=RoxyDeltaDevelopment -DMARLIN  -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR  -IC:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\sketch "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\cores\arduino" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\variants\mega" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2" "-IC:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib\src" "C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp" -o "C:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp.o"
In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:25:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h: In function 'void sync_plan_position_delta()':

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:31:92: error: call of overloaded 'plan_set_position(float&, float&, float&, float&)' is ambiguous

     plan_set_position(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS]);

                                                                                            ^

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:31:92: note: candidates are:

In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:19:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/planner.h:60:8: note: void plan_set_position(float, float, float, const float&)

   void plan_set_position(float x, float y, float z, const float& e);

        ^

In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:25:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:25:8: note: void plan_set_position(const float&, const float&, const float&, const float&)

   void plan_set_position(const float& x, const float& y, const float& z, const float& e);

        ^

In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:0:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp: In function 'void gcode_G28()':

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:71:30: error: 'mbl' was not declared in this scope

     uint8_t mbl_was_active = mbl.active;

[... cut ...]
Using library U8glib at version 1.18 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib 
Using library LiquidCrystal_I2C at version 1.1 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C 
Using library Wire at version 1.0 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2 (legacy)
exit status 1
Error compiling.

Roxy-3DPrintBoard, 2016-01-19:
And...   G28.cpp and G29.cpp both need this snippete added to the end of the #include's : 
~~~~
#if ENABLED(MESH_BED_LEVELING)
#include "mesh_bed_leveling.h"
#endif
~~~~

And G92.cpp is also guilty of not getting #define HAS_LEVELING set correctly....

Wackerbarth, 2016-01-19:
``` #define HAS_LEVELING ``` is a part of configuration. It must come from the customization/configuration/conditionals stack. It is not something that would be set directly in the code.

I suspect that the problem has to do with the fact that your leveling is not quite the same as the normal auto leveling.

Wackerbarth, 2016-01-19:
```G28``` should not be affected by leveling. It needs to operate in ```FRAME``` coordinates. Its purpose is to establish the location of the actuators, and thus the tool holder.

Wackerbarth, 2016-01-19:
Re: plan_set_position()
"Do we really need both definitions? It would seem we could always just use the first one???"
We don't **NEED** both definitions. However, the latter definition generates more efficient code for those printers that do not need the variability of "bed leveling".

Roxy-3DPrintBoard, 2016-01-19:
> G28 should not be affected by leveling. It needs to operate in FRAME coordinates. Its purpose is to establish the location of the actuators, and thus the tool holder.

Well, OK...   But...

> And Marlin_main.cpp, Homing.cpp, Probe_Management.cpp & G92.cpp are also guilty of not getting #define HAS_LEVELING set correctly....

All of these have conditional stuff based on HAS_LEVELING'.   Specifically, right now, all of these #include "movement.h" which has these lines:

~~~~
// planner motion
#if ENABLED(HAS_LEVELING)
  void plan_set_position(float x, float y, float z, const float& e);
#else
  void plan_set_position(const float& x, const float& y, const float& z, const float& e);
#endif // HAS_LEVELING
~~~~

Here is the tail end of the compile:

Arduino: 1.6.7 Hourly Build 2015/12/11 03:44 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560), Roxy's Delta Development"

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DSPECIFIC_PRINTER_ID=RoxyDeltaDevelopment -DMARLIN  -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR  -IC:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\sketch "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\cores\arduino" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\variants\mega" "-IC:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib\src" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire" "-IC:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2" "C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp" -o "C:\Users\Roxy\AppData\Local\Temp\build41e606a977895eeebcbedee6c3c36307.tmp\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp.o"
In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:28:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h: In function 'void sync_plan_position_delta()':

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:31:92: error: call of overloaded 'plan_set_position(float&, float&, float&, float&)' is ambiguous

     plan_set_position(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS]);

                                                                                            ^

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:31:92: note: candidates are:

In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:22:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/planner.h:60:8: note: void plan_set_position(float, float, float, const float&)

   void plan_set_position(float x, float y, float z, const float& e);

        ^

In file included from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\gCodes/G28.cpp:28:0,

                 from C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10\G28_link.cpp:2:

C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10/movement.h:25:8: note: void plan_set_position(const float&, const float&, const float&, const float&)

   void plan_set_position(const float& x, const float& y, const float& z, const float& e);

        ^

Using library Marlin-RoxyDev-Jan10 at version 1.2.0-dev in folder: C:\Users\Roxy\Documents\Arduino\libraries\Marlin-RoxyDev-Jan10 
Using library LiquidCrystal at version 1.0.1 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal 
Using library U8glib at version 1.18 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\U8glib 
Using library LiquidCrystal_I2C at version 1.1 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidCrystal_I2C 
Using library Wire at version 1.0 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Users\Roxy\AppData\Local\Arduino15\packages\Marlin\hardware\avr\1.0.5\libraries\LiquidTWI2 (legacy)
exit status 1
Error compiling.

Roxy-3DPrintBoard, 2016-01-19:
> #define HAS_LEVELING is a part of configuration. It must come from the customization/configuration/conditionals stack. It is not something that would be set directly in the code.

> I suspect that the problem has to do with the fact that your leveling is not quite the same as the normal auto leveling.

Where would be the correct place to #define HAS_LEVELING ?    I'll put it there and delete it from the top of the C files.  I'm guessing Customization.h would be the right place to put it???

Wackerbarth, 2016-01-19:
Yes, Customization.h is the appropriate place to insert it. However, in the future, it is likely that it will become something enabled as a result of some specific kind of leveling.  We'll just need to see how that all shakes out.

Roxy-3DPrintBoard, 2016-01-19:
More problems....

I modified some values in Customization.h but they aren't sticking.   Values in some other Configuration.h file are over riding my values.   

Specifically....   Try turning on MESH_BED_LEVELING and then try to set the values for:
~~~~
#define MESH_NUM_X_POINTS 7
#define MESH_NUM_Y_POINTS 7
~~~~
in your Customization.h file.   The values getting set are 3, but I don't know where that is coming from (because there are million files with that value in them) and I don't know how to make my values stick.

Wackerbarth, 2016-01-19:
As has been described previously, if you are using the "transitional_defaults", in the default definition, each of those entries will eventually have a "guard" around it so that your values will take precedence. However, since I am trying to factor out the commonality, those guards get in the way (and would be a lot of tedious effort, much of which will get deleted). So, you need to turn on the extra warnings in  the compiler and look at the redefinition warnings. They will tell you exactly where the values are getting redefined. Then, you can ether add the guards to those definitions, or simply delete the definitions from your copy of the defaults.  Eventually, this will become a non-issue.

Roxy-3DPrintBoard, 2016-01-19:
Can you tell me how to turn on the compiler warnings for redefinations?   That would save me a lot of effort.   Once I find where it is happening, I'll put the fences around the code to stop it.

Wackerbarth, 2016-01-19:
In the IDE preferences window, set the Compiler Warnings to "All".

Roxy-3DPrintBoard, 2016-01-19:
That wasn't too painful.   We have a couple of dozen warnings we should fix up.  I had 3 or 4 in the G27 code and a couple in M48.  I got those fixed.

a4jp-com, 2016-01-24:
If you need anything checked again @Wackerbarth, or @Roxy-3DPrintBoard, I've got my printer opened up and can see all the pins etc on the board now. I'm also finely tuning all the prints. The biggest problem was the acme thread I was given has a tiny wobble (+/- 0.1mm) at the top of it where the 3D printer company ground the rod to fit it into a bearing chase. I've kind of fixed it now but my glass bed is also a bit warped. Time to find a glazier.

I've just found 4 problems with the Marlin firmware so far while printing.

1. Feature: The LCD menu needs a stop printing option for people like me without a kill switch.
2. Long file names don't auto scroll.
3. FC on the status screen doesn't change from 100% when you change values during a print through the menu.
4. One model I found to test temperatures at different heights made the printer go into a thermal runaway at the same point every time. Is there something I should change to fix this?

Besides these issues everything is going really well.

Also a note to people using Slic3r: The calculations for infills are currently wrong so if you test Marlin and try to set your X, Y and Z values make sure the detect thin walls option is off and try not to use infill.

Any news on updates on the RC?

Roxy-3DPrintBoard, 2016-01-25:
@a4jp-com 
> If you need anything checked again @Wackerbarth, or @Roxy-3DPrintBoard, I've got my printer opened up and can see all the pins etc on the board now. 

Are you running a Delta Printer?   In a couple of weeks, I'm going to need some help shaking out some code?

a4jp-com, 2016-01-25:
I have a Bowden printer but I think @Wackerbarth might be using a Delta.

I also go a fantastically flat piece of 3 mm glass today so I guess all my prints are going to come out really well from now on. 
[image]
[image]
[image]
[image]

Roxy-3DPrintBoard, 2016-01-25:
> I have a Bowden printer but I think @Wackerbarth might be using a Delta.

Yeah... I was already planning on twisting his arm to load some code when the time comes.  

> I also go a fantastically flat piece of 3 mm glass today so I guess all my prints are going to come out really well from now on. 

You are saying the exact wrong things!   :)    When the time comes, I think the code I want help testing is going to be able to handle horribly un-flat pieces of glass.   I'm almost positive I can get a 15 x 15 mesh for compensation to run.   I think there is a good chance I can get a 20 x 20 mesh to run.

a4jp-com, 2016-01-25:
I have a piece of glass like that so I'll keep it for the tests ;)

a4jp-com, 2016-01-25:
I wanna buy a IR depth sensor next. 
Anyone here buy one and know where could I get one if they are good?

a4jp-com, 2016-02-05:
Any updates, tests or further bug fixes been done yet?

Roxy-3DPrintBoard, 2016-02-07:
I'm making progress, but the problem is I'm too busy with other things.   But I am making progress and BioPrinter is helping me with some of the issues I'm running into.   I can say I'm glad I'm off in my own file and not changing a 5000 line behemoth!

a4jp-com, 2016-02-08:
Fantastic to hear everything is moving forward. I've been doing a bit of printing and everything is looking pretty good. I'd like to get an IR z depth sensor sometime though. Any good suggestions for something that would plug into a RAMBo board and set up easily in Marlin?

Roxy-3DPrintBoard, 2016-02-08:
>   I'd like to get an IR z depth sensor sometime though.  Any good suggestions for something that would plug into a RAMBo board and set up easily in Marlin?

So, I'm not really the one to ask.   But over the last year a lot of people have come to 3DPrintBoard for help trying to get proximity probes working.   Most of them got them going.  But I'm not sure anybody got more accuracy than you can get with a micro-switch.   

Do you really need a proximity probe?   It would be good to hear from somebody that got a proximity probe going and actually saw a benefit because with a good mechanical setup and a good switch, you can get sub .01 accuracy very repeatably.  Even on my bogus RepRap kit, my Chinese (sub $1.00 micro-switch) I can get .02 mm repeat-ability without even trying very hard.

thinkyhead, 2016-04-20:
I began the process of moving files from `RCBugFix` into a layout mostly like MarlinKimbra. I even split up every GCode into its own individual .cpp file. It is a long way from compiling yet, but that is one possible way to shrink `Marlin_main.cpp` down to almost nothing. Even without applying other reorganization, we could split up all the GCode handler functions into `.h` files (so that they can be part of the same "code unit" as they are now) and then just include the lot of them from `Marlin_main.cpp` – or more likely a "master include" file that could include all the `M123.h` files.

With the g-code handlers split up into individual files (either .h or .cpp files) the other parts of `Marlin_main.cpp` can (and should) be split up into a `Robot` or `Printer` class (for the modeling of the printer) and a `Parser` class (for the GCode parser).

Anyway, random thoughts on an old topic…. As we reorganize the size of this huge main code file will naturally begin to diminish.

I will post the branches where I'm trying new organizational schemes pretty soon so we can mess around with them and see what works best.

a4jp-com, 2016-04-20:
Any reorganization that makes coding easier or bug fixes are cool ^^

Roxy-3DPrintBoard, 2016-04-20:
> I will post the branches where I'm trying new organizational schemes pretty soon so we can mess around with them and see what works best.

Great!   But I have one quick question.   Is it going to be hard to make sure we have all the changes that go into the Release Candidates if we start the move too soon?   Would it be best to wait until we freeze the last Release Candidate and promote it to Golden Status?  And then start the clean up?

jbrazio, 2016-04-20:
I would freeze Dev until this RC cycle ends, branch the existing Dev, and start over the master Dev branch from the GA release.

If we're going into a folder structure I assume we will put aside the "Arduino lib like" scheme ?

I do not warns to be in @thinkyhead shoes when we will rebase MarlinDev with RCBugfix.. so whatever makes his life easier works for me.

thinkyhead, 2016-04-23:
> start over the master Dev branch from the GA release

Yes.

>  I assume we will put aside the "Arduino lib like" scheme?

God, yes!

> I do not want to be in @thinkyhead shoes when we will rebase MarlinDev…

I'm sure it will be hours of fun!

closed 2016-05-25 by boelle