MarlinFirmware/MarlinDev #334: How to read analog inputs with Marlin

CoderSquirrel, 2016-01-20:
Hi guys
I have to use analog inputs with marlin but i'm facing troubles with it.
I've used, A3, A4, A5 ,A9, A11, A12, for testing my code.
In all of them i used analogRead to get the input, once fastio write only read digital.
And i saw that this input was changing the temperature. When the value increase it does the same w/ temp and when decrease again
Before you say something:
1. i made sure that those pins we're not being use.
2. i'm no with the printer mounted, i only working with the ramps/arduino and displays
I tried to understand the reading for TEMP_0 but i think that what i need could be more easy.
3. No i am not mixing with temp values, for make sure of it i tried only placing a simple analogRead, without saving this anywhere just for make sure it still happens.
So how should i use analog inputs properly? How should i read in the right way? Have someone saw this happening before?
Thanks

Roxy-3DPrintBoard, 2016-01-20:
Analog inputs are only available on certain pins.  That is because it takes extra circuitry to perform the analog to digital conversion.   First, pull up the spec's for the processor you are using and make sure the pin you are connecting to can be used in analog mode.

Then, take a look at this tutorial:  http://www.electroschematics.com/10053/avr-adc/     It will help you with a lot of the concepts.    First, you have to start the analog conversion.  Then you have to wait for it to finish.   Finally, you can go read the value.

This tutorial does a 'busy wait'.   It executes instructions looking to see if the conversion is finished.  That is a very simple way to do things, but it wastes the processor to accomplish its goal.  The Marlin code starts a conversion, and then at interrupt time, it goes and reads the value.  That is better but much more complicated.   Depending on what the purpose is for reading the analog pin, a 'busy wait' might work for you.

CoderSquirrel, 2016-01-20:
>@Roxy-3DPrintBoard: Analog inputs are only available on certain pins.

Yes there are only few ones and in arduino these pins are the A pins, the one which i'm using. 
It may seems dumb what i'm about to say but i have no knowledge about it and after read i have some questions, so when you have a doubt you better solve thank keep it.
Once ADC converts analog in digital i will have to read the ana log value anyway don't i? What i  saw that is the problem it the analog reading using analogRead, so in my tiny little head it seems unsufull.
BUT AGAIN, i don't have knowledge about this was just a wondering.

CoderSquirrel, 2016-01-20:
@Roxy-3DPrintBoard  i found this article http://www.gammon.com.au/adc what make me understand xD
 And i used this part of code http://www.fpaste.org/312887/
but i'm kinda worried with this part:
`ADCSRA =  bit (ADEN); `
` ADCSRA |= bit (ADPS0) |  bit (ADPS1) | bit (ADPS2); `
` ADMUX =   bit (REFS0) | (adcPin & 0x07);  // AVcc   `
in arduino code it's used as
`ADCSRA = 1<<ADEN | 1<<ADSC | 1<<ADIF | 0x07;`
i changed the one in the code for the one already in arduino, it worked, but i'm kind in doubt about the 0x07, as in the arduino code it's already there, do i need to take it of from the my code?

Roxy-3DPrintBoard, 2016-01-20:
If it is working, I wouldn't worry about it.   But here is what you do to fully understand what is going on so you can make a smart decision.   Pull up the AVR spec for the processor you are using.  Go look up the defination of the ADCSRA register.      

From the code, it looks like they are turning on the ADEN, ADSC and ADIF bits.  We don't know what bit numbers those are defined as.  But what they are doing is shifting a '1' to that location, they are or'ing them all together, and they are writing that to the ADCSRA control register.   I don't know what that 0x07 is for.   But if you look at the definition of ADCSRA in the spec, they are sticking 1's into the lowest 3 bit positions (using that 0x07).

CoderSquirrel, 2016-01-21:
So since ADCSRA is already defined at temperature.cpp I don't need to declare it again, right?

Roxy-3DPrintBoard, 2016-01-21:
> So since ADCSRA is already defined at temperature.cpp I don't need to declare it again, right?

Maybe an expert will jump in and enlighten us.  I really don't know much about this stuff.

But...   ADCSRA is never defined in Temperature.cpp.   Maybe it is in one of the .h files.   But it seems to be something the compiler knows about.  You can 'reference' it and presumably it does the right thing.   My guess is there is a definition for it buried in a header file somewhere that is getting pulled into the compilation.

To answer your question...  No!  You don't have to declare it.  Just use it.   And make sure you understand what the spec says about that register.  When you start playing with the configuration registers of the CPU, there is no telling what happens if you set a bit wrong!

CoderSquirrel, 2016-01-21:
Well, i'm using a version, i think is 1.0 that in the line 864 of temperature.cpp and has
>  ADCSRA = 1<<ADEN | 1<<ADSC | 1<<ADIF | 0x07;
Well i'm searching about registers but 'im kind busy with my actual job.
But i can make it rok now, in a parallel sketch, inside the marlin i'm still having trouble, i think due to both temperature and my analog input using the same register i need to learn more about it.
You said (i dont remember in which topic, here or marlin) that register isn't the thing you most know and so, do you know which guy worked in this part of code, here at github (the one active in the repo) so i can go to them and see if they can help me? 
But sure man you helped me a lottttttttttttt

CoderSquirrel, 2016-01-21:
Another thing:
As the temperature keep reading, like it reaally cant stop, i was thing about that may i will have to work at a sync for this situation once that both inputs will be using the same register.
Because what i saw from now is that the temperature keep the register always busy

Roxy-3DPrintBoard, 2016-01-21:
The specification for the processor you are using will have all the details.   It will be a 50 or 100 page manual.   That register will probably have a whole page of information.   You need to hunt that down and absorb it.   Setting up an ADC to work for a given application is not trivial.

Bastini, 2016-01-28:
Hello guys,

I've got a similar issue, I pretend to use a free analog input port on the prusa i3 hardware to read the value (0-5v) of a external sensor that I've got. However, when I do analogRead(A5) the values displayed are floating alot.. I guess the adc is not reading well. I've used other free ports but it still the same. (The sensor works just fine and I can read the signal if I use other separated arduino). Do you know how I can solve my issue? Thanks in advance

KiteLab, 2016-01-28:
All values read by analogRead() have always a lot of noise, on every system.
Oversampling/averaging is the answer.

CoderSquirrel, 2016-01-28:
Hi, as i hadn't time enough to study adc, i managed it by adding an arduino
nano to read my input and turn it to digital, as my use of it was only
because with analog i can have 3 types of input, positive, negative and
idle, i could solve this way. But if you gonna read the analogs and use
it's value, like temperature, percentage or any other way i think i can't
help.

One thing that i kinda figured out was that, as temperature uses the adc
continually it really cant be stopped, and the arduino only have one adc
register, so it gonna be a little hard to manage.

Bastini, 2016-01-28:
Thanks for reply. Actually, my sensor is quite good and when I connect it to a Uno or Mega I can read the value very steady without any average or oversampling. However, when I connect it to the printer's hardware (let's say analog A5) the analogRead is quite awful. I have been searching for stuff over the last 2 days but I can't figure out how to solve it. I could use external uno/mega and conect to the printer via serial but there is no serial ports available as well..

Wackerbarth, 2016-01-28:
It would be good to see if the issue is related to the firmware or to the analog components.
Perhaps someone with the problem could connect another Arduino to the same power supplies (in parallel) and also to the same analog reference voltage and see if they get similar noise if they are doing the monitoring on the alternate processor.  It could be that the use of the printer is affecting the stability of the analog references.

CoderSquirrel, 2016-01-28:
I havent done it but i think it's related to the use of the adc register in
temperature reading. Once analog read use isr to interrupt the adc to read
the analog, it kind of alter the analog reference while reading.

Bastini, 2016-01-28:
Wackerbarth, it is a plausible argument that I share as well.. I have put a good voltimeter connected parallel and the sensor signal output is clean.. 
However, if I shunt ground to the analog port A5(or any other) the reading is always steady 0... but If I shunt it to 5v the issue appears... I wonder if it is related to oversampling or average or something related to clock cycles...

I am now looking at the temperature code of marlin to see how they acquire the analog voltage and sorting how I somehow could do the same for other analog port. (In a ordinary mega, I think I can put 16 or 15 different analog voltages and read them all )

thinkyhead, 2016-04-20:
Can we close this old issue for housekeeping?

CoderSquirrel, 2016-04-20:
Yeah @thinkyhead i will close it

closed 2016-04-20 by CoderSquirrel

Bastini, 2016-04-23:
Yes it is solved. Thanks!