MarlinFirmware/MarlinDev #335: How to configure MarlinDev

htaxil, 2016-01-20:
Hi,
I constructed a Delta printer with a probe for the Bed Auto Leveling incorporated into the effector, so that the HotEnd's nozzle is the probe. I would be interested in the M48 function developed by Roxy. He told me that it was integrated in the new MarlinDev firmware, so I installed this environment on my PC. It works fine and I can compile "File->Examples->MarlinDev/MarlinFirmware" with no problem at all.

However, I am a bit confused because this new version of the firmware adopts a new method of statement of parameters that the wiki does not explain (https://github.com/MarlinFirmware/MarlinDev/wiki/Installing-Customized-Configurations)

My question: Is there a documentation, even has the state of draft somewhere? If not, can you tell me what are the new files of configuration to use (customization.h, FirmwareCustomization.h...) and how to integrate them into the firmware?

Roxy-3DPrintBoard, 2016-01-20:
Which branch did you grab?    The Delta enabled version of M48 is not in all branches.    If you want that and the Delta Physical Bed Leveling G27 function, you probably want to bring up:

https://github.com/Wackerbarth/MarlinDev/tree/Jan10      

(There is also a Jan10 branch here.  Probably it is the exact same thing but I don't know that for a fact.   The above link will give you what I'm working off of)

That is where most of the surgery of the Marlin code base is happening.    Right now, I'm working on a Delta Mesh Bed Leveling system that is based on that branch.   I encourage you to use that branch.     There are instructions on how to setup and use the branch in the Wiki here:

https://github.com/MarlinFirmware/MarlinDev/wiki

If you run into problems, just ask for guidance.  If you can figure out the issue for yourself, please update the Wiki to help others following behind.

htaxil, 2016-01-20:
I installed the dev branch on my PC using Git. The "Jan10" branch is not listed by Git : 

...\Arduino\libraries\MarlinDev> git branch -r
  origin/ABORT_ON_ENDSTOP
  origin/FAST_PWM_FAN_issue
  origin/HEAD -> origin/dev
  origin/IDE_1.6.5
  origin/IDE_1.6.6
  origin/MarlinPlatform-1.0.6
  origin/MarlinPlatform-1.6.6
  origin/MarlinPlatform-1.6.7-b
  origin/OLED_LCD_Support
  origin/TotalPrintCount
  origin/dev
  origin/master
  origin/platform
  origin/versioning

Roxy-3DPrintBoard, 2016-01-20:
I see it here:   https://github.com/MarlinFirmware/MarlinDev/tree/Jan10

But like I said, most likely this is identical to https://github.com/Wackerbarth/MarlinDev/tree/Jan10 but until we hear from Wackerbarth, it is safer to go to the 2nd link.   The Delta stuff I recently did is  in that branch.  (M48 and the G27 Delta Physical Bed Leveling)

Wackerbarth, 2016-01-20:
The Jan10 branches on Wackerbarth/ and MarlinDev/ both have the same code, but with slightly different commit sequences.  I just pushed a Jan20 branch that has a couple of minor fixes (like adding a #include that Roxy needed -- but probably doesn't affect other users). I'm still waiting for some feedback concerning a few of the commits where I was attempting to incorporate some of Roxy's code)

Roxy-3DPrintBoard, 2016-01-20:
> I'm still waiting for some feedback concerning a few of the commits where I was attempting to incorporate some of Roxy's code

I'm using the Jan10 branch and things are not too crazy.  Pretty much, everything is doing what I need it to.   The worst thing I got hit with is the EEPROM Write (and Read) routines were using a short to say how many bytes to move and that was killing me on saving and restoring the Mesh.   I suspect that is why the Cartesian Mesh code says not to go bigger than 7:

~~~~
  #define MESH_NUM_X_POINTS 3  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 3
~~~~

7*7*4 is 196.   If you move to 8x8 that is 256 and turns into a 0.    And as you know...  I'm thinking I want to be able to do 15x15 or maybe even 20x20.   So the EEPROM Read and Write had to change.

Wackerbarth, 2016-01-20:
As I suggested previously, you can easily reduce the bytes used by formatting them in another manner.
There is no reason to change the size of the EEPROM counter just to handle this one usage.

Roxy-3DPrintBoard, 2016-01-20:
> As I suggested previously, you can easily reduce the bytes used by formatting them in another manner.
> There is no reason to change the size of the EEPROM counter just to handle this one usage.

Yes, I know...   I'm going to get it working first with floating point.  And then I'm going to move to a less memory intensive int or maybe even short.   But even in a shrunk down and compressed version I'll be well over 256 bytes of Mesh data.  Going into a loop to write each element is going to be more expensive on the code footprint than just changing those routines to accept int's.   

I wish I could get GCC to generate -ii files so I could look at the generated code.   My guess is changing that parameter from short to int hardly changes the code.  But without those -ii files, I don't know that for a fact.

Wackerbarth, 2016-01-20:
I'm more concerned about the possible existence of individual elements in the configuration exceeding 255 bytes. If we ever want to have variable length values stored, the maximum size of any individual struct becomes a concern.

Roxy-3DPrintBoard, 2016-01-20:
> I'm more concerned about the possible existence of individual elements in the configuration exceeding 255 bytes. If we ever want to have variable length values stored, the maximum size of any individual struct becomes a concern.

Well, all of the Read's and Writes are done using these macros.
~~~~
#define EEPROM_WRITE_VAR(pos, value) _EEPROM_writeData(pos, (uint8_t*)&value, sizeof(value))
#define EEPROM_READ_VAR(pos, value) _EEPROM_readData(pos, (uint8_t*)&value, sizeof(value))
~~~~

I changed the underlying _EEPROM_????Data() functions to not do the wrong thing with the sizeof() operator.   We could still force the same behavior by changing the macros to:
~~~~
#define EEPROM_WRITE_VAR(pos, value) _EEPROM_writeData(pos, (uint8_t*)&value, (unsigned short) sizeof(value))
#define EEPROM_READ_VAR(pos, value) _EEPROM_readData(pos, (uint8_t*)&value, (unsigned short) sizeof(value))
~~~~
and I could call the _EEPROM_writeData() and _EEPROM_readData() functions directly.   Or, I could add code to do the nested loop to write each element.

Wackerbarth, 2016-01-20:
The implementation is addressing bytes. Changing it as you suggest doesn’t change anything concerning the storage. It only affects the stack space allocated to pass that size to the eeprom i/o routine..

However, if you do add any element which exceeds 255 bytes, then we would not be able to store the length of that element in one byte.

Generally, storage schemes work best when they use a uniform representation. This would likely imply using 2 bytes for the length of EVERY element is such a store.

Roxy-3DPrintBoard, 2016-01-20:
Another thing I was considering doing is moving the Mesh table to the very end.  Part of the reason for doing this is because I think the Mesh Table can be highly compressed.  That would leave everything else always to stay at the same address.    There are a number of things we can talk about.  Nothing is cast in stone.

Wackerbarth, 2016-01-21:
I've been considering the elimination of any FIXED location for particular entries. Particularly if we decide to allow variable length entries (to hold, for example, micro-code sequences -- think "Allen Key scripts", etc.) A key/value scheme might be more appropriate.  That would also allow us to preserve things like "Total Objects Printed" even across revisions to the firmware and changing kinematic model parameters.

Roxy-3DPrintBoard, 2016-01-21:
> I've been considering the elimination of any FIXED location for particular entries. 

I agree.   But until we get there, a lot of people are saving their configuration settings on SD-Memory cards and such.   By keeping the addresses for everything but the Mesh constant, it makes it easier for them to migrate to a new setup.    The Mesh (in its full blown incarnation) is going to burn a pile of memory.   I want to make it so if a user wants a 30x30 mesh, they can do it. (Nobody is going to have the patients to get that many points set exactly right...  But still...)    

I think from an ease of use perspective, we want every thing else at a fixed address so the user can migrate as easily as possible.   And who knows?  Maybe the Delta Mesh code has an 'Unload' feature where the data can be dumped and reloaded later if need be?   That would even cure the issue of moving to a new version with extra EEPROM settings.

You know what else makes sense???  What if we added an MCode command that dumped all the current settings (that work well for a person's printer), and a corresponding MCode to load all the settings.  If a new setting showed up in the new firmware, they at least have all of their old settings.   That would be a nice, clean transition to the model you have been talking about.   Maybe you could write it?   Otherwise, in a couple of months when I'm done with Delta Mesh Bed Leveling, I would be up for writing it!

htaxil, 2016-01-22:
Oh, I feel that the discussion has moved away from the original topic!
To get back to my question about the new organization of the parameters:
- I configured the Jav10 and Jan20 environments on my PC.
- I generated an UUID for my printer: 493abfb3-e066-4681-8ccc-fdf96dd733c2
- I have to modify the boards.txt file to add my printer to the menu tools of the IDE Arduino
- By taking back the example MarlinFirmware, I created my own couple of files MarlinFirmware.ino and FirmwareCustomization.h
The file MarlinFirmware is identical to the original, my modifications are only in the file FirmwareCustomization.h

For the moment, I contented with making an include in my FirmwareCustomization.h file towards a Customizations.h file i have created. In that Customizations.h file, I put back all the parameters specific to my printer which were before distributed in the configuration.h, configuration_adv.h and pins_RAMPS_13.h files. I doubt whether it is the right solution. That compiles, but I do not know what that compiles...

I go back to my original question :
- How do we have to implement the configuration in the new organization? Which is best practice concerning the new organization of the parameters (I repeat it, the wiki says nothing on "The Structure of a Configuration File", "Selecting the Components" or "Referencing a Configuration File"
- Where are set all other parameters which were before in configuration files configuration.h, configuration_adv.h?
- In my file Customization.h, I can define or overload a value, but how deactivate a parameter (that we commented before in configuration files)?

Here my custom configurations :
C:\Users\Hubert\AppData\Local\Arduino15\packages\Marlin_for_IDE_1.6.7\hardware\avr\0.0.2\boards.txt
>mega.menu.marlin.493abfb3=DeltaFold
>mega.menu.marlin.493abfb3.build.unique_id_flags=->DCONFIGURATION_FILE=configurations/custom/493abfb3/Customization.h
>mega.menu.marlin.493abfb3.build.unique_id_flags=-DSPECIFIC_PRINTER_ID=493abfb3
>mega.menu.marlin.493abfb3.build.unique_id_flags=-DVENDOR=Hubert
>mega.menu.marlin.493abfb3.build.unique_id_flags=-DMODEL=ModelDeltaFold

E:\Users\Hubert\Documents\Arduino\DeltaFoldMarlinFirmware\FirmwareCustomization.h
>/*
> *  Customization for a particular printer
> *
>    Although you could define all of your customization here,
>    and place the customization for that printer in the customization area.
>    You can then reference that file either from the command line or by defining the path here.
>
>    If you have multiple printers, it is suggested that you create a file for each of them.
>    You can then select one of them from a menu in the Arduino IDE.
>    Please see ... for more information
>
>    Even if you have only one, you can utilize that same mechanism to specify the id that you have assigned to that printer
>    or, you could define it here *
> */
>
>#if 0
>
>  // You can obtain a unique id by visiting https://www.uuidgenerator.net/version4
>  // version 4 UUID for DeltaFold : 493abfb3-e066-4681-8ccc-fdf96dd733c2
>
>  #define  SPECIFIC_PRINTER_ID 493abfb3
>
>#endif
>
>/*
> *
> *  Printer specification
> *
>
>    Whether specified here, or in a unique specification file, the following printer description should be followed.
>
> */
>
>#if 0
>  // If you are still using a configuration file from Marlin 1.1, you could simply include it.
>
>#endif
>
>// but it is preferable to convert to the inherited customization structure
>
>#if 0
>
>#define  UUID "493abfb3-e066-4681-8ccc-fdf96dd733c2"
>
>// Place all of your overrides first
>
[... 117 lines cut ...]
>
>//-----------------------------------------------
>// Declarations du configuration_adv.h
>//-----------------------------------------------
>#define BABYSTEP_MULTIPLICATOR 1 //faster movements
>
>#define Z_HOME_BUMP_MM 5 // deltas need the same for all three axis
>#define HOMING_BUMP_DIVISOR {10, 10, 10}  // Re-Bump Speed Divisor (Divides the Homing Feedrate)
>
>#define MANUAL_FEEDRATE_XYZ 50*60
>#define MANUAL_FEEDRATE { MANUAL_FEEDRATE_XYZ, MANUAL_FEEDRATE_XYZ, MANUAL_FEEDRATE_XYZ, 60 } // Feedrates for manual moves along X, Y, Z, E from panel
>
>// (don't use SLOWDOWN with DELTA because DELTA generates hundreds of segments per second)
>//#define SLOWDOWN

>//-----------------------------------------------
>// Declarations du pins_RAMPS_13.h
>//-----------------------------------------------
>
>//#define Z_MIN_PIN          18
>#define Z_MIN_PIN          32
>
>//#define Z_MIN_PROBE_PIN  32
>#define Z_MIN_PROBE_PIN  18
>

Wackerbarth, 2016-01-22:
Roxy,I see some problems with your proposal to put the mesh parameters “at the top” of the EEPROM.(1) There is an issue of “Where is the top?” Various processors have differing amounts of that memory resource.(2) There are other features, such as the “Total objects printed?” which, equally, want to claim a similar status that protects them from other changes in the set of values stored.(3) You will need to store the description of the size of the mesh above the mesh itself because, even if you know the end address, you will need to compute the beginning address.(4) Unlike some other parameters, the mesh corrections are sufficiently time-sensitive that they will need to be stored in RAM during operation of the printer. I question whether these AVR processors really have enough memory to store an array of the size that you describe.(5) If the printer really could make use of that many correction points, I think that there is a more serious problem with the assumption that the bed is “smooth”.  I would suggest that some other kinematic correction should be utilized.As to the idea of “Unloading” all of the stored values and being able to reload them, it is a good idea (that is already implemented for most parameters).  However, it might simplify things if we had a move consistent mechanism rather than various ad-hoc M codes for the multitude of parameters.

Wackerbarth, 2016-01-22:
Sorry for the delay in responding. My ISP had a problem which took me offline for a whole day :(

> On Jan 22, 2016, at 4:16 AM, Hubert TAXIL <notifications@github.com> wrote:
> 
> Oh, I feel that the discussion has moved away from the original topic!
> To get back to my question about the new organization of the parameters:
> 
> I configured the Jav10 and Jan20 environments on my PC.
> I generated an UUID for my printer: 493abfb3-e066-4681-8ccc-fdf96dd733c2
> I have to modify the boards.txt file to add my printer to the menu tools of the IDE Arduino

Actually, rather than modify boards.txt, you can (and probably should) place those lines in a “boards.local.txt”  Although the functionality is identical, it will be easier to handle future upgrades by keeping your personal customization of the IDE separate from the distribution.
> By taking back the example MarlinFirmware, I created my own couple of files MarlinFirmware.ino and FirmwareCustomization.h The file MarlinFirmware is identical to the original, my modifications are only in the file FirmwareCustomization.h
That is one, perfectly acceptable, approach.  However, if your intent is to customize the IDE menu, you do not need to do anything in “FirmwareCustomization.h”.   The parameters taken from the menu will direct the build to use your custom files to provide the customization information. Using this approach, I can build various configurations from the same identical source files.
> For the moment, I contented with making an include in my FirmwareCustomization.h file towards a Customizations.h file i have created. In that Customizations.h file, I put back all the parameters [...]
It is probably OK. I’d have to look at the exact “#includes” that you also utilized.

If you are using them, there is one potential “gotcha” in the, currently incomplete, “transitional_*” files. The very parameters that you want to customize will conflict with, and override, your customizations. In the future, those problems will be eliminated as we add “#ifndef” guards arounf the default specifications.  However, in the interim, I suggest that you take the following steps to avoid the issue.

(1) In the IDE preferences, turn on “ALL” for the Warning level.  Then pay strict attention to any warnings that a symbol is redefined. You need to eliminate the default version of that variable.

(2) You can do this by removing the offending entry.  A “cleaner” approach would be to copy the default files to your custom configuration directory and include that version of the file instead of the one from the temporary distribution.

As an alternative, you can simply provide ALL of the configuration variables just as you did, except for location, with the Marlin 1.1-RC structure. 

> I go back to my original question :
>
> How do we have to implement the configuration in the new organization? Which is best practice concerning the new organization of the parameters (I repeat it, the [...]
Scott’s “ENABLED/DISABLED” macro mechanism if meant to provide the solution to this. Rather than removing a default definition, you pre-define it to be logically false. Then the to-be-added #ifndef guards in the defaults will suppress the alternate.

> Here my custom configurations :
> C:\Users\Hubert\AppData\Local\Arduino15\packages\Marlin_for_IDE_1.6.7\hardware\avr\0.0.2\boards.txt
>
> mega.menu.marlin.493abfb3=DeltaFold [...]
Not quite correct. You will have only one of the unique_id_flags lines. I would use the SPECIFIC_PRINTER_ID version.
Also note that if you are using the last form, both VENDOR and MODEL will appear on the same line. (But, unless you are distributing printers (or printer kits) to others, this is not the preferred form. I would suspect that you will want to have

```
#define VENDOR   xxxxx
#define MODEL      xxxxx
```
styled lines in your Customization.h file.
> E:\Users\Hubert\Documents\Arduino\DeltaFoldMarlinFirmware\FirmwareCustomization.h
>
> /*
>
> Customization for a particular printer
> *
> Although you could define all of your customization here,
> and place [...]
If you wish to place the PRINTER_ID here (rather than in the boards menu), you will need to place it outside of the “if 0” block which is effectively “commenting out” this specification.
> #endif
>
> /*
> *
>
> Printer specification
> *
>
> Whether specified here, or in a unique specification file, the following printer description should be followed.
>
> */
>
> #if 0
> // If you are still using a [...]
Again, this line is not active.  There are alternate ways to accomplish the same result. In general, you should choose only one of them.
> // Place all of your overrides first
>
> //#define BUILD_HEIGHT_ADJUSTMENT ( -23 )
>
[... 9 lines cut ...]
<snip>

> //-----------------------------------------------
>
> // Declarations du configuration_adv.h
> //-----------------------------------------------
> #define BABYSTEP_MULTIPLICATOR 1 //faster movements [...]
// This won’t be needed if you derive your configuration from delta/generic (since it is already there in that file)
#define SLOWDOWN 0
> //-----------------------------------------------
> // Declarations du pins_RAMPS_13.h
> //-----------------------------------------------
> 
> //#define Z_MIN_PIN 18
> #define Z_MIN_PIN 32
> 
> //#define Z_MIN_PROBE_PIN 32
> #define Z_MIN_PROBE_PIN 18
> 
You also need to specify the “base” configuration that you are altering.

#include "configurations/transitional_default_configurations/delta/generic/Configuration.h”

might be the appropriate place to start.  However, as mentioned earlier, while things are undergoing reorganization, a copy of those files into
E:\Users\Hubert\Documents\Arduino\libraries\MarlinDev\configurations\custom\493abfb3\ and referencing the (edited) copy may work better for you in the interim.

htaxil, 2016-01-23:
Thanks for your answhere, it helps a lot :)

>_...place those lines in a “boards.local.txt” ..._
>_...You will have only one of the unique_id_flags lines..._

I corrected my board declaration and put it in an individual file C:\Users\Hubert\AppData\Local\Arduino15\packages\Marlin_for_IDE_1.6.7\hardware\avr\0.0.2\\**boards.local.txt** :

> \########################################
> \## Arduino Mega or Mega 2560
> \## Ajout des parametres pour DeltaFold
> \########################################

>mega.menu.marlin.493abfb3=DeltaFold
>mega.menu.marlin.493abfb3.build.unique_id_flags=-DSPECIFIC_PRINTER_ID=493abfb3
>mega.menu.marlin.493abfb3.build=-DCONFIGURATION_FILE=configurations/custom/493abfb3/Customization.h
>mega.menu.marlin.493abfb3.build=-DVENDOR=Hubert
>mega.menu.marlin.493abfb3.build=-DMODEL=ModelDeltaFold

Ok
  
>_...you do not need to do anything in “FirmwareCustomization.h”..._

In the case this one has no statement I suppose. (that is not the case at the moment)...

>_...The parameters taken from the menu will direct the build to use your custom files to provide the customization information..._

It's true that I make a reference to the same configuration file from FirmwareCustomization.h and boards.local.txt. I've thus deleted all statement that contained my FirmwareCustomization.h file, because by selecting my printer in the Arduino tools menu, I already have a reference to my file Customization.h (if I understood well).

In my file Customization.h, I almost kept nothing, but I added a #include towards my Configuration.h file that I placed in the same folder as Customization.h

Here is the contents of my Customization.h file :

>//-----------------------------------------------
>// Declarations du configuration_adv.h
>//-----------------------------------------------
>#define BABYSTEP_MULTIPLICATOR 1 //faster movements
>
>#include "configurations/custom/493abfb3/Configuration.h"

OK

>_A “cleaner” approach would be to copy the *default files* to your custom configuration directory and include that version of the file instead of the one from the *temporary distribution*._

>_As an alternative, you can simply provide ALL of the configuration variables just as you did, except for location, with the Marlin 1.1-RC structure._

I made a mix by duplicating the MarlinDev/configurations/transitional_default_configurations/delta/generic/Configuration.h  configuration file that I placed in my custom folder and in which I transferred the parameters of my MarlinRC configuration.

>_Then the to-be-added #ifndef guards in the defaults will suppress the alternate._

But for now, i need to modify the configuration file that contain this setting

During compilation I actualy get two errors :
>In file included from E:\Users\Hubert\Documents\Arduino\DeltaMarlinFirmware_Transitional\DeltaMarlinFirmware_Transitional.ino:10:0:
>
>E:\Users\Hubert\Documents\Arduino\libraries\MarlinDev/CheckForConfigurationRequirements.h:138:10: error: #error You must have a Z_MIN or Z_PROBE endstop to enable Z_MIN_PROBE_REPEATABILITY_TEST.
>         #error You must have a Z_MIN or Z_PROBE endstop to enable Z_MIN_PROBE_REPEATABILITY_TEST.
>          ^

>E:\Users\Hubert\Documents\Arduino\libraries\MarlinDev/CheckForConfigurationRequirements.h:326:4: error: #error HEATER_0_PIN not defined for this board.
>   #error HEATER_0_PIN not defined for this board.
>    ^

**Z_MIN or Z_PROBE endstop**
In my custom configuration.h file, I positioned the lines below but it did not solve the problem :
>#define Z_PROBE  18
>#define Z_MIN  18

**HEATER_0_PIN error**
The HEATER_0_PIN error refers to this test:
>#if !HAS_HEATER_0
>  #error HEATER_0_PIN not defined for this board.

And Has_HEATER_0 is set in the file conditionals.h line 429 :
>#define HAS_HEATER_0 (PIN_EXISTS(HEATER_0))

It seemed to me that __HEATER_0__ corresponded to nothing since the file pins_RAMPS_13.h (which is the one who concerns me) define sets the variable __HEATER_0_PIN__ on line 110

>#define HEATER_0_PIN     10   // EXTRUDER 1

I therefore concluded in a naming variable problem, but even by changing the conditionals.h file as this, it makes no difference :
>#define HAS_HEATER_0 (PIN_EXISTS(HEATER_0)) --> #define HAS_HEATER_0 (PIN_EXISTS(HEATER_0_PIN)) 

Any idea about these two errors ?

htaxil, 2016-01-23:
[CompilationLog20160126-1315.txt](https://github.com/MarlinFirmware/MarlinDev/files/101848/CompilationLog20160126-1315.txt)
I had to have syntax errors in my configuration files, by dint of doing modifications, i corrected the two errors. I have however new errors, but it appears that it concerns the code (see the log file attached)

Wackerbarth, 2016-01-23:
A quick glance leads me to suspect some problem with the pin assignment for the extruder.
It would be most helpful if I could accurately reproduce your exact build.

The reason I created the configurations/custom/ structure was to provide a space where we can mange multiple configurations such as this without conflicting with other configurations.

The best way to communicate the files would be to have you create a fork of MarlinDev on github and then commit your configuration files to a personal fork. Then I could accurately reproduce your exact compilation, make comparisons, etc.

BTW: your boards.local.txt file should have two, and only two, (non-comment) menu lines (for each individual printer configuration). In your case,
```
mega.menu.marlin.493abfb3=DeltaFold
mega.menu.marlin.493abfb3.build.unique_id_flags=-DSPECIFIC_PRINTER_ID=493abfb3
```
The additional lines that you have are not actually doing anything.

htaxil, 2016-01-23:
You can find my configuration here : https://github.com/htaxil/MarlinDev.git
branch Jan20_DeltaFold

In my custom configuration directory, you can find two folders containing my MarlinRC configuration (that works well) and the Arduino files configuration.

Wackerbarth, 2016-01-24:
Thanks for posting it in this manner. It made it very easy for me to reproduce your build and find the problem.

configurations/custom/493abfb3/Configuration.h <>
...
284
285
286
287
288

289
290
291
...
284
285
286

287
288
289
290
291
292
@@ -284,8 +284,9 @@
 
 #endif // PIDTEMP
 
-#define INVERT_E0_DIR
-#define INVERT_E1_DIR
+// For direct drive extruder v9 set to true, for geared extruder set to false.
+#define INVERT_E0_DIR true
+#define INVERT_E1_DIR true
 
 
 

There is a certain degree of inconsistency in the way that boolean selectors are used/defined in the Marlin 1.1 & 1.2 configuration files.

In this case, you need to define the INVERTs as either true or false

Roxy-3DPrintBoard, 2016-01-24:
> There is a certain degree of inconsistency in the way that boolean selectors are used/defined in the Marlin 1.1 & 1.2 configuration files.  In this case, you need to define the INVERTs as either true or false

Yuck!   We should clean that up.   But in the mean time, we should put a very descriptive comment on the Configuration.h lines to make sure that it always has a true or false.

htaxil, 2016-01-25:
Is there other boolean constant that should be set (DELTA, AUTO_BED_LEVELING_GRID, Z_MIN_PROBE_ENDSTOP, DISABLE_ZMIN_ENDSTOP, MANUAL_HOME_POSITIONS...) ?

Compilation bothers me because : the result (warnings, errors) vary even when I compile twice the same configurations..
eg : 
--> Firt compilation : warning: "PIDTEMP" redefined. 
--> I comment the #define PIDTEMP in my custom Configuration.h and compile again --> OK.
--> I uncomment the #define PIDTEMP in my custom Configuration.h and compile again --> OK ???

I do not like very much when the result of a command varies without explanation....
I thus wonder if my configuration files are correctly and entirely taken into account in every compilation, How to be there sure? Or is there a cache that have to be emptied ?

At present, the compilation ends with always the same message: "Low memory available, stability problems may occur.", do I have to worry about it?

Nevertheless, I updated my printer firmware to Marlin 1.2.0-dev (Yeahhhh) and was able to tested the M48 command. It is stunning. Roxy, were you able to establish statistics concerning precision of various types of Z sensors ?

Roxy-3DPrintBoard, 2016-01-25:
> I updated my printer firmware to Marlin 1.2.0-dev (Yeahhhh) and was able to tested the M48 command. It is stunning. Roxy, were you able to establish statistics concerning precision of various types of Z sensors ?

There has not been any formal study of the precision of various types of Z-Probes.   But I can tell you even the very cheap micro-switches from China perform very well.  Pretty much, any noise in the numbers from those are due to less than perfect mechanics and not the switch.  (And good quality switches from MicroSwitch company (Honeywell now) do even better.)

The inductive probes seem to be hit or miss.  Some of them are very good and give numbers close to the micro switches.  But other inductive probes still produce usable results but are not as good.

thinkyhead, 2016-05-09:
> I do not like very much when the result of a command varies without explanation

Occasionally the Arduino IDE doesn't notice you edited a file. So it may be necessary to quit and re-open Arduino IDE to compile the changes.

closed 2016-05-09 by thinkyhead

Blue-Marlin, 2016-05-09:
> The inductive probes seem to be hit or miss. Some of them are very good and give numbers close to the micro switches. But other inductive probes still produce usable results but are not as good.

To my experience inductive probes have a very good repeatability, when probing a particular coordinate multiple times. Inductive probes are very picky for the homogeneity of the inductance of the surface.
On a uncovered MK2 my probes can detect where the metal traces and the thermistor is. 
* The bed should have a solid metal cover.
* Don't probe to close to the edges of the bed.
* Don't probe to close to the bolts, clamps, ... .