MarlinFirmware/MarlinDev #336: I need help understanding:   void calculate_delta(float cartesian[3])

Roxy-3DPrintBoard, 2016-01-20:
I don't understand what is going on with 't'.   Why do the calculations for each tower have the extra term (t > 25 ? sqrt(t) : 0.0) added on?  It looks like they are trying to see if the requested position is getting too close to being fully extended (at 90 degrees) from the tower.   If this magical 't' number is less than 25, we don't move the carriage on that tower.  

Why is this?  And what happens if we cross this threshold so we don't add on the sqrt(t) and then move the nozzle back.  Won't we have shifted the position of the carriage on that tower and be goofing up the print????

~~~~
  void calculate_delta(float cartesian[3]) {

    float t = delta_diagonal_rod_2_tower_1
              - sq(delta_tower1_x - cartesian[X_AXIS])
              - sq(delta_tower1_y - cartesian[Y_AXIS]);
    delta[TOWER_1] = cartesian[Z_AXIS] + (t > 25 ? sqrt(t) : 0.0);

    t = delta_diagonal_rod_2_tower_2
              - sq(delta_tower2_x - cartesian[X_AXIS])
              - sq(delta_tower2_y - cartesian[Y_AXIS]);
    delta[TOWER_2] = cartesian[Z_AXIS] + (t > 25 ? sqrt(t) : 0.0);

    t = delta_diagonal_rod_2_tower_3
              - sq(delta_tower3_x - cartesian[X_AXIS])
              - sq(delta_tower3_y - cartesian[Y_AXIS]);
    delta[TOWER_3] = cartesian[Z_AXIS] + (t > 25 ? sqrt(t) : 0.0);
  }
~~~~

Wackerbarth, 2016-01-20:
If you cross the threshold, you will cause real problems with the mechanism getting locked up and possibly driving itself into an imverted position that can then damage the arms.

We need some better soft limits, but we certainly need something to prevent the mechanism from damaging itself.

Roxy-3DPrintBoard, 2016-01-20:
Can you confirm this is to protect against the arm reaching straight out from the tower?  The other possibility is this is trying to protect for the case where the effector gets too close to a tower.

Wackerbarth, 2016-01-20:
If you look at the equation, t is the square of the height of the triangle that is formed by the tower and arm. Therefore a small t implies that the arm is approaching horizontal. I think that that condition exists only when you get away from the towers.

> On Jan 20, 2016, at 2:29 PM, Roxy-3DPrintBoard <notifications@github.com> wrote:
> 
> Can you confirm this is to protect against the arm reaching straight out from the tower? The other possibility is this is trying to protect for the case where the effector gets too close to a tower.

Roxy-3DPrintBoard, 2016-01-20:
Thank You.  That helps me understand this.

Wackerbarth, 2016-01-20:
This also points out why we should not be trying to track the nozzle in object coordinates, but rather frame coordinates. The effect of these software limits in some sense "truncate" the object produced. But, if we are not tracking in object space, than the nozzle will be in a correct position once the designated target is again within range.

Wackerbarth, 2016-01-20:
In any case, if this comes into play, it is because of some configuration error. The entire printable area can be reached without any arm coming that close to horizontal.

KiteLab, 2016-01-21:
I'm quite sure the allowed area is strictly convex. There is no possible line through a forbidden area.
A truncation is not enough. An alert is required.

Wackerbarth, 2016-01-21:
@KiteLab -- I agree that an alarm would be in order if you are "printing". However, what I am saying here is that, because of misconfiguration, it is possible to attempt to drive the arms into an undesirable region. That region can be limited by various software "limits" which, taken together form a convex boundary.

Even if you do not cause an alarm condition, if you replace the movement that would have ended outside the boundary with one that terminates on the boundary, you can "recover" by moving to some allowable point.

The current methodology cannot do this because it "gets lost" when the software limit substitutes a different end point and the nozzle does not reach the expected point in object space.

KiteLab, 2016-01-21:
Without having looked up where these 3 equations exactly apply, is suppose in effect the results are added to the z height. Would it make sense to make the second value sqrt(25) (that could be calculated once) instead of 0, or are the splinted delta moves already small enough to to make no real difference.

What happens to the E coordinate when clipped here? Looks like an over extrusion.

Thinking about it, the __area__ described here is convex, but the build __volume__ is not - because of the 3 spheres around the tower tops.

Wackerbarth, 2016-01-21:
I think that it is unimportant.. The nozzle will be outside the print area before the arm gets that close to horizontal.  The only time that I have seen this become a problem was when the Allen probe configuration was not set up properly and the nozzle was well outside the print area.

Roxy-3DPrintBoard, 2016-01-21:
OK, I know this is going to sound simplistic.   I sort of get what you two are saying.   But here is the way I see it. 

If we detected an error condition, why are we not moving the offending carriage instead of calling kill() ?  The print is already going to be screwed up because we didn't move the carriage where it was supposed to go.  

Shouldn't we just kill the print?

thinkyhead, 2016-04-20:
Can we close this old issue for housekeeping?

closed 2016-04-20 by Roxy-3DPrintBoard