MarlinFirmware/MarlinDev #337: Fan RPM reading for improved safety

Kakao3D, 2016-01-21:
Some all-metal hotends requires active cooling on cold end with a small fan. When this fan malfunctions, they quickly overheat and tend to cause damage to plastic mount which holds them. 

A way to prevent this issue will be to assign one pin for RPM reading, and check the fan RPM every X seconds. When the RPM drops below given value, Marlin will stop heating hotend and give FAN_ERROR or something like that.

If something like this already exist in Marlin can you point me where to find it? If not, I would call it feature request.

a4jp-com, 2016-01-26:
I thought there was some code that says thermal runaway. I guess this is similar to what you want. The printer stops printing if the temperature change is to fast or a bit weird.

a4jp-com, 2016-01-26:
You can also set the maximum heat in the config file.

Kakao3D, 2016-01-29:
Yeah, but in this case, there is no "visible" heat change for the printer to see, because it only monitors temperature on the hot-side of hot-end. 

Let me do an example: With E3D-V6 hotend, the heater cartridge heats up to 240°C, and at this temperature, the cold-end side (the aluminium ribs) have just a little over 70°C in the closest point to the heater cartridge, and this temperature descend gradualy with the distance from the heater cartridge. Which leaves the hotend mount at something just over room temperature. But if the fan stops blowing air through the ribs, they can heat up to temperature over 90°C which will cause them to melt or deform the hotend mount if it is printed from ABS or PLA. But in this situation the heater cartridge temperature still doesn't exceed 240°C. And so, system will not detect any failure or dangerous situation.

thinkyhead, 2016-04-20:
Can we close this old issue for housekeeping?

Kakao3D, 2016-04-20:
If it is not solved yet, I would keep it as open feature request. Since it is safety concerning request which is easy to implement I think someone should look into it. I would write it by myself but I cant code. But I can give a rough sketch what it will propably looks like:

#enable hotend-fan-safety //if enabled the system will keep an eye on RPM of hotend cooling fan, and if by any reasons the fan stops turning, it will shut down hotend heating immediately and return "FAN ERROR". This will prevent some al- metal hotends from overheating on their cold side and melting out from their plastic mount (especialy PLA). PLEASE NOTE: This feature requires 3pin fan connected 

> define: fan_minimal_RPM = 200 //or any pre-defined value
> get RMP reading of fan1
> if fan1_RPM <= fan_minimal_RPM //indicate fan failure
> wait_seconds = 5 //wait fes seconds to see if the fan stop isnt only temporary
> if fan1_RPM <= fan_minimal_RPM //check if the fan failure still last
> then shutdown_hotend_heating 
> return "FAN ERROR"

> else continue monitoring

thinkyhead, 2016-04-22:
Just putting this link here for reference, if someone decides to implement this. http://playground.arduino.cc/Main/ReadingRPM

closed 2016-05-25 by boelle