MarlinFirmware/MarlinDev #340: stepper driver info?

coalman321, 2016-01-25:
i need some info on how the firmware handles the driver step, dir and enable pins while in 2 states.
the first is while the motor is being moved and second while that motor is not moving but others are.
third i would also like to know if it has been moved in this code. 
any help would be great!
Thanks

Roxy-3DPrintBoard, 2016-01-26:
Can we send you an invoice for our time to answer your questions while we improve this software you get for free???

closed 2016-01-26 by Roxy-3DPrintBoard

KiteLab, 2016-01-26:
The ISR in stepper.cpp is actually setting the pins.
What to do is calculated in planner.cpp.

coalman321, 2016-01-26:
I apologize but I am not attempting to copy anything if that's what you think. I am attempting to drive a laser from the 3 pins that control extruder 2 on my board. I'm trying to learn why when I hook the direction pin to it, it turns on even when it is not mentioned in the move.

KiteLab, 2016-01-26:
Search the issues here and/or in MarlinFirmware/Marlin.
We already have had this question - and got a satisfying answer.

KiteLab, 2016-01-26:
https://github.com/MarlinFirmware/Marlin/issues/2809