MarlinFirmware/MarlinDev #342: Allen probe key - kossel mini

Jan9320, 2016-01-29:
As I was told I should look up here for the answer but couldn't find any.

I have a problem with Allen probe key and auto bed levelling on my kossel mini, ramps 1.4 and arduino mega 2560. Printer already worked fine (w/o bed levelling) on older marlin software.

I'm almost sure it's just a little thing I missed thousand times...

So the thing is:
Homing (G28) works fine and M119 shows all endstops triggered.
When I try G30, hotend moves down to the side of bed (almost outside of bed area) and starts moving down. If i press allen probe key then, it keeps moving down, then suddenly decides to go up and on the opposite side (almost outside of bed area) and does the same. Then I get Z probe error on the screen.

When I try G29, hotend tries going up, crashing endstops for awhile then goes down to my coordinates to retract probe and that's it.

I spent hours on this but can't find any promising solutions. I'm not arduino or software kind of guy, I prefer hardware. 

I hope someone could help me or give me some pointers on what to do and get my printer running on marlin v1.1.

Many thanks!

KiteLab, 2016-01-29:
How many is all? Be sure not more then 4. Any unused switch shall not be triggered. So at the home position 3 endeops for the towers, plus the Allen Key can be triggered - not more.
With an Allen Key probe be also sure to have defined ENDSTOPS_ONLY_FOR_HOMING.
Be sure to have a sufficient small DELTA_PRINTABLE_RADIUS.
Be sure all Z_PROBE_ALLEN_KEY_DEPLOY_, Z_PROBE_ALLEN_KEY_STOW_ points are inside DELTA_PRINTABLE_RADIUS.
Be sure Z_RAISE_BEFORE_PROBING is high enough not to touch the bed at 0/0 during a beginning G29.

KiteLab, 2016-01-29:
The first move shoulnt go to  somewhere (almost outside of bed area). It shoud go to:
```
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y 
#define Z_PROBE_ALLEN_KEY_DEPLOY_1_Z 
```
then to:
```
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_X
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_Y
#define Z_PROBE_ALLEN_KEY_DEPLOY_2_Z
```
This move is ought to deploy you probe. If not tune the coordinates.

Then the probing should be done. When this has finished.

2 or 3 Z_PROBE_ALLEN_KEY_STOW moves are done. Tune them to stow your probe.

Jan9320, 2016-01-29:
YEAH! I did it... G29 and G30 works fine, it was exactly what you said... I had to set x_min and y_min endstops to true...
Tomorow after work I try calibrating and fine tunning... Let's hope I won't run into any trouble...
What I understand, I have to set proper probe endswitch to center hotend right? And hardware calibrating as much as posible....
Do I have to be careful anywhere else at this point or that's it?
I will report tomorow how it goes.
Thank you, as I said, was just a little mistake.

closed 2016-02-01 by Jan9320