MarlinFirmware/MarlinDev #349: How to use Z_MIN_PROBE_REPEATABILITY_TEST

htaxil, 2016-02-08:
Hi,

I'm using Marlindev branch Jan20.
I wonder how to use Z_MIN_PROBE_REPEATABILITY_TEST in Configuration.h (and maybe what it does).
Enabled or not, the G29 probe have the same behaviour. I thought that by enabling Z_MIN_PROBE_REPEATABILITY_TEST, the M48 function would be applied on each measurement point (and I was wondering where to set the M48 function parameters)

Otherwise, the M48 command works fine when I run it for a single point from the command line.

Roxy-3DPrintBoard, 2016-02-08:
No...   Turning on the Repeatability Test adds the M48 command to the list of commands that Marlin can process.    It does not affect G29 in anyway.   It is its own separate command who's goal is to report how good or bad your Z-Probe is able to measure the bed.

htaxil, 2016-02-08:
ok, thanks

closed 2016-02-08 by htaxil

Wackerbarth, 2016-02-09:
In the future we might modify some of the "leveling" commands to oversample at designated points and do the least squares fit to the the data.

htaxil, 2016-02-09:
Nice, I've always loved the future ;)