MarlinFirmware/MarlinDev #355: Behavior change: Z value not displayed when '#define DISABLE_Z' set to 'true'

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on September 27, 2015 23:40_

Hi,

This is my first issue report here so I apologize in advance if I do it somehow wrong... 

While using the 'RCBugFix' branch I've noticed the that the Z value on the display is constant "---.--" during print (from SD) and on idle.
It worked fine on the 'Release' branch but not on the 'RC' or 'RCBugFix'.
Going into 'Z axis' in 'Move axis' menu shows the right values.

Hardware:
Simple RAMPS v1.4 EEF (dual extruder) structure w/ G3D display/SD panel (based on gMax 1.5 with changes).

Trying to attach my configuration.h file failed so here is what I've changed:
```cpp
#define MOTHERBOARD BOARD_RAMPS_13_EEF
#define EXTRUDERS 2
#define TEMP_SENSOR_1 1
#define TEMP_RESIDENCY_TIME  5  // (seconds)
#define HEATER_0_MAXTEMP 245
#define HEATER_1_MAXTEMP 245
//#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
//#define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed
const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
#define DISABLE_Z true
#define Y_HOME_DIR  1
#define X_MAX_POS 365
#define Y_MAX_POS 400
#define Z_MAX_POS 250
#define DEFAULT_AXIS_STEPS_PER_UNIT   {200*16/16/2, 200*16/16/2, 200*16/0.0625/25.4, 96}
#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 25}
#define DEFAULT_MAX_ACCELERATION      {900,900,100,1000}
#define DEFAULT_ACCELERATION          900    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  900    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   900    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves
#define PLA_PREHEAT_HOTEND_TEMP 212
#define ABS_PREHEAT_HOTEND_TEMP 235
#define SDSUPPORT // Enable SD Card Support in Hardware Console
#define G3D_PANEL
```
Regards,
Aviv.

_Copied from original issue: MarlinFirmware/Marlin#2661_

Roxy-3DPrintBoard, 2016-02-10:
_From @KiteLab on September 28, 2015 1:7_

You have ether not homed Z or `#define DISABLE_Z true`.
The "z:---" means z position unknown.
When z is disabled it may/will skip to the next full step.

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on September 28, 2015 5:19_

"#define DISABLE_Z true" is defined so the Z motors can rest during layer time (and allow me to tweak layer height manually)... 
It works with exact same configuration with the 'Release' branch. 
After homing, it knows should know it at 0 at it should count from there... 
When the long print currently running will end, I'll try to change just to see if it actually solve the issue but again, it worked like that on 'Release' branch. 
/Aviv

https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-143610556

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on September 28, 2015 8:10_

Sadly we can not detect manual turning the z-axis. If wee could, it would make much sense to deactivate the z.display for that.

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on September 28, 2015 11:21_

I don't expect the software to detect manual intervention but I do expect it to display what it know. 
My Z axis based on ACME rods so even if the motors are powered off between layers, it won't lose the position. 
However, when I do manually intervene, it is mostly when I don't want the software to know about it. 

But lets leave the theoretical debate aside for a second... 
I've changed it to 'false' and indeed it start to show the Z value. 
This should be considered as bug as the behavior was changed from previous versions... 
During all the versions in the last year and an half it worked fine with this value set to 'true' (up to 1.0.x brunch) but since 'RC1' (and RCBugFix) it stopped. 
And going back to the theoretical debate, I don't see a good reason for this change. 

Make sense? 
/Aviv

https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-143668306

Roxy-3DPrintBoard, 2016-02-10:
_From @ntoff on October 3, 2015 4:38_

It's not a bug, they did it on purpose and I feel your pain but it's been there for ages and doesn't look like it's ever going back to the way it was unless you (we) modify it for ourselves. I've basically given up on marlin from here, I just cherry pick bits and pieces and bug fixes that I add into my own version of it after getting sick of all the behaviour changes going back to the whole homing debacle.

Roxy-3DPrintBoard, 2016-02-10:
_From @KiteLab on October 3, 2015 9:51_

How about a compromise?
How about making the coordinates blink (1/2Hz) when position is unknown.
That makes the numbers readable (what ever they may mean then) but clearly indicates "something is wrong here".

Roxy-3DPrintBoard, 2016-02-10:
_From @ntoff on October 3, 2015 10:24_

question that doesn't affect me but if the extruder motors are set to disable the inactive extruders, does marlin then forget where the inactive extruder is at next time it goes to use a second extruder? or is this ridiculous behaviour limited to x y and z?

Roxy-3DPrintBoard, 2016-02-10:
_From @KiteLab on October 3, 2015 12:41_

@ntoff 
I'm deeply impressed about your ability to "cherry pick" parts of the Marlin code you like.
I'm deeply impressed by the number of issues you opened in Marlin.
Even more overwhelming is the number of issues you commented.
A bit less impressive is the number of constructive contributions (PRs) you made - null.
Now an then you show us your profound knollage of the Marlin code.

Noting is forgotten, just not displayed here.
https://github.com/MarlinFirmware/MarlinDev/blob/master/dogm_lcd_implementation.h#L326
https://github.com/MarlinFirmware/MarlinDev/blob/master/ultralcd_implementation_hitachi_HD44780.h#L583

Please search for `axis_known_position[` yourselves. It's not used in conjunction with the extruders.
That would not make much sense because the extruders are not homed, but set to 0 by the slicers. Hopefully the do that when changing the tool.

When a stepper driver receives a disable signal it will unpower the coils of the stepper motor. The motor will skip to the full step it is forced to. The forces are its own magnetic field, mecanical resistances, gravity, filament pressure, ... . If the external forces are bigger than the motors own (unpowerd) holding torque, the motor will not only skip one step but begin to move. When the driver is enabled again it applies the current again. When all is working well to the same currents when disabled. Again the rotor adds the internal and external torques and skips in the forced direction. If the external forces where big enough we now have lost a full step.

So this "ridiculous behaviour" is physics - not Marlin.
And yes, as we have very limited influence on 'law of nature' this also applies to extruders.
Marlin has not forgotten anything - but does not know for sure where the motor is.

Roxy-3DPrintBoard, 2016-02-10:
_From @Wackerbarth on October 3, 2015 13:3_

@KiteLab -- Your excellent description of how the "physics" works when you remove power from a stepper, coupled with the reality that we don't really want to keep powering all of the extruder stepper motors all of the time, and the fact that the external forces on the filament are not likely to cause a significant movement of the filament while the stepper is powered off, suggests the following:

Since we typically perform a "retract" before removing power, could we not ALWAYS make THAT retraction to a full step position? I don't think that the amount of retraction is all that critical, as long as we track the amount accurately so that we can, upon reactivating that extruder, return to the proper position.

If this is a viable approach, then we might make it into a feature that gets included in a future release.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 3, 2015 16:3_

Back to the problem.

Displaying "---" was meant to force the users to home their printers. As `axis_known_position[` is only set to true by a successful homing this seemed to be a good way. Before homing we can only display coordinates relative to the (random) point where the tool was, when the machine was switched on. Notifying the user about that is still a good idea. I'm open for suggestions how to do it. Blinking is not a bad idea - i think.

While testing this patch we observed several times a "---" display after homing.  During longer waits for heating up the bed, or long lasting G29 the mechanism was triggered. Investigating this we discovered that the "shut down the steppers when the printer is unused" mechanism did not respect the DISABLE_? settings - and changed that. Sadly this prevented the  "shut down the steppers when the printer is unused" mechanism to work at all, because no one wants DISABLE_? = true. Increasing  the default DEFAULT_STEPPER_DEACTIVE_TIME could have been more efficient and or refreshing `stepper_inactive_time` during the long lasting commands would also have made the trick

Why we don't want DISABLE_? = true was explained again by @KiteLab.
The reason why we do have DISABLE_? at all, is to prevent the steppers/drivers from overheating when they have to deal with high current. Then a little break for cooling down may be preferable over the risk loosing one full step. 

What has the AUTO_BED_LEVELING_FEATURE to do with this complex?
 When `DISABLE_Z = true` z is disabled when not used and no planed moves for z are in the buffer. As with ABL z is involved in every move, the buffer is never empty, z not disabled anymore.

Extruder: 
For a geared wade one full step is below 0.03 mm/full step, that feels pretty irrelevant and will probably disappear in the backlash of the gears. The other extreme is a direct driven MK8-gear with about 0.1 mm/full step, that feels pretty relevant. That also means that the motor on a wade needs considerable less torque (and current), to transport the filament than on the MK8. 
On the other side i do not expect any relevant outer moments on the axis of a retracted extruder stepper motor. Here it's very likely we'll not loose or win a step.
My conclusion is: 
Don't care about switching off the stepper on a wade - the effect will be low if you do it or not.
To be save mount a fan on your MK8 and don't switch it off.

@Wackerbarth 
I doubt it is worth the effort. I see a counter per motor in the stepper-ISR, divisions to determine the microsteps, variables for them,  further #defines for the microstep settings, a logic to drive the stepper to a full step when disabling and even worse a similar logic before enabling the stepper again.

Roxy-3DPrintBoard, 2016-02-10:
_From @Wackerbarth on October 3, 2015 16:31_

@AnHardt -- Whether, or not, it is worth the effort is a somewhat separate issue.
However, I think that the concept can be applied in a much cleaner manner if you look at motion in the frame(actuator/stepper) domain rather than just the object domain.

In the stepper domain, it is very easy to determine the nearest full step (an integer round, or more likely a truncate because you want to retract a little extra rather than too little).

So, at the stepper level, rather than trying to do relative movements, do them in absolute form.

printing away .... 
``` Extrude to <eg> 10.3```
When we retract with an instruction ```Back off about 2```, the planner might determine that the actual target needs to be 8.25 and actually move back 2.05.
Later we resume with
``` Extrude to 10.4 ```
and let the planner determine that it needs to move forward 2.15 because it keeps track of the absolute stepper position after each operation.

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 3, 2015 16:50_

For the record, in my specific case, each Z step is less then 0.008mm.
In addition to that, statistically, sometimes we loose partial step and sometimes we gain partial step so in the entire print which have at least few hundreds of layers, we should be even. 
So in my opinion, not displaying the Z value entirely just because the software might be wrong by 0.008mm is ridiculous. 
Even more, the display have 2 digits accuracy so this tiny shift is not displayed anyway. 

Not showing Z value (or any value) before homing is more than acceptable. 
But not showing value because we asked the motor to rest during its idle time is not acceptable. 

Ps. I have prints with layers that can take up to few minutes each, so I don't see a reason to power up two Z motors for all that time just for the friction of a second of work... its more than 99.9% of power just to keep accuracy of 0.008mm which statistically will be evened anyway. 
/Aviv

What has the AUTO_BED_LEVELING_FEATURE to with this complex?

https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-145260993

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 3, 2015 20:20_

Sometimes i'd love to fork a issue-thread.

@Wackerbarth 
I have no confidence in keeping track of the steps in 'float'. 
Probably we got one's wires crossed again. 

@Aviv22 
What's you point?
In the very common case of a M5 spindel we have exact 0.004 mm/full step - nothing bad will happen if you switch of the motor current.
In the less common case of a delta with a direct driven paste extruder the effector will rush down if you disable the steppers.
Your statistics is wrong. We do not disable the stepper during a move. Ether we have system with low to no external forces like the z-axis of a prusa i3, or a retracted extruder - then we likely will not have lost or added steps at all. Or we have a system where we have external moments, like a delta with a heavy effector, or a  extruder under pressure; here the force has allays the same direction and it's likely we loose/win a step with every disable/enable cycle in the same direction.

Obviously i have not be clear enough when i admitted that blanking the coordinate display for disabled steppers was a surprising side effect. I did not try to defend blanking for disabling, but tried to explain what is going on and what went wrong.

What i need now is not further complaining - it's wrong - it's wrong - it's an issue.
What i need is some suggestion for the fixes.

issue a:
"Remember the user to home"
solutions: 
1.) Do nothing until ether homed or 'unlocked' by a special command or button (GRBL)
2.) Show blinking relative coordinates.
3.) Show a '+/-' symbol near the unhomed coordinate. (Here i would like to know where exactly)
4.) A message in the statusline you cant get rid off - except by homing.
5.) ... (Your idea)

issue b:
"Make the 'shut down the steppers when the printer is unused'-feature work again - but without the danger of a sliding down tool."
partial solutions:
1.) Make this independent off DISABLE_? again.
2.) Make a new set of defines especially for this feature - especially to not shut down z.
3.) Enlarge the default DEFAULT_STEPPER_DEACTIVE_TIME.
4.) Try to find ways to reset the `stepper_inactive_time` during long commands.
5.) All of them.
6.) ... (Your idea)

issue c:
"`DISABLE_? = true` is doing what it is supposed to do - but i don't like it"
solution:
1.) Turn it off and live with the consequences.
2.) ... (Your idea)

issue d:
"`DISABLE_? = false` is doing what it is supposed to do - but i don't like it"
solution:
1.) Turn it on and live with the consequences.
2.) ... (Your idea)

issue e:
"DISABLE_Z = true is blanking the coordinate display"
solution:
1.) Use different variables for 'axis_is_homed[]' and 'axis_known_position[]'
2.) Do we need `axis_known_position` for anything else? Investigate.
3.) ... (Your idea)

issue f:
"Users open n issues for the same problem"
solution:
1.) Users will search the issue tracker before sending a new issue.
2.) Users sending duplicates will be ignored.
3.) Duplicate issues will be closed silently.
4.) Supporters will leave the ship.
5.) ... (Your idea)

issue g:
"Users post issues without any hint about their machine, branch, config, ..."
1.) A traveling agent will paint the users ears with red lipstick.
2.) All the users harddisks are mirrored to a public server.
3.) All Ebay, Amazon, Ali, Bangood, and d3-printer-shop's orders are automagicial published to the world.
4.) All supporters will be provided with the worlds best available crystal balls.
5.) The users promise to give at least some useful hints.
6.) ... (Your idea)

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 3, 2015 21:27_

My opinion(s):
A. 3 or 2. 
B. Give the user the option to choose if to power off the tool/axis when idle or not... and he will leave with the results... 
C. 1 or 2 (you can also mark it on display with ~ sign before the value so user will know the value could be not accurate. 
D. Same as now. 
E. See A, B & C. 
F. 1 + 5 (issue will be closed and marked as duplicate). 
G. 5.

/Aviv

https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-145285845

Roxy-3DPrintBoard, 2016-02-10:
_From @clefranc on October 3, 2015 21:52_

A. 4.) When printer start, show `PLEASE USE HOME (G28)` instead of XYZ or 5.) Show question mark near unknown axis
B. Give the user the option to choose if to power off the tool/axis when idle or not... and he will leave with the results...
C. 1 or 2 (you can also mark it on display with ~ sign before the value so user will know the value could be not accurate. **Replace actual with ~000.0**)
D. Same as now.
E. See A, B & C.
F. 5 (issue will be closed and marked as duplicate **with link to duplicate**).
G. 5. and 6.) Collaborators will be polite and respectful

Roxy-3DPrintBoard, 2016-02-10:
_From @ntoff on October 4, 2015 1:54_

@KiteLab 

>A bit less impressive is the number of constructive contributions (PRs) you made - null.

Because it's easier to just make the firmware work for me how I want it than to fight all of you on simple issues. During the homing debacle it took the developer of (I think) octoprint? coming in here to say no this isn't how it should be, for the devs to get their heads out of the sand and change it back.

Lowly useless people like me have no power here so I keep all my changes to myself, as I also do with all the other things I've changed like pronterface as submitting pull requests would just result in them being denied. *However* those changes ARE on my github repo for anyone who does want them. 

Plus I am not a coder in any way shape or form so any code I do change is most likely not the most efficient way of doing it but it still works for me.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 5, 2015 23:18_

For now quite untested, but the direction i want to go.
https://github.com/AnHardt/Marlin/tree/fix2661

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 6, 2015 9:41_

To get an impression of the optics.

GLCD: All axes unhomed
https://www.youtube.com/watch?v=sO5kL7aXAI0

GLCD: Z with reduced accuracy
https://www.youtube.com/watch?v=MfxxqrRf-0c

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 6, 2015 13:35_

Looks reasonable enough...

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 6, 2015 15:32_

Because the '~' shows on the japanese 2004's as a '->' (arrow right) i switched to a ' ' (space). Looks even better.
GLCD:
https://www.youtube.com/watch?v=S4g01DHSlkc
CLCD:
https://www.youtube.com/watch?v=HIeICg93WBE
https://www.youtube.com/watch?v=z54FMYMwSD0

Roxy-3DPrintBoard, 2016-02-10:
_From @clefranc on October 6, 2015 15:55_

I like that, good work!

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 6, 2015 16:49_

Although space doesn't mean approximate like tilde sign does, it still reasonable enough and in some way even look cleaner... 
Good work and thanks.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 9, 2015 20:4_

@Aviv22  @ntoff  @clefranc 
Has anyone tested #2676 or #2686 ?
Can anyone confirm that the stepper releases now work as expected?
No feedback - no merge.
This blocks 2 other fixes.

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 9, 2015 20:18_

My plan for this weeked... will check it out in the next 24 hours...

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 10, 2015 7:8_

Tested fix2661 from your repo and it looks ok to me.
And thanks for adding the option the disable the accuracy warning... :)
However, I think the accuracy warning should be enabled by default so when someone "disable steppers" from the menu, he'll see the warning.
Thanks.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 10, 2015 7:52_

@Aviv22 
Thanks for your report.

Enabling any option by default is problematic - it needs much more effort to overwrite a already defined option. 

When enabling DISABLE_X in the config, you have right above a warning about the possible consequences and right below the option to be warned - that should be enough.

Roxy-3DPrintBoard, 2016-02-10:
_From @Aviv22 on October 10, 2015 8:13_

The reason I think the warning should be on by default is because user can be in an not-accurate situation even without disabling axis on idle... 
It can happen in 2 conditions:
1. User click disable steppers on menu. 
2. Axis is idle for more than 60 sec.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 10, 2015 10:43_

You are right  with 1. and 2.
I still think enabling the hint-option is not the best idea. Because, i suppose,  we will get massive resistance from @Wackerbarth otherwise.

At the moment we have the stepper DISABLE configuration in two files. On the long term it could make sense to unify the location. 

Edited

Roxy-3DPrintBoard, 2016-02-10:
_From @Wackerbarth on October 10, 2015 13:56_

@AnHardt -- "Enabling any option by default is problematic"  Yes, I agree. However, the solution is to create a different option to replace it. If you want the default action to be "ACTION", then create a "DISABLE_ACTION" option. So here, the option might be "DISABLE_REDUCED ACCURACY_WARNING"

Roxy-3DPrintBoard, 2016-02-10:
_From @Wackerbarth on October 10, 2015 14:14_

@AnHardt -- I'm still waiting for a proposal concerning the DOCUMENTATION. We have, and always will have, multiple versions of the firmware. Even if we have "moved on" to Marlin 3.17, the documentation for 1.1 should still be available SOMEWHERE that a user with that "old" version can utilize. (Perhaps if only to figure out what was meaning of an old configuration that they want to update.)

So, how do you propose that we maintain the various sets of documentation in a manner that a user can find the set that applies to his version of the firmware? Not that I am advocating doing so, one way is to make the documentation a collection of files in a Documentation folder that is part of the branch of the firmware to which it applies. I recognize that there are others who advocate using a wiki to make it easier for them to contribute, but, by itself, that does not address the issue of multiple versions customized to the particular branch to which the documentation applies..

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on October 12, 2015 23:29_

Providing information in a documentation folder inside the repository/branch was not so bad. For me, as a beginning user, it worked very well. (But i don't need much initial information to find the details in the code, the issues list, or the descriptions of the PR's. The ability to search seems not to be very common nowadays) Connecting/structuring information is very hard there. Folders are almost the single way to provide a structure.

For many things it is sufficient to add the new/changed behavior v.e. new parameters to a g-code in a section titled "from version xxx on" or "until version" in a wiki-article.

__Features/Concepts__ should be described in their own articles, ether in a wiki or the documentation folder in the branch they are added. When it's time to release this branch it's time to integrate the info to the wiki. In the configurations, at the top of a section (Endstops, ABL, MBL, Disabling Motors, ...) a link to the concept would be nice.

The primary source for __g-codes__  should be the description in Marlin_main.h. A short header in the top-directory. In detail where the implementation is. Until the 'release'  ideally everything should be findeble in the wiki. Again with a hint "from on" or something like that.

For __configuration options__ i'd expect a sufficient description in the configurations. A description in the wiki is optional. If a description is needing very much room a link to wiki may be an option. For old/replaced/renamed options the wiki is a good place. All what it needs is a 'renamed to that[link] at version x'

A __Media-Wiki__ would be great (when writable). The possibility to add articles to a category by a simple tag is invaluable. The information "tree" builds then much faster and better.

For sure you noticed, that i'm not the best qualified person here around, for writing doku. Don't expect more than the absolutely necessary from me. Working on the issue list is totally different from writing doku. If a can actually help, my bad language may be excused, but my sometimes 'individual' spelling and screwed up  sentences will seriously damage a honest documentation.

Roxy-3DPrintBoard, 2016-02-10:
_From @Wackerbarth on October 13, 2015 1:25_

I really find the intermixing of descriptions to be very confusing for users. It is much more readily understood if you have a document for each version that gives the proper information for that version without the reader having to try to sort out which part applies to the version that they are using.

When upgrading, a description of the inter-version changes is helpful, but otherwise, the user generally doesn't care "how it was"

I totally disagree with the idea that the primary USER documentation should be anywhere in the code itself (unless you are using a documentation processor that uses the source files as input and extracts a readable document from them.)  Far too many users have trouble even finding a file, much less possessing the ability to extract information from a dual purpose file. I think we need a collection of user documentation that can be read like a book or website which provides a comprehensive description of both how and why particular settings should be applied.

I also think that this documentation SHOULD NOT be in the configuration file. The present configuration file is already unreadable and we need to have even more explanatory material for each option.
Therefore, I suggest that there be a configuration manual (I'm not specifying how, or here, it is stored) that is organized in logical sections, each of which is of a manageable size, explaining each aspect of the configuration in detail.

Note that I recognize that this is a significant task, but it is one that I think we need as a long term goal.

Take a look at the documentation for a major project such as django. They have addressed these issues in their documentation. They support multiple versions of the code and the user gets a separate view of the documentation for each version.

Roxy-3DPrintBoard, 2016-02-10:
_From @AnHardt on November 4, 2015 13:34_

Activated the warning about possible reduced accuracy by default
The options name is now  `DISABLE_REDUCED_ACCURACY_WARNING`
(https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-147090799)

https://github.com/MarlinFirmware/Marlin/pull/2686

Roxy-3DPrintBoard, 2016-02-10:
_From @thinkyhead on February 10, 2016 5:50_

Are current solutions for this behavior sufficient to make everyone happy?

It is certainly possible to show something like `X  0?  Y  0?  Z123.45?` or `X100  Y120  ?123.45` when coordinates are actually unknown, instead of hiding the coordinate altogether. Hiding the coordinates makes the most sense before `G28` since the printer won't even move in the direction away from endstops at that point – but even then, showing what the printer "believes" to be its position can be useful to have on-screen, so long as it's clear that the value is actually "unknown". As for those times when the stepper is disabled and loses its position, using a "?" to indicate that the coordinate is questionable is actually much nicer to the user.

I propose using two flags: Keep using `axis_known_position` to indicate that an axis has been homed and its motor has not been disabled. But add another flag `axis_was_homed` which indicates only that `G28` has been performed for the axis. This flag will be used to determine whether to show coordinates _at all_, while the `axis_known_position` flag will determine whether to show it as "uncertain."

Roxy-3DPrintBoard, 2016-02-10:
> As for those times when the stepper is disabled and loses its position, using a "?" to indicate that the coordinate is questionable is actually much nicer to the user.

I think I like this idea.   

> I propose using two flags: Keep using axis_known_position to indicate that an axis has been homed and its motor has not been disabled. But add another flag axis_was_homed which indicates only that [...]

I wonder if this would be better done with just one flag that can have multiple different values.  I don't really know, but it might be better to keep all of the information in one place.

I'm trying to figure out how to move this thread over to MarlinFirmware/MarlinDev but so far I have not figured out how to do that.   The reason to move this thread is that we really should continue this discussion.  But the changes and the testing required to validate them are too extensive to be made to a Release Candidate.  (Our LCD Code is already too slow in some setups.)  

Is it even possible to move a thread?  Does anybody know?

AnHardt, 2016-02-10:
Sadly the the links from the numerous issues are not pointing to here anymore.

Roxy-3DPrintBoard, 2016-02-10:
OK...   Let's just close this one and re-open the original thread.  I guess we don't have the technology to move issues back and forth right now.

closed 2016-02-10 by Roxy-3DPrintBoard