MarlinFirmware/MarlinDev #362: FeatureRequest - Automatic Mesh Leveling

mirage335, 2016-02-15:
Please consider implementing automatic mesh leveling. Current probe-based automatic leveling support does not correct well for bumps, Y-axis twisting, X-axis sagging, or other nonlinear deformities.

Roxy-3DPrintBoard, 2016-02-15:
I'm working on it for Delta's.   If I'm careful, I might be able to make it work for Cartesian and Delta machines.

mirage335, 2016-02-16:
Much thanks. My request is for Cartesian machines, like the LulzBot Taz, Mini, etc.

adamfilip, 2016-02-28:
Yes please. im willing to donate some money if it will help get it added sooner :)

Roxy-3DPrintBoard, 2016-03-02:
Donating test time would be more helpful.   I'll let you know when there is something 'feature rich' enough and stable enough to test.

Sniffle, 2016-03-02:
I would roxy but my delta is running smoothie :-D

adamfilip, 2016-03-02:
Im ready and willing to do some testing.

mirage335, 2016-03-02:
Also ready and willing to test automatic mesh leveling.

Roxy-3DPrintBoard, 2016-03-03:
Here is the header block in front of the code I'm writing.   You should be able to understand a lot of what I'm thinking from the comment block.    Probably, you can understand why it is going to take 5 or 6 weeks to get the system fully baked and ready to hand out for people to try.

Are there any other feature requests or omissions that need to be handled?

 ~~~~
/**
   * G29: Delta Mesh Based Compensation System
   *
   * Parameters understood by this leveling system:
   *
   *  A		Activate  Activate the specified Mesh.   All loads and stores of Mesh Compensation data
   *  			  will occur at the Activated location in EEPROM.  A separate Mesh definition is 
   *  			  needed for each piece of glass you will printing on.  These pieces of glass need to
   *  			  be positioned the same during printing as during the Mesh Calibration (and
   *  			  storage) process.  Activation of a specified set of Mesh Data happens before
   *  			  any subsequent processing of the command line.   Activation of a specified set
   *  			  of Mesh Data does not implicitly Store or Load any Mesh any data to the EEPROM.  
   *  			  It also does not affect the default Mesh Data specified by the EEPROM.  It only
   *  			  affects the current session and an M500 is required to update the contents of
   *  			  the EEPROM.   The GCode command:  G29 A 3 S   performs the same as  G29 S A 3 
   *
   *  F		Fade	  Fade the amount of Mesh Based Compensation over a specified height.  At the
   *  			  specified height, no correction is applied and natural printer kenimatics
   *  			  take over.
   *
   *  L		Load	  Load the currently Active Mesh from Active EEPROM location.
   *
   *  M		Map	  Display the Mesh Map.
   *
   *  P0      	Phase 0   Zero Mesh Data and turn off the Mesh Compensation System.  This reverts the
   *                      Delta Printer to the same state it was in before the Delta Mesh Based Compensation
   *                      System was turned on.
   *
   *  P1	Phase 1	  Perform Phase 0 and continue with automatic generation of the Mesh data using
   *  			  the Z-Probe.   Depending upon the values of DELTA_PROBEABLE_RADIUS and 
   *  			  DELTA_PRINTABLE_RADIUS some area of the bed will not have Mesh Data automatically 
   *  			  generated.  This will be handled in Phase 2.  The Delta Mesh Compensation System is 
   *  			  turned on at the conclusion of Phase 1.
   *
   *  P2	Phase 2   Probe areas of the Mesh that can be automatically handled.
   *
   *  P3	Phase 3   Manually fill in the unprobed areas of the Mesh. 
   *
   *  P4	Phase 4   Fine tune the Mesh.  The Delta Mesh Compensation System assume the existance of
   *  			  an LCD Panel.  It is possible to fine tune the mesh without the use of an LCD Panel.
   *  			  The System will search for the closest Mesh Point to the nozzle.  It will move the
   *  			  nozzle to this location.  The user can use the LCD Panel to carefully adjust the nozzle
   *                      so it is just barely touching the bed.  When the user clicks the control, the System
   *                      will lock in that height for that point in the Mesh Compensation System.
   *
   *                      Phase 4 has several additional paramters that the user may find useful.  Phase 4
   *                      can be started at a specific location by specifying an X and Y parameter.  Phase 4
   *                      can be requested to continue the adjustment of Mesh Points by using the R(epeat) 
   *                      parameter.  If the Repetition count is not specified, it is assumed the user wishes
   *                      to adjust the entire matrix.  The nozzle is raised after a Mesh Point is calibrated
   *                      and before the nozzle is moved to the next location.  This height can be controled
   *                      with the H parameter.  The default value is 5mm.
   *
   *                      Phase 4's general form is G29 P4 [R # of points] [X position] [Y position] [H height]
   *
   *  P5	Phase 5	  Undefined right now.
   *
   *  S		Store	  Store the current Mesh in the Activated area of the EEPROM.
   *
   *  T		Test      Load specified Test Pattern to assist in checking correct operation of system.
   *
   *  Z		Zero      Probes to set the Z Height of the nozzle.  With no specified paramter 
   *  			  Z_PROBE_OFFSET_FROM_EXTRUDER is used without any extra correction.  If a parameter
   *  			  is specified, it is added to Z_PROBE_OFFSET_FROM_EXTRUDER for the final zero position.
   *
   */
   ~~~~

mirage335, 2016-03-03:
Thanks for working on this. Does indeed seem like quite an extensive codebase to develop.

Is the idea to store multiple meshes in EEPROM, to be recalled for use with different print surfaces? Personally, I still might want to regenerate the mesh before every job. In particular, it is becoming common practice to laminate glass with PEI or other materials, and the adhesive layer may change thickness between jobs.

Fade rate might need to change depending on the job or tools used. Exceptionally precise jobs might benefit from little or no fading.

Roxy-3DPrintBoard, 2016-03-03:
> Is the idea to store multiple meshes in EEPROM, to be recalled for use with different print surfaces? 

Yes.  There are some constraints however.  I want to be able to do up to a 15 x 15 Mesh.  So, that is a lot of data.   Depending upon the resolution (and how well I can compact the data), higher resolution Mesh's might limit how many of them can be stored in EEPROM. 

In reality, I mostly use one piece of glass.  But some times while one is in the freezer to help me break the parts free, I'll use another one to get a 2nd print started.   So for me, storing two Mesh's would be fine.

> Personally, I still might want to regenerate the mesh before every job. In particular, it is becoming common practice to laminate glass with PEI or other materials, and the adhesive layer may change thickness between jobs.

Well...  More power to you!  :)  But even with a fancy system that helps you dial in the Mesh as fast as possible (like I'm trying to do), I suspect it will take 20 or 30 minutes to get the Mesh perfect.  I kind of doubt you will want to do that for every print.

I'm also thinking of doing some other items as a follow up.  For example, suppose you get a Mesh perfectly defined.  But when you put the piece of glass down on the heat bed there is a small piece of dirt that raises an edge or corner.  It should be possible to do a 3-Point leveling on the Mesh to correct for that also.   I'm thinking we get the High Resolution Mesh Leveling out there and see what kind of things it has problems handling.  And then we will go after those next.

> Fade rate might need to change depending on the job or tools used. Exceptionally precise jobs might benefit from little or no fading.

Yes.  I can see how that could be the case.   Probably the thing to do is get it working and get some experience with the concept.  And once people get comfortable (and knowledgeable)  about it's use, we will have lots of ideas to deal with the limitations of the first implementation.

mirage335, 2016-03-03:
Well, 30 minutes is a long time. Probing a 5x5 grid takes tens of seconds, and happens before the hotend finishes heating up. Is manual intervention expected to be necessary for routine use?

Roxy-3DPrintBoard, 2016-03-03:
> Well, 30 minutes is a long time. Probing a 5x5 grid takes tens of seconds, and happens before the hotend finishes heating up. Is manual intervention expected to be necessary for routine use?

Maybe fine tuning won't be necessary.   I don't know yet because the code isn't done and working.  It is very possible that if you are content printing within the area the probe can measure things, no fine tuning will be necessary.   And if your bed is fairly flat, a 5 x 5 grid just might work fine.

I'm mostly doing this because it is an interesting exercise and I want it to handle a very poorly calibrated Delta.    Do you remember the original Auto Bed Leveling videos that showed up when the feature first came out?   Some people were printing on a bed raised a couple of inches on one side.   That was enough to get me to print some parts and order a Servo.

I want the same thing to happen except with a poorly calibrated Delta Printer.   If DELTA_RADIUS and DELTA_DIAGONAL_ROD are off by .5mm the movement becomes horribly un-flat.  I want the system to handle a huge error like that.   (That is kind of like saying I want to handle as big of an error as the original Auto Bed Leveling videos showed.)

So...   I'm way over designing the system just to make sure I can get there.  If your printer is pretty normal and well calibrated, it is very possible you don't need anything but the Phase 1 stuff if you don't want to print outside of the probed area.   If you want to print outside of the probed area, you have no choice but to manually fill in the rest of the Mesh.  But that is the whole purpose of Phase 2.  It's purpose is to let you very quickly fill in the rest of the grid manually.  It will move the nozzle to each location of the grid it needs data for and let you quickly dial the nozzle down to the bed and then it will continue to the next unfilled location of the Mesh.

mirage335, 2016-03-03:
Makes sense. Leveling a delta machine can require some interesting math to begin with, handing extreme errors like that could take a lot more precise measurements.

Filling in areas outside the mesh... would expect some squares outside of the mesh could be estimated based on nearby squares, at least for cartesian machines. Likely, the two or three points needed to define an adjacent square could be estimated rather well simply by drawing a stright line through two other points that have been probed. At least for cartesian machines with flat glass surfaces.

When this is operational, it is also possible cartesian machines will again become more competitive with delta machines, as the demand for mechancial performance is lessened.

samwyze, 2016-03-04:
I'm also very interested in this feature, though for Cartesian kinematics with a servo mounted z-min endstop probe. I have a heap of messy glass plates that are far from flat, so am willing to help with testing when the time comes.

adamfilip, 2016-04-18:
how are things progressing?

psyvision, 2016-04-18:
Willing to donate some testing time for this on a Cartesian printer with IR z-probe.

Roxy-3DPrintBoard, 2016-04-19:
> how are things progressing?

It's coming along.   And in fact, it may be possible to incorporate (unify) the 3-point and grid leveling into it.   But I keep getting sucked into my brother's software projects and that is a big time drain.

Roo123, 2016-04-19:
assign your bro a lower interrupt priority ! ;-)

Roxy-3DPrintBoard, 2016-04-20:
Yeah, that would be nice.   The problem is he seems to be a NMI.

adamfilip, 2016-04-28:
Hope to see Auto Mesh Leveling in RC7 :)

Roxy-3DPrintBoard, 2016-04-29:
> Hope to see Auto Mesh Leveling in RC7 :)

No you don't!   You want RC7 to be stable and well shook out!   But it will be available as a fork of the last Release Candidate when it goes Golden.  So if that happens, it won't be a big deal to bring up if you have already brought up the RC family.

gddeen, 2016-04-29:
My Wishes:

I've always wanted to simply turn a dial to raise the printhead up or down VERY small increments at the beginning of a print. (Cartesian printer). The idea being to adjust the first adhesion layer.  Different
materials need different heights to make them stick to the bed. Also at different temperatures they have different viscosities which changes the bead on the first layer.

MID-PRINT is would be nice to say up, up, up, up, there...  Even if the print had to pause, raise 10mm, recalculate everything, lower then resume.

As faster boards get used, it would be nice to have a PC and do any type of translation to the mesh bed level (While it is printing...)

gddeen, 2016-04-29:
Are there any specific test platforms I should build to work out MABL? I might be able to construct a bunch of tests that I can run and rerun to validate the features. (A System test.)  Are you using something
like a mega2560 or a new 32bit board?

I can also unit test cartesian if I get much more info...

Blue-Marlin, 2016-04-29:
@gddeen 
> MID-PRINT is would be nice to say up, up, up, up, there... 

Did you try `BABYSTEPPING`? Ah - sorry - not for DELTA.

adamfilip, 2016-05-18:
Is there some code I can test out, insert in RC6?
to get Auto Mesh levelling working?

thinkyhead, 2016-05-19:
@adamfilip Word on the street is that @Roxy-3DPrintBoard is working on that — making it so that we can have Automatic Mesh Leveling, and also Manual Grid Leveling. But I don't know how far along she is.

adamfilip, 2016-05-19:
I noticed that the new Reprap i3 Mk2 has automatic Mesh leveling listed as a feature
http://shop.prusa3d.com/en/3d-printers/53-original-prusa-i3-mk2-3d-printer.html

I wonder what code they are using to accomplish it.

Blue-Marlin, 2016-05-19:
At least in they public repository there is nor the slightest hint on MBL (https://github.com/prusa3d/Prusa-Firmware)

gddeen, 2016-05-20:
Adam, Mesh Bed leveling works great in RC6... (I did add a couple of code changes but I don't
think they were needed by others.)

You can do:
M502 ; reload defaults (0 mesh)
G29 S1 ; create a mesh by lowering head to the bed manually
G29 S2 ; next point till done.
M500 ; Write these values for default...

G28's are tricky... It depends where your probe is and your XY coords, etc...
Cura sticks them in my GCODE. So I always add "M420 S1" to turn mesh bed leveling
back on after ANY G28 in start.code or start2.gcode.

G29 S4 Z0.3 
G28
M420 S1 
M500 ; will raise the nozzle (lower bed) up a hair and save it for later...

Adding this to octoprint controls makes it slightly easier... I haven't tested non-absolute coordinates
like G91:G29 S4 Z+0.3 to see if THAT works. If it did you could make up/down buttons.

gddeen, 2016-05-20:
@Blue-Marlin you mentioned baby stepping... is there a LCD menu item I can select while printing
to just micro-step the Z axis up or down completely ignoring updating current_position??? If I am
printing, I usually do a G1 Z0.5 to go up a bit (on the first 0.3 layer) from the terminal window in
Octoprint. That's a pain to always lug around a laptop to the printer... Then I wait until the next layer
comes around to see how much fudging that needs to fix the missing space. (I usually increase the flow a bit) Then wait for that next layer and set the flow back to normal. That's 3 layers and always the longest layers of a print.

Or is baby-stepping something else?

Blue-Marlin, 2016-05-20:
https://www.youtube.com/watch?feature=player_embedded&v=LE15jh_Nr_A

closed 2016-05-25 by boelle