MarlinFirmware/MarlinDev #365: Support for platformiO in latest versions

thawkins, 2016-02-19:
PlatformiO have a released their IDE http://platformio.org/#!/platformio-ide which makes the platformiO system a far more capable and useable system than the existing arduino setup. Whilst maintaining the arduino framework compatability up to version 1.6.7. 

Can we revisit the lack of platformio support in the new repository layouts.

Roxy-3DPrintBoard, 2016-02-20:
Will this load and compile the current Release Candidate?     And the follow up question, can it load can compile the code in MarlinDev that has started to be split apart?

thawkins, 2016-02-20:
I'm having a go at getting that in place right now.

Roxy-3DPrintBoard, 2016-02-20:
One thing interesting in the documentation pages was you get to set compile time flags for everything.   I want to be able to declare --ii flag to display the generated code for each line of C code.   That would have been helpful and interesting when looking at the Rx and Tx buffer indexes.

thawkins, 2016-02-20:
It's a more traditional system, but also a one stop install for IDE,
compilers, libs, frameworks etc. So in that respect it's like the arduino
setup. The upside is it supports a huge range of boards and processor
types,  and will automatically download the compilers etc, plus it does
real libraries, rather than the pseudo libs that arduino does. Which are
little more than source code archives.

closed 2016-05-25 by boelle