MarlinFirmware/MarlinDev #376: Question about Arduino and the EEPROM space

Roxy-3DPrintBoard, 2016-03-03:
I'm hoping this is an easy question and I don't have to go create an account over at the Arduino forums and ask what I'm sure is a stupid question:

Q: Is there a #define'd value that tells me how big the EEPROM is on a given AVR chip?

Bonus question:

Q: Is there a #define'd value that tells me the start and end address (or index) of the EEPROM on a given AVR chip?

CONSULitAS, 2016-03-03:
@Roxy-3DPrintBoard 
Hi Roxy,

perhaps this helps:

http://playground.arduino.cc/Code/EEPROMex (see at the end)
moved: https://github.com/thijse/Arduino-EEPROMEx

http://forum.arduino.cc/index.php?topic=56336.0 (see #1 und #3)

Roxy-3DPrintBoard, 2016-03-03:
@CONSULitAS   

THANKS!!!!  Somebody else needed the same information...   At http://www.nongnu.org/avr-libc/user-manual/group__avr__io.html are #define values for the details about all the memory types.

Finally, the following macros are defined:

RAMEND      The last on-chip RAM address. 
XRAMEND    The last possible RAM location that is addressable. This is equal to RAMEND for devices
                that do not allow for external RAM. For devices that allow external RAM, this will be 
                larger than RAMEND. 
E2END         The last EEPROM address. 
FLASHEND   The last byte address in the Flash program space. 
SPM_PAGESIZE      For devices with bootloader support, the flash pagesize (in bytes) to be 
                              used for the SPM instruction.
E2PAGESIZE         The size of the EEPROM page.

Because it is a Harvard Architecture, all of the addresses start at 0x0000.  (I haven't found that said anywhere.  But because they are not saying it, or defining values for the start, I suspect that is true.)

closed 2016-05-25 by boelle