MarlinFirmware/MarlinDev #379: State machine and events

jbrazio, 2016-03-10:
--- Copy/paste from MarlinFirmware/Marlin#3118 ---

A 3D printer is in it's core a state machine:
- Or is idle,
- Or is in the planner doing a move,
- Or is heating the nozzle,
- Or is doing two, three, four things at the same time but all of them are finite.

Currently we have no exact way to know at a specific point in time what is the state of our printer unless we do a over complicated spaghetti check. I found PR #3109 from @thinkyhead where he introduces states brilliant but it has one key aspect missing, our state machine can be doing more than on thing at a time. I would propose the introduction of a global (32 bit ?) "register", we can then specify a bit per state like so [this is not by any means a final list, just food for thought]:

```
#define READY           0x01        // 0000 0000 0000 0000 0000 0000 0000 0001
#define BUSY            0x02        // 0000 0000 0000 0000 0000 0000 0000 0010
#define FAULT           0x04        // 0000 0000 0000 0000 0000 0000 0000 0100
#define HOMED           0x08        // 0000 0000 0000 0000 0000 0000 0000 1000
#define PRINTING        0x20        // 0000 0000 0000 0000 0000 0000 0001 0000
#define WAITING         0x20        // 0000 0000 0000 0000 0000 0000 0010 0000
#define WAITING_USER    0x40        // 0000 0000 0000 0000 0000 0000 0100 0000
#define IN_PLANNER      0x80        // 0000 0000 0000 0000 0000 0000 1000 0000

#define SDCARD          0x100       // 0000 0000 0000 0000 0000 0001 0000 0000
#define SDCARD_BUSY     0x200       // 0000 0000 0000 0000 0000 0010 0000 0000

#define EXTRUDING_0     0x400       // 0000 0000 0000 0000 0000 0100 0000 0000
#define EXTRUDING_1     0x800       // 0000 0000 0000 0000 0000 1000 0000 0000
#define EXTRUDING_2     0x1000      // 0000 0000 0000 0000 0001 0000 0000 0000
#define EXTRUDING_3     0x2000      // 0000 0000 0000 0000 0010 0000 0000 0000

#define HEATING_0       0x4000      // 0000 0000 0000 0000 0100 0000 0000 0000
#define HEATING_1       0x8000      // 0000 0000 0000 0000 1000 0000 0000 0000
#define HEATING_2       0x10000     // 0000 0000 0000 0001 0000 0000 0000 0000
#define HEATING_3       0x20000     // 0000 0000 0000 0010 0000 0000 0000 0000
#define HEATING_HBP     0x40000     // 0000 0000 0000 0100 0000 0000 0000 0000

#define COOLING_0       0x80000     // 0000 0000 0000 1000 0000 0000 0000 0000
#define COOLING_1       0x100000    // 0000 0000 0001 0000 0000 0000 0000 0000
#define COOLING_2       0x200000    // 0000 0000 0010 0000 0000 0000 0000 0000
#define COOLING_3       0x400000    // 0000 0000 0100 0000 0000 0000 0000 0000
```                    

By using some bitmath we can easily check if the printer is printing and heating by doing `foobar & (PRINTING | HEATING_0 | HEATING_1 | HEATING_2 | HEATING_3)`.

Why and where can this be useful ?
Events.

Another concept would be the introduction of events, when I was doing the code for PR #3113 I found myself repeating the same code and thinking "why can't I just trigger an event ?"

If we pinpoint key points of Marlin than would trigger specific events like:
- Print start
- Print end
- Heater start
- Heater cooling
- Planner full
- LCD whatever
- etc etc etc

In fact there are some global functions as of today which are already some kind of events, but my idea here is to really make them so and expand on the concept.

gddeen, 2016-03-10:
What does ‘Homed’ mean? Got X Y Z E?

jbrazio, 2016-03-10:
> @thinkyhead wrote: Best to post new concepts like this at MarlinDev for further discussion. And actually one thing about Marlin is that in a certain sense it is "stateless" and remains in "command mode" at all times. Usually the point of a full state machine is to also limit which states can transition to other states, and when. For Marlin we don't really need such complexity. Anyway, we can discuss this more over at MarlinDev!

Marlin is at "a state", it allows the state to change at any time because there are no strict flow charts (could be the console, could be user input, could be an interrupt), but if we really want it we could map all the states and their transitions as they are finite. But this philosophical aspect is not the key for this discussion, I understand what you mean by stateless but please just be aware that when I call it a state machine I do not want to make Marlin a soda dispenser, that's not the concept here.. no state transition limitation nor anything of that crap.

The concept is to know at any given time the state of Marlin in an easy way.

This could for sure lead to some some optimization and cleanup like global scope var reduction and consolidation for instance.

@gddeen Could be ZXY, at least is what G28 homes. PS please do not reply by email without cleaning the quoted stuff.

closed 2016-05-25 by boelle