MarlinFirmware/MarlinDev #395: Arduino Experts:  Question about breaking Marlin up into modules

Roxy-3DPrintBoard, 2016-03-17:
This is a question to the Arduino Experts:

Does the ability to have a sketch be comprised of separate folders allow us to break up the code into separate modules.  In the past, we had to have everything in the same directory, and in fact, currently some of the files (like Marlin_main.cpp) have kept getting bigger and bigger because we can't seem to break them up.

We started down a path of turning Marlin into a library just so we could start breaking it up into modules.  But there are a lot of limitations to going that path.  First, you can only place the sketch in C:\Users\name\Documents\Arduino and even worse, it won't work if there are other sketches there.

The release notes seem very incomplete and I can't find information within the Arduino project on how to even use the multiple folders that it supposedly supports.

Can somebody jump in and enlighten us?

MagoKimbra, 2016-03-18:
If you bring everything back as it was before in a sketch, now with Arduino IDE 1.6.7 you can include files in folders and sub-folders ...
Example:
#include "module/motion/stepper.h"
#Include "module/language/english/language.h"

Roxy-3DPrintBoard, 2016-03-18:
I don't think we have a need to do it...  Can we use paths that are up above and outside of the folder that the sketch is in?

MagoKimbra, 2016-03-18:
Exactly ... If you look at my fork is done right so ...

thinkyhead, 2016-03-18:
I want very much to organize in subfolders, and the MagoKimbra layout is a good model, but I felt that Marlin 1.1.0 should be compatible with older versions of Arduino and other IDE systems that don't have the 1.6.7 toolset yet. Basically, because the community deserves a really good release following the old paradigm before we force them to move to a new one.

Would it make sense to (1) call Marlin with the new layout 1.1.1, or 1.2.0, and (2) should we curate a version of Marlin that uses the older layout for the convenience of users who prefer or are stuck with older toolsets?

Roxy-3DPrintBoard, 2016-03-18:
> I want very much to organize in subfolders, and the MagoKimbra layout is a good model, but I felt that Marlin 1.1.0 should be compatible with older versions of Arduino and other IDE systems that [...]

@thinkyhead    When we get done with the Release Candidates, we should take a look at how hard it would be to merge all of the bug fixes in it to the file structure that MagoKimbra has.   It is probably too much to hope for, but wouldn't it be nice to have the best of both worlds???

jbrazio, 2016-03-18:
As Marlin is dependent on Arduino as it's toolkit, if the toolkit evolves I do not believe the burden of curating legacy Marlin code should fall into core team hands. But I do agree with @thinkyhead were a transition period should be somehow in place in order not to unleash user rage. A stable release shall be then marked as the final one supporting Arduino toolkit 1.x and the team must move forward following mainstream.

Roxy-3DPrintBoard, 2016-03-18:
> As Marlin is dependent on Arduino as it's toolkit, if the toolkit evolves I do not believe the burden of curating legacy Marlin code should fall into core team hands. But I do agree with @thinkyhead [...]

I'm not sure I understand what you are trying to say.   But here is what I'm thinking...   

If we can start moving functionality into separate folders it makes it much easier to start cleaning up the code.   We would still be using Arduino, and to the user the only thing that changed is the organization of the files when they unzip the .ZIP file to build the sketch.    

In fact, I think we would still try to have a configuration.h file that controlled the bulk or high level of the configuration.   Where we might want to have discussions is on topics like Auto Bed Leveling where there are so many sub-features that need to be tuned.   It may make sense to have reasonable defaults but bury those down in the folder where they get used.   And if you need to tweak a setting from its (reasonable) default, you would not find that in the main configuration.h file.   But like I say, that is just my thinking of how we might want to organize things assuming we can start breaking up the code into modules.

If we went this route, that is probably very easy for the user.  When they upgrade to a version of the firmware that is organized around the new file layout, they would do what they always do.  They would do a diff on their configuration.h file and compare it to the new configuration.h file to see if there are any new settings they need to cross over.    And then they compile and load it...

I don't think trying to cross our RCBugFix's over to the file layout that MagoKimbra has will cause issues for the users.

jbrazio, 2016-03-18:
Maybe I got it wrong, but if we change the layout then the src code is incompatible with previous versions of Arduino right ? This was the context of my answer but once again I may have got the wrong understanding.

jbrazio, 2016-03-18:
Overall I agree with your approach Roxy.

Roxy-3DPrintBoard, 2016-03-18:
It complicates things a little for people that maintain and do things in their own branch.  But that statement is true of any change we make.   I think what we would do is try to make the minimal number of changes to slice the code apart.    That would make it much easier for somebody to cross their changes over to the new code base.   And it is from this foundation that we would try to continue the development work.

I think the topic deserves discussion to make sure we hear all of the concerns and get the best thinking out on the table.    After the discussion we will do our best to minimize the impact on people trying to maintain their own forks.   But we really do need to start cleaning up the code.   Marlin_main is over 5000 lines long just because we could not slice it up and use the current tool set.

gddeen, 2016-03-19:
A user like me goes in and picks up a Marlin build. 
Their Concerns: Does it work with my printer and have all previous features.
They change Configuration.h.  Maybe make a few dozen changes elsewhere.
They learn quickly that it is difficult to keep up and just keep a configuration.h
and throw away all their other changes.

What they want is to select a Marlin release. merge dev "features 1, 2, 37" into that release.
Merge their previous configuration.h changes onto the release with the features (and save that delta). When a new base with the features 1, 2, 37 are merged into the release they want to be notified.

At that time, repeat...

1,000 fixes makes a user throw away their changes the same as breaking up files.

Roxy-3DPrintBoard, 2016-03-19:
I said:

> If we went this route, that is probably very easy for the user. When they upgrade to a version of the firmware that is organized around the new file layout, they would do what they always do. They [...]

and:

> I think what we would do is try to make the minimal number of changes to slice the code apart. That would make it much easier for somebody to cross their changes over to the new code base. And it is from this foundation that we would try to continue the development work.

What part of trying to make it as easy as possible for the user don't you understand?

But here is something for YOU to understand:  The developers here literally have put in 1000's of hours of effort to design, write, debug and maintain the code.  The code base desperately needs to be cleaned up so that it can be reliable and easier to continue development.   If you don't want that, you can just stick with an old version you know and love.   If you want the code base to evolve, you are going to have to spend an extra 15 or 20 minutes when you switch to the latest and greatest version the first time.     

If we are going to make the code better, there have to be changes made to it.  It isn't realistic to say "I want to the code to be better but you are not allowed to make any changes."

gddeen, 2016-03-19:
Yes Ma'am,
> I said:

> If we went this route, that is probably very easy for the user. When they upgrade to a version of the firmware that is organized around the new file layout, they would do what they always do. They [...]
and:

> think what we would do is try to make the minimal number of changes to slice the code apart. That would make it much easier for somebody to cross their changes over to the new code base. And it is [...]

#Serious response to your questions
Why do you care about single file with 30,000 lines or 10,000 files. With github creating items for each change they "logically" are identical.

They practically aren't identical because of the (I)DEs used during the development process by the different roles of people. So, I outlined a role. Here is a "typical 3dprinter owner firmware patcher." They use the arduino software to build. They download from github. They gave up changing because Arduino lacks the tools to allow easy merge. Outline how to use merge tools might make their job seconds, not longer.

#Reason I Resonded
I had expected in response: "Here is a typical Marlin developer role. We use github desktop for new pulls, tracking issues. We have object oriented development platforms which edit, build, search, cross reference with new files for everything.  The majority use...'kjkjhifif'"**

I had not expected: "Jane you ignorant slut. You wouldn't know a reentrant call by reference if one probed your bits! This is hard work, sit back and let the intelligent people (that are coding on 8 bit processors) work." (I hope you laughed...)
 
Evolution and extra time usually are contraindicated.

My goal was to look at all of the roles represented and see if there are tools, changes to move everyone along. You did ask for this. Or, that's how I understood your original question.

#Elaborations
So far, because that single class is only changing 1 file, you could slice and dice every line in every file.
Scripts could be used to preprocess and retrieve configuration settings into implementation descriptions. IE, you wouldn't have to include the configuration.h file. IDE platforms could be targeted. Older platforms could be cast in stone, etc.

I am not sure of all of the platforms supported, CNC machines, coke machines, arms, printers. Developers, Retail rework, high school kids, old farts, etc. The scope of the effort I heard was make it nifty for mid-level firmware developers.

Also, I don't consider the current code very difficult at all. Marlin is a big file but it is broken into 3 logical sections.  An 'OK' editor removes with multiple views, functional searching, context sensitive support makes it easy.  

Build difficulty. Because Arduino handles that, the load module is kinda guessed at. I assume multiple files are wanted to allow multiple platforms to compile along with the traditional file structures and not uniquely C/C++. I.E. support for graphical dev/build platforms for the displays, traditional networking platforms, and mathematical object platforms.

Direction strategy. Maybe you wish to co-exist, merge into, pull from Arduino, iOS, Android, unix, web apps? I dunno.

So, if anyone is voting... My vote is for me to sit back and follow. I would hope Marlin defines state of the art IDEs to be used and how each player in the game would join in. Hackers, dabblers, full blooded programmers (like me), marketers, manufacturers, students. I would vote to define the set of equipment that would want to use the Marlin firmware and its purpose of that equipment. I would vote for descriptions of the work/effort so people could engage and profit in Marlin.

**But, I am a happy go lucky intellectual optimist. I enjoy other's perspectives.**

Roxy-3DPrintBoard, 2016-03-19:
I apologize.   I took your message to mean you had some changes you added to your fork and you were not willing to spend a few minutes to move to an updated code base.    

It depends on which developer you ask, but pretty much they are all in agreement the code base needs to be cleaned up.   Different issues are affecting different things.    The size of some of the files is causing major issues with scoping right now.   There is a very real desire by other developers to shift to a class based hierarchy.  The size of files and not having functionality grouped together is complicating that shift.

The fact we are not modulized and some of the files have grown to huge sizes also makes it difficult to stay out of each other's way when there are multiple areas being worked on simultaneously.

gddeen, 2016-03-19:
That's kind of the part I'm missing, the way people are getting trounced upon. The github system handles the merges. There would be interactions when "they unfortunately picked the same spot" to add code, or changed the same content. Moving to objects "might" prevent both of these, but probabilistically only the first.

Objectizing "additional" features is a great idea but may preclude the lower power products if the objects "perform" adjustments to the base product. The code is structured as is to reduce cycles. Objects aren't inherently slower but the software industry is packing up shop to switch to a larger 64 bit product. People writing for this class of product do not normally show interest in performance, in my opinion. If you jump to use the big tools/ideas, the product will end up with the fully protected execution environment. Bang, 8 bits may be too slow. (I have no concern one way or the other.)

If I were a business, I would look to the future and see if Marlin fits into the rollout of those cheap, slow 4.3GHz SOC 8 core 64b processors while maintaining dominant presence in the current low cost, hobbyist world.

Is that future a smooth transition. Are the processors of the near term incrementally better or do they require different OS. (I see Marlin as an OS.) Then, the question is build or buy the OS, build or buy the functionality.

A brand spanking new preemptive, multi process, multi threaded "built" Marlin is AWESOMELY fun to create. (having worked on non-virtual realtime dispatcher/scheduler in the far past.) It brings in a class of people that are fun to work with.  But, it is highly selective. This is kin to writing an Amiga ('85) OS and getting all the platforms to cross compile to it. Not THAT tough. You can end up with a good real-time OS with reduced complexity but is integrated into the major development platforms. It may take months.

Or, you can "buy" (not the exchange of legal tender) one that fits perfectly or can be modified to a "Androidish" look and feel. Maybe some have been abandoned by their companies? (There are many to consider.) 

I (myself) wouldn't radically change the design without some real consideration of where you'll end up.
I think the change is coming quickly and the TTL (dual pun. TimeToLive/TransTransLogic) days are numbered. 

Unix built up into a big protected user environment. Maybe it can also build up into one that also supports a large real-time environment aimed at something slightly easier than the requirements of a unix kernel developer. I don't have difficulty at all seeing python, shell, java in the mix, but I wouldn't want it interfaced to real time hardware. So, maybe you're looking for a mix. A simple easy to use arduino like environment in the kernel and the user space.  (There are many many reasons to dig into everything I've typed. But I'm sure lots of people are interested.)

Soapbox off...

Roxy-3DPrintBoard, 2016-03-19:
> That's kind of the part I'm missing, the way people are getting trounced upon. The github system handles the merges. There would be interactions when "they unfortunately picked the same spot" to add [...]

In theory, Yes.   But in practice it doesn't.   A good example would be the Auto Bed Leveling code.    Currently, we have  None, 3-Point, Mesh, and Grid.   Each of these has various flavors and targets.   When somebody changes something simple like what happens when the Z-Probe is retracted but still triggered, chaos breaks out.    You start to layer different machine types on top of the Auto Bed Leveling and you really have a complicated stew to pick through.   And then you add the fact nobody has each machine type and it is real easy to understand how problems creep in.

The code needs to be simplified and cleaned up.

>  If you jump to use the big tools/ideas, the product will end up with the fully protected execution environment. Bang, 8 bits may be too slow. (I have no concern one way or the other.)

> If I were a business, I would look to the future and see if Marlin fits into the rollout of those cheap, slow 4.3GHz SOC 8 core 64b processors while maintaining dominant presence in the current low cost, hobbyist world.

We are pretty careful to keep the code efficient.  I think the 8-Bit AVR's still have some life left in them.   But it really doesn't matter.  The high performance 64-Bit AVR's are starting to be available and will be very inexpensive soon.   There are already Marlin ports to the 64-Bit AVR's.  We probably will be able to support both with the same code base for some time.  It may be some of the fancier stuff like full graphics LCD Panels only run well on the 64-Bit AVR's.   I don't know what compromises we will have to make, but I'm pretty sure we can adjust.

We already have some use of C++ classes  in Marlin.   They haven't been too hard on the processor.  But we will have to be careful about that as we introduce more class hierarchies.

gddeen, 2016-03-19:
Ah Ha! Short term consideration! Not strategy session.
The current arduino systems are plenty fast to handle all C++ objects. Just place them in non-realtime path as they are today.

- Appoint a Marlin god. (Probably is one.)
- Mesh/grid both use the same concept. Grid has 1 plane. Mesh has N planes.
  Make a “Plane object.” Grid will pretty much use one, mesh lots. 
  (Grid is simpler in implementation, but pretty much bent my x bar. It’s very easy to make it into object.)
  (Mesh actually works - Keeps bed from getting crashed.)
  Make 1 object everyone uses, but can select multiple squares/points, start/call mechanisms.
- Anoint a new object “Probe Object,” subclass into “Probe end stop object"
- Anoint a new "End stop Object”
- Anoint a new “Coordinate object”
- Annoint a new “Board pad mapper to logical address to hardware address mapper”
- Design a “printer feature” object. Select a feature on or off, dynamic or static.
- Design an “old horrible display” object.

Hammer out the object design. Pass them out.
Get everyone to agree on the objects and how they work.
Let people redesign their objects. Meet again.
Go grab steal the current code. Put it all together.
Require it all compile as C++.

I would assume using an Object to describe my printer is just as easy as configuration.h file. Manually no big deal for average user.

I would assume a grand object based Marlin everyone wants.

I would assume a grand poobah is needed to keep stuff working together.

I would assume a new “use this object” instead of that object would become the common desired practice.
(His object doesn’t work, hers does, etc…)

I would assume business as usual on bug fixes, enhancements, some features.

I can hardly wait!!!

jbrazio, 2016-03-20:
The problem with objects is I believe it's incompatible with the current Marlin dev workflow. Objects would require everything to be designed and planned upfront implementation, this is not the current Marlin philosophy.

Roxy-3DPrintBoard, 2016-03-20:
Wow!   

Perhaps a counter example will help the discussion:   Would a Z-Probe object require everything to redesigned prior to it's instantiation?   My guess is a Z-Probe object could be made that supported things such as Deployment(), Retraction(), Run_Z_Probe(), Report_Repeatability(), Set_Servo_Angles(), etc. without redesigning everything else. 

And if this Z-Probe object existed, it would be almost trivial to use it instead of the existing calls.   As @gddeen suggested, the Grid, Mesh and 3-Point Auto Bed Leveling have a lot of common usage of Planes in their functionality.   This is more tricky because the real time code needs fast and easy access to underlying data.  We would have to be especially careful how we constructed an Auto_Bed_Leveling object for this reason.   But I think we can see a few methods and class data that all flavors have.

thinkyhead, 2016-03-22:
Stepper, planner, temperature, and other sub-modules can today quite easily be wrapped up into singleton classes to get the encapsulation we like, and without any performance hit, because we would not be using any fancy features of C++. This is a process that will unfold gradually. The functionality and contents of the functions of the initial classes would follow the same pattern as the current code. But, as time goes on and more things –like endstops, extruders, etc.– become encapsulated into objects, the structure will begin to fall into place more. I don't think we'll see any major overhaul of the execution flow. But we will have nicer encapsulation and the code will be getting easier to follow over time.

gddeen, 2016-03-22:
jbrazio. There are currently "objects" in Marlin. It's simply c++. One does not need to think and plan to create/design objects. However, objects can inherit "stuff" that just magically changes their behavior.
Maybe that isn't the smartest thing to do on an 8bit computer. So, don't do that.

Classes, objects all sorts of programming concepts were added in 80s for good reasons.
Hacking (Making) will still survive.

CONSULitAS, 2016-03-25:
@gddeen
Perhaps you should think and have a look at examples for objects on Arduino before you disencourage people here by nonsense. Objects are not slow on an 8 bit machine! Stupid programming is slow.

If you don't believe i recommend some reading for you:
http://cosa-arduino.blogspot.com.es
http://cosa-arduino.blogspot.com.es/2013/03/benchmarking-pin-classes.html (slow objects?)
https://github.com/mikaelpatel/Cosa
http://forum.arduino.cc/index.php/topic,150299.0.html

Did you read it all?

So perhaps start contributing good code and show working concepts instead of bashing an important approach for making Marlin more maintainable and even more flexible.

BTW: I read it...

jbrazio, 2016-03-26:
I'm not saying we should move to objects, it's a pleasure to read @repetier code. I'm just saying the with the technical shift a mindset shift must also happen.

A quick illustration of what I mean: a key feature of good OOP is self containment, meaning [as an example] global vars are over, if you need to access a particular data struct then this must be published by the object's public interface and no form of "hacking" should be tolerated from the developers. There is no more coordinate changes by just editing the current position array and then informing the planner.. we must tell the planner our intent and then it's up to it to do it and then inform us of the result.

Sniffle, 2016-03-27:
Damn wheres a like button when you need it...

If your gonna go OOP do it right or not at all, maintain scope and keep
things private between objects so that the objects are just little black
boxes that take inout and give output.  I kinda have a feeling this is the
turning point for marlin.  Its either gonna grow back into the monster
powerhouse firmware it once was or its gonna fall into obsolescence.  This
is basically the step that allows marlins features to move forward onto 32
and 64 bit chips.  I realize im a very small player in this but you guys
also know i only open my mouth when i have a good point to make.

CONSULitAS, 2016-03-27:
@Sniffle It is new just in the right upper corner of any comment shown as + :smile: .

Roxy-3DPrintBoard, 2016-03-27:
These kinds of discussion have a tendency to become religious...  But...

> A quick illustration of what I mean: a key feature of good OOP is self containment, meaning [as an example] global vars are over...

No...  we need to be very careful with the words we use.   For example,  If we took the simple case of a instantiating a Probe Object, I'm pretty sure this object would be a global variable of class Probe with what ever appropriate extensions were needed by that particular hardware setup.   But...

> If your gonna go OOP do it right or not at all, maintain scope and keep things private between objects so that the objects are just little black boxes that take inout and give output. 

This hypothetical Probe Object would not have its class data available for anything else to mess around with it.   This Probe Object would be a little black box that just does what its supposed to do.

Sniffle, 2016-03-27:
<3 roxy i just saw something worth talking about and threw in my .02 like
usual :-D

jbrazio, 2016-03-27:
What happen to @Roxy-3DPrintBoard comment ?

> I'm pretty sure this object would be a global variable of class Probe

Of course we still have "global" pointers to objects, but they are no longer global, they are inside the **Controller** class.. in a nutshell what we know as `Marlin_main` will be a `Marlin` *controller* class which will take care of instancing all the other objects.

Roxy-3DPrintBoard, 2016-03-27:
> What happen to @Roxy-3DPrintBoard comment ?

I deleted it because I'm probably the only one that doesn't have a formal Computer Science training.  I figured I would end up getting picked on because there is probably some technicality where an object is not considered a variable or some such non-sense!   :)

Sniffle, 2016-03-27:
I only partially completed mine so im only good for theory.  I do
understand thats why i dont talk much.

ghumphr1, 2016-03-27:
@Roxy-3DPrintBoard - I am an engineer - and i think you should have say whatever.... just because you havent such and such piece of paper doesnt mean you have no valid input.. more so than some of the 'qualified' people most times

Roxy-3DPrintBoard, 2016-03-27:
> and i think you should have say whatever.... 

Everything is good.   And if you saw the deleted comment, the first line of it was "These types of discussions tend to turn religious..."    That is because the object orientated topic is very important with very strong beliefs on all sides.   

What we are all after is a cleaner and more easily maintained code base.  And probably some C++ objects can help on that topic.   But for me it isn't a religious matter.   It is putting the objects in places where it makes sense and helps encapsulate data and functionality.

But because Marlin has a bunch of real time code and is currently powered by a slightly under powered AVR processor, we have to be careful how we do it.  We can move in that direction, but it has to be carefully thought out.

jbrazio, 2016-03-27:
As long as we don't go into the `malloc()` road..

If you go object oriented.. you should go all the way; the problem is when people forget they are on a AVR arch and start doing "dynamic stuff".

ghumphr1, 2016-03-27:
* powered by a slightly under powered AVR processor*
lol..

Roxy-3DPrintBoard, 2016-03-27:
> As long as we don't go into the malloc() road..

Another religious topic when it comes to firmware!  malloc() and free() can be used safely but they are so controversial, it is just best to avoid them if possible.

jbrazio, 2016-03-27:
lol it's Easter, a lot of religious topics !

The "dynamic issues" come into plate when people start delegation into runtime stuff that should, in a embedded environment, be defined at compile time, some examples are number of end stops, number of extruders..

If we know upfront we how many extruder objects are needed then there is no need to start dynamically increasing the array where their pointers are stored just because we are dealing with objects, preprocessing is [and should continue to be] our valuable friend.

thinkyhead, 2016-03-28:
The initial wrapping of Marlin elements into classes will be done without any changes to the functions themselves or the code outside the classes. It will just be a convenient wrapper so that we, fore example, only expose a single "stepper" object instead of all the global variables defined in `stepper.cpp`. Callers will simply be preceding all their function calls with `stepper.` (not `stepper->`). Stepper and Planner will be the first two objects created. The purpose of the first round of object-izing will not be to improve the design patterns, only to get encapsulation and limit access to data. It is only a first step, and out of it some additional order will emerge. If done gradually, Marlin will naturally morph into a proper architecture. But we have to take it slowly starting from our current minor mess, otherwise we might as well just start with another firmware that's already got this done.

Roxy, no worries over "skills." I have zero formal training in computer-science, and am entirely self-taught. Everything you need to know can be learned right on the Internet. The key thing is to pursue your interests, learn what you need to know as you go along, and keep trying new languages and environments. After a while, all programming looks a lot the same!

Roxy-3DPrintBoard, 2016-03-28:
> Roxy, no worries over "skills." I have zero formal training in computer-science, and am entirely self-taught. Everything you need to know can be learned right on the Internet. The key thing is to [...]

Thanks for the encouragement.  Mostly what you are seeing is 'scar tissue' from previous interactions.  But that is behind us now.   I like messing with code that is 'down on the metal'.   (M100 for instance)   (and my current play toy the Delta Mesh Leveling System)

But when it comes to higher level architectural direction, my biggest contribution is going to be getting consensus from the people that really know the issues.  I am very serious about rebuilding the Marlin Team and making it a force to be reckoned with.

jbrazio, 2016-03-28:
> The initial wrapping of Marlin elements into classes will be done without any changes to the functions themselves or the code outside the classes.

Absolutely agreed. I've been talking about high level design, not about the detailed roadmap.

`<offtopic>`
> After a while, all programming looks a lot the same!

Not all of [them](https://en.wikipedia.org/wiki/Brainfuck)..
```bf
++++++++[>++++[>++>+++>+++>+<<<<-]>+>+>->>+[<]<-]>>.>---.+++++++..+++.>>.<-.<.+++.------.--------.>>+.>++.
```
(Impressive, Github has a bf syntax highlighter)
`</offtopic>`

Roxy-3DPrintBoard, 2016-03-28:
~~~~
<more offtopic>
After a while, all programming looks a lot the same!
Not all of them..
++++++++[>++++[>++>+++>+++>+<<<<-]>+>+>->>+[<]<-]>>.>---.+++++++..+++.>>.<-.<.+++.------.--------.>>+.>++.
(Impressive, Github has a bf syntax highlighter)
</more offtopic>
~~~~
So of course I get curious and start checking things out.  It says:

>A second version of the compiler used only 240 bytes.[4]

>As Aminet grew, the compiler became popular among the Amiga community, and in time it was implemented for other platforms. Several brainfuck compilers have been made smaller than 200 bytes, and one is only 100 bytes.

I was very disappointed.  It was nice that the source code for the compiler is available to look at.  But it is written in 68000 assembly language and not in BrainFuck.   It would seem to me the BrainFuck compiler should be written in BrainFuck.   And of course, this compiler had absolutely no error or syntax checking.

thinkyhead, 2016-03-29:
I spent a few years writing 680x0 assembler on the Amiga, and I really liked that platform. I wrote a couple of games. I was the author of **Dino Wars** and the Amiga port of **Bill 'n' Ted's Excellent Adventure**.

gddeen, 2016-03-29:
It WAS awesomeness.

gddeen, 2016-03-29:
I am not a religious object fan anyway.  I am for what works the best. Nothing really works the best 
except the way the code is now.  The main feature are no big dynamic data. ConsultitAS thanks for
the reading material. I do think 16MHz machines are slow when compared to 4.4GHz 32 core ones.
Programmers these days think a quad core 1.2GHz 64 bit machine is slow. It's because they have
no control over all the excess luggage they carry.  I am a firm believer in "Keep it stupid, simple."
Elegance is nice, simple is nice but it needs to be understandable.

Sniffle, 2016-03-31:
this developed into a bigger post than I intended, Ive been thinking on this a LOT lately...

/me puts foot on soapbox

So, IF marlin goes the object route or even if it doesnt, has anyone thought about laying out the current code, and deciding how it should be broken up?  The macro view of breaking the code up and planning for things to be converted, because without it, it's gonna be a dirty ugly mess as the current marlin dev code shows.  Before this happens unlike previous times there needs to be a clear plan of attack.  I know @thinkyhead probably has an idea in his head how he wants to go about this, and @Roxy-3DPrintBoard seems to be the driving force that is keeping this on track and has pushed marlin to the point it currently is and it seems she has more pulled people to the cause.  

so just for examples sake, things like how the code interacts, I'll give a code example and reasoning.  

I'm using this to learn C++ syntax and brush up on my coding skills so dont hold it to too high of a standard
'''c++
class probe{
	bool stowed;
	fake_actioner a;
	public:
		void extend(){
			a.extend();
			stowed = false;
		}
		void retract(){
			a.retract();
			stowed = true;
		}
		bool probe_state(){
			return stowed;
		}
		void probe_init(fake_actioner b){
			a = b;//needs to reference the original object not a copy
			this->retract();
		}
};
'''
Should the probe object just store its state or should it call the actioner to act upon it, or even further should there even be a probe object that stores its state, it technically is an object to be identified and fleshed out.

/me steps on soapbox
To me this is the elephant in the room, beyond breaking marlin up into modules.  You can want to break it up all you want to but without a cohesive plan of attack it's going to fall apart and I personally dont want that.  Deciding bug fix time is easy your not basically creating something new, your just fixing whats already there.  If marlin undergoes a large change like rewriting it as OO you might as well call it marlin 2.0 not 1.2 it is a complete rewrite of the old code to a new higher standard to increase code readability and functionality moving forward to be able to go to new hardware beyond the mega2560.

This is something that NEEDS to be be highly visible and enforced with big checkoffs and milestones to keep things on track no matter who comes and goes so that it will move forward no matter what, much like the coding standard that has been implemented.  I really like marlin and this community I want it to be so much more than it is...
/sigh
/me steps off soapbox

Roxy-3DPrintBoard, 2016-03-31:
@Sniffle You have several questions.   First the easy ones:

> The macro view of breaking the code up and planning for things to be converted, because without it, it's gonna be a dirty ugly mess as the current marlin dev code shows. 

We already have several proto types for breaking up the files into modules.  In MarlinDev there is a good start at that.   We probably want to have a separate folder for the files that belong in each topic area.   Please take a look at the Feb14 branch.    

But...  I'm guessing this all gets abandoned.   MarlinKimbra has Due support running and has started breaking up the files into modules similarly to what you see in the Feb14 branch.   This next idea is going to be discussed to get everybody's input prior to it happening, but a number of people are in favor of taking the changes (improvements) in the Release Cadidate (Once it goes Golden) and crossing those over to the MarlinKimbra branch.   If we did that, we would use that as the starting platform for continued MarlinDev work.

> Should the probe object just store its state or should it call the actioner to act upon it, or even further should there even be a probe object that stores its state, it technically is an object to be identified and fleshed out.

The whole purpose of moving **SOME** of the functionality to being Object Orientated is to start isolating behavior and encapsulating data.   So, "Yes!", the probe object should store its state.  And this brings up an important consideration for the conversion process.   Each objects functionality (and data) need to be very quick and efficient to access.    We will have discussions on this, but we need things like Z_Probe->is_deployed().    But we may not want to incur a function call to get that information.   We may want to define an inline macro expansion of a is_deployed() function that goes straight to the class data that knows about the probe.   

The Z-Probe is a good example.   Its actual use is fairly limited.   And with the exception of the stepper code that is constantly checking to see if an end stop has been hit, its use is not in any critical real time code.   So, with the inline function work around for Z_Probe->is_deployed() I expect a Z-Probe object will be fairly clean and easy to transplant into the code base.    We cut out all of the old code, and just start referencing the base functionality of the Z-Probe object.

And at that point...   We have a single global pointer to a Z-Probe object.  And it can be defined to be any of the known types:   None, Manually Deployed, Manually Positioned Nozzle, Allen Key, Servo, Inductive, etc.  All of the quirks of particular probe will be out of the main code base and hidden in that particular probe's code.

> To me this is the elephant in the room, beyond breaking marlin up into modules. You can want to break it up all you want to but without a cohesive plan of attack it's going to fall apart and I personally dont want that. 

Well...   There is a lot of stuff that is obviously in need of clean up.   If we start getting the obvious things cleaned up, we will have time to think about and decide how to clean up the rest.   Right now, we do have a high level plan:   Start getting things cleaned up.       Later, we will have more specific thoughts on how to continue.

Sniffle, 2016-03-31:
OK so I guess since I have a partially complete formal CS degree, I was taught that objects should be as simple as possible and should build and call upon each other to make more complex actions easier to follow and read.

I see a probe as an object that is either deployed or stowed, and it is acted upon by probe actioners that that define how that probe object is used.  Said actioners are the methodologies of deploying the probe, either by doing nothing or null object, manual deployment object, Manually Positioned Nozzle object , Allen Key object, Servo object, Inductive object, etc. and by having the probe do the calling of that actioner object, you have a probe object that can be acted upon by anything and will always store its state.  Below the probe object you would have a bed leveling object that is initialized by a general functionality object that inits a movement objects that init stepper objects... ...that init the HAL object that talks to defined hardware.  the config would be read by the general functionality object that goes on to init everything needed.  Partially going OO means that instead of referencing other objects to do tasks you are picking a spot to remove all the layers below and dump them all into that object much like how marlin_main.cpp has gotten so large, at that point in the code things suddenly get complex, bloated and hard to read with code duplication everywhere..  

Using this type of methodology for converting the code if there was ever an issue with say the allen key code you would be able to go to the allen key actioner directly and only deal with that clearly defined piece of code never actually needing to touch the probe code or anything else.  All of those objects could be either broken out into their own files or be a single probe actioner file with clearly defined and hidden methods.

I just have a hard time seeing how you can go partially OO, Its so integral to the way things are coded and how the code flows partiality will probably cause a lot of issues.  I guess where I am getting hung up is the actual coding methodology for implementation vs the concept and plan for doing so.  I'm really not trying to be a naysayer, I want it to happen and I'll help as much as I possibly can, but wrapping my head around how to do it without starting over from the bottom and working our way up without a solidly defined plan is giving me headaches.  As an idea, go look at Smoothies source and how it is broken up using full OO standards(as far as I can tell.) It is a good place to look for examples on how OO works out and why partiality is hard to do. wolfmanjm is a hard taskmaster for keeping smoothies code on track, elegant and properly developed.

again these are all observations on the assumption that the marlin developers will choose go OO

bobc, 2016-04-09:
Breaking the code into modules is a great idea, it's an essential addition to the Arduino IDE IMO.

As regards OO, I did a CS degree but also learnt through job experience that KISS and being pragmatic are more important in reality than maintaining a pure OO abstraction. I've seem some godawful code written according to a supposed "OO paradigm".

I prefer evolution rather than the big bang approach. As (I think) all of the codebase is actually compiled as C++, there is no linking issues introducing objects in stages.

The first step to producing a clean OO abstraction is to figure out what the modules are, and what the dependencies are. Some refactoring will certainly be needed.

jbrazio, 2016-04-09:
@Roxy-3DPrintBoard @thinkyhead seems that we're the only ones without a formal CS degree around here. ;-D

Sniffle, 2016-04-09:
I never finished mine :-D so you arent alone :-P

bobc, 2016-04-09:
Heh, my degree predates most of the OO stuff any way. Not much call for FORTRAN any more...

gddeen, 2016-04-09:
There is PLENTY of call for FORTRAN still (like it or not...) You could probably translate libraries to APL?(jk)
Formal degrees are only good when the person dedicates themselves to them.
In programming, daily usage is catamount. Unless you want to know if a problem is NP complete, left or right recursing grammar, or why the heck random numbers are so necessary to computational science.

CONSULitAS, 2016-04-09:
@jbrazio @Roxy-3DPrintBoard @thinkyhead 
OK: I drop my pants, too. I only have an PH D in chemistry.

But: Does some C source and Bash scripts in the appendix of the thesis (found online) count? Perhaps not. 😁 😀 

But: The formal CS degree says nothing about a programmer or IT architect. I wrote my first program down in 1980 and had my first own computer in 1981. Have there been CS degrees in germany at all? Dunno exactly. I have been much to young for such stuff 😄 .

@all
So lets get back to the topic. IMO:
* We need modules, because Marlin is not as structured as it should be.
* We need modules, because it makes Marlin more maintainable.
* Objects are great and should be used. But step by step and function by function.
* If you read about the object implementation in C++ it is just an enhanced `struct`. So a good step to start is to pack variables which belong together in structs.
* For this struct you can then start to write functions working with it aka "methods".
* If an object is not completely ready, you could make the properties you cannot access by methods public. So legacy code sees just a struct without knowing about the object.
* Big bang never works when refactoring. So find a way and start walking. Step by step.

So lets start!

A really annoying part of code which makes a lot of trouble for users is EEPROM.

Take a look at https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L148 and following.
* Imagine, how it would look like, if we had a struct with all feedrates, all accelerations and all jerks. Just a few lines left.
* A major advantage: If there is a new `max_f_jerk` e.g. for a new tool, it would work without a change!

And now https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L504 and following.
* If the struct would be extended to an object that has a method named `resetToDefaults`, you would just write `feedrates.resetToDefaults();`.

Interesting is the part https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L515 
* I would suggest `bedLevelingMesh.disable();` and if the feature is switched of, all methods are exchanged by dummies. So you wouldn't need an `#if`, because it sounds like:

```
bedLevelingMesh.disable() {
  #if ENABLED(MESH_BED_LEVELING)
    bedLevelingMesh.active = 0;
  #elsif
    // do nothing, because no action required if MBL disabled
  #endif
}
```

BTW:
* The call of the now empty method `disable()` will be optimized by the compiler. So the resulting code is the same, but shorter, better to maintain, less nested `#if`s and so on.
* I would recommend to define a naming convention for any object or struct in Marlin: Important part first and camel case e.g.
  * bedLeveling
  * bedLevelingMesh
  * bedLevelingManual

So you can better sort, find, see parts that belong together. And it fits the way C(++) uses for structs.

So https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L533 would result in a struct like
```
// anywhere
#define PLA 0
#define ABS 1

// resulting code
preheat.temp.hotend[PLA] = PLA_PREHEAT_HOTEND_TEMP;
preheat.temp.heatbed[PLA] = PLA_PREHEAT_HPB_TEMP;
preheat.speedFan[PLA]
preheat.temp.hotend[ABS]
preheat.temp.heatbed[ABS]
preheat.speedFan[ABS]
```

Than https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L206 would be funny. Just:
`EEPROM_WRITE_VAR(i, preheat);` or perhaps `preheat.eepromWrite();` which would use an EEPROM object and just inherit the code from a generic configuration object. And then every new object has to have its own logic for writing GCODE in `Config_PrintSettings`.

As an result i would extend `configuration.eepromWrite();` to `configuration.eepromWrite(mode);` with modes for EEPROM, GCODE, SD and more to come. One code fits all. And you can keep it quite simple.

Not that challenging, but a lot of work and changing the look and feel of Marlin quite a bit.

Uh! Quite a long contribution. What do you think?
* Step by step starting with structs and easy objects?
* Start with configuration variables as global variables which should be encapsulated and seem to be rather simple?

MagoKimbra, 2016-04-09:
Look this https://github.com/MagoKimbra/MarlinKimbra4due/tree/V4_3_0_new_tools/MK4due/module/tools

Test version width object Hotends and Beds...

bobc, 2016-04-09:
@MagoKimbra Ok, seriously now, what the fuck are you doing?

You've taken other people's code, removed their names and put your name on it. This is just not fucking cool.

Unless there is *very prompt* action I shall be filing copyright complaints.

jbrazio, 2016-04-10:
@CONSULitAS thanks for your wise opinion.

It's almost nothing, but I've converted the SD/Serial print timer into an [object](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stopwatch.h), I was trying to eyeball what would be the acceptance of such approach: start migration small parts of the code into dedicated objects. @thinkyhead approved the approach but I didn't received any other feedback, now I see you do support the same approach for the refactoring. :+1: 

> Imagine, how it would look like, if we had a struct with all feedrates, all accelerations and all jerks. Just a few lines left.

As entry points to the `ConfigurationStore` read and write are quite easy to pin point on Marlin, IMO this one is a perfect [next] target to be "objectified".

Now imagine we, on the new `ConfigurationStore`, implement a `ConfigStore.export();` to a SDCard and the all the SDCard stuff is still as is today.. this is my approach issue: when we have "objects" that need to access parts of Marlin which we have no refactored yet. Should we leave them as they are and later on revisit them or should we start chaining them together ?

> If the struct would be extended to an object that has a method named resetToDefaults, you would just write feedrates.resetToDefaults();.

Are you suggesting to have multiple structs, i.e. one for feedrates, another for accelaration, jerk etc ?

> and if the feature is switched of, all methods are exchanged by dummies

In order to be able to do `mbl.disable()` it means that the `mbl` object would always need to be defined, independent of `MESH_BED_LEVELING`, wouldn't this cause impact on the memory footprint if we transpose to "all" the other Marlin features ?

> I would recommend to define a naming convention for any object or struct in Marlin

Not crafted in stone but I've been told to use `UpperCamelCase` for objects and `lowercase` for file names, nevertheless I would agree `lowerCamelCase` looks better for methods.

> And then every new object has to have its own logic for writing GCODE in Config_PrintSettings.

Are you suggesting each object which has EEPROM data would require someting like `printSettings()` method to dump it's own settings and then `ConfigurationStore` would do something like this ?

```
mbl.printSettings();
preheat.printSettings();
(..)
```

CONSULitAS, 2016-04-10:
@jbrazio
> thanks for your wise opinion.

You are welcome. BTW: Thanks for all the great work you did on Marlin in the last weeks.

>It's almost nothing, but I've converted the SD/Serial print timer into an object

No, this isn't nothing, its just one step into the right direction. It will make code much more capsulated and readable. I had a short look at it and think it is rather nice now! But you should perhaps change `status` to `state`, because you implemented it as an state machine, which is desirable, because they are rather stable and easy to understand. 

>  @thinkyhead approved the approach but I didn't received any other feedback

* Have a look which members of the core team are active (and merging) in the last weeks. You've  got a good statistics. 😄
* You could say in this moment: "It is good code, if @thinkyhead merges it."
* And with every community work: If nobody complains loud, it is in most cases a great solution!
* **Perhaps just start a PR with your little nifty stopwatch and see whats happening.

> As entry points to the ConfigurationStore read and write are quite easy to pin point on Marlin, IMO this one is a perfect [next] target to be "objectified".

Thats exactly the reason why i picked this part of code. For the start of refactoring you should use simple examples, because we have to learn and exercise. So it has to be easy.

> Should we leave them as they are and later on revisit them or should we start chaining them together ?

If you try to solve every dependency, you will find, that you have to change everything in a big bang. So **leave them as they are**. This is important for bug fixing, too. If you change to much, you don't know where to search.

> Are you suggesting to have multiple structs, i.e. one for feedrates, another for accelaration, jerk etc ?

Not really. It has been a quick and dirty example just before going to bed. But thinking more about this topic, it makes sense to chain every config to the object it is needed for.

So we have to think which objects make sense. I would suppose to start with the simple ones again to make experiences.

> wouldn't this cause impact on the memory footprint if we transpose to "all" the other Marlin features ?

If we define it only with empty dummies and without properties it will be optimized away and don't hurt.

> Not crafted in stone but I've been told to use UpperCamelCase for objects and lowercase for file names, nevertheless I would agree lowerCamelCase looks better for methods.

I would agree, but for different reason:
* Objects are things with names and you write names starting with capitals like `João`. So my object should be called `Preheat` instead.
* Methods are instructions what to do and you can count them as verb. This is written lower case like `goOn`.
* `João.goOn();` is readable like a sentence, what is probably the reason why you think it "looks" better.

> Are you suggesting each object which has EEPROM data would require someting like printSettings() method to dump it's own settings and then ConfigurationStore would do something like this ?

Excactly. And the objects than should use a public method of the EEPROM object, because this is the master of the storage and encapsulates the technical interface.

**As an result you get the long searched for HAL!** If the EEPROM object knows how to interface with the hardware and the configuration items just use it, you can exchange the code as you like, if you only keep the interface.

So the EEPROM object should be renamed to something like PersistentStorage which supports channels like SD, EEPROM, GCODE, engraving to stone plates 😁 or anything else. It has just to be persistent (in a way).

But this leads to the point, where you need to redefine the way you store the information. If you want to write GCODE and binary data. Look here;
https://github.com/MarlinFirmware/MarlinDev/blob/dev/configuration_store.cpp#L614

I would prose something like:
```
Axis.printSettings();

in Axis.cpp:

Axis::printSettings() {
...
PersistentStorage.saveData("M92 X",axis_steps_per_unit[X_AXIS]);
PersistentStorage.saveData("M92 Y",axis_steps_per_unit[Y_AXIS]);
PersistentStorage.saveData("M92 Z",axis_steps_per_unit[Z_AXIS]);
PersistentStorage.saveData("M92 E",axis_steps_per_unit[E_AXIS]);
...
}
```

This looks similar to the actual code, but the result is completely different: M5xx only calls the objects to be saved and the objects care for their selves. And if anybody extends the object like adding a new F axis he will see the method and adapt it. Now nearly every time EEPROM code is forgotten.

Sorry, but i didn't have the time to make it shorter. 😀

Roxy-3DPrintBoard, 2016-04-10:
> Are you suggesting each object which has EEPROM data would require someting like printSettings() method to dump it's own settings and then ConfigurationStore would do something like this ?

I'm wrestling with a similar question for the High Resolution Delta Mesh Bed Leveling.  I'm going to be storing a matrix of numbers in the EEPROM and reloading at reboot time.   What I was going to do in order to not be very invasive was have a method inside my Auto_Bed_Leveling object that would write its data at an index passed into it.  And it would return the index of where subsequent EEPROM writes can occur.   

That would allow me to just add one line in the Configuration_Store.cpp file to save my object's data, and another similar method to restore its data.

jbrazio, 2016-04-10:
I'm on mobile right now and I need a PC to fully answer to the previous two comments but how about a EEPROM_lookup_t[] which hold data structs, each struct would define:
 - number of bytes reserved
 - function to be called for get/put data into the object

Like this ConfigurationStore would be agnostic and would not require to be edited each time a new EEPROM data block is introduced, just add it to the table.

ConfugurationStore when reading the data from EEPROM would check the offset and call the required object function which will then do whatever is required with the data.

IMO this goes into the direction @CONSULitAS was speaking about "isolating" each object and their data into a container, making each object the sole responsible for its footprint.

Roxy-3DPrintBoard, 2016-04-10:
> but how about a EEPROM_lookup_t[] which hold data structs, each struct would define:
* number of bytes reserved
* function to be called for get/put data into the object

Or...  Maybe we can free ourselves from fixed locations in the EEPROM.  Maybe each object that wants to store data can give the EEPROM (or Persistent_Storage object) the name of the data that it is storing.  Then as items get added or deleted, it just doesn't matter.  The previous values can still be found and used.

jbrazio, 2016-04-10:
At first look I see a possible issue because the order to which each object registers itself with the `ConfigurationStore` would then matter, or am I under seeing this ? But this type of "freedom" would be great I do agree with you.

[43 more replies]

closed 2016-05-25 by boelle