MarlinFirmware/MarlinDev #397: Feature Request - Add support for Tilt z-probe

stratop80, 2016-03-28:
As a follow up for PR#3190 in the Marlin RC branch, as suggested by thinky head, I put this here for consideration.

This type of z-probe can be seen in http://www.thingiverse.com/thing:1427575, mounted in a cartesian printer, an ultimaker derivative but with usual Arduino/RAMPS electronics.

It is based on a twisting probe, that deploys when G28 is performed on a specially defined Z_SAFE_HOMING point -that is  slightly outside of the print area-  using the vertical homing movement towards the bed, by acting a lever against a fixed piece.

This way it performs an initial z-homing on that point, that is provisional. Next, it moves to the center of the bed, where it homes for real, and then returns to the safe homing point to stow the probe, finishing the homing. Later calls to deploy or stow the probe by the firmware work transparently.

Though this unit is made for this special printhead, the system may be used in similar pieces applied to a Greg extruder, or any other else, as far as you can get a few mm outside your print area to the extra movement needed., where the vertical homing descent of the head is the movement that triggers the deployment. 

In my branches of Marlin I've made the modifications needed both for RC3 and RC4 -so many changes arisen!- for anybody to check. I'm not a programmer, and this is made at my best knowledge. I mean, it works, no luck involved, but for sure there will be better ways of coding this, or seeing it as a part of a wider scope, or not... I don't even know how to sync this for any future change, any comment will be appreciated.

Hope I may have explained myself, and thanks if you have a look in this!

Roxy-3DPrintBoard, 2016-03-28:
That is cool!   If this had existed when the first Z-Probing was being done, nobody would have messed with servo's.

stratop80, 2016-03-29:
Glad you like it! Yes, I think it might suit many users, at almost zero cost :-) Regards!

Roxy-3DPrintBoard, 2016-04-02:
I think there may be a volunteer to start the conversion of probe functionality to an Object Orientated design.   Support for this type of a probe would happen at the later stages when probe designs such as Allen Key probes get supported.   Some Allen Key probes get deployed (and retracted) with special paths of movement.  That same logic could deploy and retract this probe.

paulusjacobus, 2016-04-04:
That's a really cool probe. Its avantage is that you measure with the nozzle so no need for an offset. I might give this a try to satisfy my inner geek.

Blue-Marlin, 2016-04-04:
@paulusjacobus 
No offset?
You are talking about an other kind of probe.

paulusjacobus, 2016-04-04:
@Blue-Marlin actually it is www.thingiverse.com/thing:1093345 
I followed the trial around the tilt sensor and that thing on thingiverse is a real tilt sensor. The hot end is mounted in a device that pushes against an opto coupler. A very ingenius design that uses the slight nozzle pressure to tilt a switch (in this case an opto coupler).

jbrazio, 2016-04-04:
I thought tilt sensors were not very accurate taking into consideration the have like two positions.

For me the more user friendly probe design is the one Lulzbot have, nozzle tip is conductive and then it touches a "special" metal clip (holding the glass) which terminates the circuit. The only drawback is that your tip must be really clean otherwise bad thing will happen.

paulusjacobus, 2016-04-04:
@jbrazio I thought that too, the introduced gap will influence the accuracy. Having said this, the maker on thingiverse engineered the gap in the z direction. I assume that the pivot movement is very small. On a moving z axis that correct the bed, the z stability would have now also the pivot movement. So I assume you would see that in the walls of a printed object. Lulzbot feature which is also used by Prusa is actually very elegant as well. Prusa actually has integrated 4 switches in the bed. I probably stick with my proximity switch now after seeing the actual trick 😀

stratop80, 2016-04-04:
@Roxy, I agree that my kind of probe is in a certain  sense similar to the AllenKey, the logic for deploying is by means of following a certain path, so coding is someway similar.
@paulusjacobus I was to call this thing at first a 'swing' probe, maybe tilt is a bit misleading, not the exact meaning, as the tilt/swing is not the probing method, but the description of the deploying/retracting movement. Anyway, once in deployed position, measures are quite consistent, as the probing is very straight.
I saw this video [https://www.youtube.com/watch?v=e119OXzkk7s](url) a couple years ago that seems to be the same idea of the delta probing youre discussing. I was inspired by that and tried a design, (over this same printhead I use, but with the same principle) the nozzle acting as the probe, swinging/tilting as it touched the hotbed and pressing a microswicth. It needed some extra movements before probing to reach a point where a piece of woolsteel (dont know the word, sry) cleaned the tip of the nozzle before probing, But oozing was not always avoided, and some retracts of the extruder were involved, it was all a bit messy. 
[image]
The setup of the part, in a way that the hotend was free to tilt by a light pressure, made that the whole thing was keen to vibrate when hitting a protruding bit of plastic of the existing piece, or when resonating if creating infills, so, as you say quality was affected. I decided not to go on with that, and started messing with proximity sensors, and lately with this approach. I find it simple and effective.

stratop80, 2016-04-14:
Hi, @thinkyhead, 
I saw your new PR https://github.com/MarlinFirmware/Marlin/pull/3491 
Does it support this kind of probe?

thinkyhead, 2016-04-22:
@stratop80 When it comes to the correct way to configure your probe, the person to ask is probably @esenapaj.

stratop80, 2016-04-22:
Thanks, @thinkyhead, I have it working just fine with RC3,4,5.
I see there is a lot of movement about ABL and probes,  so I'll probably wait until it stabilizes to code again my stuff, then I should tell him if he is willing to check. 
Best regards!

closed 2016-05-03 by thinkyhead