MarlinFirmware/MarlinDev #398: Recommendations for a good 32 bit microprocessor to run Marlin

Roxy-3DPrintBoard, 2016-03-30:
The Atmel ATSAM3x8e on the Due board is definitely viable.   It has the benefit it can plug into the RAMPS or RADDS base board and at that point it is easy to populate it with other needed items like Step Sticks to handle the stepper motors.

But my questions is "What other companies make viable platforms to run 3D-Printers?"   Ideally, these microprocessors have good compiler support and can have similar items to the Step Sticks plugged into them.

The best case would be if these other microprocessor boards were compatible with the RAMPS and RADDS boards and could just plug into them.

jbrazio, 2016-03-30:
This is an interesting topic. Following your discussion on the other thread really got me thinking to buy one of these boards (DUE/RADDS).

Roxy-3DPrintBoard, 2016-03-30:
@jbrazio    Well...   If you are not in a super big hurry...   It may be best to hold off for a couple of weeks.   The real purpose of this thread is to try to identify a 'Reference Platform' for 32-Bit Marlin's.  @BioPrinter and I have been exchanging emails.   If we can get some consensus on 2 or 3 viable platforms, he is going to start lobbying the microprocessor vendors to see if one of them will step up and provide the development hardware for our group.

The thinking is, if we have 6 or 7 contributors with the exact same hardware being used as a reference platform, we are going to be able to make the transition faster.   And then for our users, that exact reference platform is pretty much guaranteed to always work and to have the latest and greatest feature set at all times.   The hardware vendor would get a big magnet pulling new 32-Bit 3D-Printer sales in their direction.   And given the size of the Marlin user base, that is nothing to sneeze at.

Sniffle, 2016-03-30:
I would think if we build a hardware interface layer it could be just about
anything even a smoothie...

Roxy-3DPrintBoard, 2016-03-30:
Yeah...   That would be the theory!   Maybe I'm just being negative.  But having other contributors with the exact same hardware gives me more confidence.

jbrazio, 2016-03-30:
@Roxy-3DPrintBoard no worries, I believe your approach makes perfect sense and will for sure speed up the effort.

I just see one potential problem, if we want to keep somehow the same existing toolkit and implement some sort of HAL to allow us to have the same code base for 8bit and 32bit then we must stay inside something supported by the Arduino platform ?

Roxy-3DPrintBoard, 2016-03-30:
It is very possible that Atmel will be the most logical choice at which point, everything is simple.   But really...  The users don't use the the IDE environment of Arduino to do anything but compile and upload.   This is a valid point for discussion, but it is very possible that another environment for the 32-Bit stuff will make more sense.   It may be that some version of GCC with a makefile can do everything they need.

jbrazio, 2016-03-30:
Users do not use the IDE.. but we do use the Arduino core. :-/

Roxy-3DPrintBoard, 2016-03-30:
> Users do not use the IDE.. but we do use the Arduino core. :-/

At which point we need some type of HAL to move to another vendor's microprocessor!   :)

CONSULitAS, 2016-03-30:
@Roxy-3DPrintBoard @jbrazio
see my https://github.com/MarlinFirmware/MarlinDev/issues/317#issuecomment-203663421

@Wurstnase has a running (but rather old, based on Marlin of 01. Feb. 2015, and experimental) fork for Arduino Due: https://github.com/Wurstnase/Marlin4Due
He just stopped development regarding to his readme.

Wurstnase, 2016-04-01:
Yes, i've stopped.
Anyhow, there are a lot of 32bit Controller. Take a look at mbed.org for example.
You can use the arm-none-eabi-xxx for stm32, lpc, also the Arduino Due use it. The 'problem' is the HAL or whatever you want to use. But there is no standard lib like avr-lib.

Sniffle, 2016-04-07:
I already showed this to @Roxy-3DPrintBoard 

but what do you guys think?
main board - 2 exruders+ heated bed support + x, y, z drivers
http://www.filastruder.com/collections/electronics/products/duet-controller-v0-8-5

extension board to add 4 more extruders
http://www.filastruder.com/collections/electronics/products/duex4-expansion-board

touchscreen display
http://www.filastruder.com/collections/electronics/products/paneldue

and it's already supported by reprap firmware so we have somewhere to reference for pinouts, etc.

is it just me or does this board almost seem too good to be true because it covers pretty much every thing a "moddern" printer could ask for.  I'm sure there is a gotcha somewhere but damn talk about a complete package.

jbrazio, 2016-04-07:
I like it uses the same uC as the Due but the drawback (without even deep diving into hardware specs) it's the price, the overall solution you propose is quite expensive ($285).

Sniffle, 2016-04-07:
Yeah that was my concern as well... It is just kinda the all in one
poackage aside from the 1/16th drivers.

Roxy-3DPrintBoard, 2016-04-07:
I wonder if they have an updated version of the board they are working on with 1/64 Micro-Stepping?   Other than that, and the cost...   This is a really good setup.

Wurstnase, 2016-04-07:
The firmware on this board is not fast enough to support higher
microsteppings. They can calculate Delta without segmentation. But only at
<20kHz.

um 17:24:

jbrazio, 2016-04-07:
> The firmware on this board

This is were Marlin comes in.. or which firmware are you talking about ?

Bio-Printer, 2016-04-07:
At:  https://docs.google.com/document/d/1whV779t_660hpgTxYxFomgqrOk1CpEQi-8QttTHaI-Q/pub

They say they pre-load the RepRap firmware.  They recommend:

> The DC42 version of the Firmware covers the widest range of printer configurations - cartesian, delta and core-XY and is recommended for general use. It is available for download at https://github.com/dc42/RepRapFirmware/tree/dev.

Wurstnase, 2016-04-07:
Right, that firmware is very good in what it's doing. I see no need to put
Marlin on it. Also because Marlin has no webinterface and this is one of
the main features. Anyhow, the Due is not that bad. Only good hardware for
it is hard to get.

jbrazio, 2016-04-07:
Just throwing out something out:
The DUE is shield compatible with Mega.. can't we have a "interface board" that converts CMOS(3.3V) to TTL(5V) and vice versa ? This would mean we could use existing RAMPS shields.

Bio-Printer, 2016-04-07:
> The DUE is shield compatible with Mega.. can't we have a "interface board" that converts CMOS(3.3V) to TTL(5V) and vice versa ? This would mean we could use existing RAMPS shields.

That is an interesting idea.   That might work good for things where their locations is known.  Like the Step Sticks and MOS-FET's.    But for the generic GPIO pins that are brought out to the RAMPS board you don't always know how they will be used.  They might be used as an input or they may be an output.  It might be possible to have both input and output buffers on those extra lines and the user has to manually configure the board with jumpers to say which direction the data is flowing.

jbrazio, 2016-04-07:
In terms of BOM what would be a ballpark figure to buffer all pins in a RAMPS board so it would be DUE compatible ?

CONSULitAS, 2016-04-07:
@jbrazio 
> The DUE is shield compatible with Mega.. can't we have a "interface board" that converts CMOS(3.3V) to TTL(5V) and vice versa ? 
Sorry, but it is not that easy. Please see:
https://github.com/MarlinFirmware/MarlinDev/issues/317#issuecomment-203688194

There are serious safety risks to care only for the voltage level! The due has inverted logic on startup.

This means: If you put a blink sketch (e.g. to test the Board) on the due with a RAMPS it will start heating immediately and would not stop!

Please don't even think about it...

You need a shield with a logic inverter between the pins and the MOSFETs. Anything else is to dangerous.

jbrazio, 2016-04-07:
From the efforts done in the last week(s) I see the following landscape:
 - RADDS
  - Pro: cheap, available, form factor (shield)
  - Cons: board desing flawled on some hw revisions
 - Smoothie board
  - Pro: all-in-one, avalilable, good reference
  - Cons: deeply associated with Smoothie firmware
 - DUET
  - Pro: all-in-one, available
  - Cons: expensive

Choices are short.. if we could minimize the cost of upgrading a current RAMPS setup to just a Due a couple of components..

@Wurstnase which board setup are you using ?

@CONSULitAS the IO ports are inverted, can't this be mitigated with pull-down resistors and digitalWrite(LOW) on the setup() ?

esenapaj, 2016-04-07:
>The due has inverted logic on startup.
I see, so that's what it is...
My system is RAMPS Duo(modded RAMPS) + Arduino Due that it's driven by [Marlin Duo](https://github.com/esenapaj/Marlin/tree/Duo),
thus all MOSFETs are in full throttle at uplloading a sketch.
[video clip](https://www.youtube.com/watch?v=K-Am8guqwCQ)

I prevent it by to cut off the aux power at uploading at present.

Roxy-3DPrintBoard, 2016-04-07:
- RADDS
--Pro: cheap, available, form factor (shield)
- -Cons: board design flawed on some hw revisions (will turn on all heaters upon reset)
- Smoothie board
--Pro: all-in-one, avalilable, good reference
--Cons: deeply associated with Smoothie firmware
- DUET
--Pro: all-in-one, available
--Cons: expensive, only supports 1/16th Micro-Stepping

Wurstnase, 2016-04-08:
> Cons: board design flawed on some hw revisions (will turn on all heaters upon reset)

Really? Never saw this while flashing my Due.
Radds is not cheap. A set with an Arduino Due, 4 stepsticks (RAPS128) and the shield is ~150€.

> Cons: deeply associated with Smoothie firmware

a con? Well, you can't use the Arduino IDE.

> Cons: expensive, only supports 1/16th Micro-Stepping

You can use external drivers. Only the maximum speed is limited to ~100kHz. This is anyhow more than a ATmega can reach (40kHz). The single stepping is <20kHz. Marlin has a single stepping currently at 10kHz.

AnHardt, 2016-04-08:
The inverted MOSFET problem is not a general DUE one, but unique to RAMPS-FD V1 (the ones sold by Geeetech)
This __inverting__ MOSFET __drivers__ are the reason for the trouble.
[image]
All the other boards have learned the lesson (as far as i know)!

jbrazio, 2016-04-08:
Ok we are missing RAMPS-FD on our list.

jbrazio, 2016-04-08:
Hum this one slipped my radar, [it costs](https://www.ebay.com/ulk/itm/271325537292) 20€ without the driver stamps.

Roxy-3DPrintBoard, 2016-04-08:
> Hum this one slipped my radar, it costs 20€ without the driver stamps.

I could not find a version # silk screened on this board.   Is this a RAMPS-FD V1 (the ones sold by Geeetech) ???

Wurstnase, 2016-04-08:
> I could not find a version # silk screened on this board. Is this a RAMPS-FD V1 (the ones sold by Geeetech) ???

Yes. They've never made the V2.

Roxy-3DPrintBoard, 2016-04-08:
* RAMPS-FD V1
  - Pro: Cheap, available, good form factor and options 
  - Con: board design flawed on some hw revisions (will turn on all heaters upon reset)
* RADDS
  - Pro: cheap, available, form factor (shield)
  - Cons: board design flawed on some hw revisions (What is the flaw and on which versions?)
* Smoothie board
  - Pro: all-in-one, avalilable, good reference,  proven by Smoothie usage
  - Cons: Unknown currently
* DUET
  - Pro: all-in-one, available
  - Cons: expensive, only supports 1/16th Micro-Stepping (but this can be mitigated with external drivers)

CONSULitAS, 2016-04-08:
* RAMPS-FD V1
  * Pro: Cheap, available, good form factor and options
  * Con: board design flawed on all currently (Apr 2016) sold hw revisions (will turn on all heaters upon reset and without running Firmware sketch e.g. blink.ino)

* RAMPS-FD V2
  * not currently sold (see: http://reprap.org/wiki/RAMPS-FD -> warning section)
  * Answer by V1 seller on eBay:

    >Neue Nachricht von: link-fortune  Top Rated Seller(9,057Green Star)
Sorry, we don't have the 2.0 version on sell yet.
   >>you write that this is Version 1.4.
Can you please provide Version 2.0?
1.4 is known as high safety risk and can cause fire!
See: http://reprap.org/wiki/RAMPS-FD -> warning section
Thanks for your reply in advance.
Regards

CONSULitAS, 2016-04-08:
@jbrazio 
> @CONSULitAS the IO ports are inverted, can't this be mitigated with pull-down resistors and digitalWrite(LOW) on the setup() ?

* It would, but the digitalWriite(LOW) ist the problem: If there is no sketch caring (e.g you have blink on it) you have a problem.
* And you have to shut off the internal pull up.

So best would be to invert with an TTL electronics like 74LSxx:
* you get an TTL output level for your MOSFET -> solves problem with overheating bed MOSFET
* simple with 74LS04 - just inverting
* configurable with 74LS02 - NAND with jumper for selecting or not

Additionally you need a logic level converter for every directly connectable IO line (aux, servos, endstops, i2c, PS-ON). An optical endstop with 5V output would fry your Due.

So ideal would be a RAMPS Forte which lets you choose between 3.3/5V and inverting or not...

esenapaj, 2016-04-08:
@Roxy-3DPrintBoard, in addition, there is also such a choice.
[Alligator Board](http://reprap.org/wiki/Alligator_Board)

[SMART Ramps](http://www.aliexpress.com/item/3D-print-unit-new-reference-SMART-Ramps-For-Arduino-Due-Like-Ramps-FD-RADDS-AZTEEG-MINI/32213624860.html?spm=2114.30010308.3.1.eOm2nV&ws_ab_test=searchweb201556_9,searchweb201602_1_10017_10005_10006_10034_10021_507_10022_508_10020_10018_10019,searchweb201603_6&btsid=fa9b0ac1-f3a8-4e6b-b259-58d486113bcc)
[(official repository?)](https://github.com/glennswest/SmartRamps)

Roxy-3DPrintBoard, 2016-04-08:
The Alligator Board looks good.  But a few quick searches makes it appear it isn't being produced yet.

The SMART Ramps board is interesting.  But it isn't clear why the board is 'Smart'.  Is it a version 2 board?

Here is a Smoothie compatible board and LCD I stumbled on to:

www.aliexpress.com/item/New-Version-AZSMZ-Mini-12864-LCD/32421047617.html?spm=2114.10010108.100009.2.pEtygk

jbrazio, 2016-04-08:
Are you able to find RADDS ? eBay.com shows none.

esenapaj, 2016-04-08:
Here it is, but it's out of stock at this time.
[Radds V1 5 3D Printer Controller for RepRap Mendel Prusa | eBay](http://www.ebay.com/itm/281790122423)

Roxy-3DPrintBoard, 2016-04-08:
*RADDS
  -Pro: cheap, available, form factor (shield)
  -Cons: board design flawed on some hw revisions (What is the flaw and on which versions?)

What exactly is the board design flaw that exists on some revisions of the board?

Sniffle, 2016-04-08:
Roxy i think the issue was not truly open source.

http://reprap.org/wiki/RADDS

jbrazio, 2016-04-08:
For 50€ without stamps I would say:
 - RADDS
  - Pro: ~cheap~, available, form factor (shield)
  - Cons: board design flawed on some hw revisions (What is the flaw and on which versions?)

Roxy-3DPrintBoard, 2016-04-08:
>> What exactly is the board design flaw that exists on some revisions of the board? 
>> RADDS does not power up with all of the heaters running full tilt, right?

> Roxy i think the issue was not truly open source.

If that is the only issue, that don't think that bother me.  They obviously tried to make it easy for people to evolve the design, and from a legal perspective, they would have a very difficult time trying to litigate any 'infringement'.   

As @jbrazio said, I'm kind of leaning towards the RADDS setup right now.    I wish it was more clear that all of the Step Sticks plug into it and work and that any LCD Panel can plug into it and not fry the Due.

jbrazio, 2016-04-08:
@Roxy-3DPrintBoard I've read http://reprap.org/wiki/RADDS and they have a big warning point: Not Open Source.

Roxy-3DPrintBoard, 2016-04-08:
Yes...  What that says is:

> In particular, at least these item(s) were found to be missing:
> Board layout.  There might be more items missing.

But they include the schematics for everything.  That is kind of like saying:   Marlin is not Open Source because you don't include a pre-compiled binary ready to load for every printer.   It is true that I did not find any board layout stuff.  But for $55 it lets you plug in the Due and Step Sticks to get going.   I have no intention of making a board anyway.    The one reason it would be nice to see the board layout is just in case we need to cut and jump some traces for some reason.  But if the board works...  Even that is a stretch.

jbrazio, 2016-04-08:
What I do not understand.. is all the people porting and using the DUE.. whch boards are they using. :-D

Roxy-3DPrintBoard, 2016-04-08:
I think the truth is there are not that many people (yet) using Due's.   It's just the Bleeding Edge crowd.  We can hardly even find information about the various boards.   Some of the promising boards are not even in production or available ( RAMPS-FD V2, RADDS v1.5 out of stock).

jbrazio, 2016-04-08:
Then I would say if we stick with anything which is uC compatible with the Arduino core and as long as all devs participating in the initial effort have exactly the same board (whatever that will be) we shall be good.

What is the best stamp we could use on one of these shields ?

Roxy-3DPrintBoard, 2016-04-08:
> What is the best stamp we could use on one of these shields ?

Stamp?  You mean like Step Sticks?

jbrazio, 2016-04-08:
Yes, the stepper drivers.

Roxy-3DPrintBoard, 2016-04-08:
We want 1/64th Micro-Step capability.   Preferably, we want the ability to have high drive current for the motors.   I don't know which ones fit those characteristics.

bobc, 2016-04-09:
There is another Due based system which has not been mentioned: BAM&DICE http://reprap.org/wiki/BAM%26DICE

But basically in the 32 bit world there are only Due based boards (including Duet and various shields), or Smoothieboard and it's clones. Apart from the Chinese made versions most of the boards seem to have intermittent stock problems.

Well, there is also the 4pi which is based on SAM3U https://0xfb.com/shop.html.

I guess RAMPS-FD is the most widely used board, if only because it is cheap. Although it has a design flaw, I've not heard of any causing fire. Unfortunately Geeetech jumped the gun by putting into mass production, they have never contacted me about it.

I was thinking of revisiting RAMPS-FD v2, and hoping to persuade someone to make it.

jbrazio, 2016-04-09:
BAM&DICE seems a bit exotic with fancy PCI-X like connectors..

You're the creator of RAMPS-FD ?

bobc, 2016-04-09:
They are PCI connectors - PCI connectors are not fancy or exotic! Edge connectors have been around since before computers were invented, PCI was created in 1992. Every PC has PCI connectors in them, they are made by the million so are quite cheap. Per pin, they are by far the most cost effective way to join two boards. I can't imagine why anyone would think they are fancy or exotic.

Yes, I designed RAMPS-FD, with help from Cefiar.

Thinking more on topic, if I was picking a new CPU for a printer board today, I would use at least a Cortex M4 + FPU, for the price it's hardly worth bothering with anything less.. You can get these cheaper than an AVR 8 bit. An example board might be https://developer.mbed.org/platforms/Seeed-Arch-Max/. It has also ethernet and an Arduino shield pinout, should work with existing shields. The Arch Max costs about $40 and I believe is open source. It does mean leaving behind the rather limited Arduino ecosystem though, but Mbed have some nice open source code.

However, if you want something reliable that you can buy today, there is probably only Duet or Smoothieboard, or a cheap clone like MKS-SBASE. Those boards provide a range of price options, plus different chips, so would be good reference targets. I would try for an STM32 or Kinetis target as well.

jbrazio, 2016-04-09:
>  PCI connectors are not fancy or exotic

Don't take me literally ;-)
I was joking about the fact that is not common to see PCI connectors on 3D printer boards.

Bio-Printer, 2016-04-09:
I've been sending out emails and have gotten a few responses that need to be communicated to the larger group.   First, David Crocker confirmed that the Duet can use 1/64 Micro Step Sticks on the expansion board and in fact some users are doing just that.  These users still use the on-board 1/16 Micro Step controllers on the Duet for the Extruders in their systems.

I also received an email from Tim Elmore that we should discuss.   Tim Elmore is the USA Distributor for the Duet.   He said he would provide deeply discounted pricing on boards for up to 6 of our developers that are working on porting the Marlin firmware to a Duet environment.   The pricing he will provide us is about 60% of the advertised list price.

I have a printer I can dedicate to this purpose.   I suspect @Roxy-3DPrintBoard is going to want to be involved in the 32-Bit port.   It would be really good if we could get @thinkyhead and @MagoKimbra to sign up also.   If all of these people do want a Duet system to help with the port, that still would leave two more systems for other people that would be willing to help do the port.

If we do decide to go with a Duet based Reference Platform, we need the people that get one of the six slots to really be serious and help with the port.

Sniffle, 2016-04-09:
My favorite thing to do is mod and work on printers.  I would be happy to,
or ill just buy it myself anyway...

jbrazio, 2016-04-09:
These are excellent news but we must be a bit pragmatic, currently it will be a mess to start Marlin32 inside the same repo as Marlin8 and I don't like the idea to fork a new 32bit repo. Having said this we urgently need to invest time on OO the basic stuff which will allow us to introduce an HAL.

We should organize ourselves, prioritize and built at least an high level roadmap. @thinkyhead has been really busy with the RC cycle but it's time for him to provide feedback to this thread.

bobc, 2016-04-09:
I think Duet would be a good choice, it's a solid board with good support. I have the older 0.6 version, but it is quite similar to the latest. After doing Marlin for Due/RAMPS-FD, I was going to do a version for Duet, but I got sidetracked doing "STM32 for Arduino" then kinda lost the 3d printing bug for a bit.

Because the Duet uses extra features not supported by Due, there is no direct support for it in the Arduino IDE. RepRapPro build their firmware with Eclipse. If it is desired to keep builds within the Arduino environment, then it would be good to create a new Arduino platform variant for the Duet.

Bio-Printer, 2016-04-09:
@bobc   Can you explain what extra features are used by the Duet that are not supported by Due.   It would be helpful to understand what the issues would be to go that route.

bobc, 2016-04-09:
@Bio-Printer for example there are pins the Duet uses which are not on Due. In the Arduino Due core, the pin numbers (which users know about) are mapped to physical hardware pins, but the Duet specific pins will not be in the mapping.

But it looks like there is now an Arduino core for Duet https://github.com/dc42/CoreDuet, so maybe that will work in Arduino IDE? I expect dc42 can advise on that. I'm not really up to speed with latest Arduino IDE developments, they have brought in several new features.

[51 more replies]

closed 2016-05-27 by jbrazio