MarlinFirmware/MarlinDev #400: Architectural concept for Marlin objects

CONSULitAS, 2016-04-11:
Dear developers,
@thinkyhead @jbrazio @Roxy-3DPrintBoard @MagoKimbra and others

on #395 we have a discussion about modules, objects and HAL.

I would like to propose some "first ideas" for objects and invite you to contribute, so that we can sketch a **rough** architecture for the first steps.

Ideas and thoughts for mapping parts of the robot to objects.

Note: This is written like a class hierarchy, but not every level perhaps/sure will be an object and some are only properties, methods or hints.

* Electronics
  * Console
    * Serial (mostly)
    * but how about network or even web console?
  * PersistentStorage (see https://github.com/MarlinFirmware/MarlinDev/issues/395#issuecomment-208025964)
    * EEPROM (see #376)
    * SD card
    * I2C EEPROM (https://github.com/MarlinFirmware/Marlin/issues/626 for Due and others)
    * punch cards (regards to @jbrazio 😄)
  * Sensors
    * Endstops (gives mechanical end postiion) -> HAL, you can abstract `Endstop.isHit();`
      * physical
        * mechanical/electronical
      * calculated (former known as software endstops 😄)
    * Reference switches (give zero position), poor makers often only have endstops, so they are used for reference, too
    * probes (used for leveling or calibrating, e.g. ABL), poor makers often use ZMIN endstop as probe
      * HAL: just check `Probe.isHit();` and the probe either tests its own hardware or just `EndstopZ.isHit();`
    * temperature sensors (there is some HAL via thermistor tables)
      * hotends
      * bed
  * switches
    * Heaters
      * hotends
      * bed
      * building room -> GCODE for setting build room temperature?
    * fans
      * object (standard fan for cooling freshly printed filament)
      * hotends (e.g. for E3D)
      * electronics (stepper drivers)
      * build room (cooling closed build rooms if to hot? see heaters)
    * light
      * LEDs for status
      * Illumination of building room
  * displays
    * different types and interfaces
* board
  * IO pins (called exactly like on silk screen of the board)
    * D2, A1 and so on
  * IO ports (called exactly like on silk screen of the board) -> RAMPS: X, Y, Z, E0, E1, 
    * use 2 to n IO pins
    * stepper ports
    * i2c port

* mechanics
  * objectspace
    * size for building room (can be smaller than frame sizes, e.g. because of tilt, cleaning or tool positions)
    * leveling methods (ABL, ...)
      * probing
      * transformation to frame coordinates 
    * movement
      * target for G0/G1 and others
      * tells `Frame.Axis` to move logically -> Transformation for ABL
  * frame
    * axis
      * target for `Objectspace.Movement`
      * tells corresponding actors to move physically -> HAL for different kinematics (Cartesian -> trivial, CoreXY, Delta, ...)
    * sizes for physical movement
    * current coordinates
  * actors
    * stepper (can differ from frame axis, e.g. CoreXY, Delta, Scara)
      * ratio (aka steps per mm)
      * use IO ports (e.g. RAMPS X->X, Y->Y, Z->Z1, E0->E, E2->Z2)
    * servos
    * selenoids (if used for moving)

**Please keep in mind: We shall not and will not implement all this at once! But if we have the same big picture in mind - even with the visionary parts, we will make more flexible and compatible code.**

**It is important to go on in small steps like explained in #395 . So we can learn and keep on the track.**

I think i will update this with your ideas as an collection for the information in one place. Please provide hints to useful PRs and issues for reference, too.

jbrazio, 2016-04-11:
> calculated endstops

May I suggest "virtual endstops" ?
Whats you feeling about UML and diagrams ?

CONSULitAS, 2016-04-12:
@jbrazio 
>>calculated endstops

>May I suggest "virtual endstops" ?

Good idea. Will change it.

>Whats you feeling about UML and diagrams ?

Formal descriptions are great and i am sketching diagrams all the day.

But, perhaps UML is not common for all of us. And for a first discussion and collection of ideas i thought:

Keep it simple, stupid.

jbrazio, 2016-04-12:
I was thinking to maybe start sketching the "dependencies" also.
But we would require a collaboration tool where everyone could edit and contribute.

Roxy-3DPrintBoard, 2016-04-12:
A little bit off topic perhaps...  But I'm think about how we transition.   Would it make sense for the Persistent_Storage object to initially have its own data structure of things that it needs to save and restore.  And initially...  Its structure is everything that Configuration_Store.cpp does ?   

And then, as we add new objects, we remove them from the Persistent_Storage object's structure and give control of the under lying data to the newly written object?    That might transition nicely.   Especially, if we had names that can be referenced for various object's data.   Initially...  There would just be the Persistent_Storage's data structure (with its name on all of that data) in the permanent memory.  But as new objects are created, more items would end up in the list.

jbrazio, 2016-04-12:
Yes I would agree with something on those lines or for this specify "transition issue" propose we could write a `LegacyConfigurationStore` to work with `PersistentStorage` and with `ConfigurationStore`.

CONSULitAS, 2016-04-12:
@Roxy-3DPrintBoard 
You are not really of topic. :smile: 

If i say step by step, i mean exactly approaches like this. Only care for a small corner where you have overview whats happening if you change it, but have the big picture in mind.

Your idea is great. And perhaps wrap the old Configuration_Store code which is not ready in a method like `legacy()`.

CONSULitAS, 2016-04-12:
@jbrazio Oh no!

You just type faster and shorter! 17 seconds...

jbrazio, 2016-04-12:
"Great minds think alike"
".. but some type faster and shorter" 

lol

CONSULitAS, 2016-04-12:
@jbrazio 
>I was thinking to maybe start sketching the "dependencies" also.
But we would require a collaboration tool where everyone could edit and contribute.

If you have ideas... The tool should be easy and online. How about something like google tools? Tables for structured information and slides for graphics. 

Results would be then put to the documentation project.

CONSULitAS, 2016-04-12:
@MarlinFirmware/collaborators 
No suggestions? Is this complete?

Nobody recognized that there is advance missing. Will fix it.

If there are no complains about this proposal, i suppose we have a perfect, universal and final version which should be implemented soon. :grin:

CONSULitAS, 2016-04-12:
Are there more issues / PRs for this milestone?
https://github.com/MarlinFirmware/MarlinDev/milestones/Marlin%20with%20first%20object%20orientation

gddeen, 2016-04-12:
Hmmmmmmmm...

What is an 'endstop' object? What does it do? Hmmmm.
I dunno...

CONSULitAS, 2016-04-14:
I do! 😁 

It just represents one or all physical or virtual endstop switch(es).

The easiest way to implement it in a first step would just be encapsulation of `update_endstops()` in `stepper.cpp`.

closed 2016-05-25 by boelle