MarlinFirmware/MarlinDev #401: "You’ve been removed from the Collaborators team on the MarlinFirmware organization."

nothinman, 2016-04-12:
Explain yourselves.

fmalpartida, 2016-04-12:
I got one too.

Roxy-3DPrintBoard, 2016-04-12:
> Explain yourselves.
@nothinman  Not even a 'Please' in front of that???   

Did you see you both have Write access to all three repositories?

nothinman, 2016-04-12:
@Roxy-3DPrintBoard I've noticed after opening this ticket. Well, that was only partially serious, but I guess people don't try to read the sentences the way they were meant to be read, if they're lacking emoticons :-)
'Please' is when you do something wrong and ask for forgiveness, or ask for something you didn't already have :-)

closed 2016-04-12 by nothinman

Roxy-3DPrintBoard, 2016-04-13:
A while ago GitHub changed the meaning and usage of 'Collaborator'.   Today I started to clean up some of the Team usage of MarlinFirmware.   More still needs to be done.