MarlinFirmware/MarlinDev #403: Rumba connection issue

adamfilip, 2016-04-14:
Running into an issue connecting from Pronterface or Simpliy3d to Rumba via usb
I have tried fresh installs of Marlin 1.1 (RC3, RC4 and current RC5 Bugfix) and they all show the same issue. cannot connect (using Baud 115200, Port 3)
Motherboard is set to 80
I am able to connect with Arduino IDE 1.6.8 no issue

I have flashed the board with other brands of  firmware and it works fine. but with Marlin it doesnt I cant explain it. I contacted simplify3d for support and they cant explain it.. they suggested i try to connect without the startup command. it connects this way but it wont accept gcode commands this way.

not sure where else to report this issue.

Roxy-3DPrintBoard, 2016-04-14:
Ah...   Looking at the Configuration.h file, the default baud rate is set with:

#define BAUDRATE 250000

Is that the problem???

adamfilip, 2016-04-14:
No I have it set to 115200

Endlesshunt, 2016-04-15:
Hi, 

I'm currently running a Rumba on Marlin 1. and have an issue with S3D where it would pause multiple times during a print for no reason. If I then manualy pause and continue, it carries on with the print. If I leave it, it times out, throws an error (will have to check it on a print for the exact error message) and then reconnects from S3D and carries on with the print. But that leaves a huge blob on the print. S3D recons it is a windows 10 problem but I do not agree. Running the same firmware on the Rumba but from Repitier it never gets this problem. I find that it mostly, but not always, stops at the end of a perimeter

Endlesshunt, 2016-04-15:
PS, I have tried different comms speeds right down from 38400 up to 250000 as well as different buffer sizes with no difference, S3D still randomly stops

Endlesshunt, 2016-04-15:
Not sure if the above is relevant but it does show that there is some kind of issue between Marlin and S3D. I have seen others also complain on the S3D forums of comms failures of all different kinds between Marlin and S3D

Endlesshunt, 2016-04-15:
Question, did anything change in the comms routines from V1 to V1,1?

Endlesshunt, 2016-04-15:
The next question is how does Marlin comms differ from the other brands of firmware

jbrazio, 2016-04-15:
I recommend always testing the latest Marlin version which is RCBugFix branch.

adamfilip, 2016-04-15:
When I use repetier on the rumba, I can connect to Simplify3d/Pronterface without issue, 
But if i switch back to Marlin I cannot.. I should try an earlier release to see if it changes anything. the oldest ive tried is 1.1 RC3

Endlesshunt, 2016-04-15:
I have not tried Repetier firmware, only Repetier Host with Marlin 1 and that combo does work without any issues. Only have issues when I drive M1 with S3D. I would say try Marlin 1 with S3D and see if it work or not. It will eliminate if there is maybe a problem with you Rumba board or not

Endlesshunt, 2016-04-15:
I can flash my Rumba tonight with 1.1 and see if I can connect with S3D.

adamfilip, 2016-04-15:
I tried Marlin 1.0.1 with the same issue.
I can accept that my Rumba was defective if, it didnt work with other firmware.
Arduino IDE Connects and updates fine
Repetier runs fine and connects easily, using same cable and computer
my computer is a Win 10 64bit machine

Blue-Marlin, 2016-04-15:
What do you want to achieve with "Port 3"?

adamfilip, 2016-04-15:
OMG.that was the issue! I selected Port 3 thinking it means use Com 3 when trying to connect via printrun/S3D
Changing it back to Serial_Port 0 fixed it.
Thank you.

Endlesshunt, 2016-04-15:
@adamfilip I'm onMarlin 1.0.2 and have comms with S3D. Just have issues with intermittant stops and timeouts simiral to #2000 in MarlinFirmware/Marlin . 

Should have seen that. I read Port 3 and simply assumed Com3, Good catch Blue-Marlin

Endlesshunt, 2016-04-15:
Sorry for my spelling but I have a valid excuse, I'm diagnosed dyslexic.

closed 2016-04-18 by adamfilip