MarlinFirmware/MarlinDev #404: Feature Request: Online Configurator

adamfilip, 2016-04-18:
Repetier has a very handy online configuration tool. that once filled out will output the necessary config.h files needed for a users machine. it will also input an existing .h file and let you make changes.
for novice users this is very handy to get things configured quickly.

Something like this would be great for each official release.

Pwuts, 2016-04-19:
There already was a configurator included in the Marlin v1.1.0-RC3 package but it has been removed due to problems with running it locally. There is an online version hosted by thinkyhead, but that too doesn't work anymore since the Github API was updated. It just has to be modified to suit the latest version of Marlin and Github. :)

thinkyhead, 2016-04-20:
https://github.com/thinkyhead/MarlinConfigurator

I started this configurator that parses Marlin's own `Configuration.h` and `Configuration_adv.h` files (so it can use them as the "database" for the configuration options). But it has fallen out of date, as @Pwuts points out. So I need to do some updates and then continue to work on it. It shows some promise as a starting point. After the first phase (making the "expert" configuration tool) I was planning to add the "easy step by step wizard" interface to help set the most important settings.

bobc, 2016-04-30:
Some time ago I created a Bounty asking someone to create a config tool. The Teacup ConfigTool is pretty much exactly what I had in mind http://forums.reprap.org/read.php?1,516185. It is in Python so cross-platform and always available locally, and Open Source of course.

It was written specific to Teacup, but hopefully some of the architecture can be re-used. At least it illustrates the principle.

thinkyhead, 2016-05-02:
@bobc My Javascript-based configurator runs everywhere too, with only a web browser needed. It could be reworked into other languages, but I find this is a good way to prototype and share with others more quickly. As mentioned, it currently needs some updates to work with the latest Github API, and it needs further work to parse the most recent configuration files. I will patch it up a bit tonight for the latest Github API, and get it into better working order with the most recent configurations.

https://rawgit.com/thinkyhead/MarlinConfigurator/master/src/index.html

The idea with this tool is to first get the "advanced mode" configuration mode working, and then build a simpler "wizard" to collect the core settings, and to config each add-on feature separately. Some of the core functions will also be useful for doing a "drag-and-drop" conversion tool to take 1.0.2-1 (and earlier) configuration files and move settings into a new configuration file. (That tool should be fairly straightforward, but I need to gather all the old/new settings and make a list of which need conversion and which can just move over as-is.)

gddeen, 2016-05-02:
Is it use once? Or does it keep in sync?

thinkyhead, 2016-05-02:
@gddeen I'm not sure what you mean by "does it keep in sync?".

The tool is designed to download the configuration files directly from the Github repository, so that it always has all the latest options. But it has to be configured to do so, otherwise it uses its own local copies. At the moment it still needs updates to deal with recent formatting changes (of comments) in the configs.

The tool still needs a bunch of work to be better organized, but it's coming along well.

gddeen, 2016-05-02:
A user grabs an old config file, then starts the tool to generate the new data files.

When through they never use the tool again and know how to edit the generated files?
Or do they just continue to use the config and it repopulates its interface from the generated
files?

thinkyhead, 2016-05-03:
@gddeen Ideally: The web page has a nice interface to edit config options. If you drop old configuration files onto the page it generates new configuration files for you and carries on. When you finally download the configuration files, they are still human readable and editable text files, and you don't need the tool to modify them further.

jbrazio, 2016-05-03:
The configuration must be online otherwise it's another thing we have to make sure the users update and keep in sync.

thinkyhead, 2016-05-03:
Good thing here is no back-end needed. So we can install it within the marlinfw site. Or anywhere. The link above has it loading right off Github, which is convenient.

jbrazio, 2016-05-03:
What we would require is some kind of easy hex uploader, once the configuration is up I want to start delivering pre-built hex files, people won't need to mess with Arduino IDE, all is taken care for them in the browser, the missing link on my plan is the hex uploader.

thinkyhead, 2016-05-03:
@jbrazio Something like this? http://marlinbuilder.robotfuzz.com/

closed 2016-05-25 by boelle